groeck/it87 #2: Support for ITE8786

Zlika, 2015-01-14:
Hi!
sensors-detect reports on my computer a chip ITE 0x8786 that is currently not supported by Linux (v3.16 from Debian testing) nor by this driver.
Do you plan to support it in a future release?
Best regards,
Thomas

groeck, 2015-01-14:
On Wed, Jan 14, 2015 at 03:09:37AM -0800, Zlika wrote:
> Hi!
> sensors-detect reports on my computer a chip ITE 0x8786 that is currently not supported by Linux (v3.16 from Debian testing) nor by this driver.
> Do you plan to support it in a future release?

Problem is that we don't have a datasheet, and ITE does not usually make
datasheets available, so it is somewhat unlikely that we can get the necessary
information.

Guenter

Zlika, 2015-01-16:
Hi Guenter,
Thank you for your answer. I've googled around and you're true, I wasn't able to find a datasheet for the IT8786E. WTF? How are they selling their chips without providing datasheets?

groeck, 2015-01-16:
On Fri, Jan 16, 2015 at 12:40:08PM -0800, Zlika wrote:
> Hi Guenter,
> Thank you for your answer. I've googled around and you're true, I wasn't able to find a datasheet for the IT8786E. WTF? How are they selling their chips without providing datasheets?
> 
ITE is one of the chip vendors who only provide datasheets to their
customers, ie to board vendors. In some cases it is possible to get
a datasheet under NDA, but of course that does not really help
if one wants to write an open source driver (since we can not publish
anything, including register addresses, if we get the information under
NDA).

Board vendors for the most part don't care about Linux, and thus do not
assert pressure on chip vendors to release their data sheets. So the only
thing one can really do is not to buy one of those boards. Unfortunately that
doesn't really work either since most people only discover the unsupported
part(s) _after_ buying a board :-(.

Guenter

Zlika, 2015-01-17:
Shame on them!
When I have some free time I will see if I can reverse it, I would like to be able to read temperature sensors and fans speed managed by the chip, I think it's doable. If I succeed I will send you a pull request.

groeck, 2015-01-17:
On 01/17/2015 03:22 AM, Zlika wrote:
> Shame on them!
> When I have some free time I will see if I can reverse it, I would like to be able to read temperature sensors and fans speed managed by the chip, I think it's doable. If I succeed I will send you a pull request.
>
Sure.

You can try to configure it as IT8783. Maybe the chips are similar.

Guenter

groeck, 2015-01-17:
Before working on the driver, make sure you get the latest version. I synchronized it with the upstream version.

Zlika, 2015-01-19:
If I do the same thing for it8786 than for it8783, the sensor command give the following results compared to the sensor values given by my BIOS:
* fan speed is ok
* temp1 (which seems to be system temp) is ok
* 2 others temp sensors are printed, but sensor type is disabled and the values are negative (my BIOS displays a cpu temp that is not found here)
* in0-in8 voltages are completly wrong (except maybe in3/+5V). There values is around 2V (except in3/+5V which is 4.45v and in8/vbat which is 0.37v). Voltages display in my BIOS are completly different (in particular vbat=3.146v here). Worst, the min and max voltages are wrong, min beeing nearly always greater than max!

I will continue my investigations, but if you have any advice I will be glad to know :-)

groeck, 2015-01-19:
On 01/19/2015 05:25 AM, Zlika wrote:
> If I do the same thing for it8786 than for it8783, the sensor command give the following results compared to the sensor values given by my BIOS:
>
>   * fan speed is ok
>   * temp1 (which seems to be [...]

You could try some other 'force' types, such as 8782, 8772, or 8728. After all, we are
just guessing wildly. The voltages will always be wrong to some degree, because they need
to be scaled, but at least some of them should be correct (the ones with the labels).

Overall I suspect we'll have to wait until we can find a datasheet, which may take
a while. Given that this is not an "official" chip but one of those which don't
officially exist (according to ITE's web site), that may unfortunately never happen :-(.

Can you send me the output of the following two commands ?

isadump 0x295 0x296
isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 4

You may have to replace 0x295 / 0x296 with the address reported by sensors-detect (plus 5 / 6).

Thanks,
Guenter

Zlika, 2015-01-21:
Hi!

I made some progress: I dumped the register values to locate the values corresponding to the voltage sensors and guess the LSB of the ADC by comparing with the voltage values displayed in my BIOS. It seems that the LSB of the ADC is 12 for 2 of the voltages and 24 for 2 others. Other voltage sensors seem to have more "exotic" LSB: 72, 30... So there is not a single "x2" scaling factor.
The locations of the Vin max/min values seem however to be incorrect (maybe a 1-byte offet?).
Here are the values of the registers (in bold the voltage values (3rd line) and the voltages max/min values (4th line)):
00: 19 16 FF 00 00 00 00 00 00 80 64 CD 07 B7 FF FF
10: E5 F9 A1 71 C6 81 00 00 02 FF FF DB F7 DA 19 19
20: **91 7E A6 8B A6 A6 B9 88 17** 25 C9 80 B9 33 30 E5
30: **06 F4 07 0F 1A FA 81 B1 BB C5 43 BA CB F3 77 6**
40: 2C D3 ED FF DE 17 5F 40 AD 6A D4 00 FF FF FF FF
50: FF 88 7F 7F 7F 40 00 00 90 00 04 12 60 00 00 00
60: A6 CA D6 28 20 04 0F FF 7F 7F 7F 00 00 04 0F FF
70: 7F 7F 7F 00 00 04 0F FF FF FF FF FF FF FF FF FF

For information, here are the voltage values as reported by my BIOS:
Vcore=1.740V, V-DIMM=1.512V, 12V=11.952V, 3.3V=3.336V, 5V=4.980V, VSB 3.3V=3.264V, VBAT=3.146V

Here are the outputs of the commands you asked me to type (sensors-detect does not report any adress, it only says that it found an ITE chip at 0x2e/0x2f, so I'm not the results are relevant):

sudo isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 4
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 4 using bank register 0x07.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 87 86 01 00 00 00 00 00 00 00 00 48 89 00 00 00 
30: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 20 0a 10 00 00 00 00 00 00 00 00 00 00 00 00 
70: 09 02 00 00 04 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
f0: 00 40 00 00 00 00 f0 00 03 48 00 00 00 00 00 00

groeck, 2015-01-21:
On 01/21/2015 12:34 PM, Zlika wrote:
> Hi!
>
> I made some progress: I dumped the register values to locate the values corresponding to the voltage sensors and guess the LSB of the ADC by comparing with the voltage values displayed in my [...]
That is pretty much what I was looking for.

Try to 'force' the following chips:
	0x8771
	0x8603

The scaling factor for the other voltages isn't that important - the minimum
scaling factor tells us the chip LSB, which is the important part. We'll need
to configure the remaining factors in sensors3.conf, after we got the base
values correctly.

Not really sure what to do with the min/max voltages. All other ITE chips
have the max registers at even addresses and the min registers at odd addresses.
Maybe the values in the registers are just wrong. You can try setting different
min-max values and see if you can trigger alarm bits.

> sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 4
> WARNING! Running this program can cause system crashes, data loss and worse!
> I will probe address register 0x2e and data register 0x2f.
> Probing bank 4 [...]

Nothing unexpected here.

Thanks,
Guenter

Zlika, 2015-02-04:
Hello Guenter,

You were right about the min/max values, the values in the registers were just wrong. I was able to change them and check that the alarms are then off:
```
cd /sys/class/hwmon/hwmon1/device
cat in0_alarm
> 1
cat in0_min
> 2544
cat in0_max
> 72
cat in0_input
> 1752
echo "2500" > in0_max
echo "1500" > in0_min
cat in0_alarm
> 0
```
So it8786 seems to basically work like an it8783: LSB and internal scaling factors are the same, temperatures and fan speed are ok, alarms are ok (even if on my computer the min/max values in the chip registers are wrong).
I would like to test some more advanced features of the driver like fan speed control but I don't know how to do it, could you give me some advice?

Regards,
Thomas

groeck, 2015-02-04:
On Wed, Feb 04, 2015 at 11:03:43AM -0800, Zlika wrote:
> Hello Guenter,
>
> You were right about the min/max values, the values in the registers were just wrong. I was able to change them and check that the alarms are then off:
> ```
> cd [...]

Maybe I am confused, but the earlier exchange seemed to suggest that
the chip is closer to 8771 and/or 8603. This is essential since the LSB
is different (12 mV instead of 16 mV). Can you clarify ?

> I would like to test some more advanced features of the driver like fan speed control but I don't know how to do it, could you give me some advice?
> 
Only thing I can think of is to check if changing pwm values results
in changing fan speeds.

Thanks,
Guenter

Zlika, 2015-02-04:
I used the following "features" (the same as for it8783 + the 12mv lsb): .features = FEAT_16BIT_FANS | FEAT_TEMP_OFFSET	  | FEAT_TEMP_OLD_PECI | FEAT_12MV_ADC
How can I test if the it8786 is also compatible with the FEAT_NEWER_AUTOPWM and what are the differences between FEAT_OLD_PECI and FEAT_PCI?

groeck, 2015-02-04:
On Wed, Feb 04, 2015 at 12:12:39PM -0800, Zlika wrote:
> I used the following "features" (the same as for it8783 + the 12mv lsb): .features = FEAT_16BIT_FANS | FEAT_TEMP_OFFSET	  | FEAT_TEMP_OLD_PECI | FEAT_12MV_ADC
> How can I test if the it8786 is also [...]

OLD_PECI vs. PECI determine which register bits are used to identify if a sensor
is connected to PECI. So far only 8721 has the combination of old PECI with 12mV
ADC, so I would have thought that PECI is more likely. Question is if you see
the CPU temperature with your chip; if yes, the respective sensor type should
be reported as PECI(6).

Similar, NEWER_AUTOPWM is typically associated with the 12mV ADC. With
NEWER_AUTOPWM, the pwm register has 8 bit, while it is only 7 bit for the
older version. With the newer pwm, pwm values are in registers 0x63+(8*nr),
while they are in registers 0x15+nr with the older pwm. From your register
log, NEWER_PWM seems to be more likely, but you should be able to find out
by writing the pwm value (after setting pwm control to manual mode) and
observing the impact.

Hope this helps,
Guenter

Zlika, 2015-02-05:
NEWER_AUTOPWM seems to be the right configuration: I was able to switch to manual mode (echo "1" > /sys/class/hwmon/hwmon1/pwm1_enable) and control the fan speed by writing in /sys/class/hwmon/hwmon1/device/pwm1.
However, I'm not sure that the automatic mode is working:
* I cannot see pwm1_auto_point* or temp1_auto_point* entrie in /sys/class/hwmon/hwmon1/, so there is no temperature threshold defined
* pwm1_auto_channels_temp = 2 (i.e. the pwm feedback loop is based on the temperature measured by the temp2 sensor). However, whereas temp1 sensor seems to be OK (and match the "System temperature" displayed in the BIOS, i.e. ~36°C), temp2 is obviously not working (-60°C, whereas this sensor should be what the BIOS calls "CPU temperature" whose value is ~55°C).

groeck, 2015-02-06:
On 02/05/2015 11:54 AM, Zlika wrote:
> NEWER_AUTOPWM seems to be the right configuration: I was able to switch to manual mode (echo "1" > /sys/class/hwmon/hwmon1/pwm1_enable) and control the fan speed by writing in [...]

Do you use FEAT_TEMP_PECI or FEAT_TEMP_OLD_PECI ?

I think you mentioned earlier that you used OLD_PECI,
which I suspect is not correct.

Guenter

Zlika, 2015-02-06:
I switched to FEAT_TEMP_PCI but this does not change the value for temp2 which is in either cases -60°C.

groeck, 2015-02-07:
On 02/05/2015 10:47 PM, Zlika wrote:
> I switched to FEAT_TEMP_PCI but this does not change the value for temp2 which is in either cases -60°C.
>

What is the complete output from the 'sensors' command ?

Thanks,
Guenter

Zlika, 2015-02-09:
With:
```
.features = FEAT_16BIT_FANS | FEAT_TEMP_OFFSET
| FEAT_TEMP_OLD_PECI,
.old_peci_mask = 0x4,
```
the output of the "sensors" command is:
```
acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +97.0°C)
temp2:        +29.8°C  (crit = +97.0°C)

it8786-isa-0a20
Adapter: ISA adapter
in0:          +2.34 V  (min =  +2.00 V, max =  +3.33 V)
in1:          +2.02 V  (min =  +0.24 V, max =  +2.10 V)
in2:          +2.66 V  (min =  +4.00 V, max =  +0.42 V)  ALARM
+5V:          +2.22 V  (min =  +2.32 V, max =  +2.06 V)  ALARM
in4:          +2.66 V  (min =  +3.15 V, max =  +2.99 V)  ALARM
in5:          +2.66 V  (min =  +2.96 V, max =  +0.05 V)  ALARM
in6:          +2.96 V  (min =  +3.89 V, max =  +3.25 V)  ALARM
5VSB:         +2.18 V  (min =  +1.60 V, max =  +1.39 V)  ALARM
Vbat:         +0.37 V  
fan1:         938 RPM  (min =   11 RPM)
fan2:           0 RPM  (min =   10 RPM)  ALARM
fan3:           0 RPM  (min =  158 RPM)  ALARM
temp1:        +36.0°C  (low  = -45.0°C, high = +76.0°C)  sensor = thermistor
temp2:        -56.0°C  (low  =  -1.0°C, high = +109.0°C)  sensor = disabled
temp3:       -128.0°C  (low  = +23.0°C, high = -34.0°C)  sensor = disabled
intrusion0:  ALARM

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +36.0°C  (high = +86.0°C, crit = +92.0°C)
Core 0:         +33.0°C  (high = +86.0°C, crit = +92.0°C)
Core 1:         +33.0°C  (high = +86.0°C, crit = +92.0°C)
Core 2:         +36.0°C  (high = +86.0°C, crit = +92.0°C)
Core 3:         +34.0°C  (high = +86.0°C, crit = +92.0°C)
```
With:
```
.features = FEAT_NEWER_AUTOPWM | FEAT_12MV_ADC | FEAT_16BIT_FANS
| FEAT_TEMP_OFFSET | FEAT_TEMP_PECI,
.peci_mask = 0x07,
```
the output of the "sensors" command is:
```
Adapter: Virtual device
temp1:        +27.8°C  (crit = +97.0°C)
temp2:        +29.8°C  (crit = +97.0°C)

it8786-isa-0a20
Adapter: ISA adapter
in0:          +1.75 V  (min =  +1.50 V, max =  +2.50 V)
in1:          +1.51 V  (min =  +0.18 V, max =  +1.57 V)
in2:          +1.99 V  (min =  +3.00 V, max =  +0.31 V)  ALARM
+3.3V:        +3.34 V  (min =  +3.48 V, max =  +3.10 V)  ALARM
in4:          +1.99 V  (min =  +2.36 V, max =  +2.24 V)  ALARM
in5:          +1.99 V  (min =  +2.22 V, max =  +0.04 V)  ALARM
in6:          +2.22 V  (min =  +2.92 V, max =  +2.44 V)  ALARM
3VSB:         +3.26 V  (min =  +2.40 V, max =  +2.09 V)  ALARM
Vbat:         +0.55 V  
fan1:         740 RPM  (min =   11 RPM)
fan2:           0 RPM  (min =   10 RPM)  ALARM
fan3:           0 RPM  (min =  158 RPM)  ALARM
temp1:        +35.0°C  (low  = -45.0°C, high = +76.0°C)  sensor = thermistor
temp2:        -59.0°C  (low  =  -1.0°C, high = +109.0°C)  sensor = Intel PECI
temp3:       -128.0°C  (low  = +23.0°C, high = -34.0°C)  sensor = disabled
intrusion0:  ALARM

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +33.0°C  (high = +86.0°C, crit = +92.0°C)
Core 0:         +30.0°C  (high = +86.0°C, crit = +92.0°C)
Core 1:         +33.0°C  (high = +86.0°C, crit = +92.0°C)
Core 2:         +33.0°C  (high = +86.0°C, crit = +92.0°C)
Core 3:         +32.0°C  (high = +86.0°C, crit = +92.0°C)
```
temp1 seems to match the "System temp" in the BIOS.
temp2 also is "alive" (ie. the measured value is changing from time to time, it is not static contrary to temp3) and "could" match the "CPU temp" in the BIOS (but without the "-" sign of course...).

Zlika, 2015-02-11:
Hello,
I made in interesting test. I used the sysbench tool to stress the CPU and increase its temperature.
Before using sysbench, the sensor tool returned the following results:
```
Physical ID 0: +36°C, temp1: +36°C, temp2=-56°C, fan1: 867RPM
```
Then I launched sysbench to have a CPU load of 100% on all the cores. The sensor tool then returned the following results (several executions):
```
Physical ID 0: +49°C, temp1: +37°C, temp2: -44°C, fan1: 1859RPM
Physical ID 0: +50°C, temp1: 36°C, temp2: -43°C, fan1: 2115RPM
Physical ID 0: +50°C, temp1: +36°C, temp2: -42°C, fan1: 5720RPM
Physical ID 0: +49°C, temp1: +35°C, temp2: -44°C, fan1: 5720RPM
```
Then I stopped sysbench:
```
Physical ID 0: +37°C, temp1: +35°C, temp2: -55°C, fan1: 1097RPM
Physical ID 0: +35°C, temp1: +36°C, temp2: -57°C, fan1: 751RPM
```
The conclusions are:
* the processor fan is correctly slaved on the CPU temperature
* the temp1 sensor measures the CPU temperature, but with an offset of around 92

According to you, where this offset comes from?
You can find my current version of the driver here: https://github.com/Zlika/it87/commit/58e217f0124afeca2af47d5a282d73c6e531209c

groeck, 2015-02-12:
On 02/11/2015 01:15 PM, Zlika wrote:
> Hello,
> I made in interesting test. I used the sysbench tool to stress the CPU and increase its temperature.
> Before using sysbench, the sensor tool returned the following results:
>
> |Physical ID 0: [...]
temp1 or temp2 ?

> According to you, where this offset comes from?

Look at the output of the "sensors" command, and the temperatures
it reports from the coretemp driver. What does it show as "high"
and "crit" limits ? "crit" may be close to the 92.

It should be possible to write the offset you observed (92) into
the temp2_offset attribute.

Background: PECI doesn't report the absolute temperature, but the
difference to the maximum (critical) temperature. Normally one has
to program the upper limit into a chip to see the absolute temperature,
but it looks like ITE doesn't handle it that way. The only registers
I can find are the temperature offset registers (0x56, 0x57, 0x59)
in the published datasheets.

> You can find my current version of the driver here: Zlika@58e217f <https://github.com/Zlika/it87/commit/58e217f0124afeca2af47d5a282d73c6e531209c>
>

Looks good. Are you sure that the chip only supports three fans ?

Thanks,
Guenter

Zlika, 2015-02-12:
Yes it's temp2 that has an offset of 92.
Coretemp reports 86 as high and 92 as crit temperatures, so your explanation seems good!
I'm not sure that the chip only supports 3 fans. The BIOS only reports 2 fans, the cpu fan (fan1) and a system fan at 0RPM (this is because there is in fact no system fan on my computer. It is an industrial PC with front fans managed by a different system, so only the cpu fan is connected to the motherboard).

groeck, 2015-02-12:
On 02/11/2015 11:16 PM, Zlika wrote:
> Yes it's temp2 that has an offset of 92.
> Coretemp reports 86 as high and 92 as crit temperatures, so your explanation seems good!
> I'm not sure that the chip only supports 3 fans. The BIOS only [...]

Try setting bit 4..5 of register 0x0c and read back the result.
If you can not set the bits, you'll know for sure.
Also, try setting bit 0..2 to see if 16-bit fan controls need to be
enabled in the first place (probably not).

Thanks,
Guenter

groeck, 2015-02-12:
On Wed, Feb 11, 2015 at 11:16:27PM -0800, Zlika wrote:
> Yes it's temp2 that has an offset of 92.
> Coretemp reports 86 as high and 92 as crit temperatures, so your explanation seems good!
> I'm not sure that the chip only supports 3 fans. The BIOS only [...]

Can you rebase your code to the latest version of my driver on github ?
I added a couple of feature flags to simplify adding new chips.

It is very likely that your chip only supports three fans; it seems to be
mostly used in industrial PCs which don't need more. Either case it is
probably safer to assume three fans for now (and 16-bit fan counters). 

Thanks,
Guenter

closed 2016-12-28 by groeck