groeck/it87 #4: 3.3v goes crazy and no 12B on it8620

ktsaou, 2015-10-03:
Hi,

I have a gigabyte [GA-J1900N-D3V](http://www.gigabyte.com/products/product-page.aspx?pid=4918#ov) motherboard, running gentoo with kernel 3.18.11.

So some reason the +3.3V goes crazy. It goes for 1.2V to 5V within seconds:

[image]

Also, there is no 12V in any input. Here in text:

```txt
it8620-isa-0a40
Adapter: ISA adapter
in0:          +0.76 V  (min =  +0.46 V, max =  +1.32 V)
in1:          +2.05 V  (min =  +0.10 V, max =  +0.07 V)  ALARM
in2:          +2.06 V  (min =  +0.88 V, max =  +2.23 V)
in3:          +1.85 V  (min =  +0.76 V, max =  +1.98 V)
in4:          +0.86 V  (min =  +2.80 V, max =  +0.16 V)  ALARM
in5:          +2.03 V  (min =  +2.11 V, max =  +2.34 V)  ALARM
in6:          +1.36 V  (min =  +1.69 V, max =  +2.86 V)  ALARM
3VSB:         +3.38 V  (min =  +4.03 V, max =  +3.50 V)  ALARM
Vbat:         +3.07 V
+3.3V:        +5.23 V
fan1:        1730 RPM  (min =   18 RPM)
fan2:           0 RPM  (min =   11 RPM)  ALARM
fan3:           0 RPM  (min =   11 RPM)  ALARM
temp1:        +34.0°C  (low  = -91.0°C, high = -114.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = -102.0°C, high = -84.0°C)  sensor = disabled
temp3:        -55.0°C  (low  = +15.0°C, high = +13.0°C)  sensor = disabled
temp4:        +45.0°C
temp5:        +36.0°C
temp6:        +45.0°C
intrusion0:  ALARM
```

Also, the min and max voltage values seem wrong (check in1, in4, 3VSB).

Is it a bad hardware on my side, or the module does not read the values properly?

ktsaou, 2015-10-04:
Well, i copied `it87.c` from the kernel git source tree and now I get this output:

```txt
it8620-isa-0a40
Adapter: ISA adapter
in0:          +0.74 V  (min =  +0.46 V, max =  +1.32 V)
in1:          +2.05 V  (min =  +0.10 V, max =  +0.07 V)  ALARM
in2:          +2.06 V  (min =  +0.88 V, max =  +2.23 V)
in3:          +1.85 V  (min =  +0.76 V, max =  +1.98 V)
in4:          +0.86 V  (min =  +2.80 V, max =  +0.16 V)  ALARM
in5:          +2.03 V  (min =  +2.11 V, max =  +2.34 V)  ALARM
in6:          +1.36 V  (min =  +1.69 V, max =  +2.86 V)  ALARM
3VSB:         +3.38 V  (min =  +4.03 V, max =  +3.50 V)  ALARM
Vbat:         +3.07 V
fan1:        1726 RPM  (min =   18 RPM)
fan2:           0 RPM  (min =   11 RPM)  ALARM
fan3:           0 RPM  (min =   11 RPM)  ALARM
fan6:           0 RPM  (min =   -1 RPM)  ALARM
temp1:        +34.0°C  (low  = -91.0°C, high = -114.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = -102.0°C, high = -84.0°C)  sensor = disabled
temp3:        -55.0°C  (low  = +15.0°C, high = +13.0°C)  sensor = disabled
intrusion0:  ALARM
```

It seems they have disabled `in9` which was giving crazy values.
`min` and `max` values still seem wrong.
`12V` is still not there.

groeck, 2015-10-04:
On 10/04/2015 12:13 AM, Costa Tsaousis wrote:
> Well, i copied |it87.c| from the kernel git source tree and now I get this output:
>
> |it8620-isa-0a40 Adapter: ISA adapter in0: +0.74 V (min = +0.46 V, max = +1.32 V) in1: +2.05 V (min = +0.10 V, max = +0.07 V) ALARM in2: +2.06 V (min = +0.88 V, max = +2.23 V) in3: +1.85 V (min = +0.76 V, max = +1.98 V) in4: +0.86 V (min = +2.80 V, max = +0.16 V) ALARM in5: +2.03 V (min = +2.11 V, max = +2.34 V) ALARM in6: +1.36 V (min = +1.69 V, max = +2.86 V) ALARM 3VSB: +3.38 V (min = +4.03 V, max = +3.50 V) ALARM Vbat: +3.07 V fan1: 1726 RPM (min = 18 RPM) fan2: 0 RPM (min = 11 RPM) ALARM fan3: 0 RPM (min = 11 RPM) ALARM fan6: 0 RPM (min = -1 RPM) ALARM temp1: +34.0°C (low = -91.0°C, high = -114.0°C) sensor = thermistor temp2: -55.0°C (low = -102.0°C, high = -84.0°C) sensor = disabled temp3: -55.0°C (low = +15.0°C, high = +13.0°C) sensor = disabled intrusion0: ALARM |
>
> It seems they have disabled |in9| which was giving crazy values.

I did that because it didn't work for me either. Reason unknown.

> |min| and |max| values still seem wrong.

Probably not set. Did you try ?

> |12V| is still not there.
>

That is what sensors3.conf is for, to let you enter scaling factors.
The ADC can not measure 12V directly. lm-sensors.org used to have
directions on how to calculate scaling factors, but unfortunately
it is down.

Guenter

ktsaou, 2015-10-04:
Hi,

> Probably not set. Did you try ?

Try what?

> lm-sensors.org used to have
> directions on how to calculate scaling factors, but unfortunately
> it is down.

Yes, I noticed that too.

groeck, 2015-10-04:
> Hi,
>
>     Probably not set. Did you try ?
>
> Try what?
To set the limits.

Guenter

closed 2016-12-28 by groeck

singulary, 2017-03-16:
Here is my configuration file. Works fine although it might be off a little bit. 

```
~$  cat /etc/sensors.d/it8620 
chip "it8620-isa-*"
### Voltages

    label  in0  "Vcore"
    label  in1  "+3.3V"
    label  in2  "+12V"
    label  in3  "+5V"     # Not in BIOS
    label  in4  "CPU_VAXG"
    label  in5  "-12V"
    label  in6  "DDR1_35VIO"
    label temp1       "M/B `Temp"`

    compute in1  1.635*@,@/1.635
    compute in2  @ * 6.0, @ / 6.0
    compute in3  2.7 * @, @ / 2.7
    compute in5 (7.67 * @) - 27 , (@ + 27) / 7.67  #used to be 27.3

~$  sensors
it8620-isa-0a40
Adapter: ISA adapter
Vcore:        +0.72 V  (min =  +0.40 V, max =  +0.60 V)  ALARM
+3.3V:        +3.37 V  (min =  +2.83 V, max =  +2.65 V)  ALARM
+12V:        +12.10 V  (min =  +6.26 V, max =  +2.88 V)  ALARM
+5V:          +5.05 V  (min =  +0.87 V, max =  +0.65 V)  ALARM
CPU_VAXG:     +1.04 V  (min =  +1.27 V, max =  +1.15 V)  ALARM
-12V:        -11.26 V  (min = -26.36 V, max = -14.21 V)  ALARM
DDR1_35VIO:   +1.36 V  (min =  +0.44 V, max =  +2.54 V)
3VSB:         +3.38 V  (min =  +4.46 V, max =  +5.02 V)  ALARM
Vbat:         +3.10 V  
fan1:           0 RPM  (min =   16 RPM)  ALARM
fan2:           0 RPM  (min =   23 RPM)  ALARM
fan3:           0 RPM  (min =   17 RPM)  ALARM
M/B Temp:     +47.0°C  (low  =  -8.0°C, high = +37.0°C)  sensor = thermistor
temp2:       +127.0°C  (low  = +122.0°C, high =  +2.0°C)  sensor = disabled
temp3:       +127.0°C  (low  = -36.0°C, high = -32.0°C)  sensor = disabled
temp4:        +45.0°C  
temp5:        +54.0°C  
temp6:        +46.0°C  
intrusion0:  ALARM

acpitz-virtual-0
Adapter: Virtual device
temp1:        +26.8°C  (crit = +100.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Core 0:       +59.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:       +59.0°C  (high = +105.0°C, crit = +105.0°C)
Core 2:       +62.0°C  (high = +105.0°C, crit = +105.0°C)
Core 3:       +62.0°C  (high = +105.0°C, crit = +105.0°C)

```