groeck/it87 #5: Support for IT8628E

pierretom, 2016-01-15:
Hello,

I would like to read temperatures and fan speed from an IT8628E Super I/O chip but 
it's not supported by the it87 driver.
I also contacted ITE to ask them a datasheet but without success.

Here a snipet of what sensors-detect found:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8628
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 8086:a123 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.
```

Output of `isadump 0x295 0x296`:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
Output of `isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 4`:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 28 02 09 00 80 37 00 00 00 80 50 88 80 00 00 
30: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 40 0a 30 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 02 00 00 04 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
f0: 00 40 00 00 00 00 f0 00 03 48 01 02 00 00 00 00
```
I don't have knowledge to reverse it but I'm ready to provide more information
if needed.

pierretom.

groeck, 2016-01-15:
Tricky without datasheet. You would have to open a case with Gigabyte support
(assuming your board is from Gigabyte).

You could try to load the driver and force the ID to 0x8620 or 0x8623.
The chips are supposedly quite similar.

Guenter

pierretom, 2016-01-16:
Thanks for help.
Yes it's a Gigabyte mobo.
I used ID 0x8620 and 0x8623, fan speed seems correct but voltages and temperatures
are weird if compared with values from the BIOS.
Is temp1 and temp3 should not be the same?

Values provided by BIOS program:

Name | Value
---------- | -----------
CPU Vcore | 1.212 V
CPU VCCSA | 1.068 V
DRAM Channel A/B voltage | 1.512 V
+3,3 V | 3.363 V
+5V | 5.070 V
+12V | 12.240 V
CPU VAXG | 0.012 V
CPU Temperature | 32C
System Temperature | 30C
CPU Fan Speed | 720 RPM
1st System Fan Speed | 322 RPM

Snipet of `sensors` output:
```
it8620-isa-0a40
Adapter: ISA adapter
in0:          +0.36 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.00 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.07 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.51 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.36 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.07 V  
+3.3V:        +0.58 V  
fan1:         646 RPM  (min =    0 RPM)
fan2:         322 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:       +127.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +27.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
temp4:        +86.0°C  
temp5:        +35.0°C  
temp6:        +32.0°C  
intrusion0:  ALARM
```
```
it8603-isa-0a40
Adapter: ISA adapter
in0:          +0.01 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.00 V  (min =  +0.00 V, max =  +3.06 V)  ALARM
3VSB:         +3.36 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.07 V  
+3.3V:        +0.36 V  
fan1:         624 RPM  (min =    0 RPM)
fan2:         322 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:       +127.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +27.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
intrusion0:  ALARM
```

groeck, 2016-01-18:
Can you tell me the exact motherboard model ?

Thanks,
Guenter

groeck, 2016-01-18:
Hi,

> Thanks for help.
> Yes it's a Gigabyte mobo.
> I used ID 0x8620 and 0x8623, fan speed seems correct but voltages and temperatures

Use 0x8620; chip registers are the same.

> are weird if compared with values from the BIOS.

Not really. Numbers need to be scaled with /etc/sensors3.conf.

> Is temp1 and temp3 should not be the same?
Not sure I understand the question. I would expect them to be different.

> Values provided by BIOS program:
>
> Name | Value
> ---------- | -----------
> CPU Vcore | 1.212 V
> CPU VCCSA | 1.068 V
> DRAM Channel A/B voltage | 1.512 V
> +3,3 V | 3.363 V
> +5V | 5.070 V
> +12V | 12.240 V
> CPU [...]

Usually Vcore

> in1:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
> in2:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
> in3:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)

One of those is +5V, one is 12V. One uses a scaling factor of
2.5, the other uses a scaling factor of 6.

Look at the raw voltages in /sys/class/hwmon/hwmonX/inX_input
and see if you can get a closer match to the BIOS voltages,
taking the scaling factors into account.

> in4:          +0.00 V  (min =  +0.00 V, max =  +3.06 V)

VAXG maybe ? Raw values might help.

> in5:          +1.07 V  (min =  +0.00 V, max =  +3.06 V)

VCCSA ?

> in6:          +1.51 V  (min =  +0.00 V, max =  +3.06 V)

DRAM Channel A/B

> 3VSB:         +3.36 V  (min =  +0.00 V, max =  +6.12 V)
> Vbat:         +3.07 V  
> +3.3V:        +0.58 V  

Ignore this one; it is wrong. I'll have to track that down
and maybe drop it.

> fan1:         646 RPM  (min =    0 RPM)
> fan2:         322 RPM  (min =    0 RPM)
> fan3:           0 RPM  (min =    0 RPM)
> fan4:           0 RPM  (min =    0 RPM)
> fan5:           0 RPM  (min =    0 RPM)
> temp1:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor

Board temperature ?

> temp2:       +127.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor

Not connected

> temp3:        +27.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI

CPU

> temp4:        +86.0°C  

Questionable. Probably not connected.

> temp5:        +35.0°C  
> temp6:        +32.0°C  

Might be various board temperatures. Do they change ?

pierretom, 2016-01-18:
>are weird if compared with values from the BIOS.
>>Not really. Numbers need to be scaled with /etc/sensors3.conf.

Ah, I didn't know that.

>Look at the raw voltages in /sys/class/hwmon/hwmonX/inX_input 
>and see if you can get a closer match to the BIOS voltages, 
>taking the scaling factors into account.

Ouput of `sensors -u it8620-isa-0a40 | grep input`:
```
  in0_input: 0.012
  in1_input: 2.040
  in2_input: 2.040
  in3_input: 2.040
  in4_input: 0.000
  in5_input: 1.068
  in6_input: 1.512
  in7_input: 3.360
  in8_input: 3.120
  in9_input: 0.120
  fan1_input: 606.000
  fan2_input: 307.000
  fan3_input: 0.000
  fan4_input: 0.000
  fan5_input: 0.000
  temp1_input: 30.000
  temp2_input: 127.000
  temp3_input: 27.000
  temp4_input: 86.000
  temp5_input: 35.000
  temp6_input: 32.000
```

>temp1:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
>>Board temperature ?

I think also.

>temp2:       +127.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
>>Not connected

No variations observed here.

>temp3:        +27.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
>>CPU

Yes.

>temp4:        +86.0°C
>>Questionable. Probably not connected.

No variations observed here.

>temp5:        +35.0°C
>temp6:        +32.0°C
>>Might be various board temperatures. Do they change ?

Yes, both vary. Overloading the cpu for one minute, with `stress` cause 
the increase of temp3, temp5 (+15°C) and temp6 (+6°C).
One can be the temperature of the integrated GPU?

groeck, 2016-01-18:
On Mon, Jan 18, 2016 at 11:56:15AM -0800, pierretom wrote:
> Ah, I didn't know that.
>
>
> Ouput of `sensors -u it8620-isa-0a40 | grep input`:
> ```
>   in0_input: 0.012

Maybe this is VAXG, then (whatever that is ;-)

>   in1_input: 2.040
>   in2_input: 2.040
>   in3_input: 2.040

No clear correlation here. Does your system have a second Super-IO chip ?
Also, if you monitor those over time, are there any changes ?

>   in4_input: 0.000

Looks like not connected to anything.

>   in5_input: 1.068
>   in6_input: 1.512
>   in7_input: 3.360
>   in8_input: 3.120
>   in9_input: 0.120

Not clear on this one. It is supposed to be AVCC per datasheet,
but the readings don't make sense. I have seen the same problem
with IT8620E.

I don't see a clear correlation to Vcore and VCCSA, nor to +5V and +12V.

You could try to check if there are any voltage changes if you put
the system under load. That could help with Vcore and VCCSA.

> fan1_input: 606.000
>   fan2_input: 307.000
>   fan3_input: 0.000
>   fan4_input: 0.000
>   fan5_input: 0.000
>   temp1_input: 30.000
>   temp2_input: 127.000
>   temp3_input: 27.000
>   temp4_input: 86.000 [...]

temp3 is the CPU (PECI how the CPU reports its temperature to the
Super-IO chip).

Most likely temp5 and temp6 are board temperature sensors.
Per datasheet, they are supposed to be connected to thermistors.
Anything else hard to say without schematics.

Guenter

i-frost, 2016-02-08:
Hi

I have a 6700K which I am nervous about cooking so I just wanted to check if there was any update regarding these mysterious readings.

I tried using "modprobe it87 force_id=0x8728" as suggested in this page at the bottom [https://wiki.archlinux.org/index.php/lm_sensors](https://wiki.archlinux.org/index.php/lm_sensors)

I get the roughly the same results as pierretom did. 

Just want to check if the value of in0 is correct/trusted for the Vcore temp?

>  it8620-isa-0a40
>  Adapter: ISA adapter
>  in0:          +0.36 V  (min =  +0.00 V, max =  +3.06 V)
> Usually Vcore

Thanks

groeck, 2016-02-09:
On 02/08/2016 05:06 AM, i-frost wrote:
> Hi
>
> I have a 6700K which I am nervous about cooking so I just wanted to check if there was any update regarding these mysterious readings.
>
> I tried using "modprobe it87 force_id=0x8728" as suggested [...]

It may or may not be VCORE. It depends on a power on strapping option (JP4, pin 27).
Only way to find out, as far as I know, is to monitor voltages on the pin under
various loads and compare with voltages displayed in the BIOS.

What do you get for in1 and in4 ?

Guenter

pierretom, 2016-02-12:
Hi,

Sorry for the delay.

>>in1_input: 2.040
>>in2_input: 2.040
>>in3_input: 2.040

>No clear correlation here. Does your system have a second Super-IO chip ?

I searched on the board and I found only one ITE chip. I noticed that there are
also two Nuvoton chips but I don't know if it's Super I/O.
One is referenced as NUVOTON 3103S and I can't read for the second. 8-pin both,
pin 1 referenced respectively as "MAU1" and "FABU1".

>Also, if you monitor those over time, are there any changes ?

No variations observed here, even with the last commit.

>>in5_input: 1.068
>>in6_input: 1.512
>>in7_input: 3.360
>>in8_input: 3.120
>>in9_input: 0.120

>You could try to check if there are any voltage changes if you put
>the system under load. That could help with Vcore and VCCSA.

in5, in7 and in8: no variations.
in6: has a new value of 1.500 and nothing else.
in9: vary from 3.120 to 4.560 roughly.

Values provided with the last commit (c8a5344):
Ouput of `sensors it8628-isa-0a40`:
```
it8628-isa-0a40
in0:          +0.73 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.01 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.07 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.51 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.36 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.12 V  
fan1:         585 RPM  (min =    0 RPM)
fan2:         319 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:       +127.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +25.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
intrusion0:  ALARM
```

Ouput of `sensors -u it8628-isa-0a40 | grep input`:
```
  in0_input: 0.324
  in1_input: 2.040
  in2_input: 2.040
  in3_input: 2.040
  in4_input: 0.000
  in5_input: 1.068
  in6_input: 1.512
  in7_input: 3.360
  in8_input: 3.120
  fan1_input: 582.000
  fan2_input: 320.000
  fan3_input: 0.000
  fan4_input: 0.000
  fan5_input: 0.000
  temp1_input: 30.000
  temp2_input: 127.000
  temp3_input: 25.000
```

i-frost, 2016-02-12:
HI

Thanks for the replies!

This is what I get at idle

acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

it8728-isa-0a40
Adapter: ISA adapter
in0:          +0.82 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +1.98 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.01 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.31 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.38 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.38 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.07 V  
fan1:         536 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:         475 RPM  (min =    0 RPM)
fan4:         491 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +25.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +20.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
intrusion0:  ALARM

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +20.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:         +17.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:         +19.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:         +19.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:         +17.0°C  (high = +80.0°C, crit = +100.0°C)

//////////////////////////////////////////////////////////////////////////////////////////////////
This under load

acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

it8728-isa-0a40
Adapter: ISA adapter
in0:          +1.31 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +1.98 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.02 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.31 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.38 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.38 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.07 V  
fan1:        1024 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:         475 RPM  (min =    0 RPM)
fan4:         501 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +24.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
intrusion0:  ALARM

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +55.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:         +54.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:         +55.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:         +54.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:         +55.0°C  (high = +80.0°C, crit = +100.0°C)

So I think its the correct reading as I am not over-clocked at the moment and that is what the bios says the voltage would be un-clocked.

When I was testing the other night over-clocking I did get a higher reading on the  in0_input than the current stock 1.31V don't remember exactly the reading but it wasn't to much higher when clocked at  4.6 GHz.

Thanks very much for your help guys.

Now I just got to figure out how to use the inbuilt sound card, if you have ideas...

:-)

xdarklight, 2016-04-06:
I'd like to say thanks to everyone who helped getting IT8628E supported :+1: 

Here's the output of sensors -u on my "GIGABYTE GA-Z170X-Gaming 7" in case more values are needed:
```
it8628-isa-0a40
Adapter: ISA adapter
Vcore:
  in0_input: 0.240
  in0_min: 0.552
  in0_max: 1.404
  in0_alarm: 0.000
  in0_beep: 0.000
in1:
  in1_input: 2.052
  in1_min: 0.000
  in1_max: 3.060
  in1_alarm: 0.000
  in1_beep: 0.000
+12V:
  in2_input: 12.240
  in2_min: 11.376
  in2_max: 12.600
  in2_alarm: 0.000
  in2_beep: 0.000
+5V:
  in3_input: 5.010
  in3_min: 4.740
  in3_max: 5.250
  in3_alarm: 0.000
  in3_beep: 0.000
VAXG:
  in4_input: 0.012
  in4_min: 0.000
  in4_max: 3.060
  in4_alarm: 0.000
  in4_beep: 0.000
VCCSA:
  in5_input: 1.068
  in5_min: 1.056
  in5_max: 1.104
  in5_alarm: 0.000
  in5_beep: 0.000
DRAM:
  in6_input: 1.200
  in6_min: 1.140
  in6_max: 1.260
  in6_alarm: 0.000
  in6_beep: 0.000
3VSB:
  in7_input: 3.360
  in7_min: 3.144
  in7_max: 3.456
  in7_alarm: 0.000
  in7_beep: 0.000
Vbat:
  in8_input: 3.048
CPU Fan:
  fan1_input: 714.000
  fan1_min: 10.000
  fan1_alarm: 0.000
  fan1_beep: 1.000
CPU Opt Fan:
  fan2_input: 557.000
  fan2_min: 10.000
  fan2_alarm: 0.000
  fan2_beep: 1.000
System Fan 1:
  fan3_input: 0.000
  fan3_min: 0.000
  fan3_alarm: 0.000
  fan3_beep: 1.000
System Fan 2:
  fan4_input: 508.000
  fan4_min: 0.000
  fan4_alarm: 0.000
  fan4_beep: 1.000
System Fan 3:
  fan5_input: 410.000
  fan5_min: 0.000
  fan5_alarm: 0.000
  fan5_beep: 1.000
System Temp:
  temp1_input: 32.000
  temp1_max: 70.000
  temp1_min: 0.000
  temp1_alarm: 0.000
  temp1_type: 4.000
  temp1_offset: 0.000
  temp1_beep: 1.000
PCH Temp:
  temp2_input: 46.000
  temp2_max: 70.000
  temp2_min: 0.000
  temp2_alarm: 0.000
  temp2_type: 4.000
  temp2_offset: 0.000
  temp2_beep: 1.000
CPU Temp:
  temp3_input: 25.000
  temp3_max: 70.000
  temp3_min: 0.000
  temp3_alarm: 0.000
  temp3_type: 6.000
  temp3_offset: 100.000
  temp3_beep: 1.000
```

This is the corresponding sensors3.conf section:
```
chip "it8628-*"
    set in0_min 0.55
    set in0_max 1.4

    set in2_min 12.0 * 0.95
    set in2_max 12.0 * 1.05
    compute in2 6*@ ,  @/6

    set in3_min 5.0 * 0.95
    set in3_max 5.0 * 1.05
    compute in3 2.5*@ ,  @/2.5

    set in5_min 1.05
    set in5_max 1.10

    set in6_min 1.14
    set in6_max 1.26

    set in7_min 3.3 * 0.95
    set in7_max 3.3 * 1.05

    label in0 "Vcore"
    label in1 "in1"
    label in2 "+12V"
    label in3 "+5V"
    label in4 "VAXG"
    label in5 "VCCSA"
    label in6 "DRAM"
    label in8 "Vbat"

    label fan1 "CPU Fan"
    label fan2 "CPU Opt Fan"
    label fan3 "System Fan 1"
    label fan4 "System Fan 2"
    label fan5 "System Fan 3"

    label temp1 "System Temp"
    label temp2 "PCH Temp"
    label temp3 "CPU Temp"

    ignore intrusion0
```

I am not 100% sure with in1-3 but in2 matches BIOS' +12V best and in3 matches BIOS' +5V best (using the correct multipliers).
Unfortunately I have no idea what in1 could be.

groeck, 2016-12-28:
Nothing else we can do here.

closed 2016-12-28 by groeck

jwwalton, 2017-04-27:
Not sure if reopening this is the right thing to do but I was wondering if pwm4 is supposed to be adjustable? It is not shown in sysfs.

```
Linux foobar 4.10.12-1 #1 SMP PREEMPT Tue Apr 25 01:28:33 CEST 2017 x86_64 GNU/Linux`
```

```
[    1.862998] it87: Found IT8628E chip at 0xa40, revision 2
[    1.863072] it87: Beeping is supported
```

```
/sys/devices/platform/it87.2624/hwmon/hwmon1 # grep ".*" *

alarms:16
grep: device: Is a directory
fan1_alarm:0
fan1_beep:1
fan1_input:396
fan1_min:10
fan2_alarm:0
fan2_beep:1
fan2_input:614
fan2_min:10
fan3_alarm:0
fan3_beep:1
fan3_input:564
fan3_min:10
fan4_alarm:0
fan4_beep:1
fan4_input:187
fan4_min:10
fan5_alarm:0
fan5_beep:1
fan5_input:472
fan5_min:10
in0_alarm:0
in0_beep:0
in0_input:1296
in0_max:3060
in0_min:0
in1_alarm:0
in1_beep:0
in1_input:2028
in1_max:3060
in1_min:0
in2_alarm:0
in2_beep:0
in2_input:2028
in2_max:3060
in2_min:0
in3_alarm:0
in3_beep:0
in3_input:2028
in3_max:3060
in3_min:0
in4_alarm:0
in4_beep:0
in4_input:12
in4_max:3060
in4_min:0
in5_alarm:0
in5_beep:0
in5_input:1068
in5_max:3060
in5_min:0
in6_alarm:0
in6_beep:0
in6_input:1200
in6_max:3060
in6_min:0
in7_alarm:0
in7_beep:0
in7_input:3384
in7_label:3VSB
in7_max:6120
in7_min:0
in8_input:3048
in8_label:Vbat
intrusion0_alarm:1
name:it8628
grep: power: Is a directory
pwm1:70
pwm1_auto_channels_temp:4
pwm1_auto_point1_temp:0
pwm1_auto_point1_temp_hyst:-3000
pwm1_auto_point2_temp:20000
pwm1_auto_point3_temp:65000
pwm1_auto_slope:28
pwm1_auto_start:70
pwm1_enable:2
pwm1_freq:23437
pwm2:70
pwm2_auto_channels_temp:4
pwm2_auto_point1_temp:0
pwm2_auto_point1_temp_hyst:-3000
pwm2_auto_point2_temp:20000
pwm2_auto_point3_temp:65000
pwm2_auto_slope:40
pwm2_auto_start:70
pwm2_enable:2
pwm2_freq:23437
pwm3:70
pwm3_auto_channels_temp:1
pwm3_auto_point1_temp:0
pwm3_auto_point1_temp_hyst:-3000
pwm3_auto_point2_temp:20000
pwm3_auto_point3_temp:50000
pwm3_auto_slope:28
pwm3_auto_start:70
pwm3_enable:2
pwm3_freq:23437
pwm5:70
pwm5_auto_channels_temp:8
pwm5_auto_point1_temp:0
pwm5_auto_point1_temp_hyst:-3000
pwm5_auto_point2_temp:20000
pwm5_auto_point3_temp:50000
pwm5_auto_slope:28
pwm5_enable:2
pwm5_freq:23437
grep: subsystem: Is a directory
temp1_alarm:0
temp1_beep:1
temp1_input:34000
temp1_max:127000
temp1_min:127000
temp1_offset:0
temp1_type:4
temp2_alarm:0
temp2_beep:1
temp2_input:-8000
temp2_max:127000
temp2_min:127000
temp2_offset:0
temp2_type:4
temp3_alarm:0
temp3_beep:1
temp3_input:29000
temp3_max:127000
temp3_min:127000
temp3_offset:100000
temp3_type:6
temp4_input:34000
temp5_input:36000
temp6_input:32000
```

groeck, 2017-04-27:
@jwwalton: Only if the output pin is configured accordingly. SIO register 0x28, bit 2, must be set to 0.

dhjw, 2017-07-16:
Does anyone know how to do this on Ubuntu 16.04? I want to set the LEDs for my Z170X-UD3 motherboard and get this in the kernel log when doing modprobe gpio-it87:
`Jul 16 05:02:37 rig1 kernel: [191319.115105] gpio_it87: Unknown Chip found, Chip 8628 Revision 2`

groeck, 2017-07-16:
Either install a more recent kernel (support for 8628 was added in March 2016), or extract the latest gpio-it87 driver from the kernel source and install it.

DeviousPenguin, 2017-12-14:
Hi,
I've added the temperature readings and fan speed reading to my GIGABYTE GA-Z170X-Gaming 5 motherboard, the instructions for the 'gaming 7' worked perfectly for me. thanks for adding your config file. 

I have 4 fans connected and I can read the speed correctly, would it be possible to adjust the speed somehow? Do I need to write values to certain devices it is there something similar to speedfan on Windows where the fan speeds can be automated.

groeck, 2017-12-14:
On Thu, Dec 14, 2017 at 06:15:15PM +0000, DeviousPenguin wrote:
> Hi,
> I've added the temperature readings and fan speed reading to my GIGABYTE GA-Z170X-Gaming 5 motherboard, the instructions for the 'gaming 7' worked perfectly for me. thanks for adding your config [...]
You can see various parameters in /sys/class/hwmon/hwmonX/pwmX*.
For example, on my system:

pwm1:70
pwm1_auto_channels_temp:1
pwm1_auto_point1_temp:0
pwm1_auto_point1_temp_hyst:-3000
pwm1_auto_point2_temp:20000
pwm1_auto_point3_temp:65000
pwm1_auto_slope:28
pwm1_auto_start:70
pwm1_enable:1
pwm1_freq:23437

pwm2:70
pwm2_auto_channels_temp:1
pwm2_auto_point1_temp:0
pwm2_auto_point1_temp_hyst:-3000
pwm2_auto_point2_temp:20000
pwm2_auto_point3_temp:65000
pwm2_auto_slope:28
pwm2_auto_start:70
pwm2_enable:1
pwm2_freq:187500

and so on. Most of the values can be adjusted.

Fan control should already be automatic and is usually controlled
from the BIOS. I would suggest to reconfigure the fans from there
if possible.

Guenter

DeviousPenguin, 2017-12-15:
Yes I'm going to be leave the CPU fan speed to be controlled the BIOS. I want to automate the speed of my system fans depending on GPU temperature, there are 5 fan headers on my system:

    $ find /sys/class/hwmon/hwmon2/pwm?
    /sys/class/hwmon/hwmon2/pwm1
    /sys/class/hwmon/hwmon2/pwm2
    /sys/class/hwmon/hwmon2/pwm3
    /sys/class/hwmon/hwmon2/pwm4
    /sys/class/hwmon/hwmon2/pwm5

I did try to set all set a speed of pwm3 to 50% for example

    $ echo 127 | sudo tee /sys/class/hwmon/hwmon2/pwm3
    tee: pwm3: Device or resource busy

It turns out I need to change the pwm_enable setting also:

    grep ".*" /sys/class/hwmon/hwmon2/pwm?_enable
    /sys/class/hwmon/hwmon2/pwm1_enable:2
    /sys/class/hwmon/hwmon2/pwm2_enable:2
    /sys/class/hwmon/hwmon2/pwm3_enable:2
    /sys/class/hwmon/hwmon2/pwm4_enable:2
    /sys/class/hwmon/hwmon2/pwm5_enable:2

    $ cd /sys/class/hwmon/hwmon2/
    $ echo 1 | sudo tee pwm3_enable | sudo tee pwm4_enable | sudo tee pwm5_enable 
    $ grep ".*" /sys/class/hwmon/hwmon2/pwm?
    /sys/class/hwmon/hwmon2/pwm1_enable:2
    /sys/class/hwmon/hwmon2/pwm2_enable:2
    /sys/class/hwmon/hwmon2/pwm3_enable:1
    /sys/class/hwmon/hwmon2/pwm4_enable:1
    /sys/class/hwmon/hwmon2/pwm5_enable:1

Now I'm able to adjust fan speed:

    $ echo 127 | sudo tee /sys/class/hwmon/hwmon2/pwm3
    127
    $ sleep 1
    $ sensors | grep Fan
    CPU Fan:      1735 RPM  (min =   10 RPM)
    CPU Opt Fan:     0 RPM  (min =    0 RPM)
    System Fan 1:  632 RPM  (min =    0 RPM)
    System Fan 2: 1278 RPM  (min =    0 RPM)
    System Fan 3: 1305 RPM  (min =    0 RPM)

This works great on all 3 system fans. Now I just need to read the GPU temps and put together a bash script to automate it all. Thanks dude!

groeck, 2017-12-15:
If the GPU temperature is one of the temperatures reported by the SuperIO chip,
you might also try to set pwmX_auto_channels_temp to point to it and stick
with automatic fan control. Values 1..3 should be valid for your chip.

Guenter

DeviousPenguin, 2017-12-15:
It's shame that my SuperIO is not able to read the GPU temp:

    $ sensors | grep -i temp
    System Temp:   +28.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
    PCH Temp:      +37.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
    CPU Temp:      +32.0°C  (low  =  +0.0°C, high = +80.0°C)  sensor = Intel PECI
    temp4:         +27.0°C  
    temp5:         +29.0°C  
    temp6:         +29.0°C

Otherwise I would have used the pwmX_auto_channels_temp setting; It would still work even if my script fails or the OS crashes. 

I forgot to mention I'm using more than 1 GPU, but getting the temperatures of the GPU's is easy enough:

    $ nvidia-smi --format=csv --query-gpu=index,fan.speed,temperature.gpu,name
    index, fan.speed [%], temperature.gpu, name
    0, 70 %, 58, GeForce GTX 1070
    1, 49 %, 58, GeForce GTX 1070
    2, 62 %, 57, GeForce GTX 1070
    3, 73 %, 57, GeForce GTX 1070
    4, 72 %, 58, GeForce GTX 1060 6GB
    5, 59 %, 58, GeForce GTX 1060 6GB
    $ # Get temperature of GPU0
    $ nvidia-smi --format=csv,noheader --query-gpu=temperature.gpu -i 0
    58

Each System Fan is blowing air over 2 GPU's. I'm already using a Bash script to control the GPU fan speed, and now I'll be able also change system fan speed, it will help keep things cool enough without having all system fans set to 100%.

I realise this isn't your normal setup, just want to say thanks to everyone in this thread, the information is invaluable. I've included a safety net so that if my script fails and temperatures get too high, the GPU workload stops, and in the case of kernel panic the machine reboots with system fans set to 100% speed in the BIOS.