groeck/it87 #6: Support for IT8607E

hubiktomas, 2016-03-08:
Please add support for IT8607E chip - used in Intel NUC computers.

groeck, 2016-03-09:
On 03/08/2016 08:55 AM, hubiktomas wrote:
> Please add support for IT8607E chip - used in Intel NUC computers.
>

Per information available online, this is an Embedded Controller,
meaning the only way to add support for it would be through ACPI,
which in turn would require either someone reverse engineering Intel's
DSDT or Intel releasing the necessary information to enable someone
to write a driver. Both are highly unlikely to happen, unless Intel's
Open Source team decides to do the work.

Guenter

elfhack, 2016-04-03:
Not quite. I can see it via sensors-detect.
`Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8607
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No`
 I've tried forcing it87 to load by making it treat is as a 8603
`options it87 force_id=0x8603`
This gave me something to play around with. Here's the output of sensors, however I've since written my config, and it seems some limits stick even after removing the config file and rebooting?
`it8603-isa-0a40
Adapter: ISA adapter
in0:          +0.64 V  (min =  +0.55 V, max =  +1.52 V)
in1:          +1.21 V  (min =  +1.14 V, max =  +1.26 V)
in2:          +1.99 V  (min =  +1.28 V, max =  +2.60 V)
in3:          +2.02 V  (min =  +0.50 V, max =  +1.87 V)  ALARM
in4:          +2.02 V  (min =  +1.46 V, max =  +0.06 V)  ALARM
3VSB:         +3.31 V  (min =  +3.14 V, max =  +3.46 V)
Vbat:         +2.86 V  
+3.3V:        +3.31 V  
fan1:           0 RPM  (min =   27 RPM)  ALARM
fan2:        2636 RPM  (min = 1997 RPM)
temp1:        +48.0°C  (low  =  +0.0°C, high = +95.0°C)  sensor = thermal diode
temp2:        +44.0°C  (low  = +15.0°C, high = +90.0°C)  sensor = thermal diode
temp3:        +46.0°C  (low  = +35.0°C, high = +100.0°C)  sensor = disabled
intrusion0:  ALARM`

The output of isadump 0x295 0x296 are all ff's, but here's isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 4
`     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 07 02 00 00 00 f7 ca 22 00 00 48 03 00 00 00 
30: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 40 0a 30 00 00 00 00 00 00 00 00 00 00 00 00 
70: 09 02 00 00 04 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
f0: 01 00 00 00 60 32 ff 00 03 48 04 00 00 00 00 00`

elfhack, 2016-04-03:
Not quite. I can see it via sensors-detect.
```
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family 'National Semiconductor/ITE'...               No
Trying family 'SMSC'...                                     No
Trying family 'VIA/Winbond/Nuvoton/Fintek'...               No
Trying family 'ITE'...                                      Yes
Found unknown chip with ID 0x8607
Probing for Super-I/O at 0x4e/0x4f
Trying family 'National Semiconductor/ITE'...               No
Trying family 'SMSC'...                                     No
Trying family 'VIA/Winbond/Nuvoton/Fintek'...               No
Trying family 'ITE'...                                      No
```
 I've tried forcing it87 to load by making it treat is as a 8603 with `options it87 force_id=0x8603`, and this gave me something to play around with. Here's the output of sensors, however I've since written my config, and it seems some limits stick even after removing the config file and rebooting?

```
it8603-isa-0a40
Adapter: ISA adapter
in0:          +0.64 V  (min =  +0.55 V, max =  +1.52 V)
in1:          +1.21 V  (min =  +1.14 V, max =  +1.26 V)
in2:          +1.99 V  (min =  +1.28 V, max =  +2.60 V)
in3:          +2.02 V  (min =  +0.50 V, max =  +1.87 V)  ALARM
in4:          +2.02 V  (min =  +1.46 V, max =  +0.06 V)  ALARM
3VSB:         +3.31 V  (min =  +3.14 V, max =  +3.46 V)
Vbat:         +2.86 V  
+3.3V:        +3.31 V  
fan1:           0 RPM  (min =   27 RPM)  ALARM
fan2:        2636 RPM  (min = 1997 RPM)
temp1:        +48.0°C  (low  =  +0.0°C, high = +95.0°C)  sensor = thermal diode
temp2:        +44.0°C  (low  = +15.0°C, high = +90.0°C)  sensor = thermal diode
temp3:        +46.0°C  (low  = +35.0°C, high = +100.0°C)  sensor = disabled
intrusion0:  ALARM
```
The output of `isadump 0x295 0x296` are all ff's, but here's `isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 4`
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 07 02 00 00 00 f7 ca 22 00 00 48 03 00 00 00 
30: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 40 0a 30 00 00 00 00 00 00 00 00 00 00 00 00 
70: 09 02 00 00 04 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
f0: 01 00 00 00 60 32 ff 00 03 48 04 00 00 00 00 00
```
Here's my config for NUC6i5SYH which looks to be good from what the BIOS shows, but it's missing +5V and +12V rails. Those disabled inputs might be it, but it's hard to do guesswork when the voltages don't change
```
# Sensors3 conf for Nuc6i5
# Limits from http://www.intel.com/content/www/us/en/processors/core/6th-gen-core-family-mobile-u-y-processor-lines-datasheet-vol-1.html and https://www.micron.com/products/dram/ddr4-sdram
chip "it8603-isa-0a40"
#Disable unused stuff
ignore in2
ignore in3
ignore in4
ignore fan1
ignore intrusion0

label in0 Vcore
set in0_min 0.55
set in0_max 1.52
label in1 SDRAM
set in1_min 1.2 * 0.95
set in1_max 1.2 * 1.05
set in7_min 3.3 * 0.95
set in7_max 3.3 * 1.05
label fan2 "System Fan"
set fan2_min 2000
label temp1 "Memory Temp"
set temp1_min 0
set temp1_max 95
label temp2 "Motherboard Ambient Temp"
set temp2_min 15
set temp2_max 90
label temp3 "PCH Temp"
set temp3_min 35
set temp3_max 100
```

groeck, 2016-04-07:
Interesting. I don't know if the board has +5V - does it show up in the BIOS ? The manual only mentions +5V.

Not really sure how to proceed. I am inclined to just add a note to README suggesting the force_id solution. I don't know if there is a chance to get the datasheet for the chip, which would be needed for a proper port.

elfhack, 2016-04-07:
Yeah, it does have +5V. Here's a screenshot
[image]

groeck, 2016-04-08:
Sorry, I meant to say " I don't know if the board has +12V - does it show up in the BIOS ?". Guess the answer is no. 

Can you provide raw values for the sensors (output of sensors -u) ? +5V is probably in3 or in4, with a scaling factor of 2.5. Is it possible to see the thresholds ? That might help as well.

Also please try "isadump 0xa45 0xa46".

elfhack, 2016-04-08:
Sure

```
coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:
  temp1_input: 40.000
  temp1_max: 100.000
  temp1_crit: 100.000
  temp1_crit_alarm: 0.000
Core 0:
  temp2_input: 38.000
  temp2_max: 100.000
  temp2_crit: 100.000
  temp2_crit_alarm: 0.000
Core 1:
  temp3_input: 38.000
  temp3_max: 100.000
  temp3_crit: 100.000
  temp3_crit_alarm: 0.000

it8603-isa-0a40
Adapter: ISA adapter
in0:
  in0_input: 0.648
  in0_min: 0.552
  in0_max: 1.524
  in0_alarm: 0.000
  in0_beep: 0.000
in1:
  in1_input: 1.212
  in1_min: 1.140
  in1_max: 1.260
  in1_alarm: 0.000
  in1_beep: 0.000
in2:
  in2_input: 2.004
  in2_min: 1.284
  in2_max: 2.604
  in2_alarm: 0.000
  in2_beep: 0.000
in3:
  in3_input: 2.016
  in3_min: 0.504
  in3_max: 1.872
  in3_alarm: 1.000
  in3_beep: 0.000
in4:
  in4_input: 2.016
  in4_min: 1.464
  in4_max: 0.060
  in4_alarm: 1.000
  in4_beep: 0.000
3VSB:
  in7_input: 3.312
  in7_min: 3.144
  in7_max: 3.456
  in7_alarm: 0.000
  in7_beep: 0.000
Vbat:
  in8_input: 2.880
+3.3V:
  in9_input: 3.312
fan1:
  fan1_input: 0.000
  fan1_min: 27.000
  fan1_alarm: 1.000
  fan1_beep: 0.000
fan2:
  fan2_input: 2586.000
  fan2_min: 1997.000
  fan2_alarm: 0.000
  fan2_beep: 0.000
temp1:
  temp1_input: 40.000
  temp1_max: 95.000
  temp1_min: 0.000
  temp1_alarm: 0.000
  temp1_type: 3.000
  temp1_offset: 0.000
  temp1_beep: 0.000
temp2:
  temp2_input: 37.000
  temp2_max: 90.000
  temp2_min: 15.000
  temp2_alarm: 0.000
  temp2_type: 3.000
  temp2_offset: 0.000
  temp2_beep: 0.000
temp3:
  temp3_input: 40.000
  temp3_max: 100.000
  temp3_min: 35.000
  temp3_alarm: 0.000
  temp3_type: 0.000
  temp3_offset: 100.000
  temp3_beep: 0.000
intrusion0:
  intrusion0_alarm: 1.000
```
And the isadump
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 11 15 38 00 5f ff 07 00 00 80 64 0f 00 ff 06 ff 
10: 85 52 c2 7f c0 00 92 00 ff 01 ff 60 01 ab ab ab 
20: 36 65 a7 a8 a8 a2 87 8a 78 2a 25 27 28 2a 2a 8a 
30: 7f 2e 69 5f d9 6b 9c 2a 05 7a 22 ee 8f 26 90 83 
40: 5f 00 5a 0f 64 23 5f 40 ad 6a d4 00 ff ff ff ff 
50: ff 0b 7f 7f 7f 40 00 00 90 64 00 12 e0 00 00 00 
60: 7f 7f 7f 80 00 00 0f 80 81 46 7f 4c 0c 02 0f 4c 
70: 7f 7f 7f 80 00 00 0f 80 ff ff ff ff ff ff ff ff 
80: 00 00 00 00 00 00 00 00 02 30 01 02 01 e4 e0 e0 
90: 00 00 00 00 00 00 ff ff 02 97 00 00 40 5e c0 01 
a0: c3 1c 1c 1c 1c 1c 1c ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

groeck, 2016-12-29:
Please check ToT.

closed 2017-03-10 by groeck