groeck/it87 #7: Support for IT8587

Kurnihil, 2016-06-20:
Hi, I would like to add support for the EC IT8587, of course there isn't any datasheet available.

- I've found a Preliminary Specification datasheet for the IT8511, but i really don't know if it's even related with the IT8587, thoughts?
- I have the schematics of my computer, I've sliced the paged of the IT8587: [IT8587_Schematics.pdf](https://github.com/groeck/it87/files/322977/IT8587_Schematics.pdf) hoping it helps.

```
sensors-detect

Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x8587
```
```
isadump 0x295 0x296

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```
```
isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 4 

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 85 87 06 01 00 00 00 00 00 00 00 00 01 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 00 04 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
```
I'm willing to reverse engineering it and I would like to know if it's feasible to write a kernel module and poke around with it (without worrying to break something badly).
If you want I can upload the IT8511 datasheet.
Thanks

groeck, 2016-06-20:
IT85XX are usually embedded controllers, not Super-IO chips. This is definitely true for IT8511, and most likely for IT8587 as well. Access to such chips is microcode-dependent; it is not possible to write a single driver. A separate driver would have to be written for each system using this chip.

Your best chance would probably be an ACPI driver, but to write that one would need to know how to access the chip's functionality using ACPI. Unfortunately, that is typically not documented. You could extract the DSDT and try to understand it, but that takes a lot of time. On top of that, a driver would have to be written to actually use that information.

Kurnihil, 2016-06-20:
Thanks for the advices
Just to be sure: since it's an embedded controller, the Super-IO subsystem, that is there, is mostly controlled by the EC firmware itself and not available directly to the "outside".
I've understand it correctly?

Thanks again and sorry for the noob-ish questions.

DABH, 2016-09-07:
Total noob here, but I also have a system with an unsupported IT8587 (as of Linux kernel version 4.6.0).  Is there any way to write a driver that would work for my or @Kurnihil's particular systems, since it sounds like maybe an ACPI driver would be out of reach?  How would one go about doing that?  Do the schematics that @Kurnihil posted give any clues?  I'm also willing to mess around with this to try to get it to work.  Any hints/help would be greatly appreciated!

groeck, 2016-09-07:
On 09/06/2016 11:01 PM, DABH wrote:
> Total noob here, but I also have a system with an unsupported IT8587 (as of Linux kernel version 4.6.0). Is there any way to write a driver that would work for my or @Kurnihil [...]

Not a chance, sorry. As mentioned before, this is a microcontroller. Even if
the same chip is used in two different systems, that does not mean that the
microcode is the same. You could disassemble the DSDT and try to see if there
is anything useful in it. The schematics don't tell us how to program the chip
and are thus not very useful.

Guenter

closed 2016-12-28 by groeck