groeck/it87 #8: Support for ITE IT8733

Lou-Cipher, 2016-07-10:
Hi, I'm working with a Gigabyte G1 Gaming Z170X Gaming G1 (yes, real product name) motherboard, which uses an ITE IT8628 for the system fans, but an ITE IT8733 for the dual CPU fan headers.  Currently, there is no support for the 8733, and I was hoping it might be possible to add it.  I haven't the knowledge at present to reverse engineer it myself, nor is there a data sheet that I'm aware of.  Being a very expensive board, I want to avoid poking around with isadump without knowing what I'm doing, so for the time being, here's the sensors-detect output:

```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8628
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8733
```

luw, 2016-12-28:
hello I have the same issue on Lenovo ThinkCentre M93p motherboard.

```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): y
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8733
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

```

groeck, 2016-12-29:
Please try the tip of the tree and report.

T-Hacker, 2017-02-20:
Hi. I'm having the same problem. My motherboard info is:

```
    Manufacturer: Gigabyte Technology Co., Ltd.
    Product Name: X99-Designare EX-CF
    Version: x.x
    Serial Number: Default string
    Asset Tag: Default string
    Features:
        Board is a hosting board
        Board is replaceable
    Location In Chassis: Default string
    Chassis Handle: 0x0003
    Type: Motherboard
    Contained Object Handles: 0
```

Here is the partial output from the sensors-detect command:

```
Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found `ITE IT8620E Super IO Sensors'                        Success!
    (address 0xa40, driver `it87')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8733
```

Thank you for your time!

groeck, 2017-02-20:
Did anyone try the latest version of the driver here ?

Lou-Cipher, 2017-02-20:
Somehow I missed the notification of the update.  I'll give it a try tonight, if possible, and post an update here.  Thanks!

T-Hacker, 2017-02-21:
Fair warning, I'm a noob in the driver realm. You better wait for more feedback from other ppl. :)

I've compiled and installed the master branch of this repository on Feb 20.
 
When I run dmesg command, it appears to detect the chip:

```
[ 2777.190840] it87: Found IT8620E chip at 0xa40, revision 4
[ 2777.190865] it87: Beeping is supported
[ 2777.191214] it87: Found IT8733E chip at 0xa60, revision 3
[ 2777.191240] it87: Beeping is supported
```

But when I run sensors-detect, it still fails to detect the chip.

As a curiosity, the only applications on Windows that could detect this chip where: HWiNFO and the manufacturer software (Gigabyte). From the batch I tested.

groeck, 2017-02-21:
On Mon, Feb 20, 2017 at 04:44:47PM -0800, Pedro Gomes wrote:
> Fair warning, I'm a noob in the driver realm. You better wait for more feedback from other ppl. :)
>
> I've compiled and installed the master branch of this repository on Feb 20.
>
> When I run dmesg [...]
One is the kernel driver, the other an application which can be used
to detect chips in the system. They don't have anything to do with
each other.

Question is what the "sensors" comand has to say. It should now display
values for both chips.

Guenter

T-Hacker, 2017-02-21:
Now I'm really confused... 

```
it8792-isa-0a60
Adapter: ISA adapter
in0:          +1.88 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +1.48 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
+3.3V:        +1.67 V  (min =  +0.00 V, max =  +2.78 V)
in4:          +1.38 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.38 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +1.67 V  (min =  +0.00 V, max =  +2.78 V)
Vbat:         +1.67 V  
fan1:        1106 RPM  (min =   10 RPM)
fan2:        1558 RPM  (min =   10 RPM)
fan3:         658 RPM  (min =   10 RPM)
temp1:        +26.0°C  (low  =  +0.0°C, high = +70.0°C)  sensor = Intel PECI
temp2:        +29.0°C  (low  =  +0.0°C, high = +70.0°C)  sensor = thermistor
temp3:        +36.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +31.0°C  (high = +85.0°C, crit = +95.0°C)
Core 0:         +27.0°C  (high = +85.0°C, crit = +95.0°C)
Core 1:         +22.0°C  (high = +85.0°C, crit = +95.0°C)
Core 2:         +23.0°C  (high = +85.0°C, crit = +95.0°C)
Core 3:         +20.0°C  (high = +85.0°C, crit = +95.0°C)
Core 4:         +25.0°C  (high = +85.0°C, crit = +95.0°C)
Core 5:         +22.0°C  (high = +85.0°C, crit = +95.0°C)

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +38.0°C  

it8620-isa-0a40
Adapter: ISA adapter
in0:          +1.01 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.51 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +1.99 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +1.99 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.06 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.06 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +2.51 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.36 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.12 V  
fan1:         925 RPM  (min =   10 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +39.0°C  (low  =  +0.0°C, high = +70.0°C)  sensor = thermistor
temp2:        +36.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp4:        +34.0°C  
temp5:        +46.0°C  
temp6:        +46.0°C  
intrusion0:  ALARM
```

A new chip has been detected, named it8792. Could this be the elusive it8733?

Here's a [screen capture](https://goo.gl/photos/TtKSouQeuERWLVJi8) from HWiNFO where it clearly shows both chips, and the it8792 is named it8733 in that application.

My problem is solved. However, a question arises: which piece of software in naming the chip right?!

groeck, 2017-02-21:
> Now I'm really confused...
>
> ```
> it8792-isa-0a60
> Adapter: ISA adapter
> in0:          +1.88 V  (min =  +0.00 V, max =  +2.78 V)
> in1:          +1.48 V  (min =  +0.00 V, max =  +2.78 V)
> in2: [...]

Hmm ... looks like the 3.3V inputs have internal dividers which are not handled
correctly. I'll have to look into that.

> fan1:        1106 RPM  (min =   10 RPM)
> fan2:        1558 RPM  (min =   10 RPM)
> fan3:         658 RPM  (min =   10 RPM)
> temp1:        +26.0°C  (low  =  +0.0°C, high = +70.0°C)  sensor = Intel PECI [...]

I followed the dataseet, which says IT8792 (with a chip ID of 0x8733).

Guenter

groeck, 2017-03-10:
Turns out ADC resolution is wrong, resulting in bad voltage measurements. Fix is underway.

groeck, 2017-03-10:
Should now be fixed.

closed 2017-03-10 by groeck