groeck/it87 #9: it8620 pwm4 issue

psofiterol, 2017-02-07:
I just got a Gigabyte GA-Z97M-D3H that uses the IT8620E. It has 1x cpu_fan and 3x sys_fan on board. 

With the default kernel - without adding this driver - pwmconfig could only detect pwm1 (cpu / fan1), pwm2 (fan2) and pwm3 (fan3). fan4 was properly detected too, but not pwm4. So I downloaded the driver, compiled and installed it, after which pwm5 appeared. It was not controlling fan4 however.

I then manually commented out line #2648 @ it87.c so pwm4 does not get skipped, and re-installed the driver. That made pwm4 appear as well.

pwmconfig still cannot bind it to fan4 properly, but I found out why after tinkering a bit with /etc/fancontrol directly. The pwm4 values for fan4 are inverted (you probably already know this) so when pwmconfig has all the pwm values set to 255 to check for installed fans, fan4 is not spinning and is therefore omitted (it spins full speed with pwm4 set to 0).

This fixed the issue for me, but I suspect I will have to redo the fix with every kernel update.

Thanks for looking into it,
ping me if you need more info.

groeck, 2017-02-07:
> I just got a Gigabyte GA-Z97M-D3H that uses the IT8620E. It has 1x cpu_fan and 3x sys_fan on board.
>
> With the default kernel - without adding this driver - pwmconfig could only detect pwm1 (cpu / [...]
The logic is reversed. The real fix is

-               if (!(reg & BIT(2)))
+               if (reg & BIT(2))

> pwmconfig still cannot bind it to fan4 properly, but I found out why after tinkering a bit with /etc/fancontrol directly. The pwm4 values for fan4 are inverted (you probably already know this) so [...]
No, I was not aware of this, and I don't immediately see why this would be the
case.

Guenter

psofiterol, 2017-02-08:
I've run into this [article](http://www.almico.com/sfarticle.php?id=6) while on it, from the guys at SpeedFan. Seems to be the same issue - same motherboards.

"PWM4 and PWM5, on the other hand, appear to behave differently and have no corresponding PWM MODE to set. Unfortunately, Gigabyte seems to have implemented fan control in different ways across his range of motherboards and it has been reported that changing PWM4 using SpeedFan leads to a PC power off on some of them."

groeck, 2017-02-08:
On 02/07/2017 04:22 PM, psofiterol wrote:
> I've run into this article <http://www.almico.com/sfarticle.php?id=6> while on it, from the guys at SpeedFan. Seems to be the same issue - same motherboards.
>
> "PWM4 and PWM5, on the other hand, [...]

It isn't entirely accurate. There are pwm mode registers for those fan controls, it only behaves differently
(ie some bits previously unused are now used).

Guenter

groeck, 2017-02-08:
On 02/07/2017 12:09 PM, psofiterol wrote:
> I just got a Gigabyte GA-Z97M-D3H that uses the IT8620E. It has 1x cpu_fan and 3x sys_fan on board.
>
> With the default kernel - without adding this driver - pwmconfig could only detect pwm1 (cpu / [...]

Real fix is in set_pwm_enable(). Writes into the pwm_ctrl registers must not overwrite
bit 2..6 of the register. Current code, when setting manual mode, reverses pwm polarity
for fan4, which results in the "inverted" pwm values. pwm polarity is set in bit 6 of
the fan control registers for fan 4..6.

Guenter

psofiterol, 2017-02-08:
Nice! Thank you very much for spending time with this.
Is this fix all-around (i.e. will it be patched into the driver) or case specific (i.e. I should go ahead and - try to - patch it myself)?

Thanks

groeck, 2017-02-08:
On Wed, Feb 08, 2017 at 07:42:54AM -0800, psofiterol wrote:
> Nice! Thank you very much for spending time with this.
> Is this fix all-around (i.e. will it be patched into the driver) or case specific (i.e. I should go ahead and - try to - patch it myself)?
> 
Please try ToT of the driver on github and let me know if it works
as expected.

Thanks,
Guenter

psofiterol, 2017-02-08:
Everything working as expected.

Thanks for everything,
George

closed 2017-02-08 by psofiterol

groeck, 2017-02-08:
On Wed, Feb 08, 2017 at 09:46:27AM -0800, psofiterol wrote:
> Everything working as expected.
> 
Excellent. Thanks for letting me know.

Guenter