groeck/it87 #10: Support for ITE8665(?)

sarnex, 2017-03-10:
Hi,

I have a brand new AMD X370 motherboard with a Ryzen processor. There is an unknown ITE chip detected, and forcing it doesn't work. Below is the sensors-detect output

````
# sensors-detect revision 3.4.0+git_45ffa15cf02e63f70ff3b85c23e22dfbab7e8f9c
# Board: ASUSTeK COMPUTER INC. PRIME X370-PRO
# Kernel: 4.11.0-rc1+ x86_64
# Processor: AMD Ryzen 7 1800X Eight-Core Processor (23/1/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Gentoo config file protection is active. Every file this program will
modify must be merged before the change will become active using
default Gentoo tools such as dispatch-conf, cfg-update, and etc-update.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8665
    (logical device 4 has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 1022:790b at 0000:00:14.0.
Sorry, no supported PCI bus adapters found.

Next adapter: SMBus PIIX4 adapter port 0 at 0b00 (i2c-0)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No
Client found at address 0x53
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No

Next adapter: SMBus PIIX4 adapter port 2 at 0b00 (i2c-1)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No
Client found at address 0x53
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No

Next adapter: SMBus PIIX4 adapter port 3 at 0b00 (i2c-2)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No
Client found at address 0x53
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No

Next adapter: SMBus PIIX4 adapter port 4 at 0b00 (i2c-3)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No
Client found at address 0x53
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No

Next adapter: AMDGPU i2c bit bus 0x90 (i2c-4)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AMDGPU i2c bit bus 0x91 (i2c-5)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AMDGPU i2c bit bus 0x92 (i2c-6)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AMDGPU i2c bit bus 0x93 (i2c-7)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AMDGPU i2c bit bus 0x94 (i2c-8)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AMDGPU i2c bit bus 0x95 (i2c-9)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AMDGPU i2c bit bus 0x96 (i2c-10)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AMDGPU i2c bit bus 0x97 (i2c-11)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: card0-DP-1 (i2c-12)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: card0-DP-2 (i2c-13)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: card0-DP-3 (i2c-14)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: Radeon i2c bit bus 0x90 (i2c-15)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: Radeon i2c bit bus 0x91 (i2c-16)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: Radeon i2c bit bus 0x92 (i2c-17)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: Radeon i2c bit bus 0x93 (i2c-18)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: Radeon i2c bit bus 0x94 (i2c-19)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: Radeon i2c bit bus 0x95 (i2c-20)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: Radeon i2c bit bus 0x96 (i2c-21)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: Radeon i2c bit bus 0x97 (i2c-22)
Do you want to scan it? (yes/NO/selectively): 

Sorry, no sensors were detected.
Either your system has no sensors, or they are not supported, or
they are connected to an I2C or SMBus adapter that is not
supported. If you find out what chips are on your board, check
http://www.lm-sensors.org/wiki/Devices for driver status.

````

groeck, 2017-03-10:
Thanks for letting me know. Unfortunately, ITE doesn't release datasheets, and ASUS doesn't support Linux, so it is quite unlikely that we will be able to support this chip.

sarnex, 2017-03-10:
That is unfortunate.

Thanks.

groeck, 2017-03-10:
Very much so. Let's leave this open, though - maybe someone has access to a datasheet and can make it available or is willing to provide the necessary code changes.
One question, though - you mentioned "forcing it doesn't work". What exactly did you try that didn't work ?

sarnex, 2017-03-10:
I tried modprobe it87 force_id=0x8665. Is that id the name of the chip that I want to try to force? I don't really know what I'm doing here. Thanks

groeck, 2017-03-10:
No, you have to try something like modprobe force_id=0x8628. Pick any of the already supported IDs in the 86xx range and see what gives you the best results.

sarnex, 2017-03-10:
Wow, modprobe it87 force_id=0x8628 works!

Is there some way to add this automatically in the driver?

Thanks!

groeck, 2017-03-10:
Can you provide "sensors" command output ? Also, do you see all fans ?
It would also help if you could run the test with 0x8607, 0x8620, 0x8622, 0x8623, 0x8733, and 0x8790, and let me know which resulting output best matches your hardware. Watch out for differences in reported voltages and fans when you do that.

Thanks!

sarnex, 2017-03-10:
Here is the output for 0x8628
````
it8628-isa-0290
Adapter: ISA adapter
in0:          +1.42 V  (min =  +3.05 V, max =  +1.32 V)
in1:          +2.74 V  (min =  +3.01 V, max =  +2.68 V)
in2:          +2.17 V  (min =  +2.64 V, max =  +1.78 V)
+5V:          +4.34 V  (min =  +2.06 V, max =  +5.90 V)
in4:          +0.04 V  (min =  +3.04 V, max =  +0.62 V)
in5:          +0.04 V  (min =  +2.92 V, max =  +1.33 V)
in6:          +0.04 V  (min =  +1.51 V, max =  +3.04 V)
3VSB:         +3.62 V  (min =  +0.36 V, max =  +1.51 V)
Vbat:         +3.53 V  
fan1:         885 RPM  (min =   10 RPM)
fan3:           0 RPM  (min =   14 RPM)  ALARM
fan5:        1283 RPM  (min =   -1 RPM)  ALARM
temp1:        +69.0°C  (low  = -17.0°C, high = -25.0°C)  sensor = thermistor
temp2:        +37.0°C  (low  =  -1.0°C, high = +111.0°C)  sensor = thermistor
temp3:        +39.0°C  (low  = -17.0°C, high = -53.0°C)  sensor = thermistor
temp4:        +39.0°C  
temp5:        +39.0°C  
temp6:        +39.0°C  
intrusion0:  ALARM

radeon-pci-2900
Adapter: PCI adapter
temp1:        +55.5°C  (crit = +120.0°C, hyst = +90.0°C)

amdgpu-pci-2800
Adapter: PCI adapter
fan1:         757 RPM
temp1:        +45.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

````

I think 8628,8607,8620, and 8622 are all the same, with anything after 8622 showing alarms on everything. The min/max voltages and temps are wrong with all options.

groeck, 2017-03-10:
Can you try 0x8732 as well ? The voltages look wrong in the output above (3VSB should be around 3.3V, for example).

sarnex, 2017-03-10:
Same issues

it8732-isa-0290
Adapter: ISA adapter
in0:          +1.02 V  (min =  +2.77 V, max =  +1.20 V)  ALARM
in1:          +2.48 V  (min =  +2.74 V, max =  +2.43 V)  ALARM
in2:          +1.97 V  (min =  +2.40 V, max =  +1.61 V)  ALARM
+3.3V:        +1.97 V  (min =  +0.94 V, max =  +2.68 V)
in4:          +0.03 V  (min =  +2.76 V, max =  +0.57 V)  ALARM
in5:          +0.03 V  (min =  +2.65 V, max =  +1.21 V)  ALARM
in6:          +0.03 V  (min =  +1.37 V, max =  +2.76 V)  ALARM
3VSB:         +1.65 V  (min =  +0.16 V, max =  +0.69 V)  ALARM
Vbat:         +1.59 V
fan1:         862 RPM  (min =   10 RPM)
fan3:           0 RPM  (min =   14 RPM)  ALARM
temp1:        +67.0°C  (low  = -17.0°C, high = -25.0°C)  ALARM  sensor =
thermistor
temp2:        +38.0°C  (low  =  -1.0°C, high = +111.0°C)  sensor =
thermistor
temp3:        +39.0°C  (low  = -17.0°C, high = -53.0°C)  ALARM  sensor =
thermistor
intrusion0:  ALARM

radeon-pci-2900
Adapter: PCI adapter
temp1:        +63.5°C  (crit = +120.0°C, hyst = +90.0°C)

amdgpu-pci-2800
Adapter: PCI adapter
fan1:         632 RPM
temp1:        +36.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

groeck, 2017-03-10:
Actually, it's worse to some degree. Internal voltage scaling and ADC resolution differs from both 8732 and 8628, leaving us guessing. Is in3 connected to +5V or +3.3V ? Or something else entirely ? What is the internal scaling factor ? What is the ADC resolution ?
Looks like the ADC resolution may be 10.9mV like in some other chips, but that leaves us guessing for +3.3V vs. +5V. Also, the board has either 4 or 6 fan headers, meaning fan control/sensor detection is incorrect.
Overall, this means that we don't have enough information to add support for this chip right now. You'll have to live with force_id=0x8628.

sarnex, 2017-03-10:
I understand, thanks anyway!

groeck, 2017-03-12:
Can you send me the output of the following two commands ? I would like to give it a try with the information we know.

sudo isadump 0x295 0x296
sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7

sarnex, 2017-03-12:
Unfortunately, I am on spring break and away from my Ryzen system until
Sunday. I will try to get you the results then.

Thanks,
Sarnex

TopMak, 2017-03-14:
Hey,
I got about the same issue, just different ITE chip. (Found unknown chip with ID 0x8655) Also Asus board, but Prime B350m-A (motherboad has one cpu and one case fan).

I tried to sudo modprobe it87 force_id=0x8xxx using 8628, 8732, 8620, 8623, 8790. 
Get the following in dmesg when doing that for each: 
```
[  528.043382] it87: Found IT8xxxE chip at 0x290, revision 6
[  528.043411] it87: Beeping is supported
[  528.043520] it87: Found IT8xxxE chip at 0xfff8, revision 15
[  528.043548] it87: Beeping is supported

```
However, sensors output however only shows: 
(lm-sensors package Version: 1:3.4.0-3, Ubuntu Gnome 17.04 beta 1)
```
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```

Here is my take on 
> sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
 ```
   0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 55 06 40 2d 00 00 c0 00 10 00 69 05 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 40 00 10 00 00 00 00 00 00 00 10 00 00 00 
d0: ff 00 00 0f 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c 
f0: 00 00 00 00 00 5f 0e 00 2c c0 00 00 00 00 00 00

```
And also
> sudo isadump 0x295 0x296
 ```
    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 01 1f ab 00 00 40 00 00 c0 44 00 00 41 d2 00 
10: 41 e5 00 00 7f 00 00 00 7f 00 00 00 7f 1a 00 00 
20: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 0c 3c 99 
30: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 01 01 01 
40: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 00 00 00 
50: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c 
60: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c 
70: 08 1e 05 4c 4b 00 00 00 00 00 7f 99 00 00 0f 99 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 f2 00 8c 
90: 00 00 00 00 00 00 00 ff 40 00 00 00 00 32 00 4b 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80 
b0: 80 5f 00 00 55 7d 9c 7c ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
Not sure if I am testing it right. Should sensors give some output after I do that "modprobe forceid"? 

I'd be glad to assist with these issues in any possible way. I can also open a new issue post for this chip if necessary. (Just thought this is kinda similar)
Here is a picture of the motherboard, chip is right next to the PCI-e 16x slot.
https://m.media-amazon.com/images/S/aplus-media/vc/3a32b139-d2eb-4742-9800-9ebd43a4a0b0.JPG

groeck, 2017-03-14:
Yes, you should see output from the "sensors" command. Do you have a "resource conflict" error message in the kernel log ? If so, you'll have to boot with "acpi_enforce_resources=lax" kernel parameter.

TopMak, 2017-03-15:
There is one ACPI error, exception and warning, but no conflicts that I see. I still tried with that boot parameter, but it had no effect on the sensors output.
```
[    0.000000] ACPI BIOS Warning (bug): Optional FADT field Pm2ControlBlock has valid Length but zero Address: 0x0000000000000000/0x1 (20160930/tbfadt-658)
[    0.567825] ACPI Error: Needed [Integer/String/Buffer], found [Region] ffff9ba59e8ed0d8 (20160930/exresop-425)
[    0.567830] ACPI Exception: AE_AML_OPERAND_TYPE, Could not execute arguments for [IOB2] (Region) (20160930/nsinit-412)
```
BIOS related issues I guess.

One thing I forgot to mention. My sensors-detect feed is the same as Sarnex's apart from:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8655
    (logical device 4 has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

```
that is the different chip ID, but then it also founds lm78 from the same address!
```
Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       Success!
    (confidence 6, driver `lm78')
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No
```

Everything else comes as "No". However, adding lm78 to /etc/modules doesn't still show anything apart that asus-isa-0000 with 1 fan at 0 rpm.

groeck, 2017-03-15:
I can see from the dump that sensors-detect might mis-detect the chip as LM78.

The problem with your chip (or maybe your system) is that the 0x295/0x296 register dump looks all wrong. For example, register 0x58 should report 0x90 to identify the chip as ITE chip. This causes the it87 driver to bail out - it explicitly checks for this value. No idea if this means that the chip has a problem, that the system has a problem, that the method to access the chip was changed, or that the register map in this chip is different to the register map of other ITE chips. We may have to wait for others with the same chip to confirm (or for someone to provide the datasheet). Either case, let's open a separate issue to handle this chip. I'll do that and copy the 8655 related notes into it.

CalcProgrammer1, 2017-03-16:
Oh my, this is interesting!  I just got an ASUS Prime X370 after my C6H bricked.  I've been trying to reverse engineer the ASUS Aura lighting system to add to my RGB peripheral music visualizer app, and I want to support Linux with it, not to mention Linux has great tools for accessing low level stuff.  I've already found that the G.Skill Trident RGB RAM is an i2c/smbus device but it is on the X370's SMBUS adapter.  However, the ASUS Aura on-board chip does not seem to be using this bus, as I probed it with my oscilloscope and there was no bus activity.  This leads me to believe that the Aura chip is probably connected via LPC or the Super IO chip (does it have an i2c host controller?).  The Aura chip is physically directly next to the super IO on the Prime board further leading me to believe they're related.  I'm going to do more hardware probing this weekend as I need to remove my GPU and remote into the board as the GPU covers the relevant board sections.

I am a total noob at software debugging, as I'm a hardware/embedded guy, but if anyone knows how to use i.e. Cheat Engine to view writes to the super IO registers from a particular program/DLL I'd love to know, as if my assumption is correct, the Aura software service could give us some info.

CalcProgrammer1, 2017-03-16:
ASUS X370 Prime:

 ubuntu@ubuntu:~$ sudo isadump 0x295 0x296
 WARNING! Running this program can cause system crashes, data loss and worse!
 I will probe address register 0x295 and data register 0x296.
 Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
 00: 09 f7 ff a5 00 00 00 00 00 c0 44 08 20 ff ff ff 
 10: f6 ff 55 70 c0 00 80 80 ff ff ff 77 f6 6b 00 00 
 20: 5d e5 b9 bc 03 03 03 9a 94 2f 20 1a 1a 1a 1a 9a 
 30: e2 ff f8 fd a7 37 3f db b5 df 35 8d ea 63 1d fe 
 40: d6 a3 64 ff f8 f1 35 fe ad 6a d4 00 5f 40 15 e2 
 50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
 60: 80 7f 7f ff 00 00 0f ff 80 1e 46 99 14 00 0f c3 
 70: 80 1e 46 99 14 00 0f c3 7f 7f 7f 99 00 00 0f 99 
 80: 00 00 ff ff 00 00 00 00 00 00 00 00 00 1e 00 2c 
 90: 00 00 00 ff ff 00 00 ff 42 99 00 00 01 2c e0 2a 
 a0: 7f 7f 7f 80 00 00 0f 80 80 7f 7f ff 00 00 0f ff  
 b0: 00 e6 00 00 57 ef b3 e2 ff ff ff ff ff ff ff ff 
 c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
 d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
 e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
 f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

CalcProgrammer1, 2017-03-16:
Oops...mistyped the second one, here's what you actually wanted:

ubuntu@ubuntu:~$ sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 65 06 40 3d 90 ff 01 00 95 00 49 85 08 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 90 ef 01 00 d5 00 00 00 00 00 00 00 10 00 00 01 
d0: ff 00 00 0c 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c 
f0: 00 00 00 00 00 00 0e 00 00 00 00 00 00 00 00 00

codeworkx, 2017-03-18:
Hope this helps: http://imgur.com/a/zRhJP

sarnex, 2017-03-18:
@groeck 
I'm back from break now. Here is my information

pc sarnex # sudo isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 f6 f7 bd 00 00 00 00 00 c0 44 08 20 78 ff ff 
10: ff f7 5c 70 c0 80 80 80 03 ff ff f7 df b7 00 00 
20: 75 e5 b5 b6 03 03 03 98 91 39 1f 1e 1e 1e 1e 97 
30: 6e fe df fb 94 dc f6 56 34 fd 6f f3 fd 7e 3f 0f 
40: e7 ef 6f ff cb ef 9b ba ad 6a d4 00 5f 40 f2 f8 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 14 4b 33 14 00 0f 92 80 14 4b 33 14 00 0f 92 
70: 80 14 4b 33 14 00 0f 92 7f 7f 7f 99 00 00 0f 99 
80: 00 00 6b 0a 00 00 00 00 00 00 00 00 00 e8 00 56 
90: 00 00 80 ff ff 00 00 ff 42 99 00 00 01 4f e0 34 
a0: 7f 7f 7f 80 00 00 0f 80 80 28 4b 33 00 00 0f 33 
b0: 59 a5 00 00 ef ef b6 ef ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
pc sarnex #  sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 65 06 40 3d 90 ff 01 00 95 00 49 85 08 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                                                                                                                    
c0: 90 ef 01 00 d5 00 00 00 00 00 00 00 10 00 00 11                                                                                                                    
d0: ff 00 00 0c 00 00 00 00 00 00 00 00 00 00 00 00                                                                                                                    
e0: 00 00 00 00 00 00 00 10 00 06 06 06 00 00 40 0c                                                                                                                    
f0: 00 00 00 00 00 00 0e 00 00 00 00 00 00 00 00 00 
pc sarnex #

groeck, 2017-03-22:
Initial version now supported. Questions remain about fan4 support. Also, temperature sensor types as reported may be incorrect.

sarnex, 2017-03-23:
Thanks for adding this. It looks like only the min/max and temp low/high are wrong. 

Here is the output now:

````
it8665-isa-0290
Adapter: ISA adapter
in0:          +1.34 V  (min =  +0.00 V, max =  +1.20 V)
in1:          +2.48 V  (min =  +2.74 V, max =  +2.78 V)
in2:          +1.97 V  (min =  +2.40 V, max =  +1.61 V)
in3:          +1.97 V  (min =  +0.94 V, max =  +2.68 V)
in4:          +0.03 V  (min =  +2.76 V, max =  +0.58 V)
in5:          +0.03 V  (min =  +2.67 V, max =  +1.38 V)
in6:          +0.03 V  (min =  +1.37 V, max =  +2.76 V)
3VSB:         +1.66 V  (min =  +2.26 V, max =  +0.69 V)
Vbat:         +1.59 V  
fan1:        1238 RPM  (min =   10 RPM)
fan2:           0 RPM  (min =   11 RPM)  ALARM
fan3:           0 RPM  (min =   14 RPM)  ALARM
fan5:        1273 RPM  (min =   -1 RPM)  ALARM
fan6:           0 RPM  (min =   -1 RPM)  ALARM
temp1:        +79.0°C  (low  = -17.0°C, high = -17.0°C)  sensor = thermistor
temp2:        +39.0°C  (low  = +115.0°C, high = +111.0°C)  sensor = thermistor
temp3:        +50.0°C  (low  = -17.0°C, high = -49.0°C)  sensor = thermistor
intrusion0:  ALARM

radeon-pci-2900
Adapter: PCI adapter
temp1:        +56.0°C  (crit = +120.0°C, hyst = +90.0°C)

amdgpu-pci-2800
Adapter: PCI adapter
fan1:         752 RPM
temp1:        +49.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

````

Let me know if you need anything else

sterkyr, 2017-03-24:
Hey, how did u get in0 to show correct voltage?
Built this module today and voltages seem really strange.
Also missing fan2. (since i have 2 cpu fans, it should actually show almost same rpm as fan1)

```
it8665-isa-0290
Adapter: ISA adapter
in0:          +0.73 V  (min =  +0.83 V, max =  +1.65 V)
in1:          +0.69 V  (min =  +2.55 V, max =  +2.65 V)
in2:          +2.02 V  (min =  +1.21 V, max =  +2.63 V)
in3:          +2.03 V  (min =  +2.18 V, max =  +2.60 V)
in4:          +0.57 V  (min =  +1.72 V, max =  +1.56 V)
in5:          +0.46 V  (min =  +2.40 V, max =  +1.37 V)
in6:          +0.92 V  (min =  +2.25 V, max =  +2.76 V)
3VSB:         +1.67 V  (min =  +0.00 V, max =  +2.67 V)
Vbat:         +1.60 V  
fan1:         865 RPM  (min =   20 RPM)
fan4:           0 RPM  (min =   -1 RPM)
fan5:           0 RPM  (min =   -1 RPM)
fan6:           0 RPM  (min =   -1 RPM)
temp1:        +47.0°C  (low  = -33.0°C, high =  -4.0°C)  sensor = thermistor
temp2:        +25.0°C  (low  = -88.0°C, high = +79.0°C)  sensor = thermistor
temp3:        +27.0°C  (low  = -67.0°C, high = -65.0°C)  sensor = thermistor
```

sarnex, 2017-03-24:
I didn't do anything, I just compiled and installed this version of the
module.

amastro2, 2017-03-25:
Have new Ryzen 7 1800x on an Asus ROG Crosshair VI Hero mobo.  Downloaded it87 module, did make and sudo make install.  Have done modprobe it87 force_id = 0x****  without reported error and removed each on all numbers in readme 8603 to 8790 but no luck.  Just running sensors yields
 
asus-isa-0000
Adapter: ISA Adapter
cpu_fan:  0 RPM

core voltage 1.482
memory voltage 1.220
bios version 0702 (have not upgraded to 0902 yet)

# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)                                                                                                                         
# Board: ASUSTeK COMPUTER INC. CROSSHAIR VI HERO
# Kernel: 4.4.0-66-generic x86_64
# Processor: AMD Ryzen 7 1800X Eight-Core Processor (23/1/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8665
    (logical device 4 has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

ran make with following results
compat.h  it87.c  it87.ko  it87.mod.c  it87.mod.o  it87.o  Makefile  modules.order  Module.symvers  README  TODO

ran sudo make install
cp it87.ko /lib/modules/4.4.0-66-generic/kernel/drivers/hwmon
depmod -a -F /boot/System.map-4.4.0-66-generic 4.4.0-66-generic

sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 65 06 40 3d d0 db 81 04 d4 80 49 85 38 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 02 a0 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: d0 c3 01 00 10 00 00 00 40 00 00 00 10 00 00 11 
d0: ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 06 06 06 00 00 40 0c 
f0: 00 00 00 00 00 00 0e 00 00 00 00 00 00 00 00 00 
 ~ $ sudo isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 fc f7 be 00 00 00 00 00 c0 44 08 30 be 57 ff 
10: ba 7e 71 70 c0 80 80 80 04 02 ff 39 d7 24 80 00 
20: 28 38 ba b7 2a 26 53 99 90 23 19 19 19 19 19 99 
30: b6 d5 ed eb de d3 df b3 e7 2c fc bb 9e a5 37 bd 
40: 33 c5 96 97 07 e3 fd 5e ad 6a d4 00 5f 40 2d a7 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 16 00 0f 5b 80 1e 46 99 14 00 0f a5 
70: 80 1e 46 99 14 00 0f a5 80 1e 46 99 14 00 0f a5 
80: ff ff ff ff 00 00 00 00 00 00 00 00 00 bb 00 94 
90: 00 00 00 ff ff 00 00 ff 42 99 00 00 01 09 e0 1f 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f ff 00 00 0f ff 
b0: 02 02 00 00 fe f5 b6 eb ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01                                                                                                                                
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01                                                                                                                                
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff                                                                                                                                
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  

Hope this helps out, Asus apparently using different sensor chip on my board.
Gary

groeck, 2017-03-26:
amastro2: This is an ASUS board, so you'll have to boot with acpi_enforce_resources=lax. The chip is an IT8665E, which should now be supported for the most part. Please be aware that the BIOS may access the chip on ASUS boards while Linux is running. This can result in undesirable side effects.

codeworkx, 2017-03-26:
Hacked a little applet for cinnamon together.
https://github.com/codeworkx/cinnamon-spices-applets/blob/6a266261881339d9be75e6c8f9e9837e7a515ff6/primeinfo%40codeworkx/screenshot.png

Some voltage readings are not correct, but i'm sure we'll get it fixed someday.

amastro2, 2017-03-28:
Got sensors working.  Using a Cooler Master 212 led with a Noctua 120mm pwm fan on the input side and the Cooler Master fan reversed on the output side.  Running mprime blended torture test got a max temp of  
71C on psensor which corresponds well with stated temps on overclockers.com 1800x review.  Didn't need acpi_enforce_resources=lax.  Now I can tweak a little. Thanks for the help.

Amastro2

groeck, 2017-03-28:
amastro2: Thanks for confirming. Some of the recent ASUS boards don't need acpi_enforce_resources=lax anymore. That is excellent news!

Soyasi, 2017-03-28:
```
inxi -MC
Machine:   Device: desktop Mobo: ASUSTeK model: CROSSHAIR VI HERO v: Rev 1.xx serial: xxxxxxxxxxxxxxx
           UEFI: American Megatrends v: 0902 date: 03/11/2017
CPU:       Octa core AMD Ryzen 7 1700X Eight-Core (-HT-MCP-) cache: 4096 KB 
           clock speeds: max: 3400 MHz 1: 2200 MHz 2: 2200 MHz 3: 2200 MHz 4: 2200 MHz 5: 2200 MHz 6: 2200 MHz
           7: 2200 MHz 8: 2200 MHz 9: 3000 MHz 10: 2200 MHz 11: 2200 MHz 12: 2200 MHz 13: 2200 MHz 14: 2200 MHz
           15: 2200 MHz 16: 2200 MHz
```

Cooler Master Seidon 120V R2--> cpu_opt=Radiator fan, cpu-fan=free y w_pump=Water Pump
cha_fan1 and cha_fan5= Used

**modprobe it87 force_id=0x8620**

sensors
**it8620-isa-0290**
Adapter: ISA adapter
in0:          +0.24 V  (min =  +2.48 V, max =  +2.16 V)
in1:          +0.67 V  (min =  +1.52 V, max =  +2.00 V)
in2:          +2.22 V  (min =  +1.15 V, max =  +1.51 V)
+5V:          +4.42 V  (min =  +5.93 V, max =  +3.00 V)
in4:          +0.46 V  (min =  +1.91 V, max =  +1.70 V)
in5:          +0.52 V  (min =  +2.86 V, max =  +1.49 V)
in6:          +1.01 V  (min =  +1.19 V, max =  +2.87 V)
3VSB:         +3.70 V  (min =  +4.42 V, max =  +2.52 V)
Vbat:         +3.50 V  
fan1:        1516 RPM  (min =   20 RPM)
fan5:        2205 RPM  (min =   -1 RPM)  ALARM
temp1:        +44.0°C  (low  = +104.0°C, high = -16.0°C)  sensor = thermistor
temp2:        +27.0°C  (low  = -66.0°C, high =  -5.0°C)  sensor = thermistor
temp3:        +29.0°C  (low  = +71.0°C, high = -65.0°C)  sensor = thermistor
temp5:        +29.0°C  
temp6:        +29.0°C  
intrusion0:  ALARM

amdgpu-pci-2900
Adapter: PCI adapter
fan1:         677 RPM
temp1:        +44.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

**modprobe it87 force_id=0x8622**

**it8622-isa-0290**
Adapter: ISA adapter
in0:          +0.23 V  (min =  +2.48 V, max =  +2.16 V)
in1:          +0.67 V  (min =  +1.52 V, max =  +2.00 V)
in2:          +2.22 V  (min =  +1.15 V, max =  +1.51 V)
in3:          +2.21 V  (min =  +2.96 V, max =  +1.50 V)
in4:          +0.47 V  (min =  +1.91 V, max =  +1.70 V)
in5:          +0.50 V  (min =  +2.86 V, max =  +1.49 V)
in6:          +1.01 V  (min =  +1.19 V, max =  +2.87 V)
3VSB:         +3.70 V  (min =  +4.42 V, max =  +2.52 V)
Vbat:         +3.50 V  
+3.3V:        +3.70 V  
fan1:        1520 RPM  (min =   20 RPM)
fan5:        2205 RPM  (min =   -1 RPM)  ALARM
temp1:        +44.0°C  (low  = +104.0°C, high = -16.0°C)  sensor = thermistor
temp2:        +27.0°C  (low  = -66.0°C, high =  -5.0°C)  sensor = thermistor
temp3:        +29.0°C  (low  = +71.0°C, high = -65.0°C)  sensor = thermistor
intrusion0:  ALARM

amdgpu-pci-2900
Adapter: PCI adapter
fan1:         677 RPM
temp1:        +43.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

**modprobe it87 force_id=0x8623**

**it8603-isa-0290**
Adapter: ISA adapter
in0:          +0.23 V  (min =  +2.48 V, max =  +2.16 V)  ALARM
in1:          +0.67 V  (min =  +1.52 V, max =  +2.00 V)  ALARM
in2:          +2.22 V  (min =  +1.15 V, max =  +1.51 V)  ALARM
in3:          +2.21 V  (min =  +2.96 V, max =  +1.50 V)  ALARM
in4:          +0.46 V  (min =  +1.91 V, max =  +1.70 V)  ALARM
3VSB:         +3.70 V  (min =  +4.42 V, max =  +2.52 V)  ALARM
Vbat:         +3.50 V  
+3.3V:        +3.70 V  
fan1:        1513 RPM  (min =   20 RPM)
temp1:        +44.0°C  (low  = +104.0°C, high = -16.0°C)  ALARM  sensor = thermistor
temp2:        +27.0°C  (low  = -66.0°C, high =  -5.0°C)  ALARM  sensor = thermistor
temp3:        +29.0°C  (low  = +71.0°C, high = -65.0°C)  ALARM  sensor = thermistor
intrusion0:  ALARM

amdgpu-pci-2900
Adapter: PCI adapter
fan1:         676 RPM
temp1:        +44.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

**modprobe it87 force_id=0x8628**

**it8628-isa-0290**
Adapter: ISA adapter
in0:          +0.36 V  (min =  +2.48 V, max =  +2.16 V)
in1:          +0.67 V  (min =  +1.52 V, max =  +2.00 V)
in2:          +2.22 V  (min =  +1.15 V, max =  +1.51 V)
+5V:          +4.42 V  (min =  +5.93 V, max =  +3.00 V)
in4:          +0.46 V  (min =  +1.91 V, max =  +1.70 V)
in5:          +0.50 V  (min =  +2.86 V, max =  +1.49 V)
in6:          +1.01 V  (min =  +1.19 V, max =  +2.87 V)
3VSB:         +3.70 V  (min =  +4.42 V, max =  +2.52 V)
Vbat:         +3.50 V  
fan1:        1516 RPM  (min =   20 RPM)
fan5:        2205 RPM  (min =   -1 RPM)  ALARM
temp1:        +44.0°C  (low  = +104.0°C, high = -16.0°C)  sensor = thermistor
temp2:        +27.0°C  (low  = -66.0°C, high =  -5.0°C)  sensor = thermistor
temp3:        +29.0°C  (low  = +71.0°C, high = -65.0°C)  sensor = thermistor
temp5:        +29.0°C  
temp6:        +29.0°C  
intrusion0:  ALARM

amdgpu-pci-2900
Adapter: PCI adapter
fan1:         678 RPM
temp1:        +44.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

**_only worked these ones that I have put_**

groeck, 2017-08-08:
3.3V voltages (3VSB, Vbat, +3.3V) should now be reported correctly.

groeck, 2017-08-08:
No further known issues, closing.

closed 2017-08-08 by groeck

ewigkeit, 2017-09-12:
Hi,
and thank you very much for your work! I have the problem that my case fans are missing in the sensors output, only the CPU-Fan is showing up. They are detected in the BIOS. I'm running the latest version of the it87-Module with Linux-Kernel 4.12.10.

`↪ sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

it8665-isa-0290
Adapter: ISA adapter
in0:          +0.24 V  (min =  +1.21 V, max =  +1.81 V)
in1:          +0.67 V  (min =  +2.06 V, max =  +1.71 V)
in2:          +2.01 V  (min =  +0.88 V, max =  +0.64 V)
in3:          +1.97 V  (min =  +2.69 V, max =  +2.78 V)
in4:          +0.56 V  (min =  +1.30 V, max =  +2.78 V)
in5:          +0.48 V  (min =  +2.58 V, max =  +1.29 V)
in6:          +0.89 V  (min =  +1.15 V, max =  +2.77 V)
3VSB:         +3.29 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.12 V  
+3.3V:        +3.29 V  
fan1:         696 RPM  (min =   24 RPM)
fan5:           0 RPM  (min =   -1 RPM)  ALARM
temp1:        +29.0°C  (low  = +115.0°C, high =  -7.0°C)
temp2:        +25.0°C  (low  = -76.0°C, high = -38.0°C)  sensor = AMD AMDSI
temp3:        +32.0°C  (low  = -65.0°C, high = -70.0°C)  sensor = thermistor
intrusion0:  ALARM`

[image]

groeck, 2017-09-12:
@ewigkeit: fan1 does show up. asus-ias-0000 is the ASUS acpi driver, which is a different beast. If you post the DSDT from your board it may be possible to figure out why it does not display anything useful. Please also post isadump information from the configuration registers.

ewigkeit, 2017-09-12:
Sorry, I forgot to mention I'm running a AMD Ryzen 1700 on an Asus Crosshair Hero VI. If I got you right, you need the dsdt.dsl. I put it up. Let me know if I can get more information to help solve the issue.

↪ sudo sensors-detect

Probing for Super-I/O at 0x2e/0x2f
…
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8665
    (logical device 4 has address 0x290, could be sensors)

↪ sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 65 06 40 2d d0 db 81 04 d4 80 49 85 38 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 02 a0 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: d0 db 81 04 d4 00 00 00 40 02 00 00 10 00 00 11 
d0: ff 00 00 1a 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c 
f0: 00 00 00 00 00 00 0e 00 00 00 00 00 00 00 00 00

↪ sudo isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 f8 7f bf 00 00 00 00 00 c0 44 08 30 b7 ef 2a 
10: f9 df e6 70 c0 80 80 80 03 02 04 69 f7 7f 80 00 
20: 30 3d b9 b5 33 2b 52 97 8f 1e 1a 1f 1f 1f 1f 97 
30: a6 6f 9d bd 3b 51 ff f7 ff 77 76 ed fe 69 ff 00 
40: f9 73 da b4 ba bf bb 73 ad 6a d4 00 5f 40 00 fa 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 16 00 0f 4e 80 1e 46 99 14 00 0f 99 
70: 80 1e 46 99 14 00 0f 99 80 1e 46 99 14 00 0f 99 
80: ff ff ff ff 00 00 00 00 00 00 00 00 00 b0 00 c8 
90: 00 00 00 ff ff 00 00 ff 42 99 00 00 01 0a e0 1e 
a0: 7f 7f 7f 80 00 00 0f 80 80 7f 7f ff 00 00 0f ff 
b0: 5f 03 00 00 d3 fc cb f4 ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

↪ dmesg | grep DSDT
80:[    0.000000] ACPI: DSDT 0x00000000DBC861D0 00892C (v02 ALASKA A M I    01072009 INTL 20120913)

[dsdt.zip](https://github.com/groeck/it87/files/1296717/dsdt.zip)

bilalshaikh42, 2018-01-27:
Hello, 
I am having the same issue as above. I have an Asus Crosshair VI Hero. Using modprobe forceid does not change the output of the sensors command. This is what I get:

Adapter: ISA adapter
in0:          +0.70 V  (min =  +1.46 V, max =  +2.58 V)
in1:          +0.67 V  (min =  +0.62 V, max =  +0.60 V)
in2:          +2.02 V  (min =  +2.42 V, max =  +2.41 V)
in3:          +2.00 V  (min =  +1.27 V, max =  +2.50 V)
in4:          +0.51 V  (min =  +1.69 V, max =  +2.76 V)
in5:          +0.47 V  (min =  +1.24 V, max =  +0.26 V)
in6:          +0.91 V  (min =  +1.15 V, max =  +1.18 V)
3VSB:         +3.31 V  (min =  +3.73 V, max =  +0.37 V)
Vbat:         +3.21 V  
+3.3V:        +3.31 V  
fan1:         528 RPM  (min =   76 RPM)
fan5:        2428 RPM  (min =   -1 RPM)  ALARM
temp1:        +15.0°C  (low  = -38.0°C, high = +108.0°C)
temp2:        +32.0°C  (low  = +38.0°C, high = -126.0°C)  sensor = thermistor
temp3:        +36.0°C  (low  =  -7.0°C, high = +93.0°C)  sensor = thermistor
intrusion0:  ALARM

This is missing the three chasis fans, and is reporting a lower than ambient tempreture1. Any thoughts?

groeck, 2018-01-27:
See https://github.com/groeck/it87/issues/51. There is an Embedded Controller on the board which controls the other fans. The means to access it is unpublished.

bilalshaikh42, 2018-01-27:
Got it. Thanks for the quick response!

miguev, 2018-06-05:
Hi, may I ask why I'm getting screwed-up temperatures?

I have a PRIME X370-PRO (90MB0TD0-M0EAY0) with the latest BIOS (from April 2018)
modprobe it87 force_id=0x8628 finds the sensors but many (specially temps) seem out of range:

$ sudo sensors-detect
...
Driver `to-be-written':
  * ISA bus, address 0x290
    Chip `ITE IT8665E Super IO Sensors' (confidence: 9)

Driver `k10temp':
  * Chip `AMD Family 17h thermal sensors' (confidence: 9)

Note: there is no driver for ITE IT8665E Super IO Sensors yet.
Check http://www.lm-sensors.org/wiki/Devices for updates.

Do you want to overwrite /etc/sysconfig/lm_sensors? (YES/no): 
Copy prog/init/lm_sensors.service to /lib/systemd/system
and run 'systemctl enable lm_sensors.service'
for initialization at boot time.
Unloading cpuid... OK

$ sudo modprobe it87 force_id=0x8628
# similar results with 0x8620, 0x8622, 0x8623, 0x8733, and 0x8790 

$ sensors

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

it8628-isa-0290
Adapter: ISA adapter
in0:          +0.85 V  (min =  +2.66 V, max =  +2.92 V)
in1:          +2.76 V  (min =  +2.57 V, max =  +1.79 V)
in2:          +2.21 V  (min =  +1.08 V, max =  +0.56 V)
+5V:          +4.37 V  (min =  +4.58 V, max =  +5.76 V)
in4:          +0.02 V  (min =  +1.12 V, max =  +0.74 V)
in5:          +0.02 V  (min =  +0.66 V, max =  +1.52 V)
in6:          +0.02 V  (min =  +1.52 V, max =  +0.67 V)
3VSB:         +3.65 V  (min =  +6.05 V, max =  +3.05 V)
Vbat:         +3.55 V  
in12:         +0.41 V  
fan1:         571 RPM  (min =   15 RPM)
fan3:         796 RPM  (min =   21 RPM)
fan5:           0 RPM  (min =   -1 RPM)  ALARM
temp1:        +31.0°C  (low  =  -3.0°C, high = -70.0°C)  sensor = thermistor
temp2:        +38.0°C  (low  = -66.0°C, high = +105.0°C)  sensor = thermistor
temp3:        +34.0°C  (low  = +127.0°C, high = +85.0°C)  sensor = thermistor
temp4:        +34.0°C  
temp5:        +34.0°C  
intrusion0:  ALARM

$ sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 65 06 40 3d 90 ff 01 00 95 00 49 85 08 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 90 ef 01 00 d5 00 00 00 00 00 00 00 10 00 00 10 
d0: ff 00 00 0c 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 06 06 06 00 00 40 0c 
f0: 00 00 00 00 00 00 0e 00 00 00 00 00 00 00 00 00 

$ sudo isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 25 25 25 25 25 25 25 25 25 25 25 25 20 be 64 54 
10: 36 ac b1 70 c0 80 80 80 04 02 03 af be 79 00 00 
20: 5f e5 b8 b7 02 02 02 98 94 1e 25 22 22 22 22 97 
30: f3 de 95 d6 2f 5a f0 bf 3e 5d 7f 37 38 7f 7f fc 
40: ba fd 69 be 55 7f ff ff ad 6a d4 00 5f 40 5b df 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 16 00 0f 4e 80 1e 46 99 14 00 0f 99 
70: 80 1e 46 99 14 00 0f 99 7f 7f 7f 99 00 00 0f 99 
80: 00 00 ff ff 00 00 00 00 00 00 00 00 00 ff 00 25 
90: 00 00 00 d4 01 00 00 ff 42 99 00 00 01 4c e0 32 
a0: 7f 7f 7f 80 00 00 0f 80 80 7f 7f ff 00 00 0f ff 
b0: bf 08 00 00 de 73 6d ef ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 

$ dmesg | grep DSDT
[    0.000000] ACPI: DSDT 0x00000000D45591D0 00BAD0 (v02 ALASKA A M I    01072009 INTL 20120913)

groeck, 2018-06-05:
@miguev : Why don't you install the driver from here and just use it without force_id ? When using force_id, you can never assume or expect that everything works.
Also, you might create a configuration file for your board and set the limits with "sudo sensors -s" to get more reasonable limit values.