groeck/it87 #11: Add support for IT8655E

groeck, 2017-03-15:
[From TopMak]

Hey,
I got about the same issue, just different ITE chip. (Found unknown chip with ID 0x8655) Also Asus board, but Prime B350m-A (motherboad has one cpu and one case fan).

I tried to sudo modprobe it87 force_id=0x8xxx using 8628, 8732, 8620, 8623, 8790. 
Get the following in dmesg when doing that for each: 
```
[  528.043382] it87: Found IT8xxxE chip at 0x290, revision 6
[  528.043411] it87: Beeping is supported
[  528.043520] it87: Found IT8xxxE chip at 0xfff8, revision 15
[  528.043548] it87: Beeping is supported

```
However, sensors output however only shows: 
(lm-sensors package Version: 1:3.4.0-3, Ubuntu Gnome 17.04 beta 1)
```
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```

Here is my take on 
> sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
 ```
   0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 55 06 40 2d 00 00 c0 00 10 00 69 05 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 40 00 10 00 00 00 00 00 00 00 10 00 00 00 
d0: ff 00 00 0f 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c 
f0: 00 00 00 00 00 5f 0e 00 2c c0 00 00 00 00 00 00

```
And also
> sudo isadump 0x295 0x296
 ```
    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 01 1f ab 00 00 40 00 00 c0 44 00 00 41 d2 00 
10: 41 e5 00 00 7f 00 00 00 7f 00 00 00 7f 1a 00 00 
20: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 0c 3c 99 
30: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 01 01 01 
40: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 00 00 00 
50: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c 
60: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c 
70: 08 1e 05 4c 4b 00 00 00 00 00 7f 99 00 00 0f 99 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 f2 00 8c 
90: 00 00 00 00 00 00 00 ff 40 00 00 00 00 32 00 4b 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80 
b0: 80 5f 00 00 55 7d 9c 7c ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
Not sure if I am testing it right. Should sensors give some output after I do that "modprobe forceid"? 

I'd be glad to assist with these issues in any possible way. I can also open a new issue post for this chip if necessary. (Just thought this is kinda similar)
Here is a picture of the motherboard, chip is right next to the PCI-e 16x slot.
https://m.media-amazon.com/images/S/aplus-media/vc/3a32b139-d2eb-4742-9800-9ebd43a4a0b0.JPG

groeck, 2017-03-15:
[From TopMak]

There is one ACPI error, exception and warning, but no conflicts that I see. I still tried with that boot parameter, but it had no effect on the sensors output.
```
[    0.000000] ACPI BIOS Warning (bug): Optional FADT field Pm2ControlBlock has valid Length but zero Address: 0x0000000000000000/0x1 (20160930/tbfadt-658)
[    0.567825] ACPI Error: Needed [Integer/String/Buffer], found [Region] ffff9ba59e8ed0d8 (20160930/exresop-425)
[    0.567830] ACPI Exception: AE_AML_OPERAND_TYPE, Could not execute arguments for [IOB2] (Region) (20160930/nsinit-412)
```
BIOS related issues I guess.

One thing I forgot to mention. My sensors-detect feed is the same as Sarnex's apart from:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8655
    (logical device 4 has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

```
that is the different chip ID, but then it also founds lm78 from the same address!
```
Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       Success!
    (confidence 6, driver `lm78')
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No
```

Everything else comes as "No". However, adding lm78 to /etc/modules doesn't still show anything apart that asus-isa-0000 with 1 fan at 0 rpm.

groeck, 2017-03-15:
I can see from the dump that sensors-detect might mis-detect the chip as LM78.

The problem with your chip (or maybe your system) is that the 0x295/0x296 register dump looks all wrong. For example, register 0x58 should report 0x90 to identify the chip as ITE chip. This causes the it87 driver to bail out - it explicitly checks for this value. No idea if this means that the chip has a problem, that the system has a problem, that the method to access the chip was changed, or that the register map in this chip is different to the register map of other ITE chips. We may have to wait for others with the same chip to confirm (or for someone to provide the datasheet). Either case, let's open a separate issue to handle this chip. I'll do that and copy the 8655 related notes into it.

groeck, 2017-03-16:
I have some information suggesting that the recent chips may support multiple pages of registers. The register in question is register 0x06, and the bank bits are bit 5..7. In the output above, the register suggests that bank 2 is selected, meaning it is not really surprising that the output is unexpected.

accelleon, 2017-03-18:
[IT8655E dump.txt](https://github.com/groeck/it87/files/852234/IT8655E.dump.txt)

SIV64X provides a detailed dump of the system buses. Here is a dump of the IT8655E's registers coupled with the parsed register data, might help engineering support for it.

groeck, 2017-03-18:
On Sat, Mar 18, 2017 at 12:48:33AM -0700, Anthony Wilkins wrote:
> [IT8655E dump.txt](https://github.com/groeck/it87/files/852234/IT8655E.dump.txt)
> 
> SIV64X provides a detailed dump of the system buses. Here is a dump of the IT8655E's registers coupled with the parsed register data, might help engineering support for it.
> 
You mean IT8665E, not IT8655E. At least the register dump if from an IT8665E.

Guenter

sarnex, 2017-03-18:
I've posted my dumps in the other thread, let me know if you need any more information or testing.

petrov0, 2017-03-20:
Register 0x58 is always 0x90 in the output for banks from 00 to 31.

petrov0, 2017-03-20:
That's the case here, 0x58 is different in bank 2. 0x06 by default is set to 0x40 (which should be bank 2):
```
isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n]
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 01 dd bc 00 00 40 00 00 c0 44 07 00 ff eb 02
10: 41 e5 00 00 7f 00 00 00 7f 00 00 00 7f 1a 00 00
20: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 33 ff 99
30: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 01 01 01
40: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 00 00 00
50: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c
60: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c
70: 08 1e 05 4c 4b 00 00 00 00 00 7f 99 00 00 0f 99
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 0f 00 80
90: 00 00 00 00 00 00 00 ff 40 00 00 00 00 ce 00 12
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80
b0: 81 81 00 00 eb 5d f2 5d ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
I don't know why 0x06 is set like that and I don't have access to a datasheet. isadump however allows me to modify maximum bit 5 at 0x06 and according to your comment above the bank can be selected by bits 5..7 in this register. The results are similar for any number lower than decimal 32 at 0x06 ... 0x58 remains 0x90.  http://pastebin.com/HcfkNmB5 here you can see an output from
`for i in `seq 1 30` ; do echo "0x295 0x296 $i 0x06" ; isadump -y 0x295 0x296 $i 0x06`

groeck, 2017-03-20:
@petrov0: Page bits are bit 5..7. I can see from your log that you set bit 0..5, which from my understanding of the datasheets only touches pages 0 and 1. Bit 0 to 4 enable/disable various interrupts. This matches my understanding of the datasheets I have available; according to those, page 1 only maps registers 0xc0 to 0xdc. Page 2, which would be selected with 0x06:=0x40, maps registers 0x10 to 0x76.

groeck, 2017-03-20:
isadump for register page 0
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 91 dd bc 00 00 01 00 00 c0 44 07 00 ff ec 00
10: 3a 47 3a 70 c0 80 80 80 ff 03 02 bf 7f c2 00 00
20: 55 e5 b7 bb ba b2 bb 9a 96 25 1a 20 20 20 20 99
30: 57 97 ef c8 b3 d5 a5 42 67 ba b9 5a bb 7c 74 9a
40: 75 76 27 ef ef de fb b4 ad 6a d4 00 5f 40 00 5d
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00
60: 80 14 46 33 16 00 0f 61 80 1e 46 99 14 00 0f aa
70: 80 1e 46 99 14 00 0f aa 7f 7f 7f 99 00 00 0f 99
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 0f 00 ad
90: 00 00 00 00 00 00 00 ff 42 99 00 00 01 8f e0 20
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80
b0: 81 81 00 00 eb 5d f2 5d ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

petrov0, 2017-03-20:
These are the results from isadump with bit 7 set to 1 and with bits 5..7 set to 0
```
isaset 0x295 0x296 0x06 0x80
WARNING! Running this program can cause system crashes, data loss and worse!
I will write value 0x80 to address 0x06 of chip with address register 0x295
and data register 0x296.
Continue? [Y/n]
isadump 0x295 0x296         
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 91 dd bc 00 00 80 00 00 c0 44 07 00 ff c9 e0 
10: 3a 47 3a 70 c0 80 80 80 ff 03 01 bf 7f c2 00 00 
20: 36 e5 b8 ba bb b2 ba 9a 96 28 1d 23 23 23 23 9a 
30: 57 97 ef c8 b3 d5 a5 42 67 ba b9 5a bb 7c 74 9a 
40: 75 76 27 ef ef de fb b4 ad 6a d4 00 5f 40 00 5d 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 16 00 0f 66 80 1e 46 99 14 00 0f af 
70: 80 1e 46 99 14 00 0f af 7f 7f 7f 99 00 00 0f 99 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 9d 00 25 
90: 00 00 00 00 00 00 00 ff 42 99 00 00 01 8f e0 23 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80 
b0: 81 81 00 00 eb 5d f2 5d ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 

isadump 0x295 0x296 
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 91 dd bc 00 00 00 00 00 c0 44 07 00 ff b7 d7 
10: 3a 47 3a 70 c0 80 80 80 ff 03 01 bf 7f c2 00 00 
20: 2b e5 b8 ba bb b2 ba 9a 96 28 1d 23 23 23 23 99 
30: 57 97 ef c8 b3 d5 a5 42 67 ba b9 5a bb 7c 74 9a 
40: 75 76 27 ef ef de fb b4 ad 6a d4 00 5f 40 00 5d 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 16 00 0f 6a 80 1e 46 99 14 00 0f b2 
70: 80 1e 46 99 14 00 0f b2 7f 7f 7f 99 00 00 0f 99 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 9d 00 09 
90: 00 00 00 00 00 00 00 ff 42 99 00 00 01 8f e0 23 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80 
b0: 81 81 00 00 eb 5d f2 5d ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
I have output from sensors with these (0x80 and 0x00) and it87 id 0x8628. The readings however are incorrect which does not surprise me. Can I help with something else ?

groeck, 2017-03-20:
Yes, add that output. Even if it isn't correct it will help us to move forward. For example, we'll see from the battery voltage if the ADC has a resolution of 12mV or 10.9mV.

petrov0, 2017-03-20:
The output from sensors when 0x06 is set to 0x00
```
it8628-isa-0290
Adapter: ISA adapter
in0:          +0.50 V  (min =  +1.81 V, max =  +1.04 V)  ALARM
in1:          +2.74 V  (min =  +2.40 V, max =  +2.87 V)
in2:          +2.21 V  (min =  +2.56 V, max =  +2.15 V)  ALARM
+5V:          +4.46 V  (min =  +1.58 V, max =  +3.96 V)  ALARM
in4:          +2.24 V  (min =  +2.23 V, max =  +1.24 V)  ALARM
in5:          +2.14 V  (min =  +1.08 V, max =  +2.22 V)
in6:          +2.23 V  (min =  +1.49 V, max =  +2.24 V)  ALARM
3VSB:         +3.70 V  (min =  +3.70 V, max =  +2.78 V)  ALARM
Vbat:         +3.60 V  
fan1:           0 RPM  (min =   13 RPM)  ALARM
fan3:        1436 RPM  (min =   13 RPM)
temp1:        +40.0°C  (low  = +118.0°C, high = +117.0°C)  sensor = thermistor
temp2:        +29.0°C  (low  = -17.0°C, high = +39.0°C)  sensor = thermistor
temp3:        +35.0°C  (low  = -34.0°C, high = -17.0°C)  ALARM  sensor = thermistor
temp4:        +35.0°C  
temp6:        +35.0°C  
intrusion0:  ALARM
```
Fan 1 is probably the CPU fan and can be ignored. This fan doesn't show its RPMs correctly when it's on idle even in the BIOS. I will test with something CPU intensive to see if there is any difference and will update the result.

petrov0, 2017-03-20:
After approximately 10 minutes 100% CPU load
```
it8628-isa-0290
Adapter: ISA adapter
in0:          +1.42 V  (min =  +1.81 V, max =  +1.04 V)  ALARM
in1:          +2.75 V  (min =  +2.40 V, max =  +2.87 V)
in2:          +2.17 V  (min =  +2.56 V, max =  +2.15 V)  ALARM
+5V:          +4.49 V  (min =  +1.58 V, max =  +3.96 V)  ALARM
in4:          +2.23 V  (min =  +2.23 V, max =  +1.24 V)  ALARM
in5:          +2.12 V  (min =  +1.08 V, max =  +2.22 V)
in6:          +2.24 V  (min =  +1.49 V, max =  +2.24 V)  ALARM
3VSB:         +3.70 V  (min =  +3.70 V, max =  +2.78 V)  ALARM
Vbat:         +3.62 V  
in10:         +0.80 V  
fan1:        2327 RPM  (min =   13 RPM)
fan3:        2002 RPM  (min =   13 RPM)
temp1:        +80.0°C  (low  = +118.0°C, high = +117.0°C)  sensor = thermistor
temp2:        +30.0°C  (low  = -17.0°C, high = +39.0°C)  sensor = thermistor
temp3:        +67.0°C  (low  = -34.0°C, high = -17.0°C)  ALARM  sensor = thermistor
temp5:        +67.0°C  
temp6:        +67.0°C  
intrusion0:  ALARM
```
There was a reboot between the two tests (the system was unstable at 3.85), and now I see in10. The output from isadump during the test is as follows:

```
isadump 0x295 0x296 
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 90 dd bc 00 00 00 00 00 c0 44 07 00 1e b6 4e 
10: 3a 47 3a 70 c0 80 80 80 01 02 01 bf 7f c2 00 00 
20: 72 e5 b8 ba ba b1 ba 9a 97 4a 1e 43 43 43 43 9a 
30: 57 97 ef c8 b3 d5 a5 42 67 ba b9 5a bb 7c 74 9a 
40: 75 76 27 ef ef de fb b4 ad 6a d4 00 5f 40 00 5d 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 16 00 0f ff 80 1e 46 99 14 00 0f ff 
70: 80 1e 46 99 14 00 0f ff 7f 7f 7f 99 00 00 0f 99 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 eb 00 a3 
90: 00 00 00 00 00 00 00 ff 42 99 00 00 01 8f e0 45 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80 
b0: 81 81 00 00 eb 5d f2 5d ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
At least Fan 1 looks OK at the moment. I can perform a similar test under windows if it's necessary the values to be compared.

petrov0, 2017-03-21:
Would you accept something like this considering the missing datasheet at the moment? I don't know if this is the proper way the device to be added. The chip however is detected and the values can be read by the sensors command. The scaling of the voltages should be possible to be adjusted through the config file. in10 and in11 however randomly appear in the sensors output, and also fan2 is missing.
[it87.c.patch.txt](https://github.com/groeck/it87/files/859281/it87.c.patch.txt)

groeck, 2017-03-21:
Something like that, only ToT of the driver @ github already handles
multi-page accesses and selects page 0. Also, as far as I can see,
the chip only has three fan and pwm controls. It is also somewhat
unlikely that FEAT_VIN3_5V is correct.

We have a patch for IT8665E upcoming; we might just declare IT8655E
to be a variant of it with three fan and pwm controls until we learn
otherwise.

Thanks,
Guenter

groeck, 2017-03-22:
Initial version now supported. It is assumed to be smilar to IT8665E with three instead of six fans. Temperature sensor types may be wrong.

petrov0, 2017-03-22:
Thank you! Looks very accurate:

```
it8655-isa-0290
Adapter: ISA adapter
Vcore:        +1.31 V  (min =  +1.65 V, max =  +0.95 V)  ALARM
in1:          +2.48 V  (min =  +2.18 V, max =  +2.60 V)
+12V:        +11.97 V  (min = +12.32 V, max = +11.95 V)  ALARM
+5V:          +5.12 V  (min =  +1.80 V, max =  +4.50 V)  ALARM
in4:          +2.04 V  (min =  +2.03 V, max =  +1.12 V)  ALARM
in5:          +1.94 V  (min =  +0.98 V, max =  +2.02 V)
in6:          +2.04 V  (min =  +1.35 V, max =  +2.04 V)  ALARM
3VSB:         +3.34 V  (min =  +3.36 V, max =  +2.53 V)  ALARM
Vbat:         +3.29 V
CPU Fan:     2445 RPM  (min =   13 RPM)
Bottom Fan:   964 RPM  (min =   20 RPM)
Back Fan:    1991 RPM  (min =   13 RPM)
MB CPU Temp:  +81.0°C  (low  = +118.0°C, high = +117.0°C)  sensor = thermistor
Ambient:      +29.0°C  (low  = -17.0°C, high = +39.0°C)  sensor = thermistor
temp3:        +66.0°C  (low  = -34.0°C, high = -17.0°C)  ALARM  sensor = thermistor
intrusion0:  ALARM
```

with the following config:
```

chip "it8655-*"
        label temp1 "MB CPU Temp"
        label temp2 "Ambient"
        label fan1 "CPU Fan"
        label fan2 "Bottom Fan"
        label fan3 "Back Fan"
        label in0 "Vcore"
        label in2 "+12V"
        label in3 "+5V"
        compute in2 @ + 10, @/1
        compute in7 @ * 2, @/1
        compute in8 @ * 2, @/1
        compute in3 @ * 2.5, @/1

```
when comparing the values with those measured under load in Windows.
[image]
Can't set the min/max for the fans and the voltages, but this is not a problem. Thank you again.

groeck, 2017-03-22:
"Can't set the min/max for the fans and the voltages" - what happens if you try ?

petrov0, 2017-03-22:
Actually it was my mistake.I had to update the values with sensors -s. There is something strange however with the min/max scaling of the voltages for in2 (12V) in3 (5V) and in7 (3VSB). If I try to set them as the other entries some kind of overflow occurs. They look OK with this configration:

```
        set in0_min 1.65 * 0.5
        set in0_max 1.65 * 1
        set in1_min 2.48 * 0.8
        set in1_max 2.48 * 1.1
        set in2_min 12 * 0.1
        set in2_max 12 * 0.2
        set in3_min 5 * 0.38
        set in3_max 5 * 0.45
        set in4_min 2.03 * 0.9
        set in4_max 2.03 * 1.1
        set in7_min 3 * 0.47
        set in7_max 3 * 0.57
        set temp2_min 25 * 0.5
        set temp2_max 25 * 1.7
        set temp1_min 40 * 0.5
        set temp1_max 40 * 2.2
        set temp3_min 36 * 0.5
        set temp3_max 36 * 1.7
        set fan1_min 0
        set fan2_min 700
        set fan3_min 1300
```

```
it8655-isa-0290
Adapter: ISA adapter
Vcore:        +0.96 V  (min =  +0.83 V, max =  +1.65 V)
in1:          +2.50 V  (min =  +1.98 V, max =  +2.73 V)
+12V:        +12.00 V  (min = +11.20 V, max = +12.40 V)
+5V:          +5.09 V  (min =  +4.74 V, max =  +5.61 V)
in4:          +2.04 V  (min =  +1.83 V, max =  +2.23 V)
in5:          +1.93 V  (min =  +0.98 V, max =  +2.02 V)
in6:          +2.03 V  (min =  +1.35 V, max =  +2.04 V)
3VSB:         +3.36 V  (min =  +2.81 V, max =  +3.42 V)
Vbat:         +3.29 V  
CPU Fan:        0 RPM  (min =    0 RPM)
Bottom Fan:   727 RPM  (min =  700 RPM)
Back Fan:    1470 RPM  (min = 1300 RPM)
MB CPU Temp:  +42.0°C  (low  = +20.0°C, high = +88.0°C)  sensor = thermistor
Ambient:      +29.0°C  (low  = +13.0°C, high = +43.0°C)  sensor = thermistor
temp3:        +37.0°C  (low  = +18.0°C, high = +61.0°C)  sensor = thermistor
```

but if I change them with

```
        set in2_min 12 * 0.85
        set in2_max 12 * 1.2
        set in3_min 5 * 0.8
        set in3_max 5 * 1.2
        set in7_min 3 * 0.9
        set in7_max 3 * 1.1
```

the min/max values are wrong:

```
it8655-isa-0290
Adapter: ISA adapter
Vcore:        +1.26 V  (min =  +0.83 V, max =  +1.65 V)
in1:          +2.50 V  (min =  +1.98 V, max =  +2.73 V)
+12V:        +12.00 V  (min = +12.78 V, max = +12.78 V)  ALARM
+5V:          +5.07 V  (min =  +6.95 V, max =  +6.95 V)  ALARM
in4:          +2.04 V  (min =  +1.83 V, max =  +2.23 V)
in5:          +1.94 V  (min =  +0.98 V, max =  +2.02 V)
in6:          +2.03 V  (min =  +1.35 V, max =  +2.04 V)
3VSB:         +3.36 V  (min =  +5.41 V, max =  +5.56 V)  ALARM
```

when this happens registers from 0x34 to 0x37 contain ff.

petrov0, 2017-03-22:
isadump when the min/max voltages are OK:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 90 00 b8 00 00 00 00 00 c0 44 07 00 ff a1 cc 
10: ff c4 07 70 c0 80 80 80 ff 03 01 ff 03 02 00 00 
20: 58 e5 b7 ba bb b1 ba 9a 97 2a 1d 25 25 25 25 99 
30: 97 4c fa b6 dc 6e ce ae cd a8 b9 5a bb 7c 9d 81 
40: 58 14 2b 0d 3d 12 fb b4 ad 6a d4 00 5f 40 00 5d 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 16 00 0f 6f 80 1e 46 99 14 00 0f b7 
70: 80 1e 46 99 14 00 0f b7 7f 7f 7f 99 00 00 0f 99 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 0f 00 80 
90: 00 00 00 00 00 00 00 ff 42 99 00 00 01 8f e0 25 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80 
b0: 81 81 00 00 eb 5d f2 5d ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
and when they are wrong
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 90 cc b8 00 00 00 00 00 c0 44 07 00 ff a3 cb 
10: ff c4 07 70 c0 80 80 80 ff 03 01 ff 03 02 00 00 
20: 57 e4 b8 bb bb b2 bb 99 97 2a 1d 25 25 25 25 9a 
30: 97 4c fa b6 ff ff ff ff cd a8 b9 5a bb 7c ff f8 
40: 58 14 2b 0d 3d 12 fb b4 ad 6a d4 00 5f 40 00 5d 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 16 00 0f 6f 80 1e 46 99 14 00 0f b7 
70: 80 1e 46 99 14 00 0f b7 7f 7f 7f 99 00 00 0f 99 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 20 00 42 
90: 00 00 00 00 00 00 00 ff 42 99 00 00 01 8f e0 25 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80 
b0: 81 81 00 00 eb 5d f2 5d ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

TopMak, 2017-03-24:
Forgot I haven't commented on this actual topic so I haven't received updates. However, I can confirm similar results on my board (Asus B350m-A). Works well!
```
[16233.924642] it87: Found IT8655E chip at 0x290, revision 6
[16233.924660] it87: Beeping is supported
```
This one got fan1(cpu) and fan2(case), both working fine. 
CPU temp is about 5'C higher than actual Tctl sensor in hwinfo on Windows, which to my understanding is correct.
All the voltages also and temp2-3 seems to be spot on with hwinfo reportings.
Any idea what that temp3 might be? For me it's around 40C idle and 55C under load. Maybe a temperature near CPU voltage regulators?
Thanks

petrov0, 2017-03-24:
The VRMs and the space around them is usually the hottest spot, so I doubt this sensor to be there (with some offset is possible ... temp3 for my board is 67 C under load with the new BIOS). By the way with BIOS 0513 the MB CPU Temp and Tctl are equal. The min/max overflow still occurs though.

braynshock, 2017-04-20:
THANK YOU FOR THIS! 
I can see some temps now!
BTW, I'm using an ASUS B350m-A
and my sensors output is:

> it8655-isa-0290
> Adapter: ISA adapter
> in0:          +0.82 V  (min =  +2.56 V, max =  +2.53 V)  ALARM
> in1:          +2.48 V  (min =  +1.38 V, max =  +0.65 V)  ALARM
> in2:          +2.00 V  (min =  +2.76 V, max =  +2.03 V)  ALARM
> in3:          +2.04 V  (min =  +0.42 V, max =  +1.38 V)  ALARM
> in4:          +2.04 V  (min =  +0.36 V, max =  +2.43 V)
> in5:          +1.95 V  (min =  +1.38 V, max =  +1.20 V)  ALARM
> in6:          +2.05 V  (min =  +0.82 V, max =  +1.35 V)  ALARM
> 3VSB:         +1.67 V  (min =  +2.32 V, max =  +1.56 V)  ALARM
> Vbat:         +1.60 V  
> +3.3V:        +1.67 V  
> fan1:        1646 RPM  (min =   16 RPM)
> fan2:        1036 RPM  (min =   30 RPM)
> temp1:        +41.0°C  (low  = +46.0°C, high =  -2.0°C)  ALARM
> temp2:        +45.0°C  (low  = +121.0°C, high = -32.0°C)  ALARM
> temp3:        +43.0°C  (low  =  -4.0°C, high = -88.0°C)  ALARM  sensor = thermistor
> intrusion0:  ALARM

Enigma0, 2017-08-07:
```
MOTHERBOARD:        ASUS PRIME B350M-A
    Memory:           32768MB
    Chipset:          AMD Device 1450

PROCESSOR:          AMD Ryzen 7 1700X Eight-Core @ 3.40GHz (16 Cores)
    Core Count:       8
    Thread Count:     16
    Extensions:       SSE 4.2 + AVX2 + AVX + RDRAND + FSGSBASE
    Cache Size:       512 KB
    Microcode:        0x8001126
    Scaling Driver:   acpi-cpufreq ondemand

OPERATING SYSTEM:   Ubuntu 17.04
    Kernel:           4.10.0-30-generic (x86_64)
    Desktop:          GNOME Shell 3.24.2
    Compiler:         GCC 6.3.0 20170406
```

I was getting temperature readouts in `sensors` prior to updating the BIOS to the latest version (better memory stability/support). Now I'm not getting anything. Any ideas? Tried forcing several suggested ID's with no success.

groeck, 2017-08-07:
Did you also update the kernel and forgot to update the it87 driver ?

Enigma0, 2017-08-07:
My system update script also builds/installs the it87 driver. `dmesg` appears to show that it's loading in any case.

```
:~$ dmesg | grep ACPI
[    0.000000] BIOS-e820: [mem 0x00000000d654d000-0x00000000d6564fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000dac90000-0x00000000dac98fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000dad9e000-0x00000000db15cfff] ACPI NVS
[    0.000000] efi:  ACPI 2.0=0xd654d000  ACPI=0xd654d000  SMBIOS=0xdbdaa000  SMBIOS 3.0=0xdbda9000  ESRT=0xd8e13098 
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000D654D000 000024 (v02 ALASKA)
[    0.000000] ACPI: XSDT 0x00000000D654D098 0000AC (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: FACP 0x00000000D6555250 000114 (v06 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI BIOS Warning (bug): Optional FADT field Pm2ControlBlock has valid Length but zero Address: 0x0000000000000000/0x1 (20160930/tbfadt-658)
[    0.000000] ACPI: DSDT 0x00000000D654D1E0 00806F (v02 ALASKA A M I    01072009 INTL 20120913)
[    0.000000] ACPI: FACS 0x00000000DB15AC80 000040
[    0.000000] ACPI: APIC 0x00000000D6555368 0000DE (v03 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: FPDT 0x00000000D6555448 000044 (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: SSDT 0x00000000D6562D28 001664 (v01 AMD    CPMCMN   00000001 INTL 20120913)
[    0.000000] ACPI: SSDT 0x00000000D65554E8 008C4C (v02 AMD    AMD ALIB 00000002 MSFT 04000000)
[    0.000000] ACPI: FIDT 0x00000000D655E138 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: SSDT 0x00000000D655E1D8 0020E4 (v01 AMD    AMD CPU  00000001 AMD  00000001)
[    0.000000] ACPI: CRAT 0x00000000D65602C0 000F50 (v01 AMD    AMD CRAT 00000001 AMD  00000001)
[    0.000000] ACPI: CDIT 0x00000000D6561210 000029 (v01 AMD    AMD CDIT 00000001 AMD  00000001)
[    0.000000] ACPI: SSDT 0x00000000D6561240 0017FF (v01 AMD    AMD AOD  00000001 INTL 20120913)
[    0.000000] ACPI: MCFG 0x00000000D6562A40 00003C (v01 ALASKA A M I    01072009 MSFT 00010013)
[    0.000000] ACPI: HPET 0x00000000D6562A80 000038 (v01 ALASKA A M I    01072009 AMI  00000005)
[    0.000000] ACPI: SSDT 0x00000000D6562AB8 000024 (v01 AMDFCH FCHZP    00001000 INTL 20120913)
[    0.000000] ACPI: UEFI 0x00000000D6562AE0 000042 (v01                 00000000      00000000)
[    0.000000] ACPI: BGRT 0x00000000D6562B28 000038 (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.000000] ACPI: IVRS 0x00000000D6562B60 0000D0 (v02 AMD    AMD IVRS 00000001 AMD  00000000)
[    0.000000] ACPI: SSDT 0x00000000D6562C30 0000F8 (v01 AMD    AMD PT   00001000 INTL 20120913)
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] ACPI: PM-Timer IO Port: 0x808
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0xff] high edge lint[0x1])
[    0.000000] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[    0.000000] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 low level)
[    0.000000] ACPI: IRQ0 used by override.
[    0.000000] ACPI: IRQ9 used by override.
[    0.000000] Using ACPI (MADT) for SMP configuration information
[    0.000000] ACPI: HPET id: 0x10228201 base: 0xfed00000
[    0.000030] ACPI: Core revision 20160930
[    0.007401] ACPI: 7 ACPI AML tables successfully acquired and loaded
[    0.866606] PM: Registering ACPI NVS region [mem 0xdad9e000-0xdb15cfff] (3928064 bytes)
[    0.894811] ACPI: bus type PCI registered
[    0.894815] acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
[    0.911040] ACPI: Added _OSI(Module Device)
[    0.911044] ACPI: Added _OSI(Processor Device)
[    0.911047] ACPI: Added _OSI(3.0 _SCP Extensions)
[    0.911050] ACPI: Added _OSI(Processor Aggregator Device)
[    0.911204] ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignored
[    0.911219] ACPI: Executed 2 blocks of module-level executable AML code
[    0.913095] ACPI Error: Needed [Integer/String/Buffer], found [Region] ffff8a03de8ed168 (20160930/exresop-425)
[    0.913102] ACPI Exception: AE_AML_OPERAND_TYPE, Could not execute arguments for [IOB2] (Region) (20160930/nsinit-412)
[    0.916610] ACPI: Interpreter enabled
[    0.916623] ACPI: (supports S0 S3 S4 S5)
[    0.916626] ACPI: Using IOAPIC for interrupt routing
[    0.916731] PCI: Using host bridge windows from ACPI; if necessary, use "pci=nocrs" and report a bug
[    0.921844] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-ff])
[    0.922593] pci 0000:00:01.3: System wakeup disabled by ACPI
[    0.922811] pci 0000:00:03.1: System wakeup disabled by ACPI
[    0.923031] pci 0000:00:07.1: System wakeup disabled by ACPI
[    0.923178] pci 0000:00:08.1: System wakeup disabled by ACPI
[    0.924142] pci 0000:03:00.0: System wakeup disabled by ACPI
[    0.937332] pci 0000:22:00.0: System wakeup disabled by ACPI
[    0.948259] pci 0000:25:00.0: System wakeup disabled by ACPI
[    0.960086] pci 0000:27:00.0: System wakeup disabled by ACPI
[    0.972075] pci 0000:28:00.3: System wakeup disabled by ACPI
[    0.972652] ACPI: PCI Interrupt Link [LNKA] (IRQs 4 5 7 10 11 14 15) *0
[    0.972695] ACPI: PCI Interrupt Link [LNKB] (IRQs 4 5 7 10 11 14 15) *0
[    0.972731] ACPI: PCI Interrupt Link [LNKC] (IRQs 4 5 7 10 11 14 15) *0
[    0.972777] ACPI: PCI Interrupt Link [LNKD] (IRQs 4 5 7 10 11 14 15) *0
[    0.972818] ACPI: PCI Interrupt Link [LNKE] (IRQs 4 5 7 10 11 14 15) *0
[... cut ...]
[    1.013332] system 00:01: Plug and Play ACPI device, IDs PNP0c02 (active)
[    1.013406] pnp 00:02: Plug and Play ACPI device, IDs PNP0b00 (active)
[    1.013525] system 00:03: Plug and Play ACPI device, IDs PNP0c02 (active)
[    1.013739] pnp 00:04: Plug and Play ACPI device, IDs PNP0501 (active)
[    1.013947] system 00:05: Plug and Play ACPI device, IDs PNP0c02 (active)
[    1.014195] pnp: PnP ACPI: found 6 devices
[    1.487574] ACPI: Power Button [PWRB]
[    1.487617] ACPI: Power Button [PWRF]
```

petrov0, 2017-08-07:
Are you sure that the module is really included in this kernel (Kernel: 4.10.0-30-generic (x86_64))? These don't look right:

```
[294982.883528] it87: Found IT8622E chip at 0x290, revision 6

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 01 ff 91 00 00 40 00 00 c0 44 00 00 0f 82 00 
```

Try to set 0x06 to 00:

`isaset 0x295 0x296 0x06 0x00`

and try sensors-detect/sensors again. By the way my BIOS version is 0803 and I don't have any problems with the driver (currently using 4.12.5).

Enigma0, 2017-08-07:
@petrov0 I'm not sure no. Aside from what I already posted I wouldn't know how to know for sure.

petrov0, 2017-08-07:
You can check the strings in the driver if you are afraid to use isaset :). Use:
`strings /lib/modules/<KERNEL VERSION>/kernel/drivers/hwmon/it87.ko | grep it8655`
which in your case should be:
`strings /lib/modules/4.10.0-30-generic/kernel/drivers/hwmon/it87.ko | grep it8655`
You should see it8655 if the correct driver is compiled (as a module).

Enigma0, 2017-08-07:
Yup it's in there.

Now I see this when running `sudo sensors-detect --auto`:
```
Sorry, no sensors were detected.
Either your system has no sensors, or they are not supported, or
they are connected to an I2C or SMBus adapter that is not
supported. If you find out what chips are on your board, check
http://www.lm-sensors.org/wiki/Devices for driver status.
```

groeck, 2017-08-08:
Can we please leave sensors-detect out of the picture ? It has nothing to do with the driver, and only serves as distraction.

0x8622 can only mean that the driver was loaded with force_id, which isn't very helpful since it might hide problems. The it87 dmesg output only means that the driver detected a chip, but not that it was instantiated. The more complete log is filtered and thus not very informative (except that it shows that the DSDT is buggy).

Enigma0, 2017-08-08:
Ah. Yeah I did force it at one point and it seems like it was overriding any other efforts. I ran `sudo modprobe -r it87 && sudo modprobe it87` and `sensors-detect` still says unknown chip ID but sensors is now actually showing some values:

```
it8655-isa-0290
Adapter: ISA adapter
in0:          +1.13 V  (min =  +1.36 V, max =  +1.11 V)  ALARM
in1:          +2.52 V  (min =  +2.38 V, max =  +0.38 V)  ALARM
in2:          +2.02 V  (min =  +2.26 V, max =  +0.69 V)  ALARM
in3:          +2.00 V  (min =  +0.25 V, max =  +0.52 V)  ALARM
in4:          +2.01 V  (min =  +2.56 V, max =  +2.77 V)  ALARM
in5:          +1.90 V  (min =  +2.15 V, max =  +2.43 V)  ALARM
in6:          +2.00 V  (min =  +2.39 V, max =  +2.51 V)  ALARM
3VSB:         +1.67 V  (min =  +2.18 V, max =  +2.54 V)  ALARM
Vbat:         +1.61 V  
+3.3V:        +1.68 V  
fan1:        1259 RPM  (min =   13 RPM)
fan2:        1717 RPM  (min =   10 RPM)
temp1:        +43.0°C  (low  = -49.0°C, high = -70.0°C)  ALARM
temp2:        +30.0°C  (low  = -25.0°C, high = +120.0°C)
temp3:        +33.0°C  (low  = -21.0°C, high = +116.0°C)  sensor = thermistor
intrusion0:  ALARM

amdgpu-pci-2700
Adapter: PCI adapter
fan1:        2612 RPM
temp1:        +43.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```

groeck, 2017-08-08:
Voltages always need to be scaled. That is nothing news. Super-IO chips have limited A/D conversion range which does not match real world voltages. That is what /etc/sensors3.conf is for.

Having said that, there _is_ something wrong: 3VSB, Vbat, and +3.3V should be auto-scaled. I'll need to look into that.

Repeating myself. **Can we please leave sensors-detect out of the picture ? It has nothing to do with the driver, and only serves as distraction.** Please keep in mind that insisting on such irrelevant information after you were informed that is irrelevant doesn't really help. I _hope_ you don't do this on purpose to trigger an adverse reaction. If you do, please stop. If not, please listen. If you don't understand _why_ it is irrelevant, please ask. The problem here is that your insistence on "sensors-detect still says unknown chip ID" almost caused me to go ballistic and miss the wrong 3.3V voltages, which is bad and unnecessary and, by itself, annoying.

Elaborating a bit: sensors-detect is one program, the it87 driver is another. Updating the it87 driver doesn't affect sensors-detect. If sensors-detect didn't recognize the chip before updating the it87 driver, it still won't recognize it afterwards. To use an analogy, if you update Chrome, it won't fix any problems with Firefox either, even though both are web browsers.

If the above offends you, my apologies. If so, my only defense is that it it the muted version, which is _much_ less offensive than my initial reaction and after having a beer to calm me down. I am human, and react with emotions even when I know that I should not.

Enigma0, 2017-08-08:
@groeck I only mentioned sensors-detect again in my case to show that it was in fact irrelevant. (It said it found nothing but clearly `sensors` did). I wasn't trying to assert its worth or value beyond that.

groeck, 2017-08-08:
3.3V voltages should now be correct.

groeck, 2017-08-08:
No further known issues, closing. Please open new issue if there are still problems.

closed 2017-08-08 by groeck

petrov0, 2017-08-08:
Yep, there is no need for scaling in the config file anymore. Thanks @groeck.