groeck/it87 #12: Add support for IT8686E

groeck, 2017-03-20:
Used on Gigabyte AB350-Gaming 3.

groeck, 2017-03-20:
Initial support is incomplete. Temperature sensors report wrong type; CPU temperature reports as "disabled" sensor.

qantourisc, 2017-04-03:
ID from sensor-detect (so google and github results fill find this ticket):
```
Trying family 'ITE'...                                      Yes
Found unknown chip with ID 0x8666
```

groeck, 2017-04-03:
qantourisc: ID 0x8666 would be a different chip, suggesting that this might be a different board. If so, please open a separate issue and provide additional information (board description, BIOS version, output from isadump). If not, we definitely need output from isadump, possibly repeated. I have trouble on my boards (AB350 Gaming and AB350 Gaming 3) to access ISA memory at 0x2e/0x2f, and it would be extremely helpful to get as many isadumps as possible from those boards to see if I am just lucky or if this is a systemic problem.

alley-oop, 2017-05-08:
groeck: I've just received a Gaming3 and a 1600. 
How can I help you "debugging" the sensors on this board?

groeck, 2017-05-08:
@alley-oop: Assuming you use the latest driver available here, what are your problems ?

alley-oop, 2017-05-08:
@groeck : I've cloned compiled and installed your repository.
however I dunno why when I execute $sensors this is the sensor detected:
it8792-isa-0a60

running sensors-detect:
Found unknown chip with ID 0x8686
Found unknown chip with ID 0x8733

any hint? I'm missing something?

qantourisc, 2017-05-08:
@alley-oop possibly: did you also download and compile the repository for the userland tools ?

alley-oop, 2017-05-08:
@groeck : now I've compiled & installed also lm-sensors.
this is the sensors-detect result now:

Driver `to-be-written':
  * ISA bus, address 0xa40
    Chip `ITE IT8686E Super IO Sensors' (confidence: 9)

Driver `it87' (built-in):
  * ISA bus, address 0xa60
    Chip `ITE IT8792E Super IO Sensors' (confidence: 9)

Note: there is no driver for ITE IT8686E Super IO Sensors yet.

groeck, 2017-05-08:
@alley-oop: You are running an old it87 driver or an upstream it87 driver. You will need to build and install the it87 driver from here. Then you'll need to sun "sudo modprobe -r it87; sudo modprobe it87". lm-sensors in this context is quite irrelevant; it tells you what it believes that the kernel supports, but it doesn't really know for sure.

alley-oop, 2017-05-09:
@groeck: yeah, I'm sorry. I've been using the built in "it87" from the vanilla kernel.
I got it to work now.
BTW I see 5 temps, did you find out which temperature is measuring every "sensors"? 
I think "temp5" is the CPU one, "temp3" & "temp6" probably VRM. No idea on temp1/temp2/temp4. 
On the volt side, "in0" is "Vcore", "in5" probably "Vsoc" and "in6" "Vram"?

Just an info, is it possible to have 2 more digits on the volt sensors? I mean "Vcore" 1.33 in BIOS is 1.33XX.

Thank you for your time, glad to have an out of the box support for this motherboard. 👍

groeck, 2017-05-09:
@alley-oop: That looks about right.
The voltage display is from the "sensors" command; the kernel reports temperatures and voltages in millti-degrees C / milli-Volt. Kind of hard to change since other applications may depends on the output. You can always read the raw values from the sysfs files, which would give you another digit.

alley-oop, 2017-05-23:
Hi @groeck I've been testing this sensor for 2 weeks and temperatures/volts/rpms seem fine.
Just an info, when this changes will be mainlined?

thank you.

groeck, 2017-05-23:
@alley-oop: Depends on my time, which is currently short, and on code reviewers, which are all but non-existing for hwmon drivers. Unlikely to make it into 4.13.

alley-oop, 2017-05-23:
@groeck ok, thank you for your work.

bisby, 2017-05-23:
temp3 is reporting as 17-18C (which seems unusually low) for me on a R7 1700 + Gigabyte AX370 Gaming K7. `sensors -u` says there is a -10C offset, but I can't seem to see where the offset is being set?

```
temp3:
  temp3_input: 18.000
  temp3_max: 127.000
  temp3_min: 127.000
  temp3_alarm: 0.000
  temp3_type: 5.000
  temp3_offset: -10.000
  temp3_beep: 1.000
```
It seems unlikely to be running at or below room temperature. Is that a motherboard, it87, or sensors issue? (Sorry if this is off topic, it seemed like this was the best place to put a one off it8686 question)

groeck, 2017-05-23:
Normally that would be /etc/sensors3.conf if set by the kernel, or it may be set by the BIOS. Either case it is wrong. For a 1700X/1800X you should set it to -20 degrees C, or to 0 for 1700.

groeck, 2017-08-08:
Issues should all be fixed, closing.

closed 2017-08-08 by groeck

DeadSix27, 2017-09-22:
`Linux box 4.13.3-041303-generic #201709200606 SMP Wed Sep 20 10:12:46 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

I have the same chip (apparently) but I get:

```
Driver `to-be-written':
  * ISA bus, address 0xa40
    Chip `ITE IT8686E Super IO Sensors' (confidence: 9)

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Note: there is no driver for ITE IT8686E Super IO Sensors yet.
Check http://www.lm-sensors.org/wiki/Devices for updates.
```

Unfortunately the website is down tho, 

But it seems like the other person got this working by compiling his own kernel with it87?

I use the [ubntu provided](http://kernel.ubuntu.com/~kernel-ppa/mainline/) kernels, guess those don't support it?`

groeck, 2017-09-22:
Looks ok to me. Correct, the upstream driver is not up to date. That is a resource problem; I just don't have the time to prepare and submit the necessary upstream patches.

DeadSix27, 2017-09-22:
@groeck not a big deal in my case, just wondering if the driver is actually functioning right now (for this chip)

chama-chomo, 2018-03-08:
Hey, I've just purchased Gigabyte z370 hd3p MoBo which says IT8686 should be the one, unfortunately I do have problem with getting it87 module online. On my previous MoBo Gigabyte AB350 Gaming3 I didin't have any issue with driver downloaded from your page.

Any advices? I guess this is a motherboard issue?

```
[ 4446.021782] it87: it87 driver version unknown
[ 4446.021905] it87: Found IT8686E chip at 0xa40, revision 2
[ 4446.022194] it87: Beeping is supported
[ 4446.022228] ACPI Warning: SystemIO range 0x0000000000000A45-0x0000000000000A46 conflicts with OpRegion 0x0000000000000A45-0x0000000000000A46 (\GSA1.SIO1) (20170831/utaddress-247)
[ 4446.022231] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
```

```
  /  sudo modprobe it87
modprobe: ERROR: could not insert 'it87': Device or resource busy
   /  lsmod |grep it87
   /  
```

groeck, 2018-03-08:
Please consider reading the README file, especially the note about "ignore_resource_conflict".

udovdh, 2018-05-23:
Thanks for the improvements to it87, but the modules from git has an issue:

# make
  CC [M]  /usr/src/it87/it87.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /usr/src/it87/it87.mod.o
  LD [M]  /usr/src/it87/it87.ko
# make install
mkdir -p /lib/modules/4.16.9/kernel/drivers/hwmon
cp it87.ko /lib/modules/4.16.9/kernel/drivers/hwmon/
depmod -a -F /boot/System.map-4.16.9 4.16.9
# modprobe it87
modprobe: ERROR: could not insert 'it87': Exec format error

How to fix this?