groeck/it87 #13: Sensors sometimes fails to detect

tholin, 2017-03-21:
I use a Gigabyte Z97X-Gaming G1 motherboard with a IT8620E chip. There is also another IT8790E chip but I use kernel 4.4.55 and the driver didn't get support for multiple chips until kernel 4.7. Most of the time the IT8620E chip is detected and works fine. Dmesg looks like this when the chip is found and working.

    it87: Found IT8620E chip at 0xa30, revision 4
    it87: Beeping is supported

Looking at the system log the IT8620E has been detected during 459 boots but has failed on at least 6 boots.

3 failures like this:

    it87: Unsupported chip (DEVID=0x620)

2 of these:

    it87: Found IT8620E chip at 0xa20, revision 4
    it87: Beeping is supported
    ACPI Warning: SystemIO range 0x0000000000000A25-0x0000000000000A26 conflicts with OpRegion 0x0000000000000A20-0x0000000000000A2F (\_SB_.PCI0.LPCB.SIO1.RNTR) (20150930/utaddress-254)
    ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver

1 error like this:

    it87: Found IT8620E chip at 0x230, revision 4
    it87: Beeping is supported

The failures has occurred using a whole bunch of different kernels from Sep 2014 up to present day. Something that looks a bit odd is that all those failures are 1 bit off from what it should be. 0x620 is one bit off from 0x8620, 0xa20 is one bit off from 0xa30 and so on. Is the LPC bus expected to be unreliable or is there some hardware problem on my system?

output from sensors-detect:

    Driver `it87' (built-in):
    * ISA bus, address 0xa30
      Chip `ITE IT8620E Super IO Sensors' (confidence: 9)
    * ISA bus, address 0xa40
      Chip `ITE IT8790E Super IO Sensors' (confidence: 9)

groeck, 2017-03-21:
Interesting that you mention it. I have a similar problem with another Gigabyte motherboard, AB350 Gaming 3. Difference is that entire bytes or nibbles sometimes read all ones (FF, Fx, or xF). In your case it looks like some bits sometimes read 0.
Try running something like
    sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f
multiple times and check if you can see an error pattern.

I can't say what causes the problem. I have not heard about the LPC bus being unreliable, but then I am not really an expert on it either.

tholin, 2017-03-21:
I ran `for i in $(seq 1 1000);do isadump -y -k 0x87,0x01,0x55,0x55 0x2e 0x2f >  $i; sleep 0.5; done` and looked at the output. I'm not sure what all the data means but 0x20-0x22 seems to be the chip id and revision. It read as `20: 86 20 04` 995 times and `20: 06 20 04` 5 times.

Looks like that data is getting corrupted sometimes and always the first bit.

Of the 1000 dumps 32 were unique. I don't know what data is expected to change and what should remain static. Here is a list with all the unique dumps with how many times they occurred.
http://sprunge.us/MZgM

The boot flash sits on the LPC bus so I would probably get boot failures if the LPC bus has random bitflips. The flipped bits doesn't seem to be random but I have no idea what cause the bitflips.

groeck, 2017-03-21:
On Tue, Mar 21, 2017 at 11:27:00AM -0700, tholin wrote:
> I ran `for i in $(seq 1 1000);do isadump -y -k 0x87,0x01,0x55,0x55 0x2e 0x2f >  $i; sleep 0.5; done` and looked at the output. I'm not sure what all the data means but 0x20-0x22 seems to be the chip [...]
Only the communication through 0x2e / 0x2f and 0x4e/0x4f, ie
the communication with the Super-IO chip configuration space,
appears to be affected. I don't see the problem with actual
memory space accesses.

Taking an example, register 0xc0 changes its value. This register
defines GPIO pin assignments and should not change. The same is true
for register 0xc8; it defines if the associated GPIO pins are inputs
or outputs.  Again, that value should normally not change.

Overall I suspect that the communication with the Super-IO chip(s)
is flaky. I have no idea why that would be the case, though.

Guenter

groeck, 2017-04-19:
Gigabyte fixed the problem for the AB350 boards with the latest BIOS. Do you have the F9 BIOS installed ? Probably yes, just making sure we have everything covered.

tholin, 2017-04-19:
Yes, I use the latest F9 bios.

groeck, 2017-08-03:
Current state of knowledge: There are two super-io chips on Gigabyte's AM4 boards. The first (8686) works fine until the second chip (8792) is accessed once. After that, access to the first chip is unreliable. The driver blacklists the second chip on AM4 boards, but that only helps if the second chip is not accessed otherwise. For example, running sensors-detect is enough to "kill" access to the first chip. Possibly the problem is the same on other boards.
Solution/fix is currently unknown. Leaving the issue open for now.

groeck, 2017-08-05:
The problem with IT8792E is presumably now understood and appears to be caused by a bug in the chip. Essentially all bets are off if this chip is ever programmed to exit configuration mode. It may be that the problem observed with the 8620E/8790E combination is the same, it that the it8790 has the same problem as the IT8792E. @tholin: It would be great if you can possibly confirm this. It should be easy to modify the driver to not leave configuration mode for IT8790E, similar to IT8792E. If we are lucky, the problem should then disappear.
Caveat: Running sensors-detect will disable configuration mode on the second chip, causing it to trigger the problem.

tholin, 2017-08-05:
I can do some testing. What do I need to do?

I'm still using the 4.4 kernel series without support for the 2nd super-IO chip. Will the IT8790E chip exit configuration mode when the driver doesn't use it?

I've been meaning to upgrade to a more recent kernel but I haven't gotten around to it yet.

groeck, 2017-08-06:
Hmmm ... you are right. Question is if the BIOS accesses the chip and exits configuration mode. If that is the case, and the chip has the problem, we are kind of lost. I'll think about it some more.

groeck, 2017-08-06:
This might work. It ensures that the 2nd chip is in configuration mode before checking the first. Confirmed working with IT8686E/IT8792E.

diff --git a/it87.c b/it87.c
index 1ca0d14..1ad987b 100644
--- a/it87.c
+++ b/it87.c
@@ -2859,6 +2859,7 @@ static int __init it87_find(int sioaddr, unsigned short *address,
                break;
        case IT8790E_DEVID:
                sio_data->type = it8790;
+               doexit = false;
                break;
        case IT8603E_DEVID:
        case IT8623E_DEVID:
@@ -3953,6 +3954,8 @@ static int __init sm_it87_init(void)
        if (err)
                return err;
 
+       __superio_enter(sioaddr[1]);
+
        for (i = 0; i < ARRAY_SIZE(sioaddr); i++) {
                memset(&sio_data, 0, sizeof(struct it87_sio_data));
                isa_address = 0;

tholin, 2017-08-06:
I tested the patch and I think it works. This is what I did.

1. I removed SENSORS_IT87 support from my current kernel (4.4.79), rebuilt it and booted into it.
2. Applied the patch to the current master of this repo and built it with some difficulties.
3. Loaded the new module and made sure both sensors were detected in dmesg.
4. Used my regular system monitor software to make sure the sensors are there and give reasonable values.
5. Ran the same test as before with isadump in a loop. I ran it 3000 times and the dump was identical each time.

Here is what the dump looks like:

         0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
    00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00
    10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    20: 86 20 04 00 00 00 b7 00 00 00 80 40 80 00 00 00
    30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    60: 00 00 0a 00 00 00 00 00 00 00 00 00 00 00 00 00
    70: 00 00 20 00 00 04 00 00 00 00 00 00 00 00 00 00
    80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    c0: 01 04 00 40 00 00 00 00 01 00 00 00 00 00 00 40
    d0: d1 00 08 08 00 00 00 00 00 00 00 00 00 00 00 00
    e0: 00 00 00 00 00 00 00 00 00 27 27 27 00 00 03 6c
    f0: 00 00 00 00 00 00 17 00 00 00 00 40 00 ff 00 ff

I guess this means putting the IT8790E chip in configuration mode solves the problem. If it's done before the IT8620E chip is probed it should also fix the driver loading problems I was having. I have no reliable way to test this because the driver only fails to load every few months.

Perhaps I could try unloading and reloading the module over and over?

The Makefile had some patchs I had to change. I had to set `KERNEL_BUILD   := $(KBUILD_OUTPUT)` so the build would find my kernel build dir. 

The line `cp $(DRIVER).ko $(KERNEL_MODULES)/kernel/$(MOD_SUBDIR)` will expand to something like `cp it87.ko /lib/modules/$(TARGET)/kernel/drivers/hwmon` so the module will be named "hwmon" instead of being put into the folder hwmon. Perhaps setting `MOD_SUBDIR = drivers/hwmon/` would work better and also verifying that the `/lib/modules/$(TARGET)/kernel/drivers/hwmon` folder exists.

groeck, 2017-08-06:
Let's assume the patch fixes the problem. Can you send me the output of /sys/class/dmi/id/sys_vendor and /sys/class/dmi/id/board_name ?

Thanks,
Guenter

tholin, 2017-08-06:
$ cat /sys/class/dmi/id/board_name
    Z97X-Gaming G1
    $ cat /sys/class/dmi/id/sys_vendor
    Gigabyte Technology Co., Ltd.

I'm guessing a fix will not be backported to 4.4? It's about time I upgrade.

groeck, 2017-08-06:
backport: The driver in v4.4 doesn't support two Super-IO chips, so that doesn't really make much sense from a driver perspective and would be difficult. I'll try to get it into 4.9 though.
Please test the latest version of the driver; unless I messed up it should now work for your board. Thanks!

tholin, 2017-08-07:
I tested the latest repo version using isadump looping and all dumps were identical. Looks good.

groeck, 2017-08-08:
Assuming this is fixed, closing. Please reopen if there are still issues or if the problem reappears.

closed 2017-08-08 by groeck