groeck/it87 #14: Add support for IT8606E and IT8626E

groeck, 2017-03-26:
IT8606E is known to be used on Asus Z97-P. Unknown where IT8626E is used. Information and register dumps would be appreciated.

SpiritOffice, 2017-03-28:
Hi groeck,
i have Asus Z97-P and already made a report to https://github.com/openhardwaremonitor/openhardwaremonitor/issues/773
Don't look at the parsed voltages, because they are incorrect. I did try to use 8628 parser in source code and it shows wrong voltages.

asinatra, 2017-07-10:
```
Found unknown chip with ID 0x8606
    (logical device 4 has address 0x290, could be sensors)
```

```
sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n]
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 86 06 03 00 20 41 20 00 4c 00 00 48 11 00 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60: 00 00 0a 00 00 00 00 00 00 00 00 00 00 00 00 00
70: 00 01 20 80 02 04 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
90: f0 04 f0 04 f0 04 f0 04 00 00 00 00 00 00 00 00
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
c0: 01 00 00 08 00 00 00 00 01 00 00 00 00 00 00 00
d0: 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 00 00 00 07 07 07 00 00 02 0c
f0: 00 00 00 00 00 00 00 00 00 00 00 00 7c 00 00 00
```

```
sudo isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n]
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 19 c8 5c 83 37 ff 07 37 ff 07 64 00 30 eb 72 f8
10: f3 59 8c 70 c0 00 00 00 00 01 01 4d 67 f1 00 00
20: 48 ba aa aa ba ba ba 8d 86 33 26 29 29 29 29 8d
30: e2 0f f0 01 a6 76 71 11 95 72 de a5 a8 6b ab 72
40: a3 d2 ba 6c 65 57 3b 51 ad 6a d4 00 5f 40 08 b6
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00
60: 00 7f 7f ff 00 00 0f ff 00 7f 7f ff 00 00 0f ff
70: 00 7f 7f ff 00 00 0f ff 7f 7f 7f 80 00 00 0f 80
80: ff ff ff ff 00 00 00 00 02 30 01 02 01 9b e0 de
90: 00 00 00 00 00 00 00 ff 40 00 00 00 00 ae 00 5c
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80
b0: cf c4 ff ff ff ff ff ff ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

promeneur, 2017-10-01:
Asus Z97-P

Probing for Super-I/O at 0x2e/0x2f
...
Trying family `ITE'... Yes
Found unknown chip with ID 0x8606
(logical device 4 has address 0x290, could be sensors)

sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
root's password:
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 06 03 00 20 41 20 00 4c 00 00 48 11 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 20 80 02 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: f0 04 f0 04 f0 04 f0 04 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 01 00 00 08 00 00 00 00 01 00 00 00 00 00 00 00 
d0: 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 07 07 07 00 00 02 0c 
f0: 00 00 00 00 00 00 00 00 00 00 00 00 7c 00 00 00 

sudo isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 19 c8 55 8f 37 ff 07 37 ff 07 64 00 30 d2 48 a0 
10: e9 6a 99 70 c0 80 80 80 03 04 03 49 88 d9 00 00 
20: 4a ba ab a8 ba ba ba 8d 85 23 1d 20 20 20 20 8b 
30: 2a 17 d2 43 2f 74 e8 1d a4 4d ea 9e 48 bc f4 67 
40: 99 f9 cd 4a 9d b1 82 dc ad 6a d4 00 5f 40 17 79 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 3d 00 00 0f 67 80 28 46 56 2d 00 0f 56 
70: 80 28 46 ff 00 00 0f ff 7f 7f 7f 80 00 00 0f 80 
80: ff ff ff ff 00 00 00 00 02 30 01 02 01 d1 e0 de 
90: 00 00 00 00 00 00 00 ff 40 00 00 00 00 27 00 34 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80 
b0: bf 5e ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
x@linux-bmy3:~>

groeck, 2017-10-02:
Experimental support added. Please report output of "sensors" command, and let me know if it deviates from BIOS output.

promeneur, 2017-10-02:
how to test it ?

promeneur, 2017-10-02:
> sensors
acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +105.0°C)
temp2:        +29.8°C  (crit = +105.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +34.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:         +33.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:         +33.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:         +34.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:         +29.0°C  (high = +80.0°C, crit = +100.0°C)

it8606-isa-0290
Adapter: ISA adapter
in0:          +0.89 V  (min =  +0.28 V, max =  +0.50 V)  ALARM
in1:          +2.23 V  (min =  +0.80 V, max =  +2.52 V)
in2:          +2.05 V  (min =  +1.39 V, max =  +0.56 V)  ALARM
in3:          +2.02 V  (min =  +0.35 V, max =  +2.78 V)
in4:          +2.23 V  (min =  +0.92 V, max =  +1.97 V)  ALARM
in5:          +2.23 V  (min =  +1.90 V, max =  +2.81 V)
in6:          +2.23 V  (min =  +2.26 V, max =  +0.86 V)  ALARM
3VSB:         +3.38 V  (min =  +2.47 V, max =  +5.86 V)
Vbat:         +3.19 V  
+3.3V:        +3.34 V  
fan1:         739 RPM  (min =   35 RPM)
fan2:         620 RPM  (min =   19 RPM)
fan3:         728 RPM  (min =   12 RPM)
temp1:        +36.0°C  (low  =  -7.0°C, high = -103.0°C)  ALARM  sensor = thermistor
temp2:        +29.0°C  (low  = +74.0°C, high = -51.0°C)  ALARM  sensor = thermistor
temp3:        +32.0°C  (low  = -79.0°C, high = -99.0°C)  ALARM  sensor = thermistor
intrusion0:  OK

sntr, 2017-10-02:
My result:
```

it8606-isa-0290
Adapter: ISA adapter
in0:          +0.86 V  (min =  +0.18 V, max =  +2.71 V)
in1:          +2.23 V  (min =  +0.01 V, max =  +2.88 V)
in2:          +2.04 V  (min =  +1.40 V, max =  +2.00 V)  ALARM
in3:          +2.04 V  (min =  +0.20 V, max =  +1.36 V)  ALARM
in4:          +2.23 V  (min =  +1.37 V, max =  +2.17 V)  ALARM
in5:          +2.23 V  (min =  +1.98 V, max =  +2.66 V)
in6:          +2.23 V  (min =  +1.28 V, max =  +2.02 V)  ALARM
3VSB:         +3.38 V  (min =  +2.74 V, max =  +4.10 V)
Vbat:         +3.19 V
+3.3V:        +3.38 V
fan1:        2647 RPM  (min =   33 RPM)
fan2:        1800 RPM  (min =   25 RPM)
fan3:        1303 RPM  (min =   10 RPM)
temp1:        +51.0°C  (low  = -46.0°C, high = -29.0°C)  ALARM  sensor = thermistor
temp2:        +34.0°C  (low  = +76.0°C, high = -70.0°C)  ALARM  sensor = thermistor
temp3:        +41.0°C  (low  = -107.0°C, high = +69.0°C)  sensor = thermistor
intrusion0:  OK

```

groeck, 2017-10-03:
Looks as good as it can get. I would have hoped that one of the temperature sensors is connected to the CPU, but it doesn't look like that is the case unless the chip has very unusual registers. I don't think I'll be able to add support for IT8626E, so I'll drop that and declare the problem solved.

closed 2017-10-03 by groeck

promeneur, 2017-10-03:
when can we expect a new lm-sensors ?

promeneur, 2017-10-03:
perhaps it is not the right word. sensors-detect does not know this new version of driver. we need a new one to get a good installation. no ?

groeck, 2017-10-03:
Please read the last paragraph of the README file.

groeck, 2017-10-03:
Maybe the question is when the upstream driver will be updated. Answer is that I don't know. I'll have to split the changes into 20+ (probably 30+ by now) individual patches, and I just don't have the time to do that.

groeck, 2017-10-03:
On Tue, Oct 03, 2017 at 07:11:18AM -0700, promeneur wrote:
> why not some kind of plugin technology to update when there is new chipset supported ?
> 
Feel free to submit patches introducing it to help me understand
what you mean.

Thanks,
Guenter