groeck/it87 #15: Looking for DSDT source from ASUS boards

groeck, 2017-03-26:
ASUS boards typically require acpi_enfore_resources=lax to be able to load hardware monitoring drivers in Linux. While this can probably not be avoided, it adds risk if used since the BIOS and Linux may access the Super-IO chip at the same time. Maybe there is a common means to locak the Super-IO chip for Linux while it is being accessed. The ACPI DSDT maybe useful to determine if there is a means to do that.
If you have an ASUS board, I would appreciate if you can run
    sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.dat
    iasl -d dsdt.dat
and publish the resulting dsdt.asl. That might help finding out if it is possible to lock the chip for Linux access, or maybe access it through ACPI.

codeworkx, 2017-03-26:
ASUS PRIME-X370 PRO
[dsdt.dat.zip](https://github.com/groeck/it87/files/870953/dsdt.dat.zip)

matthew-r, 2017-03-26:
ASUS PRIME B350-PLUS
[dsdt.dsl.zip](https://github.com/groeck/it87/files/871044/dsdt.dsl.zip)

groeck, 2017-03-26:
Does the ASUS PRIME B350-PLUS require acpi_enforce_resources=lax for the it87 driver to load ?

matthew-r, 2017-03-27:
Just added it to boot, no acpi_enforce_resources set in cmdline/grub - was able to load it87 successfully on boot and manually. 

[   10.417568] it87: loading out-of-tree module taints kernel.
[   10.417946] it87: Found IT8655E chip at 0x290, revision 6
[   10.417966] it87: Beeping is supported

sterkyr, 2017-03-27:
CROSSHAIR VI HERO v: Rev 1.xx UEFI: American Megatrends v: 1002
[dsdt.dsl.zip](https://github.com/groeck/it87/files/872654/dsdt.dsl.zip)

Soyasi, 2017-03-28:
inxi -M
Machine:   Device: desktop Mobo: ASUSTeK model: CROSSHAIR VI HERO v: Rev 1.xx serial: XXXXX
  UEFI: American Megatrends v: 0902 date: 03/11/2017
[dsdt.dat.zip](https://github.com/groeck/it87/files/875167/dsdt.dat.zip)

NoXPhasma, 2017-04-04:
The module will load fine for me without `acpi_enfore_resources=lax` with ASUS Prime x370-Pro (Bios 0515) on Linux Mint 18.1 (4.10.8-041008-generic).

```
[    7.888470] it87: loading out-of-tree module taints kernel.
[    7.888510] it87: module verification failed: signature and/or required key missing - tainting kernel
[    7.888678] it87: Found IT8665E chip at 0x290, revision 6
[    7.888707] it87: Beeping is supported
```

The dsdt was already posted by @codeworkx, so I skip it here, unless you need it several times.

dariusMangold, 2017-06-04:
Asus E3 PRO GAMING V5

https://gist.github.com/dariusMangold/726c4cb176f5df0df2a81246f428b658

subcreator, 2017-07-24:
Asus p10s ws
[dsdt.dsl.gz](https://github.com/groeck/it87/files/1169441/dsdt.dsl.gz)

keantoken, 2017-08-07:
$ inxi -M
Machine:   Device: desktop Mobo: ASUSTeK model: ROG STRIX B350-F GAMING v: Rev X.0x
           UEFI [Legacy]: American Megatrends v: 0223 date: 05/04/2017
I have never updated the BIOS.
[dsdt.dsl.zip](https://github.com/groeck/it87/files/1203897/dsdt.dsl.zip)

jmnel, 2017-11-20:
ASUS X370-PRO.
[dsdt.dat.zip](https://github.com/groeck/it87/files/1488957/dsdt.dat.zip)

combolek, 2017-12-20:
[dsdt.dsl.gz](https://github.com/groeck/it87/files/1573948/dsdt.dsl.gz)
Asus ROG Strix X370-F Gaming

Zmuogs, 2018-01-17:
Asus ROG Strix B350-f Gaming
[dsdt.dat.zip](https://github.com/groeck/it87/files/1640517/dsdt.dat.zip)
[dsdt.dsl.zip](https://github.com/groeck/it87/files/1640518/dsdt.dsl.zip)

ThyMYthOS, 2018-01-18:
Asus PRIME X399-A
[dsdt.dat.gz](https://github.com/groeck/it87/files/1643508/dsdt.dat.gz)
[dsdt.dsl.gz](https://github.com/groeck/it87/files/1643509/dsdt.dsl.gz)

raveit65, 2018-02-10:
Asus Prime X370-Pro with latest bios from Jan 2018
[dsdt.dsl.zip](https://github.com/groeck/it87/files/1713740/dsdt.dsl.zip)
Btw. sensors values from it87 are really different from that what i see in UEFI bios.
Only temp1 seems to be valid.

codeguy007, 2018-03-12:
Asus Prime A320M-K with current bios and Ryzen 3 2200G
[dsdt.dsl.gz](https://github.com/groeck/it87/files/1801070/dsdt.dsl.gz)

bl4uni, 2018-05-05:
It loads fine without `acpi_enfore_resources=lax`:
```
May  5 18:36:58 System-Product-Name kernel: [ 4460.988573] it87: Found IT8732F chip at 0x290, revision 6
May  5 18:36:58 System-Product-Name kernel: [ 4460.988598] it87: Beeping is supported
May  5 18:36:58 System-Product-Name kernel: [ 4460.988954] it87: Found IT8732F chip at 0xfff8, revision 15
May  5 18:36:58 System-Product-Name kernel: [ 4460.988978] it87: Beeping is supported
May  5 18:36:58 System-Product-Name systemd[1]: Starting IIO Sensor Proxy service...
May  5 18:36:58 System-Product-Name systemd[1]: Started IIO Sensor Proxy service.
```
[dsdt.dsl.tar.gz](https://github.com/groeck/it87/files/1976936/dsdt.dsl.tar.gz)

groeck, 2018-05-05:
@bl4uni: You only get those messages with the upstream driver and force_id. I would suggest to use the driver published here, and you should not use force_id unless you have to.

bl4uni, 2018-05-05:
Okay, I probably missinterpreted that. I'm quite the beginner in terms of linux so please bear with me. I only used force_id for the other issue and unloaded it afterwards as stated in the readme. Does this also mean the `dsdt.dsl` won't help you or is that output independent of the driver? If it isn't I'll upload a new one after I installed the driver.

keantoken, 2018-05-05:
It should work for this board without having to do anything funny. After compiling and make install, I have to do 

sudo modprobe it87

Maybe do sensors-detect afterward, and then I can see all the sensors in psensor.

The voltages display wrong because no one knows the multipliers for this board. I can't test because I don't have Windows. Maybe you could record the voltages from Windows and the voltages from Linux so we can figure out the scaling.

groeck, 2018-05-05:
dsdt.dsl is from the BIOS and driver independent. The data won't help anyway since the board does not require acpi_enfore_resources=lax, and thus there is no resource conflict with the BIOS.
As for sensors-detect, please read the last paragraph in README.

whatsitsname, 2018-05-06:
x370-pro -  bois rev. 4011
[dsdt.tar.gz](https://github.com/groeck/it87/files/1977733/dsdt.tar.gz)

module loaded with

`sudo modprobe it87 ignore_resource_onfilict=1 mmio=1`

I find the ignore_resource only necessary with using MMIO.
I get occasional freezes when running without MMIO.
If MMIO is loaded a boot I sometimes have issue with CPU fan going to 100%.

MMIO loaded after login seems more stable.  Though in4-6  behave strangely.
[image]

Maybe useful.
Thanks for your time.

KeithMyers, 2018-05-12:
ASUS X99-E-10G-WS

Needed the acpi_enfore_resources=lax on the kernel load command line to get the nct6775 driver to load without acpi conflicts.  Only the coretemp driver was loading correctly until I added the command.
[dsdt.dsl.zip](https://github.com/groeck/it87/files/1997991/dsdt.dsl.zip)

gvcastellon, 2018-05-15:
`inxi -M
.Machine:   Type: Desktop Mobo: ASUSTeK model: ROG STRIX B350-F GAMING v: Rev X.0x 
           serial: <root required> UEFI [Legacy]: American Megatrends v: 4011 date: 04/19/2018`
[dsdt.dat.zip](https://github.com/groeck/it87/files/2002994/dsdt.dat.zip)
[dsdt.zip](https://github.com/groeck/it87/files/2002995/dsdt.zip)

KeithMyers, 2018-05-26:
Thought with my new issue comment about the ASUS Crosshair VII Hero not being cooperative with the fan outputs with the it87 driver that you might find the dsdt outputs helpful with your investigation.
[dsdt.dat.dsl.tar.gz](https://github.com/groeck/it87/files/2041945/dsdt.dat.dsl.tar.gz)

goncalossilva, 2018-07-01:
ASUS Crosshair VII Hero

[dsdt.zip](https://github.com/groeck/it87/files/2153203/dsdt.zip)

closed 2018-07-09 by groeck

KeithMyers, 2018-07-09:
This is unfortunate as I just discovered that the driver will not load on BIOS 0702 on the ASUS Crosshair VII Hero motherboard due to ACPI conflicts. I tried the acpi_enforce_resources=lax kernel command parameter and it made no difference.  I had to revert to BIOS 0601 to get the SIO chip outputs back.

groeck, 2018-07-09:
Try loading the driver with ignore_resource_conflict=1 module parameter.

KeithMyers, 2018-07-09:
Thanks for the tip.  I will try again with BIOS 0702 and the new parameter.