groeck/it87 #16: Informations report for Ryzen 1800X with Gigabyte GA-AX370 Gaming 5

koji54, 2017-04-04:
Hi,

below some reporting with module IT87 loaded:

```
sudo sensors-detect                                                      
# sensors-detect revision $Revision$
# System: Gigabyte Technology Co., Ltd. AX370-Gaming 5 [Default string]
# Kernel: 4.10.8-1-MANJARO x86_64
# Processor: AMD Ryzen 7 1800X Eight-Core Processor (23/1/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8686
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8733
```

```
sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7                          
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 
20: 86 86 ff 00 00 20 33 20 00 00 80 40 c8 18 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 10 0a 00 0a 20 00 00 00 00 00 00 00 00 00 00 
70: 00 00 20 00 00 06 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[... 106 lines cut ...]
           driver: snd_virtuoso port: c000 bus-ID: 0b:04.0 chip-ID: 13f6:8788
           Sound: Advanced Linux Sound Architecture v: k4.10.8-1-MANJARO
Network:   Card-1: Intel I211 Gigabit Network Connection
           driver: igb v: 5.4.0-k port: e000 bus-ID: 06:00.0 chip-ID: 8086:1539
           IF: ens1 state: up speed: 1000 Mbps duplex: full mac: 1c:1b:0d:9a:4f:52
           Card-2: Qualcomm Atheros Device e0b1 driver: alx port: d000 bus-ID: 07:00.0 chip-ID: 1969:e0b1
           IF: enp7s0 state: down mac: 1c:1b:0d:9a:4f:50
Drives:    HDD Total Size: 3230.7GB (2.1% used)
           ID-1: /dev/sda model: WDC_WD10EALX size: 1000.2GB serial: WD-WCATR8950221
           ID-2: /dev/sdb model: ST3250823AS size: 250.1GB serial: 3ND1HKF2
           ID-3: /dev/sdc model: SanDisk_SDSSDHII size: 480.1GB serial: 154434400225
           ID-4: /dev/sdf model: SAMSUNG_HD103SJ size: 1000.2GB serial: S246J9CZ604842
           ID-5: /dev/sdg model: ST3500320AS size: 500.1GB serial: 9QM6Y918
Partition: ID-1: / size: 48G used: 18G (40%) fs: ext4 dev: /dev/sdc6
           ID-2: /home size: 100G used: 44G (46%) fs: ext4 dev: /dev/sdc7
           ID-3: swap-1 size: 2.37GB used: 0.00GB (0%) fs: swap dev: /dev/sdg5
Sensors:   System Temperatures: cpu: 38.0C mobo: 30.0C gpu: 35.0
           Fan Speeds (in rpm): cpu: 615 fan-3: 0 fan-4: 0 fan-5: 0
Info:      Processes: 291 Uptime: 1:48 Memory: 2402.5/16043.8MB Init: systemd v: 232 Gcc sys: 6.3.1
           Client: Shell (zsh 5.3.1 running in gnome-terminal-) inxi: 2.3.8

```

Thanks a lot.
Bye !

Knudch, 2017-04-05:
It looks strange that in the first part of your clip...it states "Found unknown chip with ID 0x8733"

In the second part it find "it8792-isa-0a60"

To my knowlegde there are only 2 ITE chips on this board it8686 and it8792

I have allmost the same board, GA-AX370-Gaming K7 with a ryzen 1700x

It find no 0x8733 chip !
It has the it8686 and the it8792 

br
Knud

koji54, 2017-04-05:
Ok with you about strange 0x8733 ID sensors-detect found.
8733 iteration didn't appears on sensors or xsensors screens.

qantourisc, 2017-04-07:
See: https://github.com/groeck/lm-sensors/issues/16
>  groeck commented 26 days ago
> 
> Money isn't an issue. Time is.
> Question though is what you are asking for. Support for the processor with sensors-detect ? That would be a couple of lines of python code, but doesn't really mean much. Support for the processors with the "sensors" command ? That has nothing to do with this repository; it would have to be added with a hardware monitoring driver, probably by extending the existing k10temp and fam15h_power drivers. That might require a few lines of C code if the functionality is the same as with family 15/16 chips, or it might require new drivers. Patches for all those changes are welcome.

qantourisc, 2017-04-07:
Btw: consider removing Ryzen 1800X from the title, as it contains no data about Ryzen and might confuse people (like me ;) ).

groeck, 2017-04-08:
To reiterate, the chip ID for IT9792E is 0x8733. ITE is causing a lot of confusion with that one ;-).

m00dawg, 2017-04-13:
I'm not sure if this is the right bug to comment on, but I noticed after upgrading to the F5 BIOS for the AX370, the CPU temp reported by 'sensors' for the `it8686-isa-0a40` seems to be too low:

`temp3:        +18.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI`

18C is below my ambient (around 24C). I'm on a Ryzen 1700 so I suspect the +20C confusion may be happening here since 38C is a lot more believable. I _think_ the BIOS is reading correctly (whereas I think previously it may have been reading too high) but it's hard to validate. Quite happy with Ryzen as a whole but why oh why AMD opted to make temperature readings so confusing is baffling.

groeck, 2017-04-13:
Yes, I noticed that the temperature offset is now set to 35 degrees C for 1700X, which is too much as well. Looks like Gigabyte added another 15 degrees C to the offset. You can override the value using the tempX_offset attribute.

roy7, 2017-04-18:
My Gaming K7 has the same:

Found unknown chip with ID 0x8686
Found unknown chip with ID 0x8733

Which is how I found this github/thread. Is there any more information you need that would be helpful in support for these sensor chipsets?

groeck, 2017-04-18:
@roy7: Can you try the latest version of the driver ? It may not be perfect, but it should work.

roy7, 2017-04-18:
@groeck Should I not be worried about the crashing reported in https://github.com/groeck/it87/issues/18 ? :)

groeck, 2017-04-18:
@roy7: Definitely for IT8792E (id 0x8733). That problem is also reported to be seen on Windows. I now disabled this chip for the AB350 Gaming boards. Can you attach the output of /sys/class/dmi/id/board_name so I can disable it for your board as well ?

Note that there is a separate problem with segmentation faults, mostly seen when running gcc. This is, as far as I can see, observed on all boards from all vendors and not related to the Super-IO chip. The problem with #18 is that it intermixes all those issues, and at this point I have no idea if it is related to the 8792, the 8686, or to the general crash problem. With the latest BIOS I have not seen a problem with the 8686 myself (yet).

roy7, 2017-04-18:
The board_name is "AX370-Gaming K7". If the problems are also seen on Windows, perhaps Gigabyte will get involved with fixing it. :) I know they don't care about Linux sadly. I bought my K7 before realizing Ubuntu didn't like Gigabyte boards, and that Gigabyte doesn't care for Linux. Live and learn!

Knudch, 2017-04-18:
The latest driver I pulled here from this site worked on my AX370 K7 board
I had no isssues with that board until it suddenly died
Waiting for replacement

1700x, 2*16GB ram(dual rank), R7-250 GPU, 90G sata SSD, 1TB Plextor NVMe
Knud

m00dawg, 2017-04-23:
On the note of the offset, the 15C does seem to be more believable. 24C at idle after making the change. Took me a while to remember how to do it so in my case, I added the following to /etc/rc.local:

```
# Offset temperature for AX370 CPU temp for Ryzen 1700
echo "15" > /sys/devices/platform/it87.2624/hwmon/hwmon0/temp3_offset
```

viktorheli, 2017-05-23:
Gigabyte AX-370 gaming k7 chip found, but sensors now display info form this chip. 
Output from sensors-detect:
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found `ITE IT8686E Super IO Sensors'                        Success!
    (address 0xa40, driver `to-be-written')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found `ITE IT8792E Super IO Sensors'                        Success!
    (address 0xa60, driver `it87')

Output from sensors:
it8686-isa-0a40
Adapter: ISA adapter
in0:          +1.27 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.06 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.25 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.90 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.12 V
fan1:         914 RPM  (min =   10 RPM)
fan2:         692 RPM  (min =    0 RPM)
fan3:         762 RPM  (min =    0 RPM)
fan4:        1527 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +39.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +46.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +45.0°C  (low  =  +0.0°C, high = +80.0°C)  sensor = AMD AMDSI
temp4:        +49.0°C  (low  = +127.0°C, high = +127.0°C)
temp5:        +60.0°C  (low  =  +0.0°C, high = -117.0°C)  sensor = thermistor
temp6:        +34.0°C  (low  = +127.0°C, high = +127.0°C)
intrusion0:  ALARM

sudo modprobe it87 force_id=0xa60
modprobe: ERROR: could not insert 'it87': No such device

:(

What can I help? 

Any debug from my motherboard?

qantourisc, 2017-05-23:
> sudo modprobe it87 force_id=0xa60
modprobe: ERROR: could not insert 'it87': No such device

This means you do not have the it87 module.
Either install it through your package manager or obtain the latest version from this repo, and apply overwrite the source files of a kernel and compile it. How to do either is rather out of scope here (imo).

viktorheli, 2017-05-23:
Not shure.
sudo modprobe it87 without force_id - doing successfully

and sensors display info from other IT* chip installed on motherboard. 
 lsmod | grep -i it87
it87                   57344  0
hwmon_vid              16384  1 it87

I think what no support IT8792E in it87 module at this time in this repo.

groeck, 2017-05-23:
Done. Load module with "sudo modprobe it87 blacklist=0" to override the blacklist. Please note that your system will hang maybe one out of 10 times when loading the module with blacklist=0. Good luck.

viktorheli, 2017-05-24:
Hi! Big thanks! I do "sudo modprobe it87 blacklist=0" > "sensors" > "sudo modprobe -r it87" > "sudo modprobe it87 blacklist=0"  15 times. System not hung - all works fine. "sensors" display all info from both chips:

sensors
it8686-isa-0a40
Adapter: ISA adapter
in0:          +1.27 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.06 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.25 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.90 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.12 V
fan1:         915 RPM  (min =   10 RPM)
fan2:         687 RPM  (min =    0 RPM)
fan3:         767 RPM  (min =    0 RPM)
fan4:        1541 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +37.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +43.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +43.0°C  (low  =  +0.0°C, high = +80.0°C)  sensor = AMD AMDSI
temp4:        +47.0°C  (low  = +127.0°C, high = +127.0°C)
temp5:        +58.0°C  (low  =  +0.0°C, high = -117.0°C)  sensor = thermistor
temp6:        +31.0°C  (low  = +127.0°C, high = +127.0°C)
intrusion0:  ALARM

it8792-isa-0a60
Adapter: ISA adapter
in0:          +1.26 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +0.59 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.04 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:        +3.33 V  (min =  +0.00 V, max =  +5.56 V)
in4:          +1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.49 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.31 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.16 V
fan1:        1247 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +51.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

 dmesg | grep "Ср май 24" | grep it87 | grep -i IT8733E | wc -l
15

If I found any problems - I write here. 

Big thanks for support.

viktorheli, 2017-05-24:
Info about my system, maybe it will be useful:

 inxi -Fxxx
System:    Host: vik-home Kernel: 4.10.3 x86_64 (64 bit gcc: 5.4.0) Console: tty 1 dm: sddm,sddm
           Distro: Ubuntu 16.04 xenial
Machine:   Mobo: Gigabyte model: AX370-Gaming K7 v: Default string serial: Default string
           Bios: American Megatrends v: F3 date: 04/07/2017
CPU:       Octa core AMD Ryzen 7 1800X Eight-Core (-HT-MCP-) cache: 4096 KB
           flags: (lm nx sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm) bmips: 57600
           clock speeds: min/max: 2200/3600 MHz 1: 3600 MHz 2: 2200 MHz 3: 3600 MHz 4: 2200 MHz 5: 3600 MHz
           6: 2200 MHz 7: 3600 MHz 8: 3600 MHz 9: 3600 MHz 10: 2200 MHz 11: 3600 MHz 12: 3600 MHz 13: 3600 MHz
           14: 2200 MHz 15: 3600 MHz 16: 2200 MHz
Graphics:  Card: NVIDIA Device 1b81 bus-ID: 09:00.0 chip-ID: 10de:1b81
           Display Server: X.org 1.18.4 driver: nvidia tty size: 112x44 Advanced Data: N/A for root out of X
Audio:     Card-1 Advanced Micro Devices [AMD] Device 1457
           driver: snd_hda_intel bus-ID: 12:00.3 chip-ID: 1022:1457
           Card-2 NVIDIA Device 10f0 driver: snd_hda_intel bus-ID: 09:00.1 chip-ID: 10de:10f0
           Card-3 Microsoft LifeCam Studio driver: USB Audio usb-ID: 001-002 chip-ID: 045e:0772
           Sound: Advanced Linux Sound Architecture v: k4.10.3
Network:   Card-1: Intel I211 Gigabit Network Connection
           driver: igb v: 5.4.0-k port: e000 bus-ID: 06:00.0 chip-ID: 8086:1539
           IF: ens1 state: up speed: 1000 Mbps duplex: full mac: 1c:1b:0d:9e:13:92
           Card-2: Qualcomm Atheros Device e0b1 driver: alx port: d000 bus-ID: 07:00.0 chip-ID: 1969:e0b1
           IF: enp7s0 state: down mac: 1c:1b:0d:9e:13:90
Drives:    HDD Total Size: 13252.6GB (0.3% used)
           ID-1: /dev/nvme0n1 model: N/A size: 1000.2GB serial: S3ETNX0J308561K
           ID-2: /dev/sda model: HGST_HUS726060AL size: 6001.2GB serial: K1H05R8D temp: 41C
           ID-3: /dev/sdb model: HGST_HUS726060AL size: 6001.2GB serial: K1H03HXD temp: 42C
           ID-4: /dev/sdc model: Samsung_SSD_850 size: 250.1GB serial: S2R4NB0J110006D temp: 0C
           ID-5: /dev/sdd model: ST31000528AS size: 1000.2GB serial: 9VP252J4 temp: 34C
Partition: ID-1: / size: 211G used: 20G (10%) fs: ext4 dev: /dev/sdc1
           ID-2: swap-1 size: 20.06GB used: 0.00GB (0%) fs: swap dev: /dev/sdc2
RAID:      System: supported: linear multipath raid0 raid1 raid6 raid5 raid4 raid10
           No RAID devices: /proc/mdstat, md_mod kernel module present
           Unused Devices: none
Sensors:   System Temperatures: cpu: 51.0C mobo: 40.0C gpu: 0.0:71C
           Fan Speeds (in rpm): cpu: 690 fan-1: 1252 fan-3: 767 fan-4: 1548 fan-5: 0
Info:      Processes: 321 Uptime: 11:04 Memory: 1883.9/64440.7MB
           Init: systemd v: 229 runlevel: 5 Gcc sys: 5.4.0 alt: 4.9
           Client: Shell (sudo running in tty 1) inxi: 2.2.35

inxi displays info about system fans not correctly.

Gnork, 2017-08-20:
Hey,

I have a Ryzen 7 1800x on a Gigabyte GA-AX370 Gaming 5 (Bios F6 AGESA 1.0.0.6a).
I installed the it87 kernel module on Fedora 26 (kernel 4.12.5) as follows:

```bash
sudo modprobe it87
```
`journalctl -kf` shows:

```bash
Aug 20 12:46:13 localhost.localdomain kernel: it87: Found IT8733E chip at 0xa60, revision 3
Aug 20 12:46:13 localhost.localdomain kernel: it87: Beeping is supported
```

`sudo sensors-detect`:

```bash
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Gigabyte Technology Co., Ltd. AX370-Gaming 5 [Default string]
# Kernel: 4.12.5-300.fc26.x86_64 x86_64
# Processor: AMD Ryzen 7 1800X Eight-Core Processor (23/1/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8686
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8733

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
[... 162 lines cut ...]
Adapter: ISA adapter
in0:          +0.81 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +0.59 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.04 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:        +1.67 V  (min =  +0.00 V, max =  +2.78 V)
in4:          +1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.49 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +1.70 V  (min =  +0.00 V, max =  +2.78 V)
Vbat:         +1.62 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +31.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +26.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
temp3:        +32.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

That's not a lot of information there. I tried running it87 with several `force_id` settings, but nothing showed more information.

Is there anything I'm missing?

Best regards,
Christoph

groeck, 2017-08-21:
> Hey,
> 
> I have a Ryzen 7 1800x on a Gigabyte GA-AX370 Gaming 5 (Bios F6 AGESA 1.0.0.6a).
> I installed the it87 kernel module on Fedora 26 (kernel 4.12.5) as follows:
Did you install the latest driver from this site ?

Guenter

Gnork, 2017-08-21:
Yes, I cloned the latest source code and ran `make clean && make && sudo make install`. Had to adjust the kernel source folder for Fedora beforehand.

groeck, 2017-10-17:
@gnork: There have been various changes in the driver since August. Can you try with the latest driver and report if there are still problems ?

Gnork, 2017-10-21:
@groeck: I tried the latest version from master. Install went a lot smoother this time, without needing to adjust the Makefile. After running `sensors-detect` again, the `sensors` command now gives me a lot more information than last time:

```
it8686-isa-0a40
Adapter: ISA adapter
in0:          +1.39 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.95 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.91 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.22 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.29 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.14 V  
fan1:         470 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +14.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +36.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +43.0°C  (low  =  +0.0°C, high = -117.0°C)  sensor = thermistor
temp6:        +29.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

it8792-isa-0a60
Adapter: ISA adapter
in0:          +1.36 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +0.59 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.04 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:        +3.33 V  (min =  +0.00 V, max =  +5.56 V)
in4:          +1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.50 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.38 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.07 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +26.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

groeck, 2017-10-22:
Thanks. Do you really have only one active fan ?

SimonDedman, 2017-12-05:
@m00dawg Tim did you get everything working successfully in the end? I have an AX370 w/ Ryzen 7 1700 as well so before I get my hards dirty it'd be good to find out whether i'm likely to fail! (and any tips/order of operations much appreciated (also your solar tracker code looks cool)).
Cheers!

m00dawg, 2017-12-05:
@SimonDedman yep and in fact I don't think I've ever had a system freeze or crash (symptoms there were reported early on). I may have just been lucky though, but so far so good - running with this setup since April. I don't know if you still need to force the AX370 to be enabled when you compile the module. If you do it's a few simple edits (keeping in mind it was commented out for a reason, so even though I'm having good luck you may not). I also run a custom kernel (4.13.10 currently) on Linux Mint.

All told though I'm quite happy. I have the most pertinent sensors I need setup on Conky and it just goes without really much fuss, other than to recompile the module on a big kernel upgrade.

chinf, 2018-01-13:
I've compiled the latest available version and run this on my system. I've fully mapped all the voltages, fans and temps against the values reported in BIOS (version F8) and through Gigabyte's own EasyTune application (in Windows), and also correlated with HWMonitor (in Windows).

Extract from my sensors output:

```
it8792-isa-0a60
Adapter: ISA adapter
DDR Vtt A/B:    +0.70 V  (min =  +0.55 V, max =  +0.80 V)
Chipset Core:   +1.05 V  (min =  +0.99 V, max =  +1.10 V)
CPU Vdd18:      +1.80 V  (min =  +1.74 V, max =  +1.85 V)
DDR Vpp A/B:    +2.52 V  (min =  +2.38 V, max =  +2.63 V)
3VSB:           +3.33 V  (min =  +3.21 V, max =  +3.40 V)
Vbat:           +3.18 V  
SYS5 fan/pump:  623 RPM  (min =  300 RPM)
SYS6 fan/pump: 2170 RPM  (min =  300 RPM)
SYS4 fan:       547 RPM  (min =  300 RPM)
PCIe X8:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
EC_temp 2:      +27.0°C  (low  =  +0.0°C, high = +60.0°C)  sensor = thermistor
System 2:       +31.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor

it8686-isa-0a40
Adapter: ISA adapter
CPU Vcore:    +1.07 V  (min =  +0.35 V, max =  +1.45 V)
+3.3V:        +3.35 V  (min =  +3.21 V, max =  +3.41 V)
+12V:        +12.17 V  (min = +11.66 V, max = +12.38 V)
+5V:          +5.07 V  (min =  +4.86 V, max =  +5.16 V)
Vcore SOC:    +1.10 V  (min =  +0.90 V, max =  +1.26 V)
CPU Vddp:     +0.90 V  (min =  +0.85 V, max =  +0.95 V)
DRAM A/B:     +1.42 V  (min =  +1.10 V, max =  +1.60 V)
CPU fan:      620 RPM  (min =  300 RPM)
SYS1 fan:     577 RPM  (min =  300 RPM)
SYS2 fan:     653 RPM  (min =  300 RPM)
SYS3 fan:     623 RPM  (min =  300 RPM)
CPUOPT fan:   567 RPM  (min =  300 RPM)
System 1:     +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
Chipset:      +40.0°C  (low  =  +0.0°C, high = +80.0°C)  sensor = thermistor
CPU:          +28.0°C  (low  = +11.0°C, high = +91.0°C)  sensor = AMD AMDSI
PCIe X16:     +32.0°C  (low  =  +0.0°C, high = +60.0°C)  sensor = thermistor
VRM MOS:      +34.0°C  (low  =  +0.0°C, high = +90.0°C)  sensor = thermistor
EC_temp 1:    +28.0°C  (low  =  +0.0°C, high = +60.0°C)  sensor = thermistor
```

I created a .conf file to apply the labels and convert the voltage values for it8792 in5 and it8686 in1, in2, in3. I configured sensors to ignore the intrusion alarm sensors, and also a few of the voltages as they are duplicated between it8792 and it8686.

I applied a temp offset for it8686 temp3 (CPU) to match the value reported in BIOS as closely as possible - I run an R7 1700 - but a different offset value may be needed for X models. The temp alarms I set in BIOS have carried through as the corresponding max thresholds - I didn't set any alarms for System 1, System 2 or PCIe X8 so these retain the default +127 thresholds. I have both external thermistor headers populated.

I set minimum fan speed alarms on all fan sensors as I have all fan headers in use.

gigabyte-ax370-gaming5.conf:
```
chip "it8792-isa-*"
	label in0 "CPU Vcore"
	label in1 "DDR Vtt A/B"
	label in2 "Chipset Core"
	label in4 "CPU Vdd18"
	label in5 "DDR Vpp A/B"

	compute in5 @ * (208/125), @ / (208/125)

	set in0_min 0.35
	set in0_max 1.45
	set in1_min 1.1 / 2
	set in1_max 1.6 / 2
	set in2_min 1.05 * 0.95
	set in2_max 1.05 * 1.05
	set in4_min 1.8 * 0.97
	set in4_max 1.8 * 1.03
	set in5_min 2.5 * 0.95
	set in5_max 2.5 * 1.05
	set in7_min 3.3 * 0.97
	set in7_max 3.3 * 1.03

	label temp1 "PCIe X8"
	label temp2 "EC_temp 2"
	label temp3 "System 2"

	label fan1 "SYS5 fan/pump"
	label fan2 "SYS6 fan/pump"
	label fan3 "SYS4 fan"

	set fan1_min 300
	set fan2_min 300
	set fan3_min 300

	ignore in0
	ignore in3
	ignore in6
	ignore intrusion0

chip "it8686-isa-*"
	label in0 "CPU Vcore"
	label in1 "+3.3V"
	label in2 "+12V"
	label in3 "+5V"
	label in4 "Vcore SOC"
	label in5 "CPU Vddp"
	label in6 "DRAM A/B"

	compute in1 @ * (33/20), @ / (33/20)
	compute in2 @ * (120/20), @ / (120/20)
	compute in3 @ * (50/20), @ / (50/20)

	set in0_min 0.35
	set in0_max 1.45
	set in1_min 3.3 * 0.97
	set in1_max 3.3 * 1.03
	set in2_min 12 * 0.97
	set in2_max 12 * 1.03
	set in3_min 5 * 0.97
	set in3_max 5 * 1.03
	set in4_min 0.9
	set in4_max 1.26
	set in5_min 0.9 * 0.95
	set in5_max 0.9 * 1.05
	set in6_min 1.1
	set in6_max 1.6

	label temp1 "System 1"
	label temp2 "Chipset"
	label temp3 "CPU"
	label temp4 "PCIe X16"
	label temp5 "VRM MOS"
	label temp6 "EC_temp 1"

	compute temp3 @ + 11, @ - 11

	label fan1 "CPU fan"
	label fan2 "SYS1 fan"
	label fan3 "SYS2 fan"
	label fan4 "SYS3 fan"
	label fan5 "CPUOPT fan"

	set fan1_min 300
	set fan2_min 300
	set fan3_min 300
	set fan4_min 300
	set fan5_min 300

	ignore in7
	ignore in8
	ignore intrusion0
```

groeck, 2018-01-14:
Configuration file added as configs/Gigabyte/GA-AX370-GAMING5.conf. Thanks a lot for the information!

closed 2018-01-14 by groeck

chinf, 2018-01-14:
@groeck thanks for picking this up. I'm not sure if you used my file verbatim or not - but as other people may not have fans installed on all headers, if you haven't already it's best to omit the "set fanX_min 300" lines.

All the other min/max values are fine for all the stock and OC configurations I've seen shared on forums like overclock.net.

groeck, 2018-01-14:
I used the file as-is. Good point regarding the fan configurations. If the matter comes up, we can make adjustments (and/or add comments to the file) as needed.

mettacrawler, 2018-04-21:
Even Bionic didn't come out with a working it87
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1740736