groeck/it87 #18: Mothboard resets when reading/setting on GA-AX370-Gaming 5 (at random)

qantourisc, 2017-04-07:
Using version https://github.com/groeck/it87/commit/75fdbbca75bf93ecbe5277998f59f77ca7319570
With BIOS F4.

Increasing the amount and frequency of PWM's/FANs you read out seem to increase your changes for a reset. (I have gotten a reset in under 6 minutes when testing as fast as the PC could handle.)

I'm not entirely sure if only reading sensors is enough to crash it.

Panic settings in case you are wondering:
> kernel.panic = 0
> kernel.panic_on_io_nmi = 0
> kernel.panic_on_oops = 0
> kernel.panic_on_rcu_stall = 0
> kernel.panic_on_unrecovered_nmi = 0
> kernel.panic_on_warn = 0
> kernel.unknown_nmi_panic = 0
> vm.panic_on_oom = 0

I'll figure out if only reading from the sensors also crashes the computer.
(But still have some things to do before I can do more testing.)

If there is anything else you would like to see me  test that can help you/me debug (and fix) this, please do tell.

Chipsets being used: it8686-isa-0a40 it8792-isa-0a60, more specifically: IT8792E and IT8686E

I have also created a ticket at Gigabyte requestion documentation and/or BIOS update.

groeck, 2017-04-08:
Gigabyte will most likely tell you that they don't support Linux, and will ask you to reproduce the problem under Windows.

Please try to identify if the reset occurs when reading data from the IT8686E, from the IT8792E, or both. Also, please provide the specific attributes which, when read, result in the reset.

I am seeing random hangups when accessing the SIO registers on the IT8792E. This may be related.

qantourisc, 2017-04-08:
Could do this. For the support ticket, not rejected so far, they say they could not reproduce it ... Personally I'm wary of this claim, so I asked what kernel they used ;)
Also I'm note even sure what software to use in Windows to reproduce it ...

Also I will run they request you ask, possibly with a new tool (probably a small script) to hammer the IC with requests. (This seems to greatly speed-up time-to-reset.)

fnarfbargle, 2017-04-10:
I'd watch this fairly closely.

I was seeing a similar thing on as ASUS X370 Prime with an it8665. If i didn't load the it87 module the machine seemed not to crash. In my case it appears the crash is actually memory related, and polling the it87 just brought it on quicker. I can still make it crash by running a looping kernel compile, prime95 and a few other background load testing tasks simultaneousy. It just takes a lot longer to crash without polling the it87.

Try unloading (or not loading the hwmon module), run mprime -t and a looping kernel compile, something like : while true ; do make distclean ; make allyesconfig ; make -j16 ; done

Let it run for a couple of hours, then stop and re-start mprime a few times over the next couple of hours and see if it crashed. Mine ends up with a couple of BUGs and then an OOPS, then it's all over.

My memory is on the motherboard QVL, but it's just not running right. Polling it87 just makes it crash faster, but even without it still crashes on a long stress test.

I've replaced the ram with some boring non-gamer, no XMP RAM suddenly it's not crashing anymore with or without it87 loaded.

qantourisc, 2017-04-10:
I'll run some stress tests, first test revealed CPU-fan-speed control from the BIOS was malfunctioning (or was misconfigured) as X11 crashed. CPU temperate got to 70°C.. fan was spinning at 300rpm ...

I will run some extra tests over the next nights.
Also please note: it only crashes when USING the hwmon devices, it doesn't crash when the module is loaded for me. But your distro might have some default background services reading from the devices.

Wish AMD would release some specs on the sensors, so I can at least check the CPU's internal sensors.

groeck, 2017-04-10:
Others, including myself, have reported segmentation faults as well, with various boards, so we can assume that there is a systemic problem with memory.
I also see two sets of problems accessing the Super-IO chips on the Gigabyte boards:
1) When accessing the SIO registers on IT8686E, sometimes a bad value is reported. Most of the time a read randomly returns 0xff, but sometimes only a nibble is wrong.
2) Accessing SIO registers on IT8792E using isadump may result in an entire row of data reading as 0xff. Also, accessing the IT8792E may hang the system.
I have observed those problems on both AB350 Gaming and AB350 Gaming 3, with two different CPUs (1700X), and two different sets of RAM. BIOS version does not make a difference.

Gigabyte tells me that I need have to reproduce the problems under Windows since they don't support Linux. Unfortunately, I don't have a spare Windows license. If anyone here can run Windows and has a Windows application to read Super-IO SIO registers similar to isadump, please run it and report the results. Thanks!

qantourisc, 2017-04-10:
@groeck, just then isadumps ? I think I can get my hands on a HDD with a legal Windows. And I might be able to get the dump with http://rweverything.com/screenshots/
If you still have the ticket open, I will post the output here when done.

To access the data multiple times or write it. What data do I read/write ? I might be able to setup a screen-clicker script that will do this using the rweverything.com app.

groeck, 2017-04-10:
@qantourisc: The equivalent of isadump of all registers on 0x2e/0x2f (no matter which page) and 0x4e/0x4f should do it. Just read the entire page multiple times. If the problem is seen, you should see random reads of 0xff, different with each read, on 0x2e/0x2f (maybe 5-10 bytes wrong per page). On 0x4e/0x4f, once in a while, a row of data should return 0xff (or the system hangs when you try ;-).

groeck, 2017-04-11:
@qantourisc: Problem on SIO address 2e/2f (IT8686E) appears to be fixed with BIOS F6 on AB350 Gaming 3. Everyone, please re-test with new BIOS when available.

qantourisc, 2017-04-11:
@everyone For non AB350 Gaming 3 people, the BIOS update does not mention this fix.
As such if there is a new version of the BIOS you might wish to try that for your motherboard too. However they might have changed nothing too.

groeck, 2017-04-11:
The BIOS update for AB350 Gaming 3 also doesn't mention it, but fixes the problem anyway. Gigabyte released new BIOS versions for all AM4 boards yesterday as far as I can see, so there is at least a chance that they fixed the problem for all boards, either in the BIOS or with the microcode update.

qantourisc, 2017-04-12:
F5 bios: Same problem

groeck, 2017-04-18:
To summarize my current status:
- I don't see a difference in segmentation faults if the it87 module is loaded or not. They happen either way. The easiest way to reproduce is to run kernel compiles in a loop.
- It may be possible that reading sensors in a tight loop (cd /sys/class/hwmon; while true; do grep . */* >/dev/null 2>&1; done) makes the problem appear, but I see no evidence that the it87 module would be the source of the problem. I have seen it on an MSI board with NCT6795D, and it was reported on various Asus boards as well.
- Access to IT8792 is generally insecure and may cause a system hang. This was also reported with HwInfo under Windows. I'll disable access to this chip for Gigabyte boards as I get the necessary DMI information.

groeck, 2017-06-07:
A note about segmentation faults and random reboots with Linux:
set "kernel.randomize_va_space = 0" in /etc/sysctl.conf and reboot, or write 0 into /proc/sys/kernel/randomize_va_space. That solves the problem in many cases. However, in my experience it doesn't solve the problem if memory is overclocked too much. AGESA 1006 / microcode revision 0x08001126 also helps to stabilize the system, but kernel.randomize_va_space = 0 is a MUST. Search for background information on the web.

closed 2017-06-07 by groeck