groeck/it87 #19: IT8686E: missing temp_max information/access (GA-AX370-Gaming 5)

qantourisc, 2017-04-07:
Sensors temp 4,5,6 have a high setting in the BIOS, yet not presented by the kernel.

qantourisc, 2017-04-07:
> Probing for Super-I/O at 0x2e/0x2f
> Found unknown chip with ID 0x8686
> 
> Probing for Super-I/O at 0x4e/0x4f
> Found unknown chip with ID 0x8733

groeck, 2017-04-10:
The command for the second chip is
    isadump -k 0x87,0x01,0x55,0xaa 0x4e 0x4f 7

Anyway, known problem. Some of the newer chips have limit registers for temp4/5/6, and I have not found the time to add support for it.

qantourisc, 2017-04-10:
No problem, take your time, if this stuff is  simply just RW some bytes to the some addres, I might be to do it myself. But I don't have a clue atm.

The output dump:
> isadump -k 0x87,0x01,0x55,0xaa 0x4e 0x4f 7
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
> 10: 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 
> 20: 87 33 03 00 00 00 ff 01 fe 18 02 00 8b 00 00 00 
> 30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 60: 00 00 0a 50 00 00 00 00 00 00 00 00 00 00 00 00 
> 70: 00 01 20 38 00 04 00 00 00 00 00 00 00 00 00 00 
> 80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> 90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> c0: 01 00 01 7e 08 00 00 00 00 00 00 14 08 00 1e 00 
> d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
> e0: 00 00 00 00 00 00 00 00 00 20 20 20 c4 3e 3e 0c 
> f0: 00 00 00 00 00 00 2c 00 00 00 00 00 ff 00 00 00

groeck, 2017-04-10:
If you send me an e-mail (to linux@roeck-us.net), I'll give you access to a spreadsheet with known register addresses.

groeck, 2017-04-11:
I pushed an iniitial attempt to support more than three limit (and offset, and type) registers. I think something is wrong with it, so take it with a grain of salt.

qantourisc, 2017-04-11:
Changelog for those who care: https://github.com/groeck/it87/commit/416cfbcd4a6250d4a112572e4e1587c8243a8240#diff-704d12f5e9167dd3948bb7d7e9dc1956

qantourisc, 2017-04-12:
You might be right ;)
I set my VRM to max 90°, the rest to 60°.
And to my knowledge CPU == temp3
VRM should be temp5 according to my research and earlier IT87 drivers.

> temp1:        +26.0°C  (low  =  +0.0°C, high = +60.0°C)  sensor = thermistor
> temp2:        +38.0°C  (low  =  +0.0°C, high = +60.0°C)  sensor = thermistor
> temp3:        +20.0°C  (low  =  +0.0°C, high = +60.0°C)  sensor = disabled
> temp4:        +28.0°C  (low  = +64.0°C, high = +95.0°C)  sensor = thermal diode
> temp5:        +37.0°C  (low  =  +0.0°C, high = +60.0°C)  sensor = thermal diode
> temp6:        +32.0°C  (low  =  +0.0°C, high = +90.0°C)  sensor = disabled

groeck, 2017-04-15:
Should now work. Reopen if there are still problems.

closed 2017-04-15 by groeck

qantourisc, 2017-04-15:
Good work, it works ! The world thanks you.