groeck/it87 #21: IT8613E: PWM control not working  (BIOSTAR Group TB250-BTC)

andreychernyshev, 2017-04-14:
hi Team,

this is issue to discuss potential support for IT8613E

andreychernyshev, 2017-04-14:
run sensors-detect

and responce ....
.....
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8613
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No

......

closed 2017-04-14 by andreychernyshev

mysticaltech, 2018-05-22:
Great! On my TB350, I needed to add `acpi_enforce_resources=lax` to "/etc/default/grub" CMD, and regenerate grub config with "grub2-mkconfig -o /boot/grub2/grub.cfg" to make it work, then reboot. Otherwise I got "modprobe: ERROR: could not insert 'it87': Device or resource busy".