groeck/it87 #22: IT8613E: PWM control not working (BIOSTAR Group TB250-BTC)

andreychernyshev, 2017-04-14:
hi Team,

this is issue to discuss potential support for IT8613E

i will run : sudo sensors-detect

and responce ....
.....
Probing for Super-I/O at 0x2e/0x2f
Trying family National Semiconductor/ITE'... No Trying familySMSC'... No
Trying family VIA/Winbond/Nuvoton/Fintek'... No Trying familyITE'... Yes
Found unknown chip with ID 0x8613
Probing for Super-I/O at 0x4e/0x4f
Trying family National Semiconductor/ITE'... No Trying familySMSC'... No
Trying family VIA/Winbond/Nuvoton/Fintek'... No Trying familyITE'... No

~$ sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 86 13 01 5b 00 08 f7 01 00 00 00 48 41 00 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60: 0a 10 0a 00 00 00 00 00 00 00 00 00 00 00 00 00
70: 00 01 20 38 00 04 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
b0: 00 02 00 00 00 00 00 00 00 00 00 00 c0 03 00 00
c0: 01 04 01 40 00 00 00 00 01 04 01 00 00 00 00 00
d0: 00 00 00 00 00 17 00 00 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 00 00 00 07 07 07 00 00 40 0c
f0: 10 40 00 00 11 00 0e 00 00 00 00 00 7c 00 00 00

~$ sudo isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] y
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

and i'm use: sudo modprobe it87 force_id=0x8622
and run sensors.

and see :
it8622-isa-0a30
Adapter: ISA adapter
in0: +0.25 V (min = +3.04 V, max = +3.01 V)
in1: +1.30 V (min = +2.27 V, max = +2.96 V)
in2: +2.16 V (min = +0.91 V, max = +2.29 V)
in3: +2.16 V (min = +0.00 V, max = +2.71 V)
in4: +2.23 V (min = +0.95 V, max = +2.68 V)
in5: +3.06 V (min = +2.46 V, max = +1.14 V)
in6: +1.18 V (min = +3.02 V, max = +1.14 V)
3VSB: +6.02 V (min = +2.71 V, max = +6.07 V)
Vbat: +3.43 V
+3.3V: +3.60 V
fan2: 1046 RPM (min = 10 RPM)
fan3: 3040 RPM (min = 12 RPM)
fan4: 0 RPM (min = -1 RPM)
fan5: 0 RPM (min = -1 RPM)
temp1: +32.0°C (low = +0.0°C, high = +98.0°C) sensor = thermistor
temp2: +31.0°C (low = -1.0°C, high = +127.0°C) sensor = thermistor
temp3: +32.0°C (low = +127.0°C, high = +127.0°C)
intrusion0: ALARM

super!!!
will need to add in source code it8613 = it8622

i'll do soon
thx

groeck, 2017-04-14:
Try sudo isadump 0xa35 0xa36.
There is no exact match to 8622, though. Main difference, as far as I know, is ADC resolution. IT8613E resolution is 11 mV, not 12 mV. This explains the high Vbat/3.3V values.

andreychernyshev, 2017-04-15:
hi Gunter ,

done

~$ sudo isadump 0xa35 0xa36

WARNING! Running this program can cause system crashes, data loss and worse!

I will probe address register 0xa35 and data register 0xa36.

Continue? [Y/n] y

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f

00: 13 d8 63 00 ff ff 9e ff ff ff 64 00 38 00 c5 84

10: f9 f6 fb 60 c0 00 7f 00 00 01 01 ff ef d6 00 00

20: 44 6c b4 b4 ba ff 62 fb 8f 2a 27 28 2a 2a 2a 96

30: fb fd f7 bd bf 4c e2 00 df 4f 5f cd 5f fc fd 71

40: 62 00 7f ff 7f 7f f1 ad ad 6a d4 00 5f 40 ef ab

50: 3f 18 7f 7f 7f 40 00 00 90 64 00 12 60 00 00 00

60: 7f 7f 7f ff 00 00 0f ff 7f 7f 7f 82 00 00 0f 82

70: 7f 7f 7f 82 00 00 0f 82 7f 7f 7f ff 00 00 0f ff

80: ff ff ff ff 00 00 00 00 02 30 01 02 01 d4 e0 27

90: 00 00 00 00 00 00 00 ff 40 00 00 00 00 fd 00 3a

a0: 00 00 00 ff 00 00 0f ff 7f 7f 7f 80 00 00 0f 80

b0: c4 fb 00 00 ff e9 f6 cf b8 ed ff ff ff ff ff ff

c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01

e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

thank you

groeck, 2017-04-19:
Please let me know if it works now.

andreychernyshev, 2017-04-21:
I’ll check today

thx

sibradzic, 2017-05-04:
@groeck, hi! Another Biostar board AM4, B350GTN (and possibly its sister X370GTN), same sensor chip, same 0x8613 ID. Checked the latest git master, and sensor-detect works fine for me;

Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family 'National Semiconductor/ITE'...               No
Trying family 'SMSC'...                                     No
Trying family 'VIA/Winbond/Nuvoton/Fintek'...               No
Trying family 'ITE'...                                      Yes
Found 'ITE IT8613E Super IO Sensors'                        Success!
    (address 0xa30, driver 'to-be-written')
Probing for Super-I/O at 0x4e/0x4f
Trying family 'National Semiconductor/ITE'...               No
Trying family 'SMSC'...                                     No
Trying family 'VIA/Winbond/Nuvoton/Fintek'...               No
Trying family 'ITE'...                                      No

Thanks!

groeck, 2017-08-03:
No further comment, assuming it works now.

closed 2017-08-03 by groeck