groeck/it87 #26: fix_pwm_polarity causes crash (particular chip is irrelevant)

manover, 2017-04-26:
**fix_pwm_polarity** parameter now crashes the driver after [this commit](https://github.com/groeck/it87/commit/3be51e20cb5458c8975d7980404dd962b021b616), since **it87_check_pwm()** uses **data->REG_PWM** and this field is initialized after **it87_check_pwm()** is called.
I tried to move **it87_check_pwm()** lower within **it87_probe()**, but it somehow causes a lock-up on driver initialization.
Please advice.

groeck, 2017-04-26:
Thanks for the heads up. Can you try again ?

manover, 2017-04-27:
Yeah, I finally found the reason of lock up, it was unrelated to the fix I made.
I ended up splitting register pointers initialization into a separate function, exactly like you did and calling it in exactly the same spot, except I pass ```struct it87_data*``` to it instead of ```struct platform_device*```. Is there any reason you would want to add an overhead of calling ```platform_get_drvdata()``` to obtain the pointer you need back?
But as for the crash, it went away, so this fix works.

manover, 2017-04-27:
In an unrelated note - lines 3370-3378 have spaces for indentation.

groeck, 2017-04-27:
Fixed the space issue (there were more); thanks for the note. No special reason for the parameter; I just kept it the same as the original function w/o thinking about it.

closed 2017-04-27 by groeck