groeck/it87 #28: CPU temps wrong with IT8665E

xpander69, 2017-05-21:
Asus X370 Prime with IT8665E chip at 0x290, revision 6
It used to show correct values few days ago but now is reporting all weird.
it was nicely 40-65°C depending on the load with my R7 1700X@3,9Ghz

but now its reporting all sorts of weird numbersfrom 50-60°C to -89°C and -128°C
Going into UEFI i see the idle temps there are nicely around 44°C

i updated upgraded it87 from 77.62042c9 -> 78.80e1901
but i also downgraded back to previous version with it still showing those weird numbers.

was there maybe a microcode update that can potentially do this? cause there was no BIOS update inbetween.

Anything else i can provide? Let me know?

My friend with different board, Asus CH6 and 1800X seems to have same problem as of late.

groeck, 2017-05-22:
Weird. Best idea I have would be to go back to 20dba18 and, if that version works, bisect from there.
80e1901 vs. 62042c9 is unrelated to it8665e.

xpander69, 2017-05-22:
will test more, but it seems to be issue with the motherboard as i have seen few reports from windows people also. it seems like after reboot it works fine for like 5-10 minutes and then it bugs out

someone on the overlockers forums said this:

> Anyone else having problems related to the temperature reporting of Asus fan control in the UEFI and the AI suite?
> I'm running the Prime X370 Pro with a 1700X @ 3.8 Ghz on all cores.
> The first thing I noticed is that the Q-Fan control in the UEFI with a custom fan curve bugs out after a while and forces the fans to blow at full speed, no matter the temperature and won't slow down. The only "fix" is to restart the computer.

groeck, 2017-05-22:
This makes me wonder if LPC communication with the chip is stable. It might be useful to run isadump on both configuration and environmental monitor register sets a number of times and observe if there are any problems.

xpander69, 2017-05-23:
i think this can be closed, problem got resolved, i dunno if it was kernel 4.11 update or bios update or something else, but i havent seen anything weird anymore. Will reopen if something weird happens again.

closed 2017-05-23 by xpander69