groeck/it87 #29: ASUS B350 Plus

thusnelda, 2017-05-22:
```
# sensors-detect revision 3.4.0+git_45ffa15cf02e63f70ff3b85c23e22dfbab7e8f9c
# Board: ASUSTeK COMPUTER INC. PRIME B350-PLUS
# Kernel: 4.11.2-gentoo x86_64
# Processor: AMD Ryzen 7 1700X Eight-Core Processor (23/1/1)

Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8655
    (logical device 4 has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f

```

```
isadump -k 0x87,0x01,0x55,0x55 0x4e 0x4f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x4e and data register 0x4f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 86 55 06 40 2d 00 20 00 00 94 00 69 05 00 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60: 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
c0: 00 20 00 00 d4 00 00 00 00 20 00 00 14 00 00 00
d0: ff 00 00 0f 00 00 00 00 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c
f0: 00 00 00 00 00 00 0e 00 00 00 00 00 00 00 00 00

```

```
isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 01 5b ae 00 00 40 00 00 c0 44 00 00 cb db ba
10: 46 e7 00 00 7f 00 00 00 7f 00 00 00 7f 1a 00 00
20: 1a a5 00 00 7f 00 00 00 7f 00 00 00 7f 00 fc 98
30: 25 b1 00 00 7f 00 00 00 7f 00 00 00 7f 01 01 01
40: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 00 00 00
50: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c
60: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c
70: 08 0a 00 4c 4b 00 00 00 00 00 7f 99 00 00 0f 99
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 a1 00 3b
90: 00 00 00 00 00 00 00 ff 40 00 00 00 00 e7 00 c1
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80
b0: 7a db 00 00 70 eb 27 32 ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

Kernel: 4.11.2-gentoo and it87 don't work. 
Sensors only output my graphic card temp.

If i can do anything please let me know.

groeck, 2017-05-22:
Your ASUS board most likely reserves the IO memory space for itself. Do you see a kernel log message when loading the driver from here ? If yes, you might need to boot with "acpi_enforce_resources=lax".

thusnelda, 2017-05-24:
I clone it again and get this dmesg and compiler warning:

> modprobe: ERROR: could not insert 'it87': Unknown symbol in module, or unknown parameter (see dmesg)

```
[  291.379431] it87: loading out-of-tree module taints kernel.
[  291.379465] it87: Unknown symbol vid_from_reg (err 0)
[  291.379476] it87: Unknown symbol vid_which_vrm (err 0)
```

```
WARNING: "vid_which_vrm" [/home/user/it87/it87.ko] undefined!
WARNING: "vid_from_reg" [/home/user/it87/it87.ko] undefined!
```

groeck, 2017-05-24:
That means that CONFIG_HWMON_VID is not set in your kernel configuration, which is odd. I'll try to work around it.

groeck, 2017-05-24:
Please try again.

thusnelda, 2017-05-26:
**dmesg**

```
[ ] it87: Found IT8655E chip at 0x290, revision 6
[ ] it87: Beeping is supported
```
**isadump 0x295 0x296 (output changed).**

```
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 90 5a 8e 00 00 00 00 00 c0 44 00 00 72 3f 3c
10: 55 a5 ff 70 c0 80 80 80 03 03 03 64 8b e7 00 00
20: 6f e5 b6 b4 b4 ab b4 98 95 31 1f 27 27 27 27 98
30: ea 3e 4b 3d d9 07 64 b3 8d af ed 1e 91 41 ea 67
40: 7b 89 ae b6 9e e8 c3 7a ad 6a d4 00 5f 40 41 0d
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00
60: 80 14 46 33 16 00 0f 82 80 1e 46 99 14 00 0f c8
70: 80 1e 46 99 14 00 0f c8 7f 7f 7f 99 00 00 0f 99
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 39 00 08
90: 00 00 00 00 00 00 00 ff 42 99 00 00 01 71 e0 37
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80
b0: 6a db 00 00 70 eb 6f 32 ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

sensors work now, thank you.

```
it8655-isa-0290
Adapter: ISA adapter
in0:          +1.21 V  (min =  +0.68 V, max =  +2.55 V)
in1:          +2.50 V  (min =  +0.67 V, max =  +0.82 V)  ALARM
in2:          +2.00 V  (min =  +0.08 V, max =  +2.37 V)
in3:          +1.96 V  (min =  +1.95 V, max =  +1.09 V)  ALARM
in4:          +1.96 V  (min =  +1.91 V, max =  +1.54 V)  ALARM
in5:          +1.86 V  (min =  +0.33 V, max =  +2.58 V)
in6:          +1.95 V  (min =  +0.71 V, max =  +1.58 V)  ALARM
3VSB:         +1.66 V  (min =  +1.12 V, max =  +2.55 V)
Vbat:         +1.62 V
+3.3V:        +1.66 V
fan1:         763 RPM  (min =   26 RPM)
fan2:         812 RPM  (min =   18 RPM)
fan3:         814 RPM  (min =   11 RPM)
temp1:        +49.0°C  (low  = -119.0°C, high = +123.0°C)
temp2:        +30.0°C  (low  = -74.0°C, high = -82.0°C)  ALARM
temp3:        +39.0°C  (low  = -24.0°C, high = -98.0°C)  ALARM  sensor = thermistor
intrusion0:  ALARM
```

I upgrade kernel 4.11.3 and BIOS is now "PRIME B350-PLUS BIOS 0613 2017/05/24"

thusnelda, 2017-05-28:
"acpi_enforce_resources=lax" was not necessary by the way.

closed 2017-05-31 by groeck

Daniel-Krebs, 2017-09-12:
I have the same Motherboard but somehow i can't get sensors to work. I cloned the git repository, edited the makefile (seems like kernel sources are in a different directory under Fedora 26), did "make clean", "make" and "sudo make install". 

sudo modprobe it87 force_id=0x8622

and various other IDs did not work, nor did ""acpi_enforce_resources=lax". Only output is:
```
sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```
One thing I noticed, is a newer BIOS Version than @thusnelda has:

Version: 0806
Release Date: 07/12/2017

Here is more Output:
```
sudo sensors-detect 
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# Board: ASUSTeK COMPUTER INC. PRIME B350-PLUS
# Kernel: 4.12.9-300.fc26.x86_64 x86_64
# Processor: AMD Ryzen 5 1600 Six-Core Processor (23/1/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8655
    (logical device 4 has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
[... 152 lines cut ...]
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 01 5f 85 00 00 40 00 00 c0 44 00 00 58 18 ff 
10: 41 e5 00 00 7f 00 00 00 7f 00 00 00 7f 1a 00 00 
20: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f fc 03 9b 
30: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 01 01 01 
40: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 00 00 00 
50: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c 
60: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c 
70: 08 0a 00 4c 4b 00 00 00 00 00 7f 99 00 00 0f 99 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 90 00 38 
90: 00 00 00 00 00 00 00 ff 40 00 00 00 00 ba 00 57 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80 
b0: 22 97 00 00 7b e6 5c f9 ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 

```
The Output seems to be different than the one from thusnelda.

I don't know what else to do atm..

Thanks,
DK

groeck, 2017-09-12:
If you need to use force_id, you don't use the latest driver from here. Some distributions started to keep modules as compressed file. Maybe that is the case with your distribution.

groeck, 2017-09-12:
asus-isa-0000 is from the ASUS ACPI driver. Posting the DSDT might help figuring out why it doesn't work.

Daniel-Krebs, 2017-09-12:
I simply cloned your git repo using "git clone https://github.com/groeck/it87.git", which should give the newest version shouldn't it?
I am using Fedora Workstation 26, I did a bit googling and couldn't find information about compressed (kernel) modules.
DSDT is attached (hope its the right file)
[dsdt.txt](https://github.com/groeck/it87/files/1296881/dsdt.txt)

groeck, 2017-09-12:
>  I simply cloned your git repo using "git clone https://github.com/groeck/it87.git", which should give the newest version shouldn't it?

Yes, but that does not mean that the driver is built and/or installed.
Consider adding a log message into the probe function and verify that
you see it in the kernel log after loading the driver. If you do,
check the kernel log for any other log messages. Either case,
don't use force_id.

Guenter

Daniel-Krebs, 2017-09-14:
Hey Guenter,
you were right, the compressed kernel modules seem to be the "problem". I noticed, that make install copies the it87.ko to /lib/modules/4.12.9-300.fc26.x86_64/kernel/drivers/hwmon/. It's the only .ko file in this folder, all other modules are compressed .ko.xz files. As there is a it87.ko.xz already, "modprobe it87" tried to load the existing (and probably older)  it87.ko.xz. Of course dmesg reported it87 anyway.
 
So the solution for me was to manually insert the new module using "insmod it87.ko". Now I got sensors working.

Would log messages be helpful for you?
Should I file a bugreport at Fedora or will your module bei included in future kernels as default? 

Daniel

groeck, 2017-09-14:
On Thu, Sep 14, 2017 at 09:22:27AM +0000, Daniel-Krebs wrote:
> Hey Guenter,
> you were right, the compressed kernel modules seem to be the "problem". I noticed, that make install copies the it87.ko to /lib/modules/4.12.9-300.fc26.x86_64/kernel/drivers/hwmon/. [...]
This is not a fedora bug. You'll just have to compress the module yourself.
Feel free to send a patch making the necessary Makefile changes if you can
figure out how to determine that modules are compressed in a given
distribution.

Sure, the it87 changes will be upstream at some point. However, I dont have
much time, and no one else seems to be willing to step in and submit upstream
patches. So it will take some time.

Guenter

Daniel-Krebs, 2017-09-15:
OK, as you see, I'm not an expert at all, in this kernel module thing, I don't think you want your makefile screwed up by me ;)
But maybe you can add a note/warning to the readme file like "If you are using a distribution with an existing it87 kernel module (Like Fedora), which is in a compressed form (.ko.xz), the original kernel module will not be overwritten. Instead the new, uncompressed it87 module ist copied in the same folder as the original, compressed module. You than can load the uncompressed module with "linsmod /PATH/TO/MODULE/it87.ko"

Hope that helps
Daniel

groeck, 2017-09-15:
On Fri, Sep 15, 2017 at 02:23:54PM +0000, Daniel-Krebs wrote:
> OK, as you see, I'm not an expert at all, in this kernel module thing, I don't think you want your makefile screwed up by me ;)
> But maybe you can add a note/warning to the readme file like "If you [...]

That doesn't solve the problem.

Try again, though be aware that there are now a couple of untested patches
related to pwm control / temperature sensor mapping in the tree.

Guenter