groeck/it87 #30: Unknown chip 0x8987 on Intel NUC6CAY

R0nd, 2017-05-28:
I'm trying to achieve software fan control on an Intel Apollo Lake NUC and found this unknown chip (possibly ITE, because other Intel NUCs use them).

sensors-detect:
```
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x8987
```

isadump -k 0x87,0x01,0x55,0x55 0x4e 0x4f 7
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 89 87 01 01 00 00 00 00 00 00 00 00 01 00 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
isadump 0x295 0x296
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

modprobe --verbose it87 force_id=0x8987
```
insmod /lib/modules/4.9.0-0.bpo.3-amd64/kernel/drivers/hwmon/hwmon-vid.ko
insmod /lib/modules/4.9.0-0.bpo.3-amd64/kernel/drivers/hwmon/it87.ko force_id=0x8987
modprobe: ERROR: could not insert 'it87': No such device
```

groeck, 2017-05-29:
The datasheet says it is an IT8987D, as usual with no datasheet available. Unusual that it is at 0x4e, though, that it is accessible with the 0x55/0x55 sequence, and that most of the registers read as 0. What is reported on 0x2e, and what do you get when using the 0x55 / 0xaa access sequence ?

groeck, 2017-05-29:
Regarding force_id, you have to force one of the existing IDs, not the one from the chip. Try force_id=0x8607, for example, since other NUCs use that chip.

R0nd, 2017-05-29:
isadump -k 0x87,0x01,0x55,0xaa 0x4e 0x4f 7 gives exactly the same values as 0x55/0x55
isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7 returns only ff bytes, same with isadump -k 0x87,0x01,0x55,0xaa 0x2e 0x2f
modprobe --verbose it87 force_id=0x8607 results in "No such device" as well.

groeck, 2017-05-29:
Are you using the upstream driver or the driver from github ?

groeck, 2017-05-29:
Some additional research suggests that the IT8987D is an embedded controller, not a Super-IO chip. If so, writing a driver for it will likely be all but impossible, since its programming is board vendor (in this case Intel) specific. There might be a chance to write an ACPI based driver for it, but that would only work if any environmental information is actually exposed through ACPI, and if someone would be able to spend the time to analyze it. The only other means would be for Intel to provide the necessary information, which is extremely unlikely to happen. I'll leave the issue open for the time being, but I'll likely close it as non-feasible. Either case, adding support for this chip will likely require a new driver.

groeck, 2017-05-30:
This is confirmed to be an EC. Access to hardware monitoring data is via Intel defined MMIO. The API to the chip is Intel specific and not released to the public. Even if data was available, someone would have to write a driver; it can not be handled as part of this driver.

closed 2017-05-30 by groeck