groeck/it87 #31: Gigabyte Z170-HD3P-CF correct detection

gsedej, 2017-06-06:
I have motherboard `Gigabyte Technology Co., Ltd. Z170-HD3P-CF`
The temperature reports are "wired", see `temp2 -55C`, also `temp4, 5 and 6 `just reporting temps
I did compile lastest "It87" from this repo and "lm-sensors" from their github on ubuntu 16.04 + 4.9 kernel (AMD's KDF kernel)

The sensors-detects says
```
$ sudo sensors-detect
...
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8628
Probing for Super-I/O at 0x4e/0x4f
...
```

Installing compiled kernel object `sudo insmod it87.ko`

```
$ sensors
...
it8628-isa-0a40
Adapter: ISA adapter
in0:          +0.37 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +1.97 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.01 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.07 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.19 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.36 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.14 V  
fan1:        1448 RPM  (min =    0 RPM)
fan2:         541 RPM  (min =    0 RPM)
fan3:         804 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
temp4:        +92.0°C  
temp5:        +36.0°C  
temp6:        +35.0°C  
intrusion0:  ALARM
...
```
```
$ sensors -u
...
it8628-isa-0a40
Adapter: ISA adapter
in0:
  in0_input: 1.056
  in0_min: 0.000
  in0_max: 3.060
  in0_alarm: 0.000
  in0_beep: 0.000
in1:
  in1_input: 2.004
  in1_min: 0.000
  in1_max: 3.060
  in1_alarm: 0.000
  in1_beep: 0.000
in2:
  in2_input: 1.968
  in2_min: 0.000
  in2_max: 3.060
  in2_alarm: 0.000
  in2_beep: 0.000
in3:
  in3_input: 2.028
  in3_min: 0.000
  in3_max: 3.060
  in3_alarm: 0.000
  in3_beep: 0.000
in4:
  in4_input: 0.012
  in4_min: 0.000
  in4_max: 3.060
  in4_alarm: 0.000
  in4_beep: 0.000
in5:
  in5_input: 1.068
  in5_min: 0.000
  in5_max: 3.060
  in5_alarm: 0.000
  in5_beep: 0.000
in6:
  in6_input: 1.188
  in6_min: 0.000
  in6_max: 3.060
  in6_alarm: 0.000
  in6_beep: 0.000
3VSB:
  in7_input: 3.360
  in7_min: 0.000
  in7_max: 6.120
  in7_alarm: 0.000
  in7_beep: 0.000
Vbat:
  in8_input: 3.144
fan1:
  fan1_input: 1448.000
  fan1_min: 0.000
  fan1_alarm: 0.000
  fan1_beep: 1.000
fan2:
  fan2_input: 542.000
  fan2_min: 0.000
  fan2_alarm: 0.000
  fan2_beep: 1.000
fan3:
  fan3_input: 804.000
  fan3_min: 0.000
  fan3_alarm: 0.000
  fan3_beep: 1.000
fan4:
  fan4_input: 0.000
  fan4_min: 0.000
  fan4_alarm: 0.000
  fan4_beep: 1.000
fan5:
  fan5_input: 0.000
  fan5_min: 0.000
  fan5_alarm: 0.000
  fan5_beep: 1.000
temp1:
  temp1_input: 35.000
  temp1_max: 127.000
  temp1_min: 127.000
  temp1_alarm: 0.000
  temp1_type: 4.000
  temp1_offset: 0.000
  temp1_beep: 1.000
temp2:
  temp2_input: -55.000
  temp2_max: 127.000
  temp2_min: 127.000
  temp2_alarm: 0.000
  temp2_type: 4.000
  temp2_offset: 0.000
  temp2_beep: 1.000
temp3:
  temp3_input: 34.000
  temp3_max: 127.000
  temp3_min: 127.000
  temp3_alarm: 0.000
  temp3_type: 6.000
  temp3_offset: 100.000
  temp3_beep: 1.000
temp4:
  temp4_input: 92.000
temp5:
  temp5_input: 36.000
temp6:
  temp6_input: 35.000
intrusion0:
  intrusion0_alarm: 1.000
...
```

I also tried different "force_id=" found within issues on this github, but I didn't get than better this

groeck, 2017-06-06:
Weird temperatures usually means that 1) nothing is connected to the respective temperature sensor pin or 2) the sensor type is wrong. It doesn't mean that the chip isn't detected correctly or that the driver does anything wrong. Do you have any information suggesting that there are additional temperature sensors on the board which are not reported ?
temp4/5/6 currently only report temperatures. However, I do see that the chip overlays voltage limit registers with temperature limits. I'll leave this issue open and let it reflect to add limit attributes for those sensors.

gsedej, 2017-06-07:
What can I do? I don't have Windows on this PC.

I did some testing using 8 times `burnP6`, for 5 minutes, checking temps for `sensors -u it8628-isa-0a40`. The temp1 (34C), temp2 (-55C), didn't change at all. The Temp4 (90C) is almost all the time 90C sometimes 91C

```
temp3_input: 34.000
temp3_input: 59.000
temp3_input: 67.000
temp3_input: 68.000
temp3_input: 71.000
temp3_input: 71.000
temp3_input: 71.000
temp3_input: 72.000
```

```
temp4_input: 91.000
temp4_input: 90.000
temp4_input: 90.000
temp4_input: 90.000
temp4_input: 90.000
temp4_input: 90.000
temp4_input: 90.000
temp4_input: 91.000
```

```
temp5_input: 36.000
temp5_input: 40.000
temp5_input: 45.000
temp5_input: 47.000
temp5_input: 52.000
temp5_input: 53.000
temp5_input: 54.000
temp5_input: 54.000
```

```
temp6_input: 34.000
temp6_input: 35.000
temp6_input: 36.000
temp6_input: 37.000
temp6_input: 41.000
temp6_input: 43.000
temp6_input: 43.000
```

Here is screenshot of "Psensor". I selected temp1-6 which are for `it8628-isa-0a40`
(when the psensor started, the cpu was under load, then i let it cool, and started again)
[image]

groeck, 2017-06-08:
We have to assume that only temp3/4/6 are connected. Does the BIOS report more temperatures ?
I added experimental temperature limit register support for temp4/5/6 on IT8628E. Please test and report.

gsedej, 2017-06-16:
Sorry for late reply.

Here is data I read it from BIOS menu (shut down->wait 2 min -> direct to bios)
```
"smart fan 5" readings
CPU:
42C
1704 RPM

System 1:
36C
559 RPM

System 2:
35C
813 RPM

Voltages:
CPU Vcore	1.188V
CPU VRIN	1.068
+3.3V		3.304V
CPU VAXG	0.012V

"Advanced voltage settings" menu
CPU Vcore 	1.205V
CPU VCCIO 	0.960V
CPU System Agent Voltage 1.060V

"Advanced memory settings"  menu
DRAM Voltage (CH A/B)		1.200V
DDRVPP Voltage (CH A/B) 	2.500V
DRAM Termination (CH A/B)	0.600V
```

I booted to linux, modprobe it87, waited few minutes to idle. 
```
$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +34.0°C  (high = +84.0°C, crit = +100.0°C)
Core 0:         +29.0°C  (high = +84.0°C, crit = +100.0°C)
Core 1:         +34.0°C  (high = +84.0°C, crit = +100.0°C)
Core 2:         +28.0°C  (high = +84.0°C, crit = +100.0°C)
Core 3:         +30.0°C  (high = +84.0°C, crit = +100.0°C)

acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

it8628-isa-0a40
Adapter: ISA adapter
in0:          +0.76 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +1.98 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.00 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.07 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.19 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.36 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +2.88 V  
fan1:        1464 RPM  (min =    0 RPM)
fan2:         562 RPM  (min =    0 RPM)
fan3:         811 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +36.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
temp4:        +92.0°C  (low  =  +0.0°C, high = +127.0°C)
temp5:        +39.0°C  (low  =  +0.0°C, high = +127.0°C)
temp6:        +38.0°C  (low  =  +0.0°C, high = +127.0°C)
intrusion0:  ALARM

amdgpu-pci-0100
Adapter: PCI adapter
fan1:         684 RPM
temp1:        +41.0°C  (crit =  +0.0°C, hyst =  +0.0°C)
```

Can i provide more data?

groeck, 2017-08-05:
looks like it is working.

closed 2017-08-05 by groeck

gsedej, 2017-08-05:
Thanks for replying and **your great work on it87.** 

Just wondering, if it possible to make:

1. disabling/hiding sensors not connected for actual motherboard 
2. labeling sensors like it's labeled on motherboard

making all this without "end" user doing it (eg in psensor changing label after figuring out, which last until reinstalling OS...). So eg I can make configuration for my motherboard (GA Z170-HD3P-CF),  while someone else makes his for own.

groeck, 2017-08-05:
@gsedej: That is exactly what /etc/sensors3.conf is for.