groeck/it87 #32: False temperature reading on AB350 Gigabyte Gaming 3

gcp, 2017-06-09:
git version 0459e7fb72281c113ab5482d936288b3eb04f5b6

[  136.626983] it87: Found IT8686E chip at 0xa40, revision 2
[  136.627016] it87: Beeping is supported
morbo@beast:~/git/it87$ sensors
it8686-isa-0a40
Adapter: ISA adapter
in0:          +0.40 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.84 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.88 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.19 V  
fan1:         409 RPM  (min =    0 RPM)
fan2:         538 RPM  (min =    0 RPM)
fan3:         564 RPM  (min =    0 RPM)
fan4:         356 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +22.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +31.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +12.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +25.0°C  (low  = +127.0°C, high = +127.0°C)
temp5:        +25.0°C  (low  =  +0.0°C, high = -124.0°C)  sensor = thermistor
temp6:        +36.0°C  (low  =  +0.0°C, high = -124.0°C)
intrusion0:  ALARM

The machine was freshly booted after some off-time. The ambient temp is ~24C so most of those readings are very reasonable, but it is not possible for temp3 to be 12.0C

gcp, 2017-06-09:
Ryzen 1700 CPU, if relevant.

groeck, 2017-06-09:
Gigabyte sets the wrong temperature offset in the BIOS. I have tried to tell them numerous times, but so far no one listened (I didn't check with the latest beta BIOS, though). You'll have to configure it yourself. Check out what it is set to (sensors -u), then reduce the value by 15 degrees C. With a Ryzen 1700, the offset is probably 15 degrees C; it should be 0.

closed 2017-06-09 by groeck