groeck/it87 #36: IT8607E : pwm not working

mickpf, 2017-07-23:
Hello Guenter,

1st question: do you speak German? It would be quiet easier to me to use German.

I've got an Intel NUC with IT8607E on board. I merged your it87 driver from github with the kernel 4.9.3x, but the pwm doesn't appear in sysfs.
I'm a computer scientist, hardware designer and software developer. In the past I wrote some linux kernel drivers for my private usage.

Can you provide me with the datasheet(s) for the IT8607E. I'd try to get pwm running by myself...
I asked ITE for it some weeks ago, but they don't answer...

Thanks in advance,
Michael

groeck, 2017-07-23:
ITE doesn't provide datasheets, or at least to no one I know. You would have to get it through the board vendor, in this case Intel. If I get datasheets, I usually I get them with an understanding that I don't pass them on, so I can not provide them either.

What exactly doesn't work ? You don't see any pwm attributes, or you don't see any output at all ? In the first case, output from isadump as well as raw sensor output (grep . /sys/class/hwmon/hwmonX/*) might help. Otherwise check dmesg to see if there is a related log message.

I do speak German, but not here. Feel free to contact me directly through my linux@ e-mail address.

mickpf, 2017-07-23:
Hello Guenter,

What do you mean with the e-mail address "linux@"?

The path "/sys/class/pwm" exists but is empty.
In the path "/sys/class/hwmon" are two devices:
- hwmon0 - acpitz
- hwmon1 - platform:coretemp

I try to get lm-sensors running (your clone found at github).
Running sensors-detect finds two devices:
- Intel digital thermal sensor...
      (driver `coretemp´)
- Trying family `ITE´...
  Found `ITE IT8607E Super IO Sensors´
      (address 0xa40, dirver `it87`)

But pwmconfig doesn*t find any pwm-capable sensor, because the path "/sys/class/pwm" is empty!
My goal at all is to control the fan.

Thanks in advance
Michael

```

groeck, 2017-07-23:
The commit log shows "Signed-off-by:" tags. Those include an e-mail address which starts with linux@. 

"pwmconfig doesn*t find any pwm-capable sensor, because the path "/sys/class/pwm" is empty"

hardware monitoring drivers don't register with the pwm subsystem. It is also news to me that pwmconfig uses /sys/class/pwm in the first place, but then I don't really know anything about it.

Anyway, the it87 driver doesn't instantiate, which is why you don't see it in /sys/class/hwmon. dmesg might give you a hint. Don't you get an error message when entering "sudo modprobe it87" ?

groeck, 2017-08-03:
problem was acpi_enforce_resources related.

closed 2017-08-03 by groeck