groeck/it87 #38: Cannot install the module.

Boiethios, 2017-07-29:
Hi, sorry if I ask a noobish question, but I cannot install the module. I typed:

```
# make
# make install
cp it87.ko /lib/modules/4.9.0-3-amd64/kernel/drivers/hwmon
depmod -a -F /boot/System.map-4.9.0-3-amd64 4.9.0-3-amd64
# modprobe it87
modprobe: ERROR: could not insert 'it87': No such device
```

My motherboard: [Tomahawk B350](https://www.msi.com/Motherboard/B350-TOMAHAWK.html#productFeature-section)
My CPU: Ryzen 1700

I searched around the internets but found nothing useful (I tried to force ID as mentioned in some sites). Maybe this is not the right place to get a driver for the sensors of my motherboard, but the researches I have done gave your repo as a solution.

groeck, 2017-08-03:
Tomahawk uses nct6795d. Wrong driver. Try the nct6775 driver.

closed 2017-08-03 by groeck

Boiethios, 2017-08-09:
Thank you. Sorry for not finding the right driver. This is difficult to find the appropriate one with the commercial name of the motherboard.