groeck/it87 #41: Is my chip supported by it87 driver ?

ikus060, 2017-08-06:
With lm-sensors, I got the following message. I understood it's related to it87 driver.

My laptop is a Gigabyte P55w v7. And I want to get sensors reading for temperature and FAN. If my chip supported ? If not how can I help to make it work ?

```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x8587

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 8086:a123 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.
Module i2c-dev loaded successfully.
```

Enigma0, 2017-08-07:
I think probably not.

groeck, 2017-08-07:
AFAIK the 8587 is an embedded controller which is not compatible to the IT86xx/87xx chips and would need its own driver (or an ACPI driver). This is, however, all but impossible since the chip documentation is not available.

closed 2017-08-07 by groeck

ikus060, 2017-08-07:
What about this line then ?
```
Found unknown SMBus adapter 8086:a123 at 0000:00:1f.4.
```
I'm gessing it87 driver is not responsible for this one either ?

groeck, 2017-08-07:
8086:a123: That would be i2c-i801. Support for that chip was added in March 2016, suggesting that you might want to consider running a more recent kernel.