groeck/it87 #42: Report for Asus ROG Strix B350F Gaming

keantoken, 2017-08-07:
Here is the hardware report for my motherboard. It seems to be wacky, but some temperature and fan reports work.

$ inxi -M
Machine:   Device: desktop Mobo: ASUSTeK model: ROG STRIX B350-F GAMING v: Rev X.0x
           UEFI [Legacy]: American Megatrends v: 0223 date: 05/04/2017

[AsusROGStrixB350FGamingReport.zip](https://github.com/groeck/it87/files/1203911/AsusROGStrixB350FGamingReport.zip)

keantoken, 2017-09-30:
What kind of feedback do you need to get correct sensor reporting for this board? I notice the voltages aren't correct.

groeck, 2017-09-30:
Voltage readings, for the most part, are never correct. Only cpu, ddr, and 3.3V readings are correct. The other voltages need to be scaled; that is what /etc/senors3.conf is for. Reason is that the ADCs on the Super-IO chips have traditionally a limited voltage range of 2V or 4V depending on the sensor.
The driver here should work fine (without force-id) for your system.

closed 2018-03-11 by groeck

bl4uni, 2018-05-05:
I'm having problems with the same mainboard.

Original `sensors` output:
```
~ $ sensors
amdgpu-pci-0900
Adapter: PCI adapter
fan1:         782 RPM
temp1:        +35.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```

`sensors-detect` output:
```
~ $ sudo sensors-detect
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# Board: ASUSTeK COMPUTER INC. ROG STRIX B350-F GAMING
# Kernel: 4.13.0-39-generic x86_64
# Processor: AMD Ryzen 5 1600X Six-Core Processor (23/1/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8665
    (logical device 4 has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
```

`isadump` output:
```
~ $ sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
[... 250 lines cut ...]

Two of the fans are detected with:
```
force_id=0x8622
force_id=0x8732
```

The reported RPM seem to be correct.

Additional Note:
The official driver by Asus in Windows also uses the wrong temperature. Instead of using the actual temperature of the CPU (Case) it uses `temp1` which is always a bit higher for the lower temperatures and a bit lower for temperatures over ~50°C.

In the syslog it says:
```
May  5 18:36:58 System-Product-Name kernel: [ 4460.988573] it87: Found IT8732F chip at 0x290, revision 6
May  5 18:36:58 System-Product-Name kernel: [ 4460.988598] it87: Beeping is supported
May  5 18:36:58 System-Product-Name kernel: [ 4460.988954] it87: Found IT8732F chip at 0xfff8, revision 15
May  5 18:36:58 System-Product-Name kernel: [ 4460.988978] it87: Beeping is supported
May  5 18:36:58 System-Product-Name systemd[1]: Starting IIO Sensor Proxy service...
May  5 18:36:58 System-Product-Name systemd[1]: Started IIO Sensor Proxy service.
```

Why does it detect a different chip than `sensors-detect`? Or am I missunderstanding something? `force_id=0x8732` is one of the drivers that detects only two fans, so is this "detection" a mistake?

Thank you for your great work. I hope that this information is somewhat helpful.

groeck, 2018-05-05:
@bl4uni: You are reporting problems with the upstream driver. That won't help much here.
Also, as for sensors-detect, please read the last paragraph of the README file. On top of that, you are loading the driver with force_id. That forces the driver to use (and thus report) a wrong ID. The difference to the real chip ID found by sensors-detect should not be surprising.
For the future, please open a new issue to report a new problem. Attaching a report to a closed issue might not always trigger a reply.

bl4uni, 2018-05-05:
I naively believed that "if I have the driver it surely is the newest version". Apparently, installing the driver from here more or less fixed my issue. I will remember to open a new issue in the future.