groeck/it87 #43: Information report: Gigabyte AB350 Gaming 3

CRCinAU, 2017-08-19:
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Gigabyte Technology Co., Ltd. Default string [Default string]
# Board: Gigabyte Technology Co., Ltd. AB350-Gaming 3-CF
# Kernel: 4.12.5-300.fc26.x86_64 x86_64
# Processor: AMD Ryzen 7 1700X Eight-Core Processor (23/1/1)

Output of sensors-detect:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no):
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8686
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8733
```

```
# isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 86 86 02 00 00 20 33 20 00 00 00 40 c8 18 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60: 0a 10 0a 00 0a 20 00 00 00 00 00 00 00 00 00 00
70: 00 00 20 00 00 06 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
a0: 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 00
b0: 00 00 00 00 00 00 00 00 00 00 20 c0 00 00 00 00
c0: 01 00 20 40 00 ff ff ff 01 00 20 00 00 18 00 00
d0: b1 00 08 0a 00 fd ff ff 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 dc 00 1d 20 12 49 00 00 0a 1c
f0: 00 00 00 00 00 00 0d 00 00 00 49 40 00 ff 00 9f
```

... and the second chip which seems wrong?
```
# isadump -k 0x87,0x01,0x55,0x55 0x4e 0x4f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x4e and data register 0x4f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[... 171 lines cut ...]
temp2:        +45.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +46.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = disabled
intrusion0:  OK

it8732-isa-0a60
Adapter: ISA adapter
in0:          +0.39 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +0.95 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.11 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:        +1.67 V  (min =  +0.00 V, max =  +2.78 V)
in4:          +1.33 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.23 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +1.66 V  (min =  +0.00 V, max =  +2.78 V)
Vbat:         +1.54 V
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +27.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +32.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
temp3:        +32.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

Please let me know if you need anything else....

CRCinAU, 2017-08-19:
For what its worth, when they are detected:
fan1 = CPU fan
fan2 = SYS_FAN1

CRCinAU, 2017-08-22:
For what its worth, I realised I may have messed up a little - as it87.ko wasn't being inserted in favour of it87.ko.xz :)

I've fixed this up and now see the following:
```
it8686-isa-0a40
Adapter: ISA adapter
in0:          +0.98 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.05 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.92 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.90 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.02 V
fan1:        2777 RPM  (min =    0 RPM)
fan2:         612 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +26.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +44.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:         +9.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +36.0°C  (low  = +127.0°C, high = +127.0°C)
temp5:        +42.0°C  (low  =  +0.0°C, high = -124.0°C)  sensor = thermistor
temp6:        +48.0°C  (low  =  +0.0°C, high = -124.0°C)
intrusion0:  OK

it8792-isa-0a60
Adapter: ISA adapter
in0:          +1.37 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +1.26 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.24 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:        +3.33 V  (min =  +0.00 V, max =  +5.56 V)
in4:          +1.35 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.26 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.31 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.10 V
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +26.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +31.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

fan1 is still the CPU fan, fan2 is still SYS_FAN1.

With temp3 being the CPU temp, is that directly from the CPU itself or via a sensor under the chip? I presume its still a 20 degree offset in temperatures for this.

CRCinAU, 2017-08-23:
Added BIOS details + sensors output:

```
it8686-isa-0a40
Adapter: ISA adapter
in0:          +0.43 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.91 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.91 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.02 V
fan1:        2812 RPM  (min =   10 RPM)
fan2:         553 RPM  (min =   10 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +25.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +36.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:         +8.0°C  (low  =  +0.0°C, high = +80.0°C)  sensor = AMD AMDSI
temp4:        +33.0°C  (low  = +127.0°C, high = +127.0°C)
temp5:        +38.0°C  (low  =  +0.0°C, high = -124.0°C)  sensor = thermistor
temp6:        +41.0°C  (low  =  +0.0°C, high = -124.0°C)
intrusion0:  OK

it8792-isa-0a60
Adapter: ISA adapter
in0:          +0.39 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +0.99 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.17 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:        +3.33 V  (min =  +0.00 V, max =  +5.56 V)
in4:          +1.36 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.29 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.31 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.12 V
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +25.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

[image]

CRCinAU, 2017-08-23:
Added system voltages:

[image]

CRCinAU, 2017-08-26:
I believe the following is correct for the AB350 Gaming 3 mainboard
```
chip "it8686-isa-0a40"
        label fan1 "CPU_FAN"
        label fan2 "SYS_FAN1"
        label fan3 "SYS_FAN2"
        label fan4 "SYS_FAN3"
        label fan5 "CPU_OPT"
        label temp1 "Mainboard"
        label temp2 "PCH Temp"
        label temp3 "CPU Temp"
        label temp4 "PCI-E Temp"
        label temp5 "VRM Temp"
        label temp6 "vSOC Temp"
        compute temp3 @+15,@+15
```

The compute for temp3 is because the default seems to be -35 offset - which seems off. The correct offset seems to be -20c - so this corrects that.

groeck, 2017-08-27:
On Sat, Aug 26, 2017 at 03:02:27PM +0000, Steven Haigh wrote:
> I believe the following is correct for the AB350 Gaming 3 mainboard
> ```
> chip "it8686-isa-0a40"
>         label fan1 "CPU_FAN"
>         label fan2 "SYS_FAN1"
>         label fan3 "SYS_FAN2"
>         label [...]
Sorry for not getting back earlier. Makes sense, and you are right -
the default offset set by the BIOS is wrong.

Guenter

CRCinAU, 2017-08-27:
No problems - just figured I'd try to gather as much info as I could to help in the meantime :)

RonGeorge, 2017-09-06:
@CRCinAU Steven, I think my board is quite close to yours. Gigabyte AB350M-D3H. My sensors come back as the same location as  yours.

```
Found unknown chip with ID 0x8686
Probing for Super-I/O at 0x4e/0x4f

Found unknown chip with ID 0x8733
```

and then, 

```
[ 9142.757674] it87: Found IT8733E chip at 0xa40, revision 2
[ 9142.757699] it87: Beeping is supported
[ 9142.758124] it87: Found IT8733E chip at 0xa60, revision 3
```

But I noticed that I am getting results similar to yours in the beginning, with nothing usable. Then noticed that I am inserting the .xz version instead of the real version? Could you help me correct mine as well?

```
root@Tower:~# sudo modprobe --verbose it87 force_id=0x8733
insmod /lib/modules/4.12.3-unRAID/kernel/drivers/hwmon/hwmon-vid.ko.xz
insmod /lib/modules/4.12.3-unRAID/kernel/drivers/hwmon/it87.ko.xz force_id=0x8733
```

I've gotten this far by reading all these posts and tons of google searches. I was hesitant to post here because I don't know how to make and make install to update the package. Since UnRaid is a branch of Slackware I didn't want to mess with it if I didn't have to because I have no idea what I'm doing, lol.

CRCinAU, 2017-09-06:
@RonGeorge I do a make && make install, then go into /lib/modules/xxxx/drivers/hwmon and do the following:
1) rm -f it87.ko.xz
2) xz -9 it87.ko
3) dracut -f

This will remove the distro provided module, we compress the one we compiled here to be used, and using dracut will update the module included in the initramfs. It's a bit of a pain, but workable.

I use Fedora 26 if that matters - so your instructions may be slightly different.

groeck, 2017-09-06:
Also, do not use force_id. If the driver does not recognize the module, it is the wrong version of the driver.

esotericnonsense, 2017-09-12:
I have the Gigabyte AB350N-Gaming and the settings above reported by CRCinAU seem accurate to me. Thank you.

groeck, 2018-01-14:
Does anyone happen to have configuration settings for the second Super-IO chip ? It would be useful to know if the settings for GA-AX370-GAMING5 and GA-AX370-Gaming K7 apply here as well.

jranke, 2018-02-20:
I installed it87 from the master branch and drafted a config file for my own use over here

https://github.com/jranke/lm-sensors/blob/gigabyte_ab350/configs/Gigabyte/GA-AB350-GAMING3.conf

Compared to the master branch, I renamed some entries for the main chip it8686 to match the BIOS names and added most of the config for it8792 from the AX370 configuration file. Is there a way  to check the labels for the it8792 chip without installing windows on the machine (I have a win10 vm)?
The temperatures reported by that chip are not shown in the BIOS and I have no idea how to check the labels and calculations for the voltages.

CRCinAU, 2018-02-20:
Seems to work ok for the AB350 Gaming 3.... A few tweaks though...

You can ignore fan5 on it8686-isa-0a40 - as I'm pretty sure it doesn't exist.

Is it possible to line up the voltage tags a little more with the BIOS? I'm trying to figure out how to screen capture it for further info!

CRCinAU, 2018-02-20:
In fact... this is looking quite a bit wrong once I've dug a little further... trying to figure it out..

it8686-isa-0a40:
in0 = CPU vCore voltage.... I'm seeing that go between 0.70v and 1.38v (1.375v) which is correct for my Pstate overclocking...
in6 = Possibly DDR voltage?

I currently have:
```
$ sensors
k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +37.9°C  (high = +70.0°C)

it8686-isa-0a40
Adapter: ISA adapter
in0:            +1.38 V  (min =  +0.00 V, max =  +3.06 V)
in1:            +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in2:            +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:            +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in4:            +0.94 V  (min =  +0.00 V, max =  +3.06 V)
in5:            +0.89 V  (min =  +0.00 V, max =  +3.06 V)
in6:            +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:           +3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:           +3.05 V
CPU_FAN:       2376 RPM  (min =    0 RPM)
SYS_FAN1:       613 RPM  (min =   10 RPM)
SYS_FAN2:         0 RPM  (min =    0 RPM)
SYS_FAN3_PUMP:    0 RPM  (min =    0 RPM)
System 1:       +37.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
Chipset Temp:   +50.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
CPU Temp:       +42.0°C  (low  = +15.0°C, high = +95.0°C)  sensor = AMD AMDSI
PCI-EX16 Temp:  +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
VRM MOS Temp:   +45.0°C  (low  =  +0.0°C, high = -124.0°C)  sensor = thermistor
vSOC MOS Temp:  +53.0°C  (low  =  +0.0°C, high = -124.0°C)  sensor = thermistor
intrusion0:    ALARM

it8792-isa-0a60
Adapter: ISA adapter
DDR Vtt A/B:   +0.91 V  (min =  +0.55 V, max =  +0.80 V)  ALARM
Chipset Core:  +1.06 V  (min =  +0.99 V, max =  +1.10 V)  ALARM
CPU Vdd18:     +1.29 V  (min =  +1.74 V, max =  +1.85 V)  ALARM
DDR Vpp A/B:   +1.96 V  (min =  +2.38 V, max =  +2.63 V)  ALARM
3VSB:          +3.31 V  (min =  +3.21 V, max =  +3.40 V)
Vbat:          +3.14 V
PCIe X4:       +29.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
EC_temp 2:     +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
System 2:      +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
```

jranke, 2018-02-20:
Do you have a CPU_OPT fan header? I did not find one, neither in the manual nor on the board...

in0 of it868 gives me 0.41 V. Did you apply any computation there?

CRCinAU, 2018-02-20:
Yep - I have both CPU_FAN and next to it, CPU_OPT.

As seen on the picture:
http://static.gigabyte.com/Product/2/6166/2017021011371789_src.png

jranke, 2018-02-20:
OK, so we have different editions of the board. The heatsink for the VRM section also looks different on my board. I also noticed that running lm-sensors reports a bit differently for my board.

jranke, 2018-02-20:
```
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Gigabyte Technology Co., Ltd. AB350-Gaming [Default string]
# Board: Gigabyte Technology Co., Ltd. AB350-Gaming-CF
# Kernel: 4.14.0-0.bpo.3-amd64 x86_64
# Processor: AMD Ryzen 7 1700 Eight-Core Processor (23/1/1)
```

CRCinAU, 2018-02-20:
Ah - yep... The AB350-Gaming isn't the same as the AB350 Gaming 3.

Yours is this one:
https://static.gigabyte.com/Product/2/6225/2017021515553213_src.png

jranke, 2018-02-20:
while you have AB350-Gaming 3-CF. OK, I got it. I have a different board. Confusing naming scheme...

CRCinAU, 2018-02-20:
The full range is:
* AB350N Gaming Wifi
* AB350M-HD3
* AB350M-D3H
* AB350M-Gaming 3
* AB350-Gaming (yours)
* AB350-Gaming 3 (mine)

Clear as mud, right? :)

jranke, 2018-02-20:
I notice you can attach RGB LED strips which I can't :( Hope you enjoy them :)

Also, you have an optical S/PDIF out connector and the CPU_OPT fan header. Apart from that they seem to be almost equal. Thanks for your feedback and sorry for the noise!

CRCinAU, 2018-02-20:
Yeah - but you can't control the LEDs in linux at all... I've been trying to get some people excited to try and support that entire subsystem in Linux with no luck as yet! :)

I bought this one for the S/PDIF more than anything - I've had Logitech optical speakers / amp combo for years - I'm not sure I could live without them anymore ;)

The VRM heatsink is a pile of garbage when using an 8 core CPU - I can easily get it to 110c under moderate load. I 3D printed a 'cowl' for the rear SYS_FAN that blows air onto the VRM heatsink - that seems to cut ~20c from the peak temps... Apparently if you don't watercool, the airflow from the CPU fan keeps things a little more under control... but eh...

General noise is fine sometimes - as it reminded me that this thread existed... I should attempt to create an accurate one for the AB350 Gaming 3 - seeing as you have one for the AB350 Gaming :)

jranke, 2018-02-20:
Nice, I have never seen anybody used S/PDIF. Must be nice not to worry about interference or ground loops. 

Yes, also in my board the VRM easily has the highest temperatures. BTW, I also had these spontaneous reboots with MCE error bea0*0108. Just disabled C-states, so I am hopeful they will not come back.

CRCinAU, 2018-02-20:
Ok, so I've been digging further and further - and I can't see what the it8792-isa-0a60 is actually measuring...

I can see that there are 6 temp sensors on the AB350-Gaming 3. All seem to be hanging off the it8686-isa-0a40 chip.

Looking specifically at the it8792:
* in0 seems to roughly track the same voltages as in0 on the it8686. I'm not sure if the subsystem would measure the two chips at difference times - hence looking at the output of ``watch sensors`` would show different values, but the ranges displayed seem to show them both tracking the same voltage ranges.

* in1-in6 don't seem to really track anything that I can nail down...

* temp1, temp2, and temp3 I'm not sure do anything. The ranges don't seem to track any other sensor. When stressing the CPU, these change by maybe 1-2 degrees.

On the it8686:
* in1, in2, in3 are currently unknown... I can't see anything in the BIOS that would run with a 2v signal.
* in0 is CPU Vcore
* in4 and in5 - one is CPU VDDP, one is VCORE SOC. I have no idea which is which - and not quite sure how to tell at the moment...
* in6 is the DRAM voltage - confirmed by changing it in the BIOS to 1.25v, saw the expected change here.

As such, here is my current config for the it8686:
```
chip "it8686-isa-0a40"
  label fan1 "CPU_FAN"
  label fan2 "SYS_FAN1"
  label fan3 "SYS_FAN2"
  label fan4 "SYS_FAN3_PUMP"
  label fan5 "CPU_OPT"
  label temp1 "System 1"
  label temp2 "Chipset"
  label temp3 "CPU Socket"
  label temp4 "PCI-EX16"
  label temp5 "VRM MOS"
  label temp6 "vSOC MOS"
  compute temp3 @+15,@+15

  set in0_min 0.6
  set in0_max 1.45
  set in6_min 1.2 * 0.97
  set in6_max 1.2 * 1.2
  set in7_min 3.3 * 0.97
  set in7_max 3.3 * 1.05

  label in0 "CPU Vcore"
  label in4 "CPU VDDP"
  label in5 "VCORE SOC"
  label in6 "DRAM A/B"
```

jranke, 2018-02-20:
Nice. I have worked a bit on the conf file for the AB350 Gaming as well and I put your changes into the conf file for the AB350 Gaming 3 in my branch of lm-sensors.

How did you match the other voltages? I could verify the DRAM channel voltage by changing it in the BIOS as you did. However, looking at the BIOS the CPU Vcore voltage is stable between 1.092 V and 1.104 V. When I boot the system, the voltage reported for in0 varies between 0.40 V and 1.31 V! I hope this is not the real voltage for my CPU, otherwise I would not be surprised about spontaneous reboots...

in1, in2 and in3 are stable at 2.03 V. in4 and in5 are around 0.91V and would therefore roughly match the voltage reported by the BIOS for CPU VDDP and VCore SOC.

jranke, 2018-02-20:
Wow, this is interesting. When I run

```
sudo python zenstates.py --c6-disable
```

(requires the python script from rm0n/ZenStates-Linux and the msr module loaded),
the voltage on in0 only varies between 0.84 and 1.12 V. When I run

```
sudo python zenstates.py --c6-enable
```
it varies between 0.40 V and 1.38 V!

So maybe it really is the CPU core voltage and it just gets destabilised by deep power saving states???

When I stress the CPU with stress -c 16, I get a very stable voltage on in0 between 1.09 and 1.10 V, regardless whether C6 is enabled or not. So it does seem to me that in0 is the CPU core voltage and my PSU is not able to provide stable voltage when the system is idle and the CPU core(s) enter C6. Although it explicitly says on the web page that it does support Intel C6/C7 states:

https://www.bequiet.com/en/powersupply/928

see Technical data.

jranke, 2018-02-20:
Or maybe the VRM is to blame. I got the idea of the PSU problem on this thread

https://community.amd.com/message/2841858#comment-2841858

Sorry that I am sidetracking this issue again, but it is related to my attempts to identify the voltage sensors. If there is a better method without install Windows, I would be glad to learn it.

groeck, 2018-02-20:
One way to identify voltages is to observe voltage jumps in the BIOS and compare those to the voltages reported in Linux. Those reflect the LSB of the respective ADC. With the physical LSB being with 2mV or 4mV, you can then determine the associated voltage channels. For example, in2 is often connected to +12V on Gigabyte boards. If so, the expected granularity in the BIOS should be 12mV (since the raw input is 2.02V), and the voltage reported in the BIOS should be around 12.12V.
Oh, and, yes, in0 is the CPU voltage, and I do see voltages below 0.5V on both Gigabyte and MSI boards when the system is more or less idle. If this is an error as suggested earlier, it would happen on boards from multiple vendors with different Super-IO chips.

jranke, 2018-02-20:
I don't understand much of all this, but it just occurred to me that such large jumps in the voltage could entail nondeterministic behaviour.

jranke, 2018-02-20:
Mhm. So somewhere we have exactly 2.02 V, associated with a granularity of 2 mV, so +12V should have a granularity of 12 mV. I get that. In the BIOS I see 12.168V for +12V which rounds to 12.12V. When I watch the sensors of my it8686 chip, and particularly in2, I only see a stable signal of 2.03 V, same as for ln1 and in3. So my guess would be that they correspond to +3.3V, +5 V and +12V, but is it possible to know when they are just stable at these values?

groeck, 2018-02-20:
You might want to consider looking at the raw voltages; chance is that those change over time which you would not notice with the normal output from the "sensors" command. For example, I do see in2_input change from 2.028 to 2.016 once in a while on my AB350M-Gaming 3. Coincidentally, 2.028*6 = 12.168. Also, at least on my system, the voltages are _not_ exactly the same. I have in1_input: 1.992, in2_input: 2.028, and in3_input: 2.016. While those are small differences, they are most of the time sufficient to match voltages to the BIOS.
Not sure I understand what you mean with "such large jumps in the voltage could entail nondeterministic behaviour". Please elaborate.

CRCinAU, 2018-02-21:
The BIOS does actually have a 3.3, 5 and 12v indications - so it would make sense that these are at least available...

CRCinAU, 2018-02-21:
In other news, I swiped this from HWINFO64..

[image]

CRCinAU, 2018-02-21:
My sensors output just about matches those values now:
```
# sensors
k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +32.0°C  (high = +70.0°C)

it8686-isa-0a40
Adapter: ISA adapter
CPU Vcore:      +0.40 V  (min =  +0.60 V, max =  +1.45 V)
+3.3v:          +3.37 V  (min =  +0.00 V, max =  +5.05 V)
+12v:          +12.10 V  (min = +18.36 V, max = +18.36 V)
+5v:            +5.04 V  (min =  +0.00 V, max =  +7.65 V)
CPU Vcore SOC:  +0.90 V  (min =  +0.00 V, max =  +3.06 V)
CPU Vddp:       +0.91 V  (min =  +0.00 V, max =  +3.06 V)
DRAM A/B:       +1.21 V  (min =  +1.16 V, max =  +1.44 V)
3VSB:           +3.26 V  (min =  +3.19 V, max =  +3.46 V)
Battery:        +3.05 V
CPU_FAN:       2351 RPM  (min =    0 RPM)
SYS_FAN1:       465 RPM  (min =   10 RPM)
SYS_FAN2:         0 RPM  (min =    0 RPM)
SYS_FAN3_PUMP:    0 RPM  (min =    0 RPM)
CPU_OPT:          0 RPM  (min =    0 RPM)
System 1:       +37.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
Chipset:        +51.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
CPU Socket:     +37.0°C  (low  = +15.0°C, high = +95.0°C)  sensor = AMD AMDSI
PCI-EX16:       +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
VRM MOS:        +39.0°C  (low  =  +0.0°C, high = -124.0°C)  sensor = thermistor
vSOC MOS:       +45.0°C  (low  =  +0.0°C, high = -124.0°C)  sensor = thermistor
intrusion0:    ALARM

it8792-isa-0a60
Adapter: ISA adapter
CPU Vcore:     +0.39 V  (min =  +0.60 V, max =  +1.45 V)  ALARM
DDR VTT:       +0.92 V  (min =  +0.00 V, max =  +2.78 V)
Chipset Core:  +1.06 V  (min =  +0.00 V, max =  +2.78 V)
VIN3:          +3.33 V  (min =  +0.00 V, max =  +5.56 V)
CPU Vdd18:     +1.29 V  (min =  +0.00 V, max =  +2.78 V)
DDR Vpp:       +1.19 V  (min =  +0.00 V, max =  +2.78 V)
3VSB:          +3.06 V  (min =  +0.00 V, max =  +3.06 V)  ALARM
Battery:       +3.31 V  (min =  +3.21 V, max =  +3.40 V)
PCI-EX8:       +29.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
Temp 2:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
System 2:      +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:   ALARM
```

I've updated the config file to:
```
# The temp3 temperature offset depends on the CPU type and needs to be
# adjusted. Compare against the output of k10temp for the correct value.

chip "it8686-isa-0a40"
  label temp1 "System 1"
  label temp2 "Chipset"
  label temp3 "CPU Socket"
  label temp4 "PCI-EX16"
  label temp5 "VRM MOS"
  label temp6 "vSOC MOS"
  label in0 "CPU Vcore"
  label in1 "+3.3v"
  label in2 "+12v"
  label in3 "+5v"
  label in4 "CPU Vcore SOC"
  label in5 "CPU Vddp"
  label in6 "DRAM A/B"
  label in7 "3VSB"
  label in8 "Battery"
  label fan1 "CPU_FAN"
  label fan2 "SYS_FAN1"
  label fan3 "SYS_FAN2"
  label fan4 "SYS_FAN3_PUMP"
  label fan5 "CPU_OPT"

  compute temp3 @+15,@+15
  compute in1 @*1.65,@*1.65
  compute in2 @*6,@*6
  compute in3 @*2.5,@*2.5

  set in0_min 0.6
  set in0_max 1.45
  set in2_min 1.97
  set in2_max 2.05
  set in6_min 1.2 * 0.97
  set in6_max 1.2 * 1.2
  set in7_min 3.3 * 0.97
  set in7_max 3.3 * 1.05

chip "it8792-isa-0a60"
  label temp1 "PCI-EX8"
  label temp2 "Temp 2"
  label temp3 "System 2"
  label in0 "CPU Vcore"
  label in1 "DDR VTT"
  label in2 "Chipset Core"
  label in3 "VIN3"
  label in4 "CPU Vdd18"
  label in5 "DDR Vpp"
  label in6 "3VSB"
  label in7 "Battery"

  compute in6 @*1.1,@*1.2

  set in0_min 0.6
  set in0_max 1.45
  set in7_min 3.3 * 0.97
  set in7_max 3.3 * 1.03

  ignore fan1
  ignore fan2
  ignore fan3
  ignore in8
```

jranke, 2018-02-21:
Good job! I think I will not get that far for my board - the basic functions now work, so I may as well  leave it at that.
@groeck: I meant that I could not imagine that it could be OK for the voltage on the CPU core to vary that much and thought that this might be the cause for the spontaneous reboots and MCE errors I was observing. And since disabling C6 states is known to be a cure for such reboots, and at the same time dimishes the voltage drops, I still think these may be related.
Thanks for the hint to look at the raw voltages. You mean the ones listed under /sys/class/hwmon/...?

groeck, 2018-02-21:
... or the voltages reported with sensors -u if you don't want to access sysfs attributes directly.
The CPU can request whatever voltage it wants, and the speed of voltage changes is well specified. Unless you are claiming that the CPU violates that specification, which you would have to back up with evidence, it is the responsibility of the board to provide the voltage (and voltage changes) requested by the CPU. As such, the board, its power infrastructure, and the power supply, are the primary suspects for power delivery related issues.

jranke, 2018-02-21:
Ah, I will have a look at `watch sensors -u` when I am back home. OK, I didn't know that the CPU can request different voltages, and no, I do not want to claim that the CPU violates any specifications. BTW, do you disable C6 states on your Ryzen systems?

CRCinAU, 2018-02-21:
Nope - and I even use custom Pstates now for that lovely overclocking factor... 3.9GHz @ 1.375v on a 1700x.

Not quite stable enough for prime95 and the max torture stress test, but hasn't hung under normal workload yet...

jranke, 2018-02-23:
The voltages on my AB-350 Gaming in in1, in2, and in3 as reported by sensors -u are all stable at 2.028 V. I will just leave the conf files as is. @groeck Feel free to pull from my lm-sensors branch if you want Steves file listed above and my current file.

groeck, 2018-03-11:
Fork seems to have been removed, or at least I don't find it, so I can not pull. Closing.

closed 2018-03-11 by groeck