groeck/it87 #44: Information report: GA-AX370-Gaming K7

mdlayher, 2017-08-20:
I'm using this motherboard with the latest BIOS: F4.  Header for `sensors-detect`:

```
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Gigabyte Technology Co., Ltd. AX370-Gaming K7 [Default string]
# Kernel: 4.10.0-32-generic x86_64
# Processor: AMD Ryzen 7 1700 Eight-Core Processor (23/1/1)
```

Super I/O sensors:

```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8686
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8733
```

As a note my machine locked up when I attempted to run `sensors-detect` after adding and removing the `it87` module compiled from this repo.  This information was gathered with the module unloaded after a reboot.

I'm unsure if this is relevant, but something appears to be detected here on several of my I2C/SMBus adapters.  The same "Client found" message appears at i2c-12 through i2c-15.

```
Next adapter: SMBus PIIX4 adapter port 0 at 0b00 (i2c-12)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x28
Probing for `National Semiconductor LM79'...                No
Probing for `National Semiconductor LM80'...                No
Probing for `National Semiconductor LM96080'...             No
Probing for `Winbond W83781D'...                            No
Probing for `Winbond W83782D'...                            No
Probing for `Nuvoton NCT7802Y'...                           No
Probing for `Winbond W83627HF'...                           No
Probing for `Winbond W83627EHF'...                          No
Probing for `Winbond W83627DHG/W83667HG/W83677HG'...        No
Probing for `Asus AS99127F (rev.1)'...                      No
Probing for `Asus AS99127F (rev.2)'...                      No
Probing for `Asus ASB100 Bach'...                           No
Probing for `Analog Devices ADM1029'...                     No
Probing for `ITE IT8712F'...                                No
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No
Client found at address 0x53
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No
```

Finally, I see "no sensors were detected" at the end of the output.

Here's the output of the requested isadump commands.

```
$ sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
[... 330 lines cut ...]
DDRVpp A/B | 2.521V
+3.3V | 3.264V
+5V | 4.980V
CHIPSET Core | 1.067V
+12V | 12.168V
VCORE SOC | 0.900V
CPU Fan | 1562 RPM
CPU OPT | N/A
1st System Fan | 614RPM
2nd System Fan | 572RPM
3rd System Fan | N/A
4th System Fan | N/A
5th System Fan (Pump) | N/A
6th System Fan (Pump) | N/A
CPU Temperature | 33.0C
System 1 | 34.0C
System 2 | 36.0C
Chipset | 46.0C
PCIEX16 | 38.0C
VRM MOS | 42.0C
PCIEX8 | 36.0C
EC_TEMP1 | N/A
EC_TEMP2 | N/A

Hope this all helps, and thank you for your time.

groeck, 2017-09-22:
Regarding the duplicate i2c detection: That is a kernel bug. I submitted a patch for it some time ago, but it has not been accepted due to lack of feedback. See https://patchwork.kernel.org/patch/9842539/.

groeck, 2017-12-03:
It would be useful to know if the system still locks up after loading the latest version of the driver. There have been a number of improvements to prevent that from happening.

mdlayher, 2017-12-04:
Sure, I can give it a try.  Do you recommend any particular forced ID for this board?

groeck, 2017-12-04:
A forced ID should not be needed.

mdlayher, 2017-12-04:
Just built and installed latest master:

```
matt@servnerr-3:~⟫ sudo sensors
[sudo] password for matt: 
No sensors found!
Make sure you loaded all the kernel drivers you need.
Try sensors-detect to find out which these are.
matt@servnerr-3:~/tmp/it87⟫ uname -a
Linux servnerr-3 4.13.0-17-generic #20~16.04.1-Ubuntu SMP Mon Nov 6 14:18:00 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
matt@servnerr-3:~/tmp/it87⟫ make
  CC [M]  /home/matt/tmp/it87/it87.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/matt/tmp/it87/it87.mod.o
  LD [M]  /home/matt/tmp/it87/it87.ko
matt@servnerr-3:~/tmp/it87⟫ sudo make install
mkdir -p /lib/modules/4.13.0-17-generic/kernel/drivers/hwmon
cp it87.ko /lib/modules/4.13.0-17-generic/kernel/drivers/hwmon/
depmod -a -F /boot/System.map-4.13.0-17-generic 4.13.0-17-generic
matt@servnerr-3:~/tmp/it87⟫ sudo modprobe it87
matt@servnerr-3:~/tmp/it87⟫ sensors
it8686-isa-0a40
Adapter: ISA adapter
in0:          +0.40 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +1.99 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +1.99 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.90 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.07 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.24 V  
fan1:        1051 RPM  (min =   10 RPM)
fan2:         625 RPM  (min =    0 RPM)
fan3:         592 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +46.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +22.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +36.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +41.0°C  (low  =  +0.0°C, high = -117.0°C)  sensor = thermistor
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

it8792-isa-0a60
Adapter: ISA adapter
in0:          +0.39 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +0.60 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.06 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:        +3.40 V  (min =  +0.00 V, max =  +5.56 V)
in4:          +1.80 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.51 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.40 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.27 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +37.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

```

Seems okay as of now, though a couple of those temperature sensors are still very odd.  Will report back if anything funny happens.

groeck, 2017-12-04:
The sensors reporting negative temps are not connected. The CPU temperature is probably off - Gigabyte configures a funky temperature offset. You can see the offset in /sys/class/hwmon/hwmonX/temp3_offset (where X is probably 1 in your case). On my AB350M-Gaming 3 it is set to -30 degrees C (the correct offset would be -20 degrees C).

mdlayher, 2017-12-04:
```
matt@servnerr-3:~⟫ cat /sys/class/hwmon/hwmon1/temp3_input 
37000
matt@servnerr-3:~⟫ cat /sys/class/hwmon/hwmon1/temp3_offset 
0

matt@servnerr-3:~⟫ cat /sys/class/hwmon/hwmon0/temp3_input 
20000
matt@servnerr-3:~⟫ cat /sys/class/hwmon/hwmon0/temp3_offset
-10000
```

Looks like just hwmon0 has an offset configured.

groeck, 2017-12-04:
On Mon, Dec 04, 2017 at 08:55:50PM +0000, Matt Layher wrote:
> ```
> matt@servnerr-3:~⟫ cat /sys/class/hwmon/hwmon1/temp3_input
> 37000
> matt@servnerr-3:~⟫ cat /sys/class/hwmon/hwmon1/temp3_offset
> 0
>
> matt@servnerr-3:~⟫ cat /sys/class/hwmon/hwmon0/temp3_input [...]
Yes, that would be the CPU. Assuming your CPU is a Ryzen 3 or 5,
this means the reported temperature is 10 degrees C too low.

Guenter

chinf, 2018-01-13:
The AX370 Gaming K7 and AX370 Gaming 5 motherboards are very similar (the only significant difference being the BCLK chip). They have the same ITE chips, fan headers and temp sensors, so anyone with a K7 should find [the Gaming 5 mappings](https://github.com/groeck/it87/issues/16#issuecomment-357402061) applicable.

groeck, 2018-01-14:
Configuration file added as configs/Gigabyte/GA-AX370-GAMING5.conf with remark that it also applies to this board.

closed 2018-01-14 by groeck