groeck/it87 #46: PWM4 and PWM5 working backwards

adityaxavier, 2017-09-03:
Greetings,
In my Motherboard - Gigabyte G1 Sniper B7, I have connected (3) of the available Case PWM Fan controllers to Noctua FANs. 
Am controlling my PWM fans using a python script. Intermittently I see FAN4 and FAN5 at 0 RPM, even though the other fan was at full speed. 

On checking the value of PWM4 and PWM5 in /sys/class/hwmon/hwmon2/ I saw both of them were at 255. When I set the PWM value to 60, the FANs were working on full speed. This happens after every few boots.

`it8628-isa-0a40
Adapter: ISA adapter
in0:          +0.23 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.05 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.01 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.07 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.36 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.17 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:        1537 RPM  (min =    0 RPM)
fan3:        1480 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +36.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
temp4:        +36.0°C  (low  =  +0.0°C, high = +127.0°C)
temp5:        +34.0°C  (low  =  +0.0°C, high = +127.0°C)
temp6:        +33.0°C  (low  =  +0.0°C, high = +127.0°C)
intrusion0:  OK

acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +33.0°C  (high = +84.0°C, crit = +100.0°C)
Core 0:        +27.0°C  (high = +84.0°C, crit = +100.0°C)
Core 1:        +31.0°C  (high = +84.0°C, crit = +100.0°C)

root@home:/sys/class/hwmon/hwmon2# cat pwm4
255
root@home:/sys/class/hwmon/hwmon2# cat pwm5
255

root@home:/sys/class/hwmon/hwmon2# echo 60 > pwm4
root@home:/sys/class/hwmon/hwmon2# echo 60 > pwm5

root@home:/sys/class/hwmon/hwmon2# sensors
it8628-isa-0a40
Adapter: ISA adapter
in0:          +0.01 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.05 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.01 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.07 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.36 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.17 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:        1544 RPM  (min =    0 RPM)
fan3:        1486 RPM  (min =    0 RPM)
fan4:        1305 RPM  (min =    0 RPM)
fan5:        1245 RPM  (min =    0 RPM)
temp1:        +36.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
temp4:        +36.0°C  (low  =  +0.0°C, high = +127.0°C)
temp5:        +34.0°C  (low  =  +0.0°C, high = +127.0°C)
temp6:        +33.0°C  (low  =  +0.0°C, high = +127.0°C)
intrusion0:  OK

acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +119.0°C)
temp2:        +29.8°C  (crit = +119.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +33.0°C  (high = +84.0°C, crit = +100.0°C)
Core 0:        +27.0°C  (high = +84.0°C, crit = +100.0°C)
Core 1:        +30.0°C  (high = +84.0°C, crit = +100.0°C)
`

groeck, 2017-09-03:
Can you provide register dumps for the working and non-working cases ? Something like "sudo isadump 0xa45 0xa46" should do it.
Background" pwm polarity is controlled by register 0x14[7] and, for pwm 4..6, registers 0x7f[6], 0xa7[6], and 0xaf[6]. If those register bits change value between the working and non-working case, we know what happens.

adityaxavier, 2017-09-03:
Sure,

root@home:~# isadump 0xa45 0xa46
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0xa45 and data register 0xa46.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 19 00 00 00 ff ff 18 20 ff 98 60 00 30 ff a7 b6 
10: ff ff ff 75 c7 82 82 02 ff 01 01 ff ff ff ff ff 
20: 24 ab a8 a7 01 59 65 8c 84 24 c9 21 24 22 22 bf 
30: ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 
40: 7f 7f 7f 7f 7f 7f 5f 40 ad 6a d4 00 00 00 00 00 
50: ff d8 7f 7f 7f 40 00 00 90 64 1f 12 65 00 00 00 
60: 00 14 41 46 1c 03 00 05 00 00 00 00 00 00 00 ff 
70: 00 14 3c ff 1c 03 00 3e 00 14 3c ff 1c 03 00 03 
80: ff ff ff ff ff ff ff ff 02 30 01 02 01 f0 e0 65 
90: 23 00 a3 00 32 00 a3 00 40 97 00 00 40 38 c1 dd 
a0: 00 14 3c ff 1c 03 00 04 7f 7f 96 80 00 00 0f 80 
b0: bd 02 8b 82 7f 00 7f 00 7f 00 3c 3c 3c 3c 3c 3c 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 

root@home:~# devmem2 0x7f6
/dev/mem opened.
Memory mapped at address 0x7fd2ad768000.
Value at address 0x7F6 (0x7fd2ad7687f6): 0x0
root@home:~# devmem2 0xa76
/dev/mem opened.
Memory mapped at address 0x7fc0e796d000.
Value at address 0xA76 (0x7fc0e796da76): 0x0
root@home:~# devmem2 0xaf6
/dev/mem opened.
Memory mapped at address 0x7fd449831000.
Value at address 0xAF6 (0x7fd449831af6): 0x0

After few consecutive reboots it started working again:

root@home:~# isadump 0xa45 0xa46
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0xa45 and data register 0xa46.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 19 00 00 00 ff ff 18 20 ff 98 60 00 30 ff 97 ce 
10: ff ff ff 75 c7 82 82 00 ff 01 01 ff ff ff ff ff 
20: 01 ab a8 a7 01 59 65 8c 84 25 c9 23 25 23 22 bf 
30: ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 
40: 7f 7f 7f 7f 7f 7f 5f 40 ad 6a d4 00 00 00 00 00 
50: ff d8 7f 7f 7f 40 00 00 90 64 1f 12 65 00 00 00 
60: 00 14 41 46 1c 03 00 05 00 00 00 00 00 00 00 ff 
70: 00 14 3c ff 1c 03 00 3e 00 14 3c ff 1c 03 00 40 
80: ae 01 bc 01 ff ff ff ff 02 30 01 02 01 f0 e0 e5 
90: 23 00 a3 00 32 00 a3 00 40 97 00 00 40 4a c1 3d 
a0: 00 14 3c ff 1c 03 00 40 7f 7f 96 80 00 00 0f 80 
b0: bf 02 8b 82 7f 00 7f 00 7f 00 3c 3c 3c 3c 3c 3c 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 

root@home:~# devmem2 0x7f6
/dev/mem opened.
Memory mapped at address 0x7fa12393a000.
Value at address 0x7F6 (0x7fa12393a7f6): 0x0
root@home:~# devmem2 0xa76
/dev/mem opened.
Memory mapped at address 0x7fc217947000.
Value at address 0xA76 (0x7fc217947a76): 0x0
root@home:~# devmem2 0xaf6
/dev/mem opened.
Memory mapped at address 0x7f658633b000.
Value at address 0xAF6 (0x7f658633baf6): 0x0

Am not sure, if that was helpful. Please do let me know if I can get you more information.

groeck, 2017-09-03:
As suspected, registers 0x7f and 0xa7 are being reconfigured. The polarity enable bit is cleared, and the temperature source is changed as well as a reserved bit. Most of the written bit values are reserved and thus invalid.
What is your configuration sequence, or in other words which values does your script write into which attributes ?

adityaxavier, 2017-09-03:
The script I does the following.

It first enables pwm configuration by writing 1 to the following files.
/sys/class/hwmon/hwmon2/device/pwm2_enable,/sys/class/hwmon/hwmon2/device/pwm3_enable,/sys/class/hwmon/hwmon2/device/pwm4_enable

Then it goes ahead and write the value between 60 ~ 255 to the files
/sys/class/hwmon/hwmon2/device/pwm2,/sys/class/hwmon/hwmon2/device/pwm3,/sys/class/hwmon/hwmon2/device/pwm4

Am not sure there if there is anything wrong here.. My script is available on GitHub..
https://github.com/adityaxavier/storagefancontrol

Unfortunately because the problem is sporadic.

groeck, 2017-09-03:
Initial assessment is that there is a data path where the cached pwm registers are not initialized from the chip on driver startup, but left at their initial values (ie 0 for the control register). I'll just have to find that data path. On top of that, there is a bug where a wrong temperature source is selected under certain conditions.

groeck, 2017-09-03:
... and confirmed. Writing pwmX_enable before reading any attribute will cause the problem. Workaround until I have a fix available should be to read pwmX_enable prior to writing it.

adityaxavier, 2017-09-03:
Awesome, thanks for the workaround.. will modify my script..

Hoping for the fix soon.. Great work!

groeck, 2017-09-03:
Please let me know if the workaround helps.

groeck, 2017-09-03:
Should now be fixed. Reopen if there are still issues.

closed 2017-09-03 by groeck