groeck/it87 #47: Gigabyte AORUS X399 Gaming 7 - Unable to ModProbe IT87

raz4725, 2017-09-05:
Information:

Fresh Git Clone of IT87 and LM-Sensors, compiled via make and make install. This is on Linux Mint 18.2 on a Threadripper 1950x CPU, with kernel 4.13RC7.

---
Steps to reproduce:

# sudo modprobe it87
modprobe: ERROR: could not insert 'it87': Device or resource busy

---
DMesg Output is:

[ 2530.034052] it87: Found IT8686E chip at 0xa40, revision 2
[ 2530.034091] it87: Beeping is supported
[ 2530.034099] ACPI Warning: SystemIO range 0x0000000000000A45-0x0000000000000A46 conflicts with OpRegion 0x0000000000000A45-0x0000000000000A46 (\GSA1.SIO1) (20170531/utaddress-247)
[ 2530.034106] ACPI: If an ACPI driver is available for this device, you should  use it instead of the native driver

---
Quesiton:

Would setting a lax ACPI flag upon boot possibly work around this issue?

raz4725, 2017-09-05:
Note: acpi_encorce_resources=lax will disable wired ethernet.

groeck, 2017-09-05:
acpi_enforce_resources=lax would be the only means to get this working right now. Unusual that this would cause Ethernet to fail. Can you send me a copy of the dsdt ? Also, I would appreciate a copy of /proc/cpuinfo.

Thanks,
Guenter

raz4725, 2017-09-05:
It would certainly be my pleasure, Guenter. 

Here's the /proc/cpuinfo and the DSDT of a Gigabyte AORUS X399 Gaming 7 board, converted via iasl.

[cpuinfo_1950x.txt](https://github.com/groeck/it87/files/1275977/cpuinfo_1950x.txt)
[x399-aorus_gaming_7_dsdt.zip](https://github.com/groeck/it87/files/1275982/x399-aorus_gaming_7_dsdt.zip)

raz4725, 2017-09-07:
@groeck 

It looks like upgrading to 4.13 from 4.13 RC7 may have fixed the issue --- no lax acpi enforce required.

Here's a sample output:

it8792-isa-0a60
Adapter: ISA adapter
in0:          +1.22 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +1.49 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.02 V  (min =  +0.00 V, max =  +2.78 V)
in3:          +2.01 V  (min =  +0.00 V, max =  +2.78 V)
in4:          +1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.49 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +1.66 V  (min =  +0.00 V, max =  +2.78 V)
Vbat:         +1.56 V
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
temp3:        +36.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

groeck, 2017-09-07:
It may seem so, but as part of the upgrade you reverted to the upstream driver, which doesn't support the IT8686, which is using the conflicting area (0xa40..0xa4f).

groeck, 2017-09-07:
What is really bad is that I don't find a legitimate reason in the dsdt why the address space in question should be reserved. I can see that it is reserved, but I don't see why.

raz4725, 2017-09-07:
You're right. Upon reboot, it shows as busy again. Weird how it worked once and only once.

bsodmike, 2017-09-20:
Chaps, I just tried the dumps on my X399 board with 1950X,

```
[root@threadripper ~]# isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 86 86 02 00 00 20 33 00 00 00 80 40 c8 18 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60: 0a 10 0a 00 0a 20 00 00 00 00 00 00 00 00 00 00
70: 00 00 20 00 00 06 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
b0: 00 00 00 00 00 00 00 00 00 00 00 c0 00 00 00 00
c0: 01 01 00 40 00 ff ff ff 01 00 00 00 00 18 00 00
d0: 91 00 08 0a 00 fd ff ff 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 dc 00 1d 20 12 49 00 00 0a 1c
f0: 00 00 00 00 00 00 0d 00 00 00 49 40 00 ff 00 0f

[root@threadripper ~]# isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

I haven't tried loading this (module) yet, does the above help?

bsodmike, 2017-09-20:
@raz4725 I'm getting the same output as you - here are the steps that I followed
https://forum.level1techs.com/t/threadripper-lm-sensors-halp/119487/8?u=bsodmike

`sensors-detect` pick up the chip on my Aorus X399 as `ID 0x8733`.  1950X CPU temp is reported as -55C (lol!)

groeck, 2017-10-01:
Latest version might improve the situation; please test and report.

raz4725, 2017-10-01:
@groeck ---

This does appear to work better, at least for me. I have the output from a fresh checkout of lm-sensors and it87 and modprobing with ignore_resource_conflict=1 on kernel 4.13.4 below. 

It looks like, on this board, it8622-isa-0a40 shows a more accurate CPU temperature than 0a60. I'm going to go further and say that 0a40 is more accurate all-around. It's interesting how the values differ between both, too. I'm assuming Temp1 and Temp3 are somewhere on the board and Temp 2 is the CPU itself? Also, I wonder why the WiFi adapter appears to have a temperature value 😆 

---
# modprobe it87 ignore_resource_conflict=1
# sensors
it8622-isa-0a60
Adapter: ISA adapter
in0:          +1.45 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +1.64 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +1.13 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.21 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.96 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.64 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +3.06 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.65 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.50 V
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +37.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

iwlwifi-virtual-0
Adapter: Virtual device
ERROR: Can't get value of subfeature temp1_input: I/O error
temp1:            N/A

it8622-isa-0a40
Adapter: ISA adapter
in0:          +1.36 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.24 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.12 V
fan1:        2272 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +47.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +34.0°C  (low  = +127.0°C, high = +127.0°C)
intrusion0:  OK

groeck, 2017-10-01:
Are you using force_id or are there really two IT8622 chips on that board ? If not, is that a different board ?

groeck, 2017-10-07:
Going to close. Please report if there are still issues.

closed 2017-10-07 by groeck