groeck/it87 #48: Gigabyte AB350N WiFi

Luke-Nukem, 2017-09-17:
Adding some info for this board:

Using current git driver
```
it8792-isa-0a60
Adapter: ISA adapter
in0:          +0.38 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +0.96 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.12 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:        +3.31 V  (min =  +0.00 V, max =  +5.56 V)
in4:          +1.32 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.22 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.31 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.21 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +18.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +29.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +46.0°C  

it8686-isa-0a40
Adapter: ISA adapter
in0:          +0.84 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.05 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +1.99 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.90 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.92 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.24 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.14 V  
fan1:        2295 RPM  (min =    0 RPM)
fan2:         685 RPM  (min =    0 RPM)
temp1:        +38.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +30.0°C  (low  =  +0.0°C, high = +80.0°C)  sensor = AMD AMDSI
temp4:        +16.0°C  (low  =  +0.0°C, high = +127.0°C)
temp5:        +47.0°C  (low  =  +0.0°C, high = -120.0°C)  sensor = thermistor
temp6:        +49.0°C  (low  =  +0.0°C, high = -120.0°C)
intrusion0:  ALARM
```

Using current git lm_sensors:
```
Found `ITE IT8686E Super IO Sensors'                        Success!
    (address 0xa40, driver `to-be-written')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found `ITE IT8792E Super IO Sensors'                        Success!
    (address 0xa60, driver `it87')
```

isadump (not sure if I've done this right):
```
[Sun Sep 17 02:05 PM] 
 [luke@localhost ~/src/it87]$ sudo isadump 0xa65 0xa66
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0xa65 and data register 0xa66.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 19 10 40 00 ff ff 00 00 00 80 48 00 30 ff ff ff 
10: ff ff ff 77 c7 00 00 00 ff ff ff ff ff ff ff ff 
20: 23 59 69 99 7b 72 ff 98 93 10 1c 20 80 1f 3a 3a 
30: ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 
40: 7f 7f 7f 7f 7f 7f 5f 40 ad 6a d4 00 ff ff ff ff 
50: ff 38 7f 7f 7f 80 00 00 90 00 09 12 60 00 00 00 
60: 00 14 41 ff 1c 03 00 ff 00 14 41 ff 1c 03 00 ff 
70: 00 14 41 ff 1c 03 00 ff ff ff ff 80 ff ff ff ff 
80: ff ff ff ff ff ff ff ff 00 00 00 00 00 bd 00 9a 
90: ff 00 00 00 ff 00 00 00 01 97 00 00 40 53 c1 42 
a0: e7 61 56 8c 5d fe 5f 3f 3e ff af ab 3f 3e 00 00 
b0: 6e 6d 6e 6b 6c 6b fb fd ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

id 0x8665 gives similar results:
```
[Sun Sep 17 02:06 PM] 
 [luke@localhost ~/src/it87]$ sudo modprobe it87 force_id=0x8665

[Sun Sep 17 02:08 PM] 
 [luke@localhost ~/src/it87]$ sensors
it8792-isa-0a60
Adapter: ISA adapter
in0:          +0.39 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +0.98 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.15 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:        +3.31 V  (min =  +0.00 V, max =  +5.56 V)
in4:          +1.34 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.24 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.31 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.21 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +15.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +27.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +31.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +39.0°C  

it8686-isa-0a40
Adapter: ISA adapter
in0:          +0.40 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.05 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +1.98 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.90 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.91 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.24 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.14 V  
fan1:        2280 RPM  (min =    0 RPM)
fan2:         686 RPM  (min =    0 RPM)
temp1:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +50.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +25.0°C  (low  =  +0.0°C, high = +80.0°C)  sensor = AMD AMDSI
temp4:        +16.0°C  (low  =  +0.0°C, high = +127.0°C)
temp5:        +41.0°C  (low  =  +0.0°C, high = -120.0°C)  sensor = thermistor
temp6:        +42.0°C  (low  =  +0.0°C, high = -120.0°C)
intrusion0:  ALARM
```

Is there anything else I can add or test here?

groeck, 2017-09-20:
Looks ok to me. Any concerns ?

Luke-Nukem, 2017-09-21:
No concerns. Just thought it might be helpful information even if it only
confirms that things are working fine. :)

closed 2017-09-21 by groeck