groeck/it87 #49: Gigabyte AORUS X399 Gaming 7 - Fedora 26, 1950X // CPU temp reported as -55C for Chip ID 0x8733

bsodmike, 2017-09-20:
Hi all,

Having compiled and loaded the module, my findings are documented here
https://forum.level1techs.com/t/threadripper-lm-sensors-halp/119487/8

```
[root@threadripper it87]# sensors
iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +38.0°C

it8792-isa-0a60
Adapter: ISA adapter
in0:          +1.38 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +1.51 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.05 V  (min =  +0.00 V, max =  +2.78 V)
in3:          +1.98 V  (min =  +0.00 V, max =  +2.78 V)
in4:          +1.79 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.50 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +1.68 V  (min =  +0.00 V, max =  +2.78 V)
Vbat:         +1.57 V
fan1:         814 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:        1024 RPM  (min =    0 RPM)
temp1:        +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
temp3:        +39.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

Why the incorrect reading?

qzed, 2017-09-20:
This might be relevant: https://images.anandtech.com/doci/11685/gigablkdiag.png . I'm not sure about the accuracy though, as the IT8792 does not seem to be present in the diagram (there is an IT8795E however, maybe a typo?).

You could also verify the readings on Windows using HWINFO (I plan to do so, but I have more pressing metters atm so it may take me a while).

groeck, 2017-09-20:
The sensor on the 8792 is not connected to the CPU and probably not configured (or not configured correctly) by the BIOS. The CPU is connected to the IT8686E, which as reported in the other case does not load due to an ACPI resource conflict.

groeck, 2017-09-20:
Please try the latest driver and load it with "modprobe it87 ignore_resource_conflict=1"

groeck, 2017-09-20:
There is a continued perception that updating the kernel module would have impact on the output of sensors-detect. However, just like changing the oil in a car won't fix its flat tires, updating a kernel module will have no impact on the output of sensors-detect.

qzed, 2017-09-20:
Using `modprobe it87 ignore_resource_conflict=1` both the IT8792 and IT8686 show up, and as you said the IT8686 shows the CPU temp (temp3). The only thing I see missing now is the `SYS_FAN4` fan header readout.

Here is the output of `sensors`:
```
it8792-isa-0a60
Adapter: ISA adapter
in0:          +1.33 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +1.50 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.04 V  (min =  +0.00 V, max =  +2.78 V)
in3:          +1.98 V  (min =  +0.00 V, max =  +2.78 V)
in4:          +1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.50 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.33 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.18 V
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +36.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +37.0°C

it8686-isa-0a40
Adapter: ISA adapter
in0:          +1.37 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.38 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.13 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.37 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.29 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.10 V
fan1:        2109 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +49.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +31.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +37.0°C  (low  = +127.0°C, high = +127.0°C)
temp5:        +39.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)
```

bsodmike, 2017-09-20:
@qzed cheers for the adding the Fedora support - I just rebuild the latest and tried this, 

```
[root@threadripper it87]# vim Makefile
[root@threadripper it87]# make
Makefile:940: "Cannot use CONFIG_STACK_VALIDATION, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  CC [M]  /opt/modules/it87/it87.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /opt/modules/it87/it87.mod.o
  LD [M]  /opt/modules/it87/it87.ko
[root@threadripper it87]# make copy
mkdir -p /lib/modules/4.12.13-300.fc26.x86_64/kernel/drivers/hwmon
cp it87.ko /lib/modules/4.12.13-300.fc26.x86_64/kernel/drivers/hwmon/
[root@threadripper it87]# modprobe it87 ignore_resource_conflict=1
[root@threadripper it87]# lsmod | grep it
it87                   57344  0
hwmon_vid              16384  1 it87
ip6table_security      16384  1
iptable_security       16384  1
ip6_tables             28672  5 ip6table_mangle,ip6table_filter,ip6table_security,ip6table_raw,ip6table_nat
[root@threadripper it87]# sensors-detect
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Gigabyte Technology Co., Ltd. X399 AORUS Gaming 7 [Default string]
# Kernel: 4.12.13-300.fc26.x86_64 x86_64
# Processor: AMD Ryzen Threadripper 1950X 16-Core Processor (23/1/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): yes
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): yes
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8686
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8733
```

This IT8686 is now being detected but `sensors` doesn't show any output (from that sensor).  I'm still just getting the exact same info as before... hmm!!!

qzed, 2017-09-20:
With regards to the -55°C sensor readings: I just connected some thermal sensors to the motherboard headers and it seems that
- `EC_TEMP1` is mapped to the `temp6` input of the IT8686
- `EC_TEMP2` is mapped to the `temp2` input of the IT8792

qzed, 2017-09-21:
FYI I'm using Arch Linux and a package (https://aur.archlinux.org/packages/it87-dkms-git/) so I'm not manually compiling the module. I can't see any significant differences between the package's build script and what you are manually doing though.

groeck, 2017-09-21:
I added the following note to README.

"There is a persistent perception that changes in this driver would have impact on the output of sensors-detect. This is not the case. sensors-detect is an independent application. Changes in this driver do not affect sensors-detect, and changes in sensors-detect do not affect this driver."

Maybe it helps.

As for output from the "sensors" command, it may help to have a look into the kernel log.

groeck, 2017-09-21:
@qzed: How many fans do you have connected in your system ? The IT8686E supports 6 fans; I'll have another look into the datasheet to see if I missed an enable bit somewhere.

qzed, 2017-09-21:
I have two fan headers connected `CPU_FAN` and `SYS_FAN4`. The `CPU_FAN` seems to be mapped to the `fan1` input on the IT8686 and seems to show correct values. I however use a PWM fan splitter on `SYS_FAN4`, could this influence the tachometer readings?

groeck, 2017-09-21:
@qzed: Yes, a pwm splitter will only send the output from one of the connected fans back to the source.

qzed, 2017-09-21:
Okay, I just noticed that those fans are not running. I'll try to find out why.

groeck, 2017-09-21:
A note on Fedora 26: My understanding is that it compresses kernel modules. Copying an uncompressed module into the modules directory will have no effect since modprobe will still install the compressed version of the driver.

qzed, 2017-09-21:
Okay, I think loading the module changes something in the controller. I have observed that:
- `SYS_FAN4` maps to `fan3` on the IT8792
- `/sys/class/hwmon/hwmon1` maps to the IT8686 (5 pwm controls)
- `/sys/class/hwmon/hwmon2` maps to the IT8792 (3 pwm controls)
- loading the module (`modprobe it87 ignore_resource_conflict=1`) stops `SYS_FAN4`
- executing `cat /sys/class/hwmon/hwmon1/pwm1_enable` temporarily spins `SYS_FAN4` until I run `sensors`, after which they stop again (they seem to spin a bit faster for a second before stopping). And yes this is `hwmon1` which maps to the IT8686 to which `SYS_FAN4` is _not_ connected.

Furthermore
- `cat /sys/class/hwmon1/pwm{1-5}_enable` all print 2 and
- `cat /sys/class/hwmon2/pwm{1-3}_enable` all print 1

Am I correct that `2` indicates bios-control?

qzed, 2017-09-21:
Loading the module using `modprobe it87 force_id=0x8733 ignore_resource_conflict=1` does not change the output of `cat /sys/class/hwmon2/pwmX_enable`, however it does not stop the fans (subsequent `sensor` also don't stop the fans either). The IT8686 is now detected as IT8792, of course.

I think it looks like the IT8686 specific code influences something on the IT8792.

bsodmike, 2017-09-21:
@groeck thanks for the tip on compressing the module, I fixed my changes so that post-copy it compresses it.

Also for Fedora, I load the driver as `modprobe it87 force_id=0x8686 ignore_resource_conflict=1`

```
[root@threadripper it87]# modprobe it87 force_id=0x8686 ignore_resource_conflict=1
[root@threadripper it87]# sensors
it8686-isa-0a60
Adapter: ISA adapter
in0:          +1.50 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +1.66 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +1.15 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.18 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.97 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.66 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +3.06 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.70 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.46 V
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:        1095 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +43.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +41.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp4:       -128.0°C  (low  = +106.0°C, high = +106.0°C)
temp5:        -51.0°C  (low  = +62.0°C, high = -74.0°C)  sensor = thermistor
temp6:        +60.0°C  (low  =  -1.0°C, high =  -1.0°C)
intrusion0:  ALARM

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +41.0°C

it8686-isa-0a40
Adapter: ISA adapter
in0:          +1.38 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.97 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.31 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.05 V
fan1:        1430 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +44.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +54.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +43.0°C  (low  = +127.0°C, high = +127.0°C)
temp5:        +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)
intrusion0:  ALARM
```

qzed, 2017-09-21:
@bsodmike is the `fan3` on the first chip connected to the `SYS_FAN4` header on your board?  And if so, does the fan speed match? Btw. are there any reasons you used `force_id`?

bsodmike, 2017-09-21:
@qzed just couldn't get it to provide any info beyond what I was seeing with the original driver without using `force_id`.

Yes, I have a fan on SYS_FAN4 and another on SYS_FAN5_PUMP.  Corsair H115i is hooked into CPU_FAN (just the tachymeter-output).  I'll jump into the BIOS to confirm.

qzed, 2017-09-21:
Are both fans spinning? I think there should be 3 fan inputs  > 0 rpm in your case (I'm just trying to find out if you can confirm the problem I'm having with the one fan stopping after loading the module).

And it should work without using `force_id`, at least for me it works. Using this parameter forces one of the chips to be detected incorrectly, i.e. one chip should be IT8686 and the other IT8792, `force_id` forces both to be detected as the same (specified) type.

groeck, 2017-09-21:
Quick update: I am trying to understand what is going on with the stopped fan, but so far I have not found a problem in the code. Hope there is no interference with the BIOS accessing the chip after all.

groeck, 2017-09-21:
@qzed:  Can you check if reading a temperature sensor type attribute after reading _/sys/class/hwmon/hwmon1/pwm1_enable_ stops the fans ?

Answering earlier question: pwmX_enable=2 configures automatic fan control, pwmX_enable=1 configures manual fan control.

groeck, 2017-09-21:
@bsodmike: Sorry, I don't understand. If you have to use force_id, something is wrong.

bsodmike, 2017-09-21:
@groeck 

> I'm just trying to find out if you can confirm the problem I'm having with the one fan stopping after loading the module

Hang on - do you mean that your fan stops spinning or is the fan spinning but reported as 0 rpm?

In my case both fans are spinning and CPU_FAN header has the Corsair pump attached; so that would get the pump's rpm reading.

bsodmike, 2017-09-21:
Success!!  Just needed a reboot...

```
[root@threadripper ~]# sensors
it8792-isa-0a60
Adapter: ISA adapter
in0:          +0.82 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +1.50 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.05 V  (min =  +0.00 V, max =  +2.78 V)
in3:          +1.98 V  (min =  +0.00 V, max =  +2.78 V)
in4:          +1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.50 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.33 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.14 V
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:        1125 RPM  (min =    0 RPM)
temp1:        +44.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +44.0°C

it8686-isa-0a40
Adapter: ISA adapter
in0:          +0.84 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.97 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.31 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.05 V
fan1:        1430 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +45.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +45.0°C  (low  = +127.0°C, high = +127.0°C)
temp5:        +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)
intrusion0:  ALARM
```

bsodmike, 2017-09-21:
@groeck I can confirm that I'm seeing your issue as well now.  One of those fans has stopped.... hmmm

groeck, 2017-09-21:
@bsodmike: I don't see the problem. That was @qzed. I am trying to understand it. Which fan stopped spinning ?

Same question: If 'cat /sys/class/hwmon/hwmon1/pwm1_enable' causes the fan(s) to start spinning again, as @qzed reported, does 'cat  /sys/class/hwmon/hwmon1/temp1_type' cause it to stop ?

[ it might me hwmon2 in your case ]

bsodmike, 2017-09-21:
@groeck - Previously, the fan that had stopped was `SYS_FAN5_PUMP`.

Here's a new run of mine:

- Fresh boot
- `modprobe it87 ignore_resource_conflict=1` // I haven't run `sensors` yet.
- `SYS_FAN4` stopped.
- Ran `cat /sys/class/hwmon/hwmon1/pwm1_enable` result = `2` // `SYS_FAN4` starts running again!
- Ran `sensors`, only showing fan output

```
fan1:        1430 RPM  (min =   10 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
```
Having checked in BIOS the rpm for the H115i looks about right.

Both fans were set to ramp on a fan-curve based on CPU temp, but with CPU temp ~30degC they were reporting ~890rpm.

```
temp1:        +41.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +50.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +28.0°C  (low  =  +0.0°C, high = +70.0°C)  sensor = AMD AMDSI
temp4:        +40.0°C  (low  = +127.0°C, high = +127.0°C)
temp5:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)
```

On the plus side, at least the range for `temp3` looks sensible.  I would be expecting `fan4` and `fan5` to show up in the 860-950 rpm range.

Hope this helps!!

qzed, 2017-09-21:
@bsodmike The IT8792 has some fan headers too and `SYS_FAN4` seems to be connected to it (`fan3`). So you might need to look at this chip, the output above seems to be from the IT8686.

@groeck `cat /sys/class/hwmon/hwmon1/temp3_type` causes it to stop. `temp1_type` and `temp2_type` did so too unitl (I think, I'll try to reproduce that after rebooting) I ran `sensors`. Afterwards only `temp3_type` stops the fan.

bsodmike, 2017-09-21:
> @bsodmike The IT8792 has some fan headers too and SYS_FAN4 seems to be connected to it (fan3). So you might need to look at this chip, the output above seems to be from the IT8686.

That makes total sense!!

From the 8792

```
fan1:        1004 RPM  (min =    0 RPM) // What's this?
fan2:           0 RPM  (min =    0 RPM)
fan3:         760 RPM  (min =    0 RPM) // SYS_FAN4?
```

From 8686

```

```

**Regarding Voltages**

As per the Gigabyte block-diagram, only the IT8795EC (which we are considering as the 8792??) is connected to the CPU, so...

Do we have a mapping of these?

From the 8792

```
in0:          +0.82 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +1.51 V  (min =  +0.00 V, max =  +2.78 V) // Closes to VCore; UEFI reports 1.38 - 1.40vdc
in2:          +1.05 V  (min =  +0.00 V, max =  +2.78 V)
in3:          +1.98 V  (min =  +0.00 V, max =  +2.78 V)
in4:          +1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.50 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.36 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.14 V
```

From 8686

Are these invalid then?

```
in0:          +0.84 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.97 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.34 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.07 V
```

qzed, 2017-09-21:
@groek I ran all the `cat` calls on the 8686, the fan stopping is on the 8792.

`isadump`: 
[isadump_0xa45_0xa46_sysfan4_running.txt](https://github.com/groeck/it87/files/1321222/isadump_0xa45_0xa46_sysfan4_running.txt)
[isadump_0xa45_0xa46_sysfan4_stopped.txt](https://github.com/groeck/it87/files/1321225/isadump_0xa45_0xa46_sysfan4_stopped.txt)
[isadump_0xa65_0xa66_sysfan4_running.txt](https://github.com/groeck/it87/files/1321224/isadump_0xa65_0xa66_sysfan4_running.txt)
[isadump_0xa65_0xa66_sysfan4_stopped.txt](https://github.com/groeck/it87/files/1321223/isadump_0xa65_0xa66_sysfan4_stopped.txt)

`acpidump`
[acpidump.txt](https://github.com/groeck/it87/files/1321242/acpidump.txt)

qzed, 2017-09-21:
@bsodmike As @groeck mentioned in another thread, the IT8792E and the IT8795E have the same ID (which IMHO is absolutely stupid), so sensors detects it as IT8792. If we are lucky there are no significant differences between these chips and we can treat it as IT8792. In reality the board seems to use the IT8795 tough.

As for the voltages, I think you've got that backwards with the chips (the IT8686 is connected to the CPU). I also don't think they are necessarily invalid, just connected to something else. As for the mapping, on the IT8686 I have:
```
in0:          +1.36 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.38 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.24 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.38 V  (min =  +0.00 V, max =  +3.06 V)
```

where `in0` seems to be Vcore (or at least matches what I expect Vcore to be).

bsodmike, 2017-09-21:
@qzed Ah thanks for the clarification.

Oddly enough though, my IT8686 voltages are tad different to yours.  Especially `in0`.

```
it8686-isa-0a40
Adapter: ISA adapter
in0:          +0.84 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.97 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
```

qzed, 2017-09-21:
That `in0` difference is weird. Did you make any voltage-related changes in the bios? I set Vcore to normal (from auto) and added an offset (something around -0.09V) to keep Vcore down a bit, maybe that influences the readings. I don't think that the CPU would run with 0.84V Vcore though, even if the automatic mode would somehow reduce the voltage more.

qzed, 2017-09-21:
With regards to the termistors: there are multiple readings in the bios, such as chipset etc. I have no idea about the mapping though. Maybe there are some VRM temperature readings too.

bsodmike, 2017-09-21:
@qzed mine's set at `auto` and I noticed in the UEFI even whilst idling `Vcore` would fluctuate and even spike to 1.4-1.41vdc.  I'll try setting mine to normal and checking if this makes any changes.  Noted re. the thermistors.

groeck, 2017-09-21:
Note that Vcore fluctuations are normal (except possibly if the system is overclocked). On an idle system with Ryzen I have seen voltages as low as 0.47V.
@everyone: Still trying to understand the problem with the stopped fans.

groeck, 2017-09-21:
Fan control on the 8792 is configured to manual. In the working case, its pwm registers are > 0, and connected fans are running. In the non-working case, the pwm registers are 0, and the fans are stopped.
After reading tempX_type, the driver currently leaves the 8686's bank select register at 2, meaning the register values are different.
If there is some BIOS code which reads temperatures from the 8686 and, based on it, writes fan control values into the 8792/8795, it might become confused and stop the fans in the 8792. Not sure yet what to do about that, or if there is anything we can do in the first place.

qzed, 2017-09-21:
@groeck Thanks for investigating, is there anything we can do to help?

groeck, 2017-09-21:
I wrote some code to improve bank selection. Please try and report. Note that the code is untested (I don't have HW where I am right now).

groeck, 2017-09-21:
@qzed: Yes, sure, find a fix :-)

qzed, 2017-09-21:
@groeck yeah, I think I need to understand how the whole SuperIO Linux subsystem or what's behind this first :-) 

Your comment about the 8792 beeing configured to manual got me thinking: The previous `isadump` with the stopped fans were created after the module has been loaded. Here are some dumps of the initial system without the module loaded:

[isadump_0xa45_0xa46_initial.txt](https://github.com/groeck/it87/files/1322917/isadump_0xa45_0xa46_initial.txt)
[isadump_0xa65_0xa66_initial.txt](https://github.com/groeck/it87/files/1322918/isadump_0xa65_0xa66_initial.txt)

I'll try the update now.

qzed, 2017-09-21:
Awesome, seems like the update fixed it. I loadet the module, ran `sensors` and the fans are still spinning.

groeck, 2017-09-21:
I am not really happy, though. What that means is that the BIOS (or an EC in the system) does access the chips. This in turn means that each chip access adds some risk because it may change the bank register value, thus messing up access for the other accessor. I'll rework the code a bit more so it doesn't re-read the temperature type registers after initial startup to further limit that risk.
Do you see temperature type attributes for all temperatures on IT8686 now ?

qzed, 2017-09-21:
Hmm yeah, that does sound like it may cause some trouble.

As for the `tempX_type`, on the IT8686 it's still not there for 4 and 6.

groeck, 2017-09-22:
Can you try again ? There was a bug in the code.

qzed, 2017-09-22:
Sure. It seems to be working now. Every `tempX_type` is present on the IT8686 and `sensors` seems to  display them correctly (both previously missing get detected as thermistor).

qzed, 2017-09-22:
Glad I could help, thank you for supporting this board.

bsodmike, 2017-09-22:
Just to confirm, things are looking good and even `in0` seems to be registering correctly now.  It may have been an `idling` issue.

```
[root@threadripper it87]# modprobe it87 ignore_resource_conflict=1
[root@threadripper it87]# sensors
it8792-isa-0a60
Adapter: ISA adapter
in0:          +1.38 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +1.50 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.04 V  (min =  +0.00 V, max =  +2.78 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +2.78 V)
in4:          +1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.50 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.33 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.14 V
fan1:        1138 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:        1048 RPM  (min =    0 RPM)
temp1:        +48.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +45.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +46.0°C

it8686-isa-0a40
Adapter: ISA adapter
in0:          +1.39 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.97 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.34 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.07 V
fan1:        1430 RPM  (min =   10 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +48.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +57.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +37.0°C  (low  =  +0.0°C, high = +70.0°C)  sensor = AMD AMDSI
temp4:        +47.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +43.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

groeck, 2017-09-29:
To follow up on this, it appears that there is an Embedded controller accessing the chips in parallel, which can cause all kinds of havoc. This embedded controller may also provide some of the fan speeds and other data visible in the BIOS. I don't have the information necessary to access this data.
I am working on a driver update which should limit the impact on the it87 driver.

bsodmike, 2017-09-29:
@qzed Have you experienced any strange shutdowns once the it87 driver is loaded?  I had one earlier today and recall that only happened after I checked my temps.

Fresh reboot, without the module loaded and my box has been running for over 9-hours.

keantoken, 2017-09-29:
I've had spontaneous reboots that left an MCE error in the logs. IIRC it is a CPU hardware defect in some Ryzen chips.

bsodmike, 2017-09-29:
@groeck thoughts on the shutdowns? Thanks @keantoken.  Would you mind sharing the exact MCE error you got?

groeck, 2017-09-30:
There is _some_ possibility that the driver is responsible. After all, we _do_ know that there is an Embedded Controller accessing the chip as well. You could try to establish a correlation by running "sensors" in a loop, but I am not sure if I want to recommend that at this time.

bsodmike, 2017-09-30:
It also does make sense that when our loading of the driver potentially blocks the Embedded Controller accessing chip -- the system may determine shutting down is the best course of action, i.e. if It can't receive CPU Temp data....?

groeck, 2017-09-30:
@bsodmike: No, the Embedded Controller isn't blocked. It accesses the chip through an I2C interface. The workaround (already used by PC products) is to disable that interface while the CPU accesses the chip. The EC might get confused, though, if accesses happen in parallel, since the chip has a paged register set.

keantoken, 2017-09-30:
@bsodmike do you know a simple way of searching through system logs for mce: lines? It might still be there somewhere, but my logs are being heavily spammed by evbug: lines for some reason I don't understand. Here is the thread about the MCE error:

https://community.amd.com/thread/216084

qzed, 2017-09-30:
@bsodmike hmm, I didn't have any problems in this direction. To be honest, I'm not checking sensors frequently, I did run them in a watch-loop once or twice without problems though.

A follow-up regarding the voltages you posted a while ago: I'm running the powersave-governor at the moment and getting 0.8V for vcore (ramping up as soon as I change that to schedutil), so that seems to be possible on idle.

@keantoken is this a new hardware problem or is this the "segfault during compilation" stuff (that _should_ not happen with threadripper)?

bsodmike, 2017-09-30:
@keantoken try `journalctl --list-boots` if you want to track down the last boots.  You can then try `journalctl --boot=-1 -p mce` or drop the p flag if you want to scroll through.  I've tried `-p err` in the past and that has sort of helped.

@groeck Ah noted.  I used to work with I2c with microcontrollers (PIC) back in the day, so now I do have a better idea of what you're referring to.

keantoken, 2017-09-30:
This is a completely different issue than the compile segfaults. For me it's rare, it might happen once a week. I don't know if the new CPU will fix the problem, but that often doesn't seem to be the case.

BTW, I don't have the hardware for this thread, I just jumped in to mention the MCE bug.

[22 more replies]

closed 2017-12-03 by groeck