groeck/it87 #50: commit aeafb7be7c845d3a60691a32c6e1d86eca623691 breaks fan control on IT8665

fnarfbargle, 2017-09-21:
I manually set up my fan control on my ASUS X370 Prime (IT8665).

IT87=`find /sys/devices/platform/ -name it87*`
if [ -n "$IT87" ] ; then
	pushd $IT87/hwmon/hwmon*
	echo 1 > pwm1_enable
	echo 90 > pwm1
	echo 30000 > pwm1_auto_point2_temp
	echo 65000 > pwm1_auto_point3_temp
	echo 50 > pwm1_auto_slope
	echo 2 > pwm1_enable
	popd
fi;

test:/sys/devices/platform/it87.656/hwmon/hwmon1# grep . pwm1*
pwm1:255
pwm1_auto_channels_temp:1
pwm1_auto_point1_temp:-128000
pwm1_auto_point1_temp_hyst:-128000
pwm1_auto_point2_temp:30000
pwm1_auto_point3_temp:65000
pwm1_auto_slope:50
pwm1_auto_start:255
pwm1_enable:2
pwm1_freq:23437

Commit referenced above  : "Rework pwm temperature mapping" breaks things so no matter what I write to pwm1_enable it remains set to "2", and thus a write to pwm1 results in "./bootsensors: line 8: echo: write error: Device or resource busy".

Reverting that commit allows it to work again.

Have I missed something I'm supposed to be configuring to allow me to manually tweak this like I used to?

groeck, 2017-09-21:
Nah, I probably messed it up. Thanks for letting me know.

groeck, 2017-09-21:
Can you try again ?

fnarfbargle, 2017-09-21:
Yep, that appears to have fixed it.

Thanks.

closed 2017-09-21 by groeck