groeck/it87 #51: Wrong temp Crosshair VI Hero Kern 4.11 ?

Extarys, 2017-09-22:
lm-sensors shows wrong infos? Or bad oconfig on my end? I'm no pro

Install process
```
cd /opt
git clone https://github.com/groeck/it87.git
cd it87
make -j4
sudo make install
sudo modprobe it87
sudo sensors-detect
sensors
```
Output:
```
$ sensors
it8665-isa-0290
Adapter: ISA adapter
in0:          +0.70 V  (min =  +2.60 V, max =  +0.50 V)
in1:          +0.68 V  (min =  +2.47 V, max =  +0.91 V)
in2:          +2.01 V  (min =  +1.34 V, max =  +2.08 V)
in3:          +1.98 V  (min =  +2.17 V, max =  +2.54 V)
in4:          +0.57 V  (min =  +2.35 V, max =  +1.38 V)
in5:          +0.49 V  (min =  +2.26 V, max =  +1.43 V)
in6:          +0.95 V  (min =  +1.02 V, max =  +2.10 V)
3VSB:         +3.31 V  (min =  +1.83 V, max =  +4.86 V)
Vbat:         +3.16 V  
+3.3V:        +3.31 V  
fan1:         756 RPM  (min =   21 RPM)
fan5:           0 RPM  (min =   -1 RPM)  ALARM
temp1:        +44.0°C  (low  = -114.0°C, high = -14.0°C)
temp2:        -86.0°C  (low  = +123.0°C, high = -49.0°C)  sensor = AMD AMDSI
temp3:        +37.0°C  (low  = -89.0°C, high = -67.0°C)  sensor = disabled
intrusion0:  ALARM
```

Temp 3 never changes, temp2 is wrong I'm not with nitrogen.
It was working until I reinstall Mint last week. No idea what changed.

I'm using Crosshair VI Hero with 1800x.

**/proc/version**
`Linux version 4.11.0-14-generic (buildd@lcy01-08) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) ) #20~16.04.1-Ubuntu SMP Wed Aug 9 09:06:22 UTC 2017`

**$ cat /etc/*-release**
```
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=18.2
DISTRIB_CODENAME=sonya
DISTRIB_DESCRIPTION="Linux Mint 18.2 Sonya"
NAME="Linux Mint"
VERSION="18.2 (Sonya)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 18.2"
VERSION_ID="18.2"
HOME_URL="http://www.linuxmint.com/"
SUPPORT_URL="http://forums.linuxmint.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/linuxmint/"
VERSION_CODENAME=sonya
UBUNTU_CODENAME=xenial
```

Extarys, 2017-09-22:
Oh damn I'm tired, the issue just under this one is the same -_-, closing.

Extarys, 2017-09-22:
Oops closed too early, no fix for me on the other issue.

groeck, 2017-09-22:
For sensors-detect, please read README. For the other problem, please try the latest version of the driver. Make sure that the new driver is actually loaded. I can't say anything about the temperatures, but in your output there are only two fans, but the board has 5 fan headers. That is definitely not correct (unless the other fans are disabled in sensors3.conf).

Extarys, 2017-09-23:
I installed it87 yesterday cloning the repo.
For the rest, no idea what you are talking about :\

I previously only clone installed and it always worked, that is why I'm now opening a ticket, because my temps aren't correct.

groeck, 2017-09-23:
I'll need output of:

sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
sudo isadump 0x295 0x296

Extarys, 2017-09-24:
Thanks dude! I'm on OC so I had to back up until I can get readings.
> sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
```
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 65 06 40 3d d0 db 81 04 d4 80 49 85 38 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 02 a0 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: d0 d3 81 04 d4 00 00 00 40 00 00 00 10 00 00 11 
d0: ff 00 00 1a 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c 
f0: 00 00 00 00 00 00 0e 00 00 00 00 00 00 00 00 00
```
>  sudo isadump 0x295 0x296
```
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 7a 7b bf 00 00 08 00 00 c0 44 08 30 19 ff 36 
10: fa 4e dc 70 c0 80 80 98 03 ff 04 79 56 dc c0 00 
20: 40 3e b8 b6 34 2d 52 98 90 27 1e 27 27 27 27 98 
30: 2e ef 53 e3 bf 7b e9 c7 7f d8 83 cf c1 5e df 54 
40: f2 8e cf 7b bd a7 1d 6b ad 6a d4 00 5f 40 dd 63 
50: ff 00 7f 7f 7f 08 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 9d 00 0f 7c 80 1e 46 99 14 00 0f b2 
70: 80 14 4b 33 8f 00 0f 56 80 1e 46 99 14 00 0f b2 
80: ff ff ff ff 00 00 00 00 00 00 00 00 00 f7 00 78 
90: 00 00 00 ff ff 00 00 ff 42 99 00 00 01 4b e0 27 
a0: 7f 7f 7f 80 00 00 0f 80 80 7f 7f ff 00 00 0f ff 
b0: f5 15 00 00 dc 8e 27 94 ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

Looks Kryptonese to me

groeck, 2017-09-24:
I am kind of baffled. The board has lots of fan headers, but according to the above data almost all the fans tachometers on this chip are disabled. Also, the temperature registers (registers 0x29..0x2e in the second dump) are all fine and show temperatures of 37 (0x27) or 30 (0x1e) degrees C. Can you try the latest version of the driver ? It has some changes regarding register bank selection. Also, does the board have a second Super-IO chip, by any chance ?

Extarys, 2017-09-24:
Cloned it in my home folder.

```
make -j4
sudo make install
reboot
```
I have modeprobe it87 in my config file.
> sensors
```
it8665-isa-0290
Adapter: ISA adapter
in0:          +0.70 V  (min =  +2.60 V, max =  +0.50 V)  ALARM
in1:          +0.68 V  (min =  +2.47 V, max =  +0.91 V)  ALARM
in2:          +2.01 V  (min =  +1.34 V, max =  +2.08 V)
in3:          +1.98 V  (min =  +2.17 V, max =  +2.54 V)  ALARM
in4:          +0.57 V  (min =  +2.35 V, max =  +1.38 V)  ALARM
in5:          +0.77 V  (min =  +2.26 V, max =  +1.43 V)  ALARM
in6:          +0.91 V  (min =  +1.02 V, max =  +2.10 V)  ALARM
3VSB:         +3.29 V  (min =  +1.83 V, max =  +4.86 V)
Vbat:         +3.16 V  
+3.3V:        +3.31 V  
fan1:         478 RPM  (min =   21 RPM)
temp1:        +27.0°C  (low  = -114.0°C, high = -14.0°C)  ALARM
temp2:       -109.0°C  (low  = +123.0°C, high = -49.0°C)  sensor = thermal diode
temp3:        +41.0°C  (low  = -89.0°C, high = -67.0°C)  ALARM  sensor = thermistor
intrusion0:  ALARM

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```

Should I try to revert to an older kernel? As long as it's above or equal to 4.10 for Ryzen multi-threading I'm fine for now.

Which test/commands/settings do you need?

groeck, 2017-09-25:
It must be the latest driver because now it doesn't show fan5 anymore, and I changed that (it is disabled as well). Temperature limit values are weird but correct (ie they match register values). Still no idea why the temperature outputs are wrong. 
You would see if there is a second Super-IO chip by running the "sensors" command. Also, dmesg would tell you.
Changing the kernel version should not make a difference.

Can you run "sudo isadump 0x295 0x296" multiple times and check if the register value at 0x2a changes its value ?

groeck, 2017-09-25:
One more question: Did you at some point update the BIOS ?

Extarys, 2017-09-25:
Hmmm, yeah! I updated to Version 1402 but can't remember when. I totally forgot my bad. Just saw Version 1602 is out though. Should I update?

I have 2x16 GB of 3000MHz ram, Ryzen don't like them that's why I update every new bios version. Since last update I can now run them @ 2933 MHz without issue. @ 4.05 GHz :D

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 7a 7b 80 00 00 13 00 00 c0 44 08 30 96 ff 14 
10: fa 4e dc 70 c0 80 80 98 07 ff 07 79 56 dc 80 00 
20: 40 3e b8 b6 41 47 52 98 91 b1 ba 93 93 93 93 98 
30: 2e ef 53 e3 bf 7b e9 c7 7f d8 83 cf c1 5e df 54 
40: f2 8e cf 7b bd a7 1d 6b ad 6a d4 00 5f 40 dd 63 
50: ff 13 7f 7f 7f 13 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 9d 00 0f 33 80 1e 46 99 14 00 0f 99 
70: 80 14 4b 33 8f 00 0f 33 80 1e 46 99 14 00 0f 99 
80: ff ff ff ff 00 00 00 00 00 00 00 00 00 f7 00 78 
90: 00 00 00 ff ff 00 00 ff 42 99 00 00 01 4b e0 29 
a0: 7f 7f 7f 80 00 00 0f 80 80 7f 7f ff 00 00 0f ff 
b0: f5 15 00 00 dc 8e 27 94 ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 7a 7b 80 00 00 13 00 00 c0 44 08 30 96 ff 17 
10: fa 4e dc 70 c0 80 80 98 07 ff 07 79 56 dc 80 00 
20: 40 3e b8 b6 41 47 53 97 91 b1 ba 93 93 93 93 98 
30: 2e ef 53 e3 bf 7b e9 c7 7f d8 83 cf c1 5e df 54 
40: f2 8e cf 7b bd a7 1d 6b ad 6a d4 00 5f 40 dd 63 
50: ff 13 7f 7f 7f 13 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 9d 00 0f 33 80 1e 46 99 14 00 0f 99 
70: 80 14 4b 33 8f 00 0f 33 80 1e 46 99 14 00 0f 99 
80: ff ff ff ff 00 00 00 00 00 00 00 00 00 2b 00 25 
90: 00 00 00 ff ff 00 00 ff 42 99 00 00 01 4b e0 27 
a0: 7f 7f 7f 80 00 00 0f 80 80 7f 7f ff 00 00 0f ff 
b0: f5 15 00 00 dc 8e 27 94 ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 7a 7b 80 00 00 13 00 00 c0 44 08 30 98 ff 17 
10: fa 4e dc 70 c0 80 80 98 07 ff 07 79 56 dc 80 00 
20: 40 3e b8 b6 41 47 52 97 91 b1 ba 93 93 93 93 98 
30: 2e ef 53 e3 bf 7b e9 c7 7f d8 83 cf c1 5e df 54 
40: f2 8e cf 7b bd a7 1d 6b ad 6a d4 00 5f 40 dd 63 
50: ff 13 7f 7f 7f 13 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 9d 00 0f 33 80 1e 46 99 14 00 0f 99 
70: 80 14 4b 33 8f 00 0f 33 80 1e 46 99 14 00 0f 99 
80: ff ff ff ff 00 00 00 00 00 00 00 00 00 50 00 70 
90: 00 00 00 ff ff 00 00 ff 42 99 00 00 01 4b e0 28 
a0: 7f 7f 7f 80 00 00 0f 80 80 7f 7f ff 00 00 0f ff 
b0: f5 15 00 00 dc 8e 27 94 ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```
;)

groeck, 2017-09-25:
Please update the BIOS. Your latest 295/296 logs show all negative temperatures (registers 0x29..0x2e). That is all bad (negative temperatures) and different to your original log.

Extarys, 2017-09-25:
Np will do and keep you posted.

groeck, 2017-09-26:
Can you also send me the full set of ACPI tables ? That is the output of "sudo acpidump".

Extarys, 2017-09-26:
[acpidump.txt](https://github.com/groeck/it87/files/1331738/acpidump.txt)

Note that I didn't update the BIOS yet, didn't have time to reboot - too many things opened.

Extarys, 2017-09-27:
Nevermind, after reboot it stopped working... What the....

groeck, 2017-09-27:
They seem to be doing some odd things with temperature sensor mapping and fan control. There is definitely some information missing - the board has how many fans ? 6 ? 8 ? And only one shows up. This doesn't really make sense. I am out of ideas right now, sorry.

Extarys, 2017-09-27:
No problem, at least we got some temps reading! ;) I'll let you decide if you want to close this or leave it open or whatever. Thanks again for all your time

groeck, 2017-09-27:
Let's leave it open for a few more days. Maybe I can find out some more.

Extarys, 2017-09-27:
Sure thing.

groeck, 2017-09-28:
The board also has an Embedder Controller which provides the missing information and continuously accesses the Environmental Controller. The observed problems are caused by interference between Embedded Controller read/write operations and driver read/write operations. There is a workaround but it is complex to implement. I'll provide test code once available.

groeck, 2017-10-01:
Latest version should solve the Embedded Controller controller synchronization problems. I was not able to test with 8665, though, only with 8686 and 8732/8792 (FWIW with kernel version v4.13.4 on Gigabyte AB350M-Gaming 3). New module parameter mmio provides optimized access but is completely untested. Test feedback would be welcome.

Extarys, 2017-10-03:
Everything works fine for temp monitoring thanks.

```
$ sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

it8665-isa-0290
Adapter: ISA adapter
in0:          +0.72 V  (min =  +2.52 V, max =  +0.51 V)  ALARM
in1:          +0.68 V  (min =  +2.47 V, max =  +0.91 V)  ALARM
in2:          +2.01 V  (min =  +1.34 V, max =  +2.08 V)
in3:          +1.98 V  (min =  +2.26 V, max =  +2.54 V)  ALARM
in4:          +0.57 V  (min =  +2.35 V, max =  +1.38 V)  ALARM
in5:          +0.49 V  (min =  +2.60 V, max =  +1.43 V)  ALARM
in6:          +0.95 V  (min =  +1.01 V, max =  +2.10 V)  ALARM
3VSB:         +3.31 V  (min =  +1.68 V, max =  +4.86 V)
Vbat:         +3.14 V  
+3.3V:        +3.31 V  
fan1:        1036 RPM  (min =   10 RPM)
temp1:        +57.0°C  (low  = -122.0°C, high = -14.0°C)  ALARM
temp2:        +30.0°C  (low  = +123.0°C, high = -49.0°C)  ALARM  sensor = thermistor
temp3:        +34.0°C  (low  = -89.0°C, high = -71.0°C)  ALARM  sensor = thermistor
intrusion0:  ALARM
```

```
$ uname -a
Linux jer-ryzen 4.13.4-041304-generic #201709270931 SMP Wed Sep 27 13:35:03 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

groeck, 2017-10-07:
Closing. Please report if there are still issues.

closed 2017-10-07 by groeck

Extarys, 2017-10-07:
Thank you buddy!