groeck/it87 #53: Error! Bad return status for module build on kernel

testsieger73, 2017-09-28:
Installing via yaourt gives me the following error:

(1/2) Install DKMS modules
==> dkms install it87/105.4d29188 -k 4.9.51-1-lts
Error! Bad return status for module build on kernel: 4.9.51-1-lts (x86_64)
Consult /var/lib/dkms/it87/105.4d29188/build/make.log for more information.
==> dkms install it87/105.4d29188 -k 4.13.3-1-ARCH
Error! Bad return status for module build on kernel: 4.13.3-1-ARCH (x86_64)
Consult /var/lib/dkms/it87/105.4d29188/build/make.log for more information.
(2/2) Arming ConditionNeedsUpdate...

I have already made a couple of installations of this driver on Manjaro and Antergos, but this error is new.

make.log:

DKMS make.log for it87-105.4d29188 for kernel 4.13.3-1-ARCH (x86_64)
Do 28. Sep 16:29:57 CEST 2017
make: Verzeichnis „/usr/lib/modules/4.13.3-1-ARCH/build“ wird betreten
  CC [M]  /var/lib/dkms/it87/105.4d29188/build/it87.o
/var/lib/dkms/it87/105.4d29188/build/it87.c:77:10: schwerwiegender Fehler: version.h: Datei oder Verzeichnis nicht gefunden
 #include "version.h"
          ^~~~~~~~~~~
Kompilierung beendet.
make[1]: *** [scripts/Makefile.build:309: /var/lib/dkms/it87/105.4d29188/build/it87.o] Fehler 1
make: *** [Makefile:1512: _module_/var/lib/dkms/it87/105.4d29188/build] Fehler 2
make: Verzeichnis „/usr/lib/modules/4.13.3-1-ARCH/build“ wird verlassen

groeck, 2017-09-28:
Did you use the provided Makefile or your own ? version.h is created dynamically from Makefile.

testsieger73, 2017-09-29:
I used yaourt since it worked perfectly until now. Manual downloading and compiling comes with the following error:

[xyz@workstation it87-master]$ make clean
rm -f version.h
[xyz@workstation it87-master]$ make
fatal: Kein Git-Repository (oder irgendein Elternverzeichnis): .git
  CC [M]  /home/xyz/Downloads/it87-master/it87.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/xyz/Downloads/it87-master/it87.mod.o
  LD [M]  /home/xyz/Downloads/it87-master/it87.ko
[xyz@workstation it87-master]$ sudo make install 
[sudo] Passwort für xyz: 
mkdir -p /lib/modules/4.13.3-1-ARCH/kernel/drivers/hwmon
cp it87.ko /lib/modules/4.13.3-1-ARCH/kernel/drivers/hwmon/
depmod -a -F /boot/System.map-4.13.3-1-ARCH 4.13.3-1-ARCH
depmod: FATAL: could not load /boot/System.map-4.13.3-1-ARCH: No such file or directory
make: *** [Makefile:65: modules_install] Fehler 1

groeck, 2017-09-29:
Info on the web suggests that arch linux doesn't use the System.map file, so maybe you can just remove the depmod lines from Makefile.

testsieger73, 2017-09-29:
I just made a fresh installation of Manjaro and sensors-detect now detects "k10temp" out of the box, so I can at least monitor the calculated CPU temps without the need to install additional drivers. I just wanted to let you know that the installation of it87-dkms-git via yaourt used to be flawless which is not the case anymore.

closed 2017-09-29 by testsieger73

groeck, 2017-09-29:
On Fri, Sep 29, 2017 at 09:06:05AM +0000, testsieger73 wrote:
> I just made a fresh installation of Manjaro and sensors-detect now detects "k10temp" out of the box, so I can at least monitor the calculated CPU temps without the need to install additional [...]
What does sensors-detect have to do with the ability to monitor the CPU
temperature ?

Guenter