groeck/it87 #54: The good and the bad, Kernel 4.15

Extarys, 2017-09-28:
So I installed 4.14.3 stable, it87 doesn't work.
The good news is that we are supposed to get temp under 4.15 as per [this article](https://www.phoronix.com/scan.php?page=news_item&px=AMD-Zen-Temps-Hwmon-Next) yay.

Just to let you know.

groeck, 2017-09-29:
The driver here is more advanced than the upstream driver.

closed 2017-09-29 by groeck

Extarys, 2017-09-29:
Well not that advanced since it only work on 4.11 and before for me ;)

groeck, 2017-09-29:
Sorry, you lost me pretty much completely. Looking through other cases, you also opened #51, and my last comment there tries to explain what is going on. So what is the point of #54 ? Can you elaborate ? What does all this have to do with the kernel version, and what do you mean with "does not work" besides the problems discussed in #51 ?

Extarys, 2017-09-29:
I'm just saying that it worked after bios upgrade in 4.11 but not in 4.13.

**Does not work**:
```
$ sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```
This one is supposed to be good news as we wont have to open issue here and debate whether or not there's a point. It will just work.

If you check issue 51, it was about kernel 4.11. This is about 4.13. Not the same thing, since now it's not working, obviously. I'm not the pro here I don't know, there is a tab called Issues. I have an issue with the it87 so I open an issue. If it's not about the kernel then change the title of the issue #51 and that's that.

I just wanted to share the news because before finding it87 I spent hours trying to find a solution and now there will be 2 solutions - if thats not great news I dont know what it is. Also because it's now not working again and there is no other bios update :P (Won't make the same mistake twice)

Just delete this if this is a problem.

groeck, 2017-09-29:
On Fri, Sep 29, 2017 at 06:23:42AM +0000, Jeremy wrote:
> I'm just saying that it worked after bios upgrade in 4.11 but not in 4.13.
>
> **Does not work**:
> ```
> $ sensors
> asus-isa-0000
> Adapter: ISA adapter
> cpu_fan:        0 RPM
> ```
> This one is supposed to be [...]

The driver and kernel versions are (or should be) independent of each other. #51
is not only for kernel version 4.11, it is for the driver runnign on any kernel
version.

Unfortunately you still don't say what the actual problem is. Output from
dmesg, for example, might help a lot. But all you say is "Does not work".
That is not at all helpful. Again, how do you expect _anyone_ to help you
with such vague information ?

Guenter