groeck/it87 #55: Incorrect naming

adityaxavier, 2017-10-07:
Hi,

It seems the newer patches have changed the nomenclature..
Previously, the sensors output used be like this -
`it8628-isa-0a40
Adapter: ISA adapter
in0:          +0.02 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.05 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.01 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.07 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.36 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.17 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:        1917 RPM  (min =    0 RPM)
fan3:        1467 RPM  (min =    0 RPM)
fan4:        1591 RPM  (min =    0 RPM)
fan5:        1503 RPM  (min =    0 RPM)
temp1:        +41.0 C  (low  = +127.0 C, high = +127.0 C)  sensor = thermistor
temp2:        -55.0 C  (low  = +127.0 C, high = +127.0 C)  sensor = thermistor
temp3:        +37.0 C  (low  = +127.0 C, high = +127.0 C)  sensor = Intel PECI
temp4:        +41.0 C  
temp5:        +37.0 C  
temp6:        +36.0 C  
intrusion0:  OK`

and now its this -
`it8628-isa-0a40
Adapter: ISA adapter
in0:          +0.32 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.05 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.01 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.07 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.36 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.17 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:        1912 RPM  (min =    0 RPM)
fan3:        1477 RPM  (min =    0 RPM)
fan4:        1588 RPM  (min =    0 RPM)
fan5:        1486 RPM  (min =    0 RPM)
temp1:        +41.0 C  (low  = +127.0 C, high = +127.0 C)  sensor = thermistor
temp2:        -55.0 C  (low  = +127.0 C, high = +127.0 C)  sensor = thermistor
temp3:        +36.0 C  (low  = +127.0 C, high = +127.0 C)  sensor = AMD AMDSI
temp4:        +41.0 C  (low  =  +0.0 C, high = +127.0 C)
temp5:        +37.0 C  (low  =  +0.0 C, high = +127.0 C)
temp6:        +36.0 C  (low  =  +0.0 C, high = +127.0 C)
intrusion0:  OK
`

It is an Intel Machine..

groeck, 2017-10-07:
Please try again.

adityaxavier, 2017-10-10:
Yes its fixed. Thanks !

closed 2017-10-10 by adityaxavier