groeck/it87 #56: Fan issues after suspend on Gigabyte GA-AX370 Gaming 5 + Ubuntu

jjaksic, 2017-10-15:
I'm having issues with fans that I've been tying to fix for months, with no luck.

After a fresh boot, everything works fine. Fan control by BIOS works fine and so do hwmon/pwn, pwmconfig, fancontrol, sensors etc.

But after resuming from suspend to memory, fans start spinning very fast, regardless of whether IT87 is loaded or not. And if I try to control them through IT87 (either directly or indirectly), I notice that PWMs have become inverted, i.e. low values make a fan spin fast, high values make it spin slow.

Here are some additional observations:
* I thought that fix_pwm_polarity would help with inverted PWM, but it seems to make no difference.
* When this happens (every time after resume), some other IT87-related issues also pop up: temp4 shows -55C, whereas after fresh boot it works fine. And IT8686 gets detected as only having 3 fans instead of the normal 6 (I was able to patch it87 to fix this particular issue, but not anything else).
* Using different kernels (even mainline ones) in the same Ubuntu setup seems to make no difference.
* This doesn't seem to happen in Suse (one distro that I've tried where I can STR out of the box); I was able to STR multiple times without the fans spinning up, though I didn't try loading it87.
* BIOS version makes no difference.
* If I remember correctly, there have been times when this issue didn't occur after a resume, which would suggest that the issue is caused by some non-deterministic factor like timing or concurrency. However, these cases are so rare that I'm not 100% sure if they really happened, much less can I reproduce them.

I'm not sure if this is a bug in IT87. I'm guessing the issue originates somewhere else, but I'm hoping that at least IT87 might be able to handle or fix it.

Any help or advice will be greatly appreciated. Let me know if I should post some debug info.

groeck, 2017-10-15:
You suspect that this may be a bug in the driver, but at the same time you mention that the problem happens if the driver is loaded or not. If so, can you explain how you think the driver can be responsible for the problem ? I don't claim that the driver is innocent. I am trying to understand your problem, and it is difficult for me to blame a driver which isn't loaded for a problem.
What driver are you running, the driver from here or the upstream driver ?
What is STR ?
How did you patch the driver to fix the fan count problem ?
Do you have it87_wdt loaded ?

groeck, 2017-10-15:
Re the fan polarity problem: That is due to bit 6 of registers 0x15, 0x16, 0x17, 0x7f, 0xa7, 0xaf having the wrong value. Question is how the wrong value gets there.

jjaksic, 2017-10-17:
I believe the problem starts out elsewhere, not in the driver, because it happens without the driver even being installed. I'm suspecting that it has something to do with Ubuntu, the motherboard, some other driver... I don't know if this issue is an anomaly in my case or it happens everywhere with similar configurations.

IT87 starts misbehaving only after the issue already happens, so I don't know if this is something that should be fixed in IT87.

I'm using the driver from here.

STR is Suspend-to-RAM, a.k.a. S3 (https://www.kernel.org/doc/Documentation/power/states.txt). This is what causes the issue, and the quickest way to trigger it is by running ```sudo rtcwake -m mem -s 10 -v``` (this causes it to suspend and resume after 10 seconds).

Here's my workaround patch: https://github.com/jjaksic/it87/commit/6e70867f7fef4d8517430ef521cf68efdee13fcc
Please be aware that I have no idea what it really does. I just printed and compared various values before and after the issue, and then changed the driver code so that it detects the issue and sets the broken bits back to their "good" values. It works ok for me, but would possibly have bad side-effects if used elsewhere.

I didn't know about it87_wdt until now. I do have it and I can load it, but I don't know what it does or what to do with it.

Yes, I think that some registers get overwritten by some bogus values, and I have no idea how that happens. Is there a way to track these changes?

groeck, 2017-10-17:
Your workaround patch enables fan tachometers 1..5 (but not 6). Which makes me wonder what you mean with "detected". Do you only see three fans, or do you see five (or six) fans, and the speed is displayed as 0 ?
Any chance you can post good and bad "sensors" output as well as good and bad register dumps for both SIO and envmon registers ? If only registers 0x0c and 0x13 are affected it might be possible to add suspend/resume support and handle the problem in the resume code. Ultimately you might have a BIOS problem, though.

groeck, 2017-12-03:
Any update ? Did you check if there is a more recent BIOS available which possibly fixes the problem ?

groeck, 2018-01-14:
No further feedback. Closing.

closed 2018-01-14 by groeck

niceiceman, 2018-03-13:
it seems this problem spread across other gigabyte motherboard

the same issues found in Z370 HD3 / hd3P as well