groeck/it87 #58: Loading it87 sets motherboard display to 00

oyvinds, 2017-10-27:
Gigabyte AX370 Gaming 5 has a two digit 80s alarmblock display on the motherboard. This display will show some kind of temperature number such as "24". 

modprobe it87 sets this display to 00. It also lets me get sensors to output some random numbers under the title "it8792-isa-0a60" and some useful information which I suspect is accurate under the title under "it8686-isa-0a40".

While it is great that it87 now makes it possible to get temperature information on the screen it is unfortunate that it removes the ability see it on the motherboard display. This is specially a luxury problem for me since I have a windowed computer case with RGB lights like a 12 year old.

Please let me know if you need me to run some kind of magic commands to get useful information from this motherboard that could aid in fixing this problem.

groeck, 2017-10-29:
Disconnected sensors will show "random" or nonsense data. Disable them in sensors3.conf.
Unfortunately, I don't know what may cause the board temperature display to display zero. At this time the only remedy I can recommend, until we get further information, is to not load the it87 driver.

groeck, 2018-07-09:
Won't fix. Please also see driver removal notice in README file.

closed 2018-07-09 by groeck