groeck/it87 #59: Support for ITE IT8768 3port Serial

masosi, 2017-11-03:
I'm not sure if this is the right place, but there doesn't seem to be a driver for the ITE IT8768. I have this running 3 RS232 serial ports and cannot communicate with any of them. I have confirmed these serial ports do work on Windows 10 though so it doesn't seem to be a hardware issue.

Ubuntu reports the following:
`hac@hac:~$ sudo setserial -g /dev/ttyS[0123]
/dev/ttyS0, UART: 16550A, Port: 0x03f8, IRQ: 4
/dev/ttyS1, UART: 16550A, Port: 0x02f8, IRQ: 3
/dev/ttyS2, UART: 16550A, Port: 0x03e8, IRQ: 10
/dev/ttyS3, UART: 16550A, Port: 0x02e8, IRQ: 11`

I can use ttyS0 because it is on the IT8758, but the other 3 serial ports are on the IT8768 and I get nothing when I use a loopback.

Please see the following thread for more info:
https://www.linuxquestions.org/questions/showthread.php?p=5776159#post5776159

I think it is just a driver issue, but am not sure what else to try. Could someone point me in the right direction please?

Thank you,
masosi.

groeck, 2017-11-03:
On Fri, Nov 03, 2017 at 07:11:30PM +0000, masosi wrote:
> I'm not sure if this is the right place, but there doesn't seem to be a driver for the ITE IT8768. I have this running 3 RS232 serial ports and cannot communicate with any of them. I have confirmed [...]
Can't help you there, sorry.

FWIW, PC serial ports do not need any special Linux support; they are
PC standard. In this context it is misleading to claim that "Linux does
not support IT8768", especially without knowing what else if anything
the chip might support. It does support its serial ports.

There is little information available about this chip. Maybe it is
misconfigured in the BIOS. Of course, it _might_ be that the serial
ports are really non-standard, and that Windows knows about it and
works around it. But that is impossible to know without support from
the board vendor.

Guenter

masosi, 2017-11-04:
Guenter,

Thank you for the explanation.
I understand what you are saying and appreciate your time on this.

I will continue talking with the vendor to find a solution.

Thank you again,
masosi.

closed 2017-11-04 by masosi