groeck/it87 #60: Report for Asus B350M-A

3esmit, 2017-11-12:
```console
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8655
    (logical device 4 has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

root@ricardo-server:~/it87# sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 55 06 44 2d 00 00 c0 00 10 00 69 05 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 40 00 10 00 00 00 00 00 00 00 10 00 00 00 
d0: ff 00 00 0f 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c 
f0: 00 00 00 00 00 5f 0e 00 00 c0 00 00 00 00 00 00 

root@ricardo-server:~/it87# sudo isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 01 ef b1 00 00 40 00 00 c0 44 00 00 76 69 00 
10: 41 e5 00 00 7f 00 00 00 7f 00 00 00 7f 1a 00 00 
20: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f cc 3c 97 
30: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 01 01 01 
40: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 00 00 00 
50: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c 
60: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c 
70: 08 0a 00 4c 4b 00 00 00 00 00 7f 99 00 00 0f 99 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 c2 00 79 
90: 00 00 00 00 00 00 00 ff 40 00 00 00 00 d8 00 1e 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80 
b0: 2d b0 00 00 89 9f a1 49 ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 

root@ricardo-server:~/it87# sudo modprobe it87 force_id=0x8622
root@ricardo-server:~/it87# sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

root@ricardo-server:~/it87# sudo modprobe -r it87
root@ricardo-server:~/it87# sudo modprobe it87 force_id=0x8628
root@ricardo-server:~/it87# sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

root@ricardo-server:~/it87# sudo modprobe -r it87
root@ricardo-server:~/it87# sudo modprobe it87 force_id=0x8728
root@ricardo-server:~/it87# sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

root@ricardo-server:~/it87# sudo modprobe -r it87
root@ricardo-server:~/it87# sudo modprobe it87 force_id=0x8732
root@ricardo-server:~/it87# sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

root@ricardo-server:~/it87# sudo modprobe -r it87
root@ricardo-server:~/it87# 
```

3esmit, 2017-11-12:
related to #11 <- working after reading this.

```
root@ricardo-server:~/it87# sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

it8655-isa-0290
Adapter: ISA adapter
in0:          +1.00 V  (min =  +1.24 V, max =  +1.43 V)  ALARM
in1:          +2.47 V  (min =  +1.33 V, max =  +0.01 V)  ALARM
in2:          +1.97 V  (min =  +2.76 V, max =  +0.85 V)  ALARM
in3:          +2.00 V  (min =  +0.84 V, max =  +1.08 V)  ALARM
in4:          +1.98 V  (min =  +0.87 V, max =  +2.52 V)
in5:          +1.91 V  (min =  +0.44 V, max =  +1.33 V)  ALARM
in6:          +1.98 V  (min =  +0.32 V, max =  +1.11 V)  ALARM
3VSB:         +3.31 V  (min =  +1.00 V, max =  +3.16 V)  ALARM
Vbat:         +3.16 V  
+3.3V:        +3.29 V  
fan1:        1066 RPM  (min =   13 RPM)
fan2:        1326 RPM  (min =   12 RPM)
temp1:        +40.0°C  (low  =  +5.0°C, high = +12.0°C)  ALARM
temp2:        +33.0°C  (low  = +61.0°C, high = +79.0°C)  sensor = thermistor
temp3:        +30.0°C  (low  = +67.0°C, high = +63.0°C)  sensor = thermistor
intrusion0:  ALARM
```

closed 2017-12-03 by groeck

cab404, 2018-01-25:
well, sensors log above doesn't really looks working to me - voltages are all over the place, and I bet temp readings are jumping like crazy

here's my dump for B350M-A
```
$ sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 55 06 44 2d 00 00 c0 00 10 00 69 07 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 40 00 10 00 00 00 00 00 00 00 10 00 00 00 
d0: ff 00 00 0f 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c 
f0: 00 00 00 00 00 5f 0e 00 00 c0 00 00 00 00 00 00
$ sudo isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 90 39 bf 00 00 00 00 00 c0 44 00 04 92 fb 00 
10: cb bf ad 30 c0 80 80 00 03 03 00 b7 63 ed 00 00 
20: 2c e6 ba bc bd b4 bd 99 94 22 25 24 24 24 24 99 
30: f4 2b f7 39 eb 36 92 9c 7f 23 20 bd ff 65 ca 7c 
40: 2b ac f1 cb fc 6d ba 8f ad 6a d4 00 5f 40 2f f6 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 00 4b 7f 00 00 0f 7f 80 00 4b bc 00 00 0f bc 
70: 7f 7f 7f 99 00 00 0f 99 7f 7f 7f 99 00 00 0f 99 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 d6 
90: 00 00 00 00 00 00 00 ff 42 99 00 00 01 f3 e0 22 
a0: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 80 00 00 0f 80 
b0: fa 63 00 00 e4 9f f6 df ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

sensors tells something like this:
```
it8655-isa-0290
Adapter: ISA adapter
in0:          +1.27 V  (min =  +0.47 V, max =  +2.66 V)
in1:          +2.51 V  (min =  +0.62 V, max =  +2.69 V)
in2:          +2.02 V  (min =  +0.59 V, max =  +2.56 V)
in3:          +2.04 V  (min =  +1.70 V, max =  +1.59 V)  ALARM
in4:          +2.04 V  (min =  +0.38 V, max =  +1.38 V)  ALARM
in5:          +1.96 V  (min =  +2.06 V, max =  +0.35 V)  ALARM
in6:          +2.05 V  (min =  +1.10 V, max =  +2.78 V)
3VSB:         +3.33 V  (min =  +2.70 V, max =  +4.40 V)
Vbat:         +3.21 V  
+3.3V:        +3.33 V  
fan1:        1173 RPM  (min =   14 RPM)
fan2:         664 RPM  (min =   26 RPM)
temp1:        +33.0°C  (low  = -84.0°C, high = +43.0°C)
temp2:        +39.0°C  (low  = -53.0°C, high = -15.0°C)  ALARM  sensor = thermistor
temp3:        +36.0°C  (low  = +109.0°C, high =  -4.0°C)  ALARM  sensor = thermistor
intrusion0:  ALARM
```

I forced RPMs on fans in BIOS to be roughly static to check if they're right - they are
3.3V is good too
there's nothing that looks like cpu voltage (~1.308V) or memory voltage (~1.225V), or 5/12V, but IDK if those should be handled here - at least I've seen those at bios .-.
there's 3 temp readings, although bios shows only two, but i guess bios can hide some thermocouples from gui - anyways, temp ranges here are really strange .-.
temp 2 is very close to actual mobo temp, so I guess it is mobo temp 
temp 1 jumps all over the place, suddenly changing by ±2-5 degrees, but roughly ([very, VERY roughly](https://user-images.githubusercontent.com/6453661/35370494-34cd017c-019f-11e8-967f-5aab74bdda36.png)) follows CPU temp

do I understand correctly, that all is needed is to reverse engineer output from `sudo isadump 0x295 0x296`? I could've tried to help .-.