groeck/it87 #61: Support for ASUS X370-PRO with Ryzen 1700X

jmnel, 2017-11-20:
What is the current status of support for the ITE 8665 chip? I have been unsuccessful in getting sensor readouts on a new Ryzen build.

Output from `sensors-detect` gives the following relevent result:

```
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                                  No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                                       Yes
Found unknown chip with ID 0x8665
    (logical device 4 has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                                  No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                                        No
```

I have force-installed drivers all ids including 0x8622, 0x8628, 0x8728, and 0x8732.

Following the README, here are some dumps.

`isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7`

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 86 65 06 40 3d 90 ff 01 00 95 00 49 85 08 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60: 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
c0: 90 ef 01 00 d5 00 00 00 00 00 00 00 10 00 00 00
d0: ff 00 00 0c 00 00 00 00 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c
f0: 00 00 00 00 00 00 0e 00 00 00 00 00 00 00 00 00
```

`isadump -k 0x295 0x296`

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 f0 f2 9f 00 00 00 00 00 c0 44 08 20 76 fb 09
10: 76 27 fe 70 c0 00 00 00 01 01 02 89 9b bb 00 00
20: 7b e4 b7 b8 04 04 04 98 92 27 1f 1d 1d 1d 1d 98
30: e9 59 6e 6d cd 2f ee 7e 4c eb 7f fb 7e d7 5c 10
40: 3f e7 f3 0a b2 ef ab ec ad 6a d4 00 5f 40 6f fe
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00
60: 80 7f 7f ff 00 00 0f ff 80 7f 7f ff 00 00 0f ff
70: 80 7f 7f ff 00 00 0f ff 7f 7f 7f 99 00 00 0f 99
80: 00 00 26 01 00 00 00 00 00 00 00 00 00 bc 00 71
90: 00 00 00 ff ff 00 00 ff 42 99 00 00 01 ce e0 30
a0: 7f 7f 7f 80 00 00 0f 80 80 7f 7f ff 00 00 0f ff
b0: 91 fa 00 00 d7 9b 7a 7a ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

Please let me know if I can help further with the driver. Keep up the good work. Thanks.

groeck, 2017-11-20:
On Mon, Nov 20, 2017 at 07:59:45PM +0000, Jacques Nel wrote:
> What is the current status of support for the ITE 8665 chip? I have been unsuccessful in getting sensor readouts on a new Ryzen build.
> 
Did you try the driver from here ? What do you see in dmesg ?

> Output from `sensors-detect` gives the following relevent result:
>
> ```
> Trying family `National Semiconductor/ITE'...               No
> Trying family [...]
See comment at end of https://github.com/groeck/it87/blob/master/README.

jmnel, 2017-11-22:
After `sudo modprobe it87` `dmesg' gives

```
[  304.313000] it87: it87 driver version v1.0-25-g6582b47
[  304.313063] it87: Found IT8665E chip at 0x290, revision 6
[  304.313097] it87: Beeping is supported
```

groeck, 2017-11-22:
and there is no other (error) message, yet there is no output from the "sensors" command ?

jmnel, 2017-11-22:
Yes exactly, and when I manually set device ID, Sensors output is blank. I tried every possible ID from the readme. It seems like the driver is loading correctly, according to dmesg. Does anyone have any suggestion to what else I could try?

jmnel, 2017-11-22:
Did you ever manage to find a data sheet for the IT8665E?

groeck, 2017-11-22:
Please provide complete output of dmesg. Yes, I do have a datasheet, but that is besides the point here. The most likely culprit would be an ACPI device conflict, but since you say there is none this is a bit puzzling. We may have to add some debug messages to the driver to find out why it is not instantiating.

groeck, 2017-12-03:
Driver analysis suggests that either modprobe should fail to load the driver, or there should be output from the 'sensors' command. Without additional information, it is impossible to determine what is wrong. Closing issue. Please reopen if/when additional information is available.

closed 2017-12-03 by groeck