groeck/it87 #64: Trying to insert module gives "No such device"

ashkitten, 2017-12-08:
This is on an ASUS PRIME B350-PLUS motherboard
`sensors-detect` gives the ID as 0x655
The package is built from `it87-dkms-git` in the AUR
Last working version is unknown, but building the `v1.0` tag works

groeck, 2017-12-08:
Information provided is not actionable. I had no influence in or knowledge about dkms. I don't even know what AUR stands for. I have no idea what the "v1.0" tag references. Please submit a problem report with dkms maintainers.

closed 2017-12-08 by groeck

bsilvereagle, 2017-12-09:
@groeck Hopefully this is more useful. If `force_id` is used as a parameter the module fails to load. It is not apparent to me if this is an it87 specific problem or something else. 

Kernel 4.15.0-rc2 on Debian sid

```
# modinfo it87
filename:       /lib/modules/4.15.0-rc2custom/kernel/drivers/hwmon/it87.ko
license:        GPL
description:    IT8705F/IT871xF/IT872xF hardware monitoring driver
author:         Chris Gauthron, Jean Delvare <jdelvare@suse.de>
depends:        hwmon-vid
intree:         Y
name:           it87
vermagic:       4.15.0-rc2custom SMP mod_unload modversions 
parm:           force_id:Override the detected device ID (ushort)
parm:           update_vbat:Update vbat if set else return powerup value (bool)
parm:           fix_pwm_polarity:Force PWM polarity to active high (DANGEROUS) (bool)
```

This has no issues:

```
# modprobe it87
```

This fails:

```
# modprobe -v it87 force_id=0x8686
insmod /lib/modules/4.15.0-rc2custom/kernel/drivers/hwmon/it87.ko force_id=0x8686
modprobe: ERROR: could not insert 'it87': No such device
```
Which doesn't make any sense:

```
$ ls /lib/modules/4.15.0-rc2custom/kernel/drivers/hwmon/ | grep it87
it87.ko
```

groeck, 2017-12-10:
It does make sense. You are not using the driver from github but the upstream driver, which is not up to date and doesn't support the latest chips yet.

pterodragon, 2017-12-29:
For future reference, I use Arch Linux as well.

mati865, 2018-01-18:
Arch package have it's own `dkms.conf` but it was overriden by `dkms.conf` merged to this repo which doesn't work for Arch.
You can apply [this patch](https://hastebin.com/zoyasunuli.pl) to the PKGBUILD until maintainer pushes fix.