groeck/it87 #65: Information Report: Gigabyte X399 Designare EX

roemer2201, 2018-01-03:
Hello,

following your hints about "reporting information for unsupported chips" I like to post it here:

```
sudo sensors-detect 
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)
# System: Gigabyte Technology Co., Ltd. X399 DESIGNARE EX [Default string]
# Board: Gigabyte Technology Co., Ltd. X399 DESIGNARE EX-CF
# Kernel: 4.13.0-21-generic x86_64
# Processor: AMD Ryzen Threadripper 1920X 12-Core Processor (23/1/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
**Found unknown chip with ID 0x8686**
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
**Found unknown chip with ID 0x8733**

......
```

```
sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 86 02 00 00 20 33 00 00 00 80 40 c8 18 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 ff 00 00 00 00 00 
60: 0a 10 0a 00 0a 20 00 00 00 00 00 00 ff 00 00 00 
70: 00 00 20 00 00 06 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 c0 00 00 00 00 
c0: 01 01 00 40 00 ff ff ff 01 00 00 00 00 18 00 00 
d0: b1 00 08 0a 00 fd ff ff 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 dc 00 1d 20 12 49 00 00 0a 1c 
f0: 00 00 00 00 00 00 0d 00 00 00 49 40 00 ff 00 0f 
```

I did not run the second line from your example, as my sensors-detect did not show a line like "logical device 4 has address 0x290, could be sensors".

```
sudo modprobe it87 force_id=0x8732
modprobe: ERROR: could not insert 'it87': Device or resource busy
```

I'm pretty sure a reboot will help to mount the module, I'll post updates ASAP. My first try was to load the module without a force_id (no sensors where detected), then I "rmmod"'ed it, but it seems not to be fully unloaded. Is there a better way to remove it from the kernel?

Regards

groeck, 2018-01-03:
"but it seems not to be fully unloaded" - this is not supported by the information you provided. If you have additional information suggesting that there is a module unload problem, please provide relevant logging information.

Note that chip id 0x8686 is supported by the driver, so that isn't the problem. Please check README for the ignore_resource_conflict module parameter, and provide relevant logging information from dmesg.

roemer2201, 2018-01-04:
Thank you for comming back to me!

"but it seems not to be fully unloaded" was indeed not the case, even after a reboot I could not load the module with "force_id=" (see my second command below). So here is what I've tried today:

```
sudo modprobe it87 force_id=0x8686
modprobe: ERROR: could not insert 'it87': No such device
```
dmesg does not report anything for this command. What am I doing wrong when you say 0x8686 is supported?

So I tried again the closest match for the found unknown chip 0x8733:
```
sudo modprobe it87 force_id=0x8732
modprobe: ERROR: could not insert 'it87': Device or resource busy
```
Dmesg reports the following for this command:
```
Jan  4 08:03:56 hostname kernel: [74141.594062] it87: Found IT8732F chip at 0xa40, revision 2
Jan  4 08:03:56 hostname kernel: [74141.594087] it87: Beeping is supported
Jan  4 08:03:56 hostname kernel: [74141.594097] ACPI Warning: SystemIO range 0x0000000000000A45-0x0000000000000A46 conflicts with OpRegion 0x0000000000000A45-0x0000000000000A46 (\GSA1.SIO1) (20170531/utaddress-247)
Jan  4 08:03:56 hostname kernel: [74141.594104] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
Jan  4 08:03:59 hostname gnome-shell[2176]: JS ERROR: Gio.IOErrorEnum: Fehler beim Lesen aus Datei: Vorgang wird nicht unterstützt#012Thermal<.refresh/<@/home/username/.local/share/gnome-shell/extensions/system-monitor@paradoxxx.zero.gmail.com/extension.js:1748:28
```

Third try was the suggested parameter "ignore_resource_conflict":
```
sudo modprobe it87 force_id=0x8732 ignore_resource_conflict
modprobe: ERROR: could not insert 'it87': Device or resource busy
```
Dmesg reports the following for this command:
```
Jan  4 08:04:15 hostname kernel: [74160.891620] it87: unknown parameter 'ignore_resource_conflict' ignored
Jan  4 08:04:15 hostname kernel: [74160.891754] it87: Found IT8732F chip at 0xa40, revision 2
Jan  4 08:04:15 hostname kernel: [74160.891777] it87: Beeping is supported
Jan  4 08:04:15 hostname kernel: [74160.891786] ACPI Warning: SystemIO range 0x0000000000000A45-0x0000000000000A46 conflicts with OpRegion 0x0000000000000A45-0x0000000000000A46 (\GSA1.SIO1) (20170531/utaddress-247)
Jan  4 08:04:15 hostname kernel: [74160.891794] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
```

Thank you for your help!

groeck, 2018-01-04:
You are running the upstream driver, not the driver frm here. Install the driver from here, then try again.

ThyMYthOS, 2018-01-08:
I can confirm that this driver works on an Asus X399-A out of the box. Reports 7 unnamed voltages (in0: 0.44V, in1: 0.60V, in2: 2.0V, in3: 2.0V, in4: 0.55V, in5: 0.60V in6: 0.91V), 3VSB, Vbat, +3.3V, 5 fans and 3 temperatures.

roemer2201, 2018-01-09:
Thank you for all the help. I finally build the driver from this repository using dkms and after loading the following messages are printed by the kernel:

```
sudo modprobe it87 ignore_resource_conflict
```

```
modinfo it87
filename:       /lib/modules/4.13.0-21-generic/updates/dkms/it87.ko
version:        v1.0-32-g5edd7bb
license:        GPL
description:    IT8705F/IT871xF/IT872xF hardware monitoring driver
author:         Chris Gauthron, Jean Delvare <jdelvare@suse.de>
srcversion:     176C4FDC44E7EB17BFB477D
depends:        hwmon-vid
name:           it87
vermagic:       4.13.0-21-generic SMP mod_unload 
parm:           force_id:Override the detected device ID (ushort)
parm:           ignore_resource_conflict:Ignore ACPI resource conflict (bool)
parm:           mmio:Use MMIO if available (bool)
parm:           update_vbat:Update vbat if set else return powerup value (bool)
parm:           fix_pwm_polarity:Force PWM polarity to active high (DANGEROUS) (bool)
```

```
[ 9328.613722] it87: it87 driver version v1.0-32-g5edd7bb
[ 9328.613795] it87: Found IT8686E chip at 0xa40, revision 2
[ 9328.613833] it87: Beeping is supported
[ 9328.613851] ACPI Warning: SystemIO range 0x0000000000000A45-0x0000000000000A46 conflicts with OpRegion 0x0000000000000A45-0x0000000000000A46 (\GSA1.SIO1) (20170531/utaddress-247)
[ 9328.613859] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[ 9328.615135] it87: Found IT8792E/IT8795E chip at 0xa60, revision 3
[ 9328.615160] it87: Beeping is supported

```

After running "yes yes | sensors-detect" I see the following:
```
it8792-isa-0a60
Adapter: ISA adapter
in0:          +1.29 V  (min =  +0.00 V, max =  +2.78 V)
in1:          +1.50 V  (min =  +0.00 V, max =  +2.78 V)
in2:          +1.04 V  (min =  +0.00 V, max =  +2.78 V)
in3:          +2.03 V  (min =  +0.00 V, max =  +2.78 V)
in4:          +1.79 V  (min =  +0.00 V, max =  +2.78 V)
in5:          +1.51 V  (min =  +0.00 V, max =  +2.78 V)
in6:          +2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:         +3.36 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:         +3.21 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +41.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

it8686-isa-0a40
Adapter: ISA adapter
in0:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.02 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +0.86 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.21 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.17 V  
fan1:           0 RPM  (min =   10 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:         783 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +36.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +66.0°C  (low  =  +0.0°C, high = +70.0°C)  sensor = AMD AMDSI
temp4:        +46.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +49.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

Is there a way to find out what temperature reading represents the CPU, chipset, ...?

groeck, 2018-01-09:
For the most part it is guesswork, though "AMD ANDSI" is known to be the CPU. -55 means nothing is connected (or the sensor is misprogrammed, which should not happen).
If you run the system under load, one of the sensors will show a really high temperature (close to or even exceeding 100 degrees C). That is going to be the CPU power converters, and the temperature to watch besides the CPU temperature.

roemer2201, 2018-01-11:
Today I digged a bit more into the sensor readings, please check the attached picture.

[image]

I found out, that "EC_TEMP1" is temp6 on the "it8686-isa-0a40" adapter. This is a wired sensor that was not connected, when I took this picture. This explains the -55°C. So I guess "EC_TEMP2" equals temp2 on "it8792-isa-0a60". So I'll try to map them in /etc/sensors.conf.

Thank you for your help. If there is some information I can provide to help improve the kernel module, let me know.

closed 2018-01-14 by groeck