groeck/it87 #66: Dell is cockblocking my IT8718F

weedy, 2018-01-04:
```
kernel: [55485.921918] it87: unknown parameter 'ignore_resource_conflict' ignored 
kernel: [55485.922154] it87: Found IT8718F chip at 0x290, revision 3 
kernel: [55485.922184] ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000295-0x0000000000000296 (\IP) (20170728/utaddress-247) 
kernel: [55485.922198] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver 
```

What should I do or what do you need so I can try and get some stats?

groeck, 2018-01-04:
You are not running the driver from here.

weedy, 2018-01-09:
Sorry, didn't realize intree and this repo were so different.
```
linux # modprobe it87
modprobe: ERROR: could not insert 'it87': Device or resource busy
 kernel: [332676.299215] it87: Found IT8718F chip at 0x290, revision 3 
 kernel: [332676.299250] ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000295-0x0000000000000296 (\IP) (20170728/utaddress-247) 
 kernel: [332676.299264] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver 

it87 # make; make install
  CC [M]  /usr/src/linux-4.14.11-gentoo/it87/it87.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /usr/src/linux-4.14.11-gentoo/it87/it87.mod.o
  LD [M]  /usr/src/linux-4.14.11-gentoo/it87/it87.ko
mkdir -p /lib/modules/4.14.11-gentoo/kernel/drivers/hwmon
cp it87.ko /lib/modules/4.14.11-gentoo/kernel/drivers/hwmon/
depmod -a -F /boot/System.map-4.14.11-gentoo 4.14.11-gentoo
it87 # modprobe it87 ignore_resource_conflict=1
 kernel: [332745.309217] it87: loading out-of-tree module taints kernel. 
 kernel: [332745.311759] it87: it87 driver version v1.0-32-g5edd7bb 
 kernel: [332745.311866] it87: Found IT8718F chip at 0x290, revision 3 
 kernel: [332745.311895] ACPI Warning: SystemIO range 0x0000000000000295-0x0000000000000296 conflicts with OpRegion 0x0000000000000295-0x0000000000000296 (\IP) (20170728/utaddress-247) 
 kernel: [332745.311905] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver 

it87 # sensors
coretemp-isa-0000
Adapter: ISA adapter
Core 0:       +46.0°C  (high = +86.0°C, crit = +100.0°C)
Core 1:       +46.0°C  (high = +86.0°C, crit = +100.0°C)

it8718-isa-0290
Adapter: ISA adapter
in0:          +1.15 V  (min =  +0.00 V, max =  +4.08 V)
in1:          +3.02 V  (min =  +0.00 V, max =  +4.08 V)
in2:          +3.31 V  (min =  +0.00 V, max =  +4.08 V)
+5V:          +2.99 V  (min =  +0.00 V, max =  +4.08 V)
in4:          +3.01 V  (min =  +0.00 V, max =  +4.08 V)
in5:          +0.05 V  (min =  +0.00 V, max =  +4.08 V)
in6:          +0.10 V  (min =  +0.00 V, max =  +4.08 V)
5VSB:         +2.96 V  (min =  +0.00 V, max =  +4.08 V)
Vbat:         +1.42 V  
fan1:        1713 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +33.0°C  (low  =  -1.0°C, high = +127.0°C)  sensor = thermal diode
temp2:        +34.0°C  (low  =  -1.0°C, high = +127.0°C)  sensor = thermistor
temp3:        -10.0°C  (low  =  -1.0°C, high = +127.0°C)  sensor = thermistor
cpu0_vid:    +1.219 V
intrusion0:  OK
```
Am I now stuck needing the out of tree version forever?

groeck, 2018-01-09:
Please feel free to submit patches to bring the upstream driver up-to-date. Remember, one logical change per patch, and please follow kernel coding style as well as Documentation/hwmon/submitting-patches. If you do that, I'll be happy to apply your patches to the upstream kernel.

closed 2018-01-09 by groeck