groeck/it87 #67: Information report: Asus ROG Strix X370-F Gaming, BIOS 3401 x64

skropf, 2018-01-05:
```
########## terminal ##########

[user@host]: ~/Downloads/it87$ make clean

[user@host]: ~/Downloads/it87$ make
  CC [M]  /home/user/Downloads/it87/it87.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/user/Downloads/it87/it87.mod.o
  LD [M]  /home/user/Downloads/it87/it87.ko

[user@host]: ~/Downloads/it87$ sudo make install
[sudo] password for user:
mkdir -p /lib/modules/4.14.11-1-ARCH/kernel/drivers/hwmon
cp it87.ko /lib/modules/4.14.11-1-ARCH/kernel/drivers/hwmon/
depmod -a -F /boot/System.map-4.14.11-1-ARCH 4.14.11-1-ARCH
depmod: FATAL: could not load /boot/System.map-4.14.11-1-ARCH: No such file or directory
make: *** [Makefile:65: modules_install] Error 1

[user@host]: ~/Downloads/it87$ sudo modprobe it87

[user@host]: ~$ sudo sensors-detect
[sudo] password for user:
# sensors-detect revision $Revision$
# Board: ASUSTeK COMPUTER INC. ROG STRIX X370-F GAMING
# Kernel: 4.14.11-1-ARCH x86_64
# Processor: AMD Ryzen 7 1700X Eight-Core Processor (23/1/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no):
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no):
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8665
    (logical device 4 has address 0x290, could be sensors)
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
[... 316 lines cut ...]
########## bios ##########

- CPU
Temp: 40 °C
Voltage: 1.405 V
Fan1: 720 RPM
Fan2: 730 RPM

- MB
Temp: 23 °C

- PCH
Temp: 43 °C

- Chassis
Fan1: 976 RPM
Fan2: 690 RPM

- Voltages:
3.3 V: 3.291 V
5 V: 4.986 V
12 V: 12.099 V
```

All the infos from my system. Hope this helps somewhat.

skropf, 2018-01-05:
Ok, sorry didn't look properly through closed issues, just found https://github.com/groeck/it87/issues/28 -> Closing.

closed 2018-01-05 by skropf

groeck, 2018-01-05:
FWIW, "make install" failed, meaning you are _not_ using the driver from here. I don't know what arch linux expects in respect to directory structure - obviously it does not consistently use 4.14.11-1-ARCH. I can't help you there, since I don't have arch linux installed and have no idea what it is doing.