groeck/it87 #71: IT8665E (ASUS Prime x370-Pro) commit 4d3dcea1 breaks fan5 if water pump is connected

NoXPhasma, 2018-01-19:
Since commit https://github.com/groeck/it87/commit/4d29188e4fc9066866304259f6584caa4d3dcea1 fan5 (AIO_PUMP) doesn't show up any more. I have my water pump of my AIO Water cooling connected.

Since https://github.com/groeck/it87/commit/4d29188e4fc9066866304259f6584caa4d3dcea1 :
```
$ sensors
it8665-isa-0290
Adapter: ISA adapter
in0:          +1.37 V  (min =  +1.36 V, max =  +2.05 V)
in1:          +2.51 V  (min =  +2.02 V, max =  +2.60 V)
in2:          +2.00 V  (min =  +2.77 V, max =  +1.36 V)  ALARM
in3:          +2.01 V  (min =  +2.76 V, max =  +2.65 V)  ALARM
in4:          +0.07 V  (min =  +0.04 V, max =  +0.09 V)
in5:          +0.07 V  (min =  +0.00 V, max =  +1.29 V)
in6:          +0.07 V  (min =  +0.04 V, max =  +0.09 V)
3VSB:         +3.36 V  (min =  +3.01 V, max =  +3.51 V)
Vbat:         +3.31 V  
+3.3V:        +3.36 V  
fan1:         770 RPM  (min =  399 RPM)
fan2:           0 RPM  (min =  399 RPM)  ALARM
fan3:           0 RPM  (min =   10 RPM)  ALARM
fan6:           0 RPM  (min =   -1 RPM)  ALARM
temp1:        +34.0°C  (low  = +10.0°C, high = +75.0°C)
temp2:        +28.0°C  (low  = +10.0°C, high = +40.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = +10.0°C, high = +75.0°C)  sensor = thermistor
intrusion0:  ALARM

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +34.0°C  (high = +70.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```

Before:
```
$ sensors
it8665-isa-0290
Adapter: ISA adapter
in0:          +1.38 V  (min =  +1.36 V, max =  +2.05 V)
in1:          +2.51 V  (min =  +2.02 V, max =  +2.60 V)
in2:          +2.01 V  (min =  +2.77 V, max =  +1.36 V)
in3:          +2.01 V  (min =  +2.76 V, max =  +2.65 V)
in4:          +0.07 V  (min =  +0.04 V, max =  +0.09 V)
in5:          +0.07 V  (min =  +0.00 V, max =  +1.29 V)
in6:          +0.07 V  (min =  +0.04 V, max =  +0.09 V)
3VSB:         +3.36 V  (min =  +3.01 V, max =  +3.51 V)
Vbat:         +3.31 V  
+3.3V:        +3.36 V  
fan1:         767 RPM  (min =  399 RPM)
fan2:           0 RPM  (min =  399 RPM)  ALARM
fan3:           0 RPM  (min =   10 RPM)  ALARM
fan5:        5037 RPM  (min = 1000 RPM)
fan6:           0 RPM  (min =   -1 RPM)  ALARM
temp1:        +35.0°C  (low  = +10.0°C, high = +75.0°C)
temp2:        +27.0°C  (low  = +10.0°C, high = +40.0°C)  sensor = thermistor
temp3:        +31.0°C  (low  = +10.0°C, high = +75.0°C)  sensor = thermistor
intrusion0:  ALARM

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +34.0°C  (high = +70.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM
```

Linux Mint 18.3 x64
Kernel 4.15-rc8

groeck, 2018-01-19:
Sorry for the trouble. Please try again. I hope I got it correct this time around.

NoXPhasma, 2018-01-19:
Yes, this works, thank you.

closed 2018-01-19 by NoXPhasma