groeck/it87 #72: Information report: ASUS ROG CROSSHAIR VI HERO

G-XVI, 2018-01-26:
Hello
Can you please support my motherboard?

```
OS: Arch Linux
Kernel: x86_64 Linux 4.14.15-1-ARCH
Board: ASUS ROG CROSSHAIR VI HERO AMD X370 ATX
CPU: AMD Ryzen 7 1800X Eight-Core @ 16x 4GHz
```
Reporting information for unsupported chips

```
sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7

I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 65 06 40 2d d0 db 81 04 d4 80 49 85 38 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 02 a0 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: d0 db 81 04 d4 00 00 00 40 00 00 00 10 00 00 10 
d0: ff 00 00 1a 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c 
f0: 00 00 00 00 00 00 0e 00 00 00 00 00 00 00 00 00

sudo isadump 0x295 0x296

I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 f8 ff b8 00 00 00 00 00 c0 44 08 30 df b2 c7 
10: f5 df bb 70 c0 80 80 80 06 02 03 5e 3d ff 80 00 
20: 41 3d b6 b7 30 29 56 98 8d 23 1c 21 21 21 21 98 
30: fe d3 f4 bf 8f e4 fe fb f6 f7 d3 bf ab bd d5 da 
40: 6e 7f 7b f3 3b d6 dd 8b ad 6a d4 00 5f 40 56 af 
50: ff 38 7f 7f 7f 40 00 00 90 00 00 12 60 00 00 00 
60: 80 14 46 33 16 00 0f 5b 80 1e 46 99 14 00 0f a5 
70: 80 1e 46 99 14 00 0f a5 80 1e 46 99 14 00 0f a5 
80: c3 03 22 01 00 00 00 00 00 00 00 00 00 26 00 20 
90: 00 00 00 ff ff 00 00 ff 42 99 00 00 01 5d e0 37 
a0: 7f 7f 7f 80 00 00 0f 80 80 7f 7f ff 00 00 0f ff 
b0: af f3 00 00 bf 7e ff fd ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

---------------------------------------------------------------

sudo modprobe it87 force_id=0x8622

it8622-isa-0290
Adapter: ISA adapter
in0:          +0.78 V  (min =  +2.53 V, max =  +3.05 V)
in1:          +0.74 V  (min =  +2.29 V, max =  +2.93 V)
in2:          +2.20 V  (min =  +2.74 V, max =  +1.72 V)
in3:          +2.20 V  (min =  +3.01 V, max =  +3.05 V)
in4:          +0.58 V  (min =  +2.96 V, max =  +2.95 V)
in5:          +0.49 V  (min =  +2.29 V, max =  +2.53 V)
in6:          +1.03 V  (min =  +2.27 V, max =  +2.05 V)
3VSB:         +3.65 V  (min =  +5.23 V, max =  +5.11 V)
Vbat:         +3.38 V  
+3.3V:        +3.67 V  
fan1:         383 RPM  (min =   27 RPM)
fan5:        2327 RPM  (min =   -1 RPM)  ALARM
temp1:        +35.0°C  (low  = +127.0°C, high = +110.0°C)  sensor = thermistor
temp2:        +28.0°C  (low  = -13.0°C, high = +123.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = -42.0°C, high = +59.0°C)  sensor = thermistor
temp4:        +33.0°C  
intrusion0:  ALARM

it8628-isa-0290
Adapter: ISA adapter
in0:          +0.78 V  (min =  +2.53 V, max =  +3.05 V)
in1:          +0.74 V  (min =  +2.29 V, max =  +2.93 V)
in2:          +2.20 V  (min =  +2.74 V, max =  +1.72 V)
+3.3V:        +4.39 V  (min =  +6.02 V, max =  +6.10 V)
in4:          +0.58 V  (min =  +2.96 V, max =  +2.95 V)
in5:          +0.49 V  (min =  +2.29 V, max =  +2.53 V)
in6:          +1.04 V  (min =  +2.27 V, max =  +2.05 V)
3VSB:         +3.65 V  (min =  +5.23 V, max =  +5.11 V)
Vbat:         +3.38 V  
+3.3V:        +3.65 V  
in10:         +0.40 V  
in11:         +0.40 V  
fan1:         379 RPM  (min =   27 RPM)
fan5:        2319 RPM  (min =   -1 RPM)  ALARM
temp1:        +35.0°C  (low  = +127.0°C, high = +110.0°C)  sensor = thermistor
temp2:        +28.0°C  (low  = -13.0°C, high = +123.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = -42.0°C, high = +59.0°C)  sensor = thermistor
temp6:        +33.0°C  (low  =  -1.0°C, high =  -1.0°C)
intrusion0:  ALARM

it8728-isa-0290
Adapter: ISA adapter
in0:          +0.78 V  (min =  +2.53 V, max =  +3.05 V)
in1:          +0.74 V  (min =  +2.29 V, max =  +2.93 V)
in2:          +2.20 V  (min =  +2.74 V, max =  +1.72 V)
+3.3V:        +4.39 V  (min =  +6.02 V, max =  +6.10 V)
in4:          +0.58 V  (min =  +2.96 V, max =  +2.95 V)
in5:          +0.49 V  (min =  +2.29 V, max =  +2.53 V)
in6:          +1.03 V  (min =  +2.27 V, max =  +2.05 V)
3VSB:         +3.65 V  (min =  +5.23 V, max =  +5.11 V)
Vbat:         +3.38 V  
fan1:         436 RPM  (min =   27 RPM)
fan5:        2319 RPM  (min =   -1 RPM)  ALARM
temp1:        +37.0°C  (low  = +127.0°C, high = +110.0°C)  sensor = thermistor
temp2:        +28.0°C  (low  = -13.0°C, high = +123.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = -42.0°C, high = +59.0°C)  sensor = thermistor
intrusion0:  ALARM

it8732-isa-0290
Adapter: ISA adapter
in0:          +0.71 V  (min =  +2.30 V, max =  +2.77 V)  ALARM
in1:          +0.68 V  (min =  +2.08 V, max =  +2.66 V)  ALARM
in2:          +1.98 V  (min =  +2.48 V, max =  +1.56 V)  ALARM
+3.3V:        +2.00 V  (min =  +2.74 V, max =  +2.77 V)  ALARM
in4:          +0.52 V  (min =  +2.69 V, max =  +2.68 V)  ALARM
in5:          +0.45 V  (min =  +2.08 V, max =  +2.30 V)  ALARM
in6:          +0.95 V  (min =  +2.06 V, max =  +1.86 V)  ALARM
3VSB:         +1.66 V  (min =  +2.38 V, max =  +2.32 V)  ALARM
Vbat:         +1.54 V  
fan1:         371 RPM  (min =   27 RPM)
fan4:         709 RPM  (min =   -1 RPM)
temp1:        +34.0°C  (low  = +127.0°C, high = +110.0°C)  sensor = thermistor
temp2:        +28.0°C  (low  = -13.0°C, high = +123.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = -42.0°C, high = +59.0°C)  sensor = thermistor
intrusion0:  ALARM
```
dsdt.dsl
[dsdt.dsl.txt](https://github.com/groeck/it87/files/1666602/dsdt.dsl.txt)

BIOS Picture
[picture.zip](https://github.com/groeck/it87/files/1666606/picture.zip)

All the infos from my system. Hope this helps somewhat.

groeck, 2018-01-26:
The chip on your motherboard is already supported by this driver. What else are you asking for ? Please explain.

G-XVI, 2018-01-26:
Ooh, oke sorry.
Why gives me then the command "sudo sensor-detect" the message "Sorry, No sensors were detected." out?

I have 5 fans installed in the case (CPU FAN, CPU OPT FAN, FAN1, FAN2 and FAN3).
With "force_id = 0x8622" I have FAN1 and FAN5, but FAN5 is the AIO water cooling.
Can you add the 5 fans and change the water cooling?

And then I would have another stupid question. I have 4 temperatures in the BIOS (CPU, MotherBoard, VRM and PCH), how do i see what temperature is what?

groeck, 2018-01-26:
Please consider reading "A note on sensors-detect" in the README file.

Unless you run the driver from here, you can not expect to get support for all fans (or anything, really). Using force_id only helps so much but does not guarantee anything. As for temperature sensor association, you'll have to figure that out yourself unless you can find someone who already did and made the information public.

G-XVI, 2018-01-26:
That's all right, thanks!

closed 2018-01-26 by G-XVI